greearb/ath10k-ct #80: Problems with Nest E thermostat - it does not like "CT" firmware

timkgh, 2019-03-25:
Description of the problem (how to configure, how to reproduce, how often it happens).
Nest E thermostat has a hard time staying connected. I suspect it's the power save mode that it gets into when its screen turns off. When the screen comes back on, the expectation is that it starts responding to ping requests, but most of the time it does not, though sometimes it does. When the screen is off, many times it doesn't even respond to ARP requests.
This is not the behavior when using what's known as the "old" ath10k driver/firmware in OpenWRT builds. The Nest behaves correctly with "old" Atheros and also with an Asus Broadcom based router, the problems start only when I flash to the "CT" builds.

Software (OS, Firmware version, kernel, driver, etc)
OpenWRT from https://forum.openwrt.org/t/build-for-netgear-r7800/316
ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fW-012-38a5514c3 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT crc32 7de96fb7

Hardware (NIC chipset, platform, etc)
Netgear R7800 router
Nest E thermostat

Logs (dmesg, maybe supplicant and/or hostap)
I don't see any obvious errors in the logs, except these occasional crashes reported in #79 though not sure if related.

greearb, 2019-03-25:
Please see if this helps, and if not, are you willing to bisect this problem if I build you a series of images to test?
[firmware-5-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/3005300/firmware-5-full-community.bin.gz)

timkgh, 2019-03-25:
I can try it, provided I get some instructions how to do it. Which OpenWrt build should I try the firmware above with? There's the driver also, not sure whether I can mix them, if it matters.
How do I replace the firmware? Just replace `/lib/firmware/ath10k/QCA9984/hw1.0/firmware-5.bin` and reboot or is there a simpler way without rebooting which can be rather slow?

greearb, 2019-03-26:
Yes, replace that firmware-5.bin file (maybe save it out of the way first), and then reboot.  Probably there is quicker way to rmmod/modprobe and restart wifi, but I bet rebooting is easier.  Looks like you will get some log spam since the driver is not yet updated to match firmware...I'm working on getting this resolved.

timkgh, 2019-03-29:
I tried firmware-5-full-community.bin.gz from above (in fact it's the exact same file already in master-r9733-e1444ab59c-20190326-ct here https://forum.openwrt.org/t/build-for-netgear-r7800/316)
Unfortunately it doesn't help, I don't see any improvement.
Here's a full dmesg of a crash, not sure if related, caused by the Nest client or some other client.
[dmesg.log](https://github.com/greearb/ath10k-ct/files/3020708/dmesg.log)

greearb, 2019-05-01:
I found another tx-hang issue in the wave-2 firmware and I think I have that fixed now.  Could you try the attached firmware and see if it helps?

[firmware-5-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/3135280/firmware-5-htt-mgt-community.bin.gz)

timkgh, 2019-05-04:
No improvement unfortunately. The Nest refuses to respond to arp requests or pings when the screen wakes up.

timkgh, 2019-05-06:
I also had a problem with a Mac getting a frozen wifi connection, could not pass any packets anymore. Had to disconnect and reconnect. I believe it corresponds to this crash in the router log.
[crash.log](https://github.com/greearb/ath10k-ct/files/3149984/crash.log)

timkgh, 2019-06-28:
I tried a new R7800 build using 10.4b-ct-9984-fW-012-e80202737
It seems the Nest E improved somewhat but still not as good as with the "old" builds.

And with this version I have a new problem with an IoT device based on MediaTek LinkIt Smart 7688. It seems to suffer from some packet loss, if I keep pinging it, every 5th ping times out. Also, typing via a ssh connection to it feels laggy. I don't see these behaviors with the "old" builds.

There's gotta be something about these IoT devices, low power/power save mode that doesn't work well with CT.

timkgh, 2019-10-27:
FWIW these problems continue with firmware ver 10.4b-ct-9984-fW-012-54863dff2 from R7800 hnyman's build r11329-b9d58f7e06

Also noticing that an ssh connection between two Macbook Pros on the same wifi network is very lagging/bursty. Typing feels like they are far apart over a lossy network. Not seeing these issues with the "old" ath10k firmware, so there is something going on here.

If it matters, I use 3 wifi VLANs to isolate devices.

greearb, 2019-10-29:
Let me know if you want to bisect this.  I am building a series of images for the r7800 now for another bug, they will be ready in a day or two.  You might also try w/out vlans to see if the problem persists?  For this type of behaviour (and where I cannot reproduce locally), I most likely cannot make any improvement unless someone can bisect to a bad commit or otherwise narrow down the problem area significantly.

timkgh, 2019-10-30:
I can try various firmwares with a default setup if you give me instructions how to switch firmwares.
As for VLANs, the trouble devices are on the "main" VLAN that gets set up out of the box, the additional VLANs get other devices that many times are not even present, e.g. guest.

greearb, 2019-10-31:
Here are notes on debugging and specifically, on how to bisect:  https://www.candelatech.com/ath10k-bugs.php
Please see this link for a series of all 9984 CT firmware images:
http://www.candelatech.com/downloads/ath10k-9984-10-4b/bisect/

timkgh, 2019-11-25:
I have not tried to bisect yet because it requires a lot of downtime.

As another data point, it seems `mDNS` has issues also with the ct firmware. For example, I use the 1password app which can sync passwords between 2 devices, phone and Mac in my case, over WLAN. The phone app uses mDNS it seems to discover other devices on the same LAN. More often than not, the sync does not work with ct firmware, the phone app cannot find the laptop. I noticed that if I disconnect the phone from WiFi, wait a bit, reconnect again and try a sync right away it works, but then after a while the sync no longer works until I repeat the process.
I do not have these issues with "old" firmware.

To me it seems that the issues are around broadcast/multicast with devices that go to sleep/enter low power mode (phones, other smart/IoT devices). I see problems with ICMP pings, ARP (devices don't respond to ARP when they should), now mDNS.

greearb, 2019-11-25:
Would you be able to get a second AP so you can connect only your NEST to it?  I assume it should be pretty low traffic, so I can give you a build that can enable some verbose tx/rx path debugging once the problem hits.  Verbose debugging in a busy environment probably will be too difficult to capture the right frames and debug, so that is why I suggest a second device.

timkgh, 2019-11-25:
I can't get a similar AP due to cost. But I have an older one around, I suppose I could revive it and move all the devices to it while I troubleshoot with the R7800. But a laptop would also need to be connected to the same AP so I can try things against the Nest. A laptop would generate more traffic.

One question: do you think the VLANs may have anything to do with this or does that logic sit above the firmware? As a first troubleshooting step I suppose I can try a simple R7800 out of the box setup, no extra VLANs, just 2 devices on it.

timkgh, 2019-11-26:
If you have a debug build, I can put it on for a bit and just have 2 devices on it, see what we can get.

timkgh, 2019-11-28:
I ran a simple test: reset my R7800 to OpenWRT image defaults, with the default VLAN. Just the Nest and Mac connected to it. I can reproduce the problems with the default config: Nest does not respond to ARP, if on the Mac I run a `ping <nest_ip>` followed by `arp -a` for the Nest I don't see a MAC address but instead `(incomplete)`. In general I see the same issue with other low power devices that don't stay active all the time, like my Android phone.
I also tried removing the default VLAN, set all ports untagged. Same issue, it does not help.
At least I know it is not my somewhat complicated config with 3 SSIDs and 3 VLANs.

timkgh, 2019-11-29:
One other data point: trying a recent R7800 OpenWRT "ct" build but I swapped in the "old" firmware version `10.4-3.10-00047` and that seems to work well so far, so it is a firmware issue. It's good that the ct driver works with the old firmware at least.

greearb, 2020-01-23:
timkgh, please see issue #50 and see if build 1084 works for you?  If so, see if bug 1085 causes the problem?  If neither, please consider bisecting with firmware linked earlier in that bug.

timkgh, 2020-01-25:
I tried firmware htt from https://github.com/greearb/ath10k-ct/issues/50#issuecomment-578259688

With this version, the ssh lag between devices I've been seeing with other ct builds seems to be gone.

But the Nest issues remain, I cannot ping it all even when the screen is on and the app complains every few minutes that it goes on and off the network.

For devices I can ping though, ping times are all over the place, there's a high variance in ping times over WiFi which I don't observe with the "old" firmware. Normally I see constant 2-3ms pings with some rare spikes, but with this version it's all over the place from 3-100ms, with most around 10-12ms.

Also I reported before the issues with the 1Password app on Android, even with this build it still cannot discover the desktop app running on my MBP so it can't sync passwords. Not sure why the Bonjour traffic doesn't make it through.

greearb, 2020-02-19:
Can you please test the y6 and z7 builds from bug 114 and let me know how the work compared to previous images you tested?

timkgh, 2020-02-22:
y6: flaky connection to MediaTek LinkIt Smart 7688, slow to start responding to pings, high pings when it works, packet loss pattern every 7-8 packets. I ran it for about 12 hours, have not seen a crash like y3, just the regular WARNING trace that seems to appear with all builds.

z7: cannot ping Nest when it comes out of sleep.

greearb, 2020-02-22:
I'm going to try to get one of these to test locally.  Is there any specific model or version information on your Nest E?

timkgh, 2020-02-22:
Nest E (https://store.google.com/us/product/nest_thermostat_e)
I've had mine for 2+ yrs now, not sure whether there are multiple revs out there using different wifi chips.
This is what mine shows:
```
Model: Display-4.8
Backplate model: Backplate-7.6
Software: 6.1-9
Updated: Feb 12, 2020 7:24 AM
```
Many times you can get these for free after a rebate from your utilities company.

greearb, 2020-02-29:
I got my new Nest-E thermostat and connected to a 9984 AP running my software/firmware.  I am able to ping it fine as far as I can tell.  I cannot really use it because I have no HVAC to hook up (it is just sitting on my desk with external power).  I have some patches I want to try reverting to see if that has some good effect....maybe someone will have time to test them.

timkgh, 2020-02-29:
The test is to let the screen go to sleep, then start pinging it continuously from another client. The pings should always fail while the screen is off. Then if you wave your hand close to the screen, it should wake up and start responding to pings.

greearb, 2020-03-01:
I tried that and it worked fine in my test.  Here is a new commit that backs out some patches previously shown to be problematic.  Please let me know how this works.
[firmware-5-full-htt-mgt-community.bin-x1.gz](https://github.com/greearb/ath10k-ct/files/4271913/firmware-5-full-htt-mgt-community.bin-x1.gz)

timkgh, 2020-03-01:
Depending on the firmware version, sometimes it works, other times it does not ever work or it's intermittent. There are also cases where the Nest goes off the network a few times. On my network I have a Fingbox that monitors my devices and alerts me when they are off the network. I believe it does it by sending ARP requests to the entire subnet every few minutes. It does not ping the devices, just ARP. When I get the alerts I try to ping the Nest, wake up its screen and it doesn't respond to pings, it appears that it's not connected to the AP at those times.

And as mentioned before, the Nest is not the only issue, just one of them. I have problems on "ct" firmware with a MediaTek LinkIt Smart 7688 client, slow connections between 2 Macs on the same WLAN, software such as 1Password or Spotify Connect not being able to find devices on the networks (mDNS broadcast issues?). Many times a couple of these work, but the others don't. I have yet to find a "ct" build where all my devices worked OK.

I'll try the x1 build later today.

timkgh, 2020-03-01:
x1 feedback:
Mac to Mac (on the same WLAN) choppy/bursty, high pings.
Nest responds but also very high pings, here's a sample:
```
64 bytes from 192.168.x.x: icmp_seq=12 ttl=64 time=5194.019 ms
64 bytes from 192.168.x.x: icmp_seq=13 ttl=64 time=4189.167 ms
64 bytes from 192.168.x.x: icmp_seq=14 ttl=64 time=3184.572 ms
64 bytes from 192.168.x.x: icmp_seq=15 ttl=64 time=2183.557 ms
64 bytes from 192.168.x.x: icmp_seq=16 ttl=64 time=1180.005 ms
64 bytes from 192.168.x.x: icmp_seq=17 ttl=64 time=176.019 ms
64 bytes from 192.168.x.x: icmp_seq=18 ttl=64 time=1170.464 ms
64 bytes from 192.168.x.x: icmp_seq=19 ttl=64 time=165.363 ms
64 bytes from 192.168.x.x: icmp_seq=20 ttl=64 time=2013.794 ms
64 bytes from 192.168.x.x: icmp_seq=21 ttl=64 time=1013.115 ms
Request timeout for icmp_seq 27
64 bytes from 192.168.x.x: icmp_seq=24 ttl=64 time=4209.364 ms
64 bytes from 192.168.x.x: icmp_seq=25 ttl=64 time=3239.007 ms
64 bytes from 192.168.x.x: icmp_seq=26 ttl=64 time=2234.400 ms
64 bytes from 192.168.x.x: icmp_seq=27 ttl=64 time=1233.464 ms
```

greearb, 2020-03-02:
Any chance you have a really large beacon interval set?

timkgh, 2020-03-02:
I'm just using the default beacon interval in OpenWRT which seems to be 100.

greearb, 2020-03-05:
In case someone is interested in bisecting, I put up a new series of 9984 firmware images.  This fixes some bugs found in earlier bisect attempts.  I suspect there are bugs in the last 4 commits as well.
https://www.candelatech.com/downloads/ath10k-9984-10-4b/bisect/all_builds-9984-H-march-5-2020.tar.gz

greearb, 2020-04-30:
I am hoping this is fixed in latest openwrt, please re-open if you see more problems.

closed 2020-04-30 by greearb

timkgh, 2020-05-02:
I tried the 2020-04-24 .018 version from https://www.candelatech.com/downloads/ath10k-9984-10-4b/?C=M;O=D and the Nest E seems to be happy now, I haven't seen it have trouble staying connected over the past 3 days.
I still see some burst/lag issues between 2 Macs on the same AP with this version, so it's not as reliable as the "old" firmware, but it seems to be moving in the right direction.