greearb/ath10k-ct #82: QCA9984 firmware crash

ironpillow, 2019-03-29:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
Firmware is crashing on openWRT 18.06.2 and NOT on current snapshot. All clients are de-authenticated and can't re-authenticate.

When more than 4 clients that are connected to the access point are playing video (youtube or twitch), firmware crashes and all clients are de-authenticated.

Software (OS, Firmware version, kernel, driver, etc)
openWRT 18.0.6.2
ath10k-firmware-qca9984-ct - 2018-05-12-952afa49-1
kmod-ath10k-ct - 4.9.152+2018-03-16-30827f7d-3

Hardware (NIC chipset, platform, etc)
QCA9984

Logs (dmesg, maybe supplicant and/or hostap)
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.187277] ath10k_pci 0000:00:00.0: firmware crashed! (uuid f80a774a-b7c8-49bc-91d9-0091667cab40)
Fri Mar 29 04:20:37 2019 kern.info kernel: [  358.196417] ath10k_pci 0000:00:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
Fri Mar 29 04:20:37 2019 kern.info kernel: [  358.206506] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
Fri Mar 29 04:20:37 2019 kern.info kernel: [  358.226288] ath10k_pci 0000:00:00.0: firmware ver 10.4-ct-9984-fW-011-cf79c7f api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,cust-stats-CT crc32 25783e66
Fri Mar 29 04:20:37 2019 kern.info kernel: [  358.251904] ath10k_pci 0000:00:00.0: board_file api 2 bmi_id 0:1 crc32 dd6d039c
Fri Mar 29 04:20:37 2019 kern.info kernel: [  358.259342] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.271600] ath10k_pci 0000:00:00.0: firmware register dump:
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.277353] ath10k_pci 0000:00:00.0: [00]: 0x0000000A 0x000015B3 0x009A3A67 0x00975B31
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.285392] ath10k_pci 0000:00:00.0: [04]: 0x009A3A67 0x00060B30 0x00000005 0x00000001
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.293418] ath10k_pci 0000:00:00.0: [08]: 0x00000003 0x000000FF 0x000000FF 0x004189A0
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.301453] ath10k_pci 0000:00:00.0: [12]: 0x00000009 0x00000000 0x009A3E44 0x009A3E5D
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.309496] ath10k_pci 0000:00:00.0: [16]: 0x00973AB0 0x009606CA 0x00973AB0 0x00000000
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.317531] ath10k_pci 0000:00:00.0: [20]: 0x409A3A67 0x004066AC 0x000000FF 0x61770778
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.325574] ath10k_pci 0000:00:00.0: [24]: 0x809A3E8E 0x0040670C 0x00000000 0xC09A3A67
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.333609] ath10k_pci 0000:00:00.0: [28]: 0x809A6828 0x0040673C 0x00000009 0x00406789
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.341652] ath10k_pci 0000:00:00.0: [32]: 0x809A781A 0x0040676C 0x00406860 0x00051697
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.349695] ath10k_pci 0000:00:00.0: [36]: 0x809A784A 0x0040685C 0x0042EDF8 0x004068EC
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.357737] ath10k_pci 0000:00:00.0: [40]: 0x80985B17 0x004068AC 0x00000000 0x00458D44
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.365773] ath10k_pci 0000:00:00.0: [44]: 0x8099311B 0x004068CC 0x0043004C 0x00458D44
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.373807] ath10k_pci 0000:00:00.0: [48]: 0x8098ED50 0x004068EC 0x0043004C 0x00000000
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.381850] ath10k_pci 0000:00:00.0: [52]: 0x80963AD3 0x00406A7C 0x0043004C 0x0098ED48
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.389893] ath10k_pci 0000:00:00.0: [56]: 0x80960E80 0x00406A9C 0x0000001F 0x00400000
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.397933] ath10k_pci 0000:00:00.0: Copy Engine register dump:
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.403943] ath10k_pci 0000:00:00.0: [00]: 0x0004a000  11  11   3   3
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.410490] ath10k_pci 0000:00:00.0: [01]: 0x0004a400  30  30 292 293
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.417038] ath10k_pci 0000:00:00.0: [02]: 0x0004a800   6   6  69  70
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.423576] ath10k_pci 0000:00:00.0: [03]: 0x0004ac00   5   5   7   5
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.430123] ath10k_pci 0000:00:00.0: [04]: 0x0004b000 7901 7901 143 104
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.436847] ath10k_pci 0000:00:00.0: [05]: 0x0004b400  20  20 371 372
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.443385] ath10k_pci 0000:00:00.0: [06]: 0x0004b800  19  19  19  19
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.449932] ath10k_pci 0000:00:00.0: [07]: 0x0004bc00   1   1   1   1
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.456479] ath10k_pci 0000:00:00.0: [08]: 0x0004c000   0   0 127   0
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.463017] ath10k_pci 0000:00:00.0: [09]: 0x0004c400   1   1   1   1
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.469565] ath10k_pci 0000:00:00.0: [10]: 0x0004c800   0   0   0   0
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.476112] ath10k_pci 0000:00:00.0: [11]: 0x0004cc00   0   0   0   0
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.484669] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x423910  dropped: 0
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.493273] ath10k_pci 0000:00:00.0: [0] next: 0x423928 buf: 0x418ba0 sz: 1500 len: 92 count: 4 free: 0
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.503816] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.510096] ath10k: [0000]: 59160500 58580010 12310000 20104300 0A000000 00000000 59160500 58580010
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.519280] ath10k: [0008]: 12310000 20104300 0E000000 00000000 98160500 014CFC0B EDDD1091 FF000000
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.528466] ath10k: [0016]: 98160500 0100FC17 673A9A00 B3150000 B3150000 9C654000 69451091
Fri Mar 29 04:20:37 2019 kern.err kernel: [  358.536848] ath10k_pci 0000:00:00.0: ATH10K_END
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.542457] ath10k_pci 0000:00:00.0: [1] next: 0x423910 buf: 0x419190 sz: 1500 len: 0 count: 0 free: 0
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.599406] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8f2f4e00 vdev: 0 addr: 14:10:9f:d2:64:7f
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.610586] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8f1cee00 vdev: 0 addr: 70:48:0f:a5:7a:0a
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.621718] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8e61d400 vdev: 0 addr: b8:27:eb:51:37:79
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.632835] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8f149c00 vdev: 0 addr: b8:27:eb:0a:29:19
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.643959] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8f14d400 vdev: 0 addr: b8:27:eb:30:51:d5
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.655082] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8f4d3400 vdev: 0 addr: b8:27:eb:41:0d:0e
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.666205] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8e61de00 vdev: 0 addr: b8:27:eb:cb:dd:8f
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.677327] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8e5d9800 vdev: 0 addr: ac:cf:5c:59:a8:2d
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.688451] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8f2bce00 vdev: 0 addr: c4:43:8f:f6:50:78
Fri Mar 29 04:20:37 2019 kern.warn kernel: [  358.699575] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8f2f4800 vdev: 0 addr: b8:e8:56:75:d8:54
[... 19 lines cut ...]
Fri Mar 29 04:20:43 2019 kern.err kernel: [  364.930854] ath10k_pci 0000:00:00.0: [28]: 0x00000000 0x00000000 0x00000000 0x00000000
Fri Mar 29 04:20:43 2019 kern.err kernel: [  364.938897] ath10k_pci 0000:00:00.0: [32]: 0x00000000 0x00000000 0x00000000 0x00000000
[... cut ...]
Fri Mar 29 04:20:43 2019 kern.warn kernel: [  365.081740] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x423910  dropped: 0
Fri Mar 29 04:20:43 2019 kern.warn kernel: [  365.090356] ath10k_pci 0000:00:00.0: [0] next: 0x423928 buf: 0x418ba0 sz: 1500 len: 0 count: 0 free: 0
Fri Mar 29 04:20:43 2019 kern.warn kernel: [  365.100818] ath10k_pci 0000:00:00.0: [1] next: 0x423910 buf: 0x419190 sz: 1500 len: 0 count: 0 free: 0
Fri Mar 29 04:20:48 2019 kern.warn kernel: [  369.803722] ath10k_pci 0000:00:00.0: wmi service ready event not received

greearb, 2019-03-29:
Since latest drivers/firmware fix this, you will need to either run latest code, or backport driver and firmware to your older release.  I asked on IRC, and it seems some OpenWRT devs may be interested in backporting drivers to the stable release, so maybe that will just happen sometime soon.

ironpillow, 2019-04-05:
Thank you Ben!
I will be using latest openWRT snapshot, as version 19 will be release soon anyway.

I will keep testing and let you know if a crash occurs.

closed 2019-04-05 by ironpillow