greearb/ath10k-ct #85: QCA9984 160Mhz enable fail

NathanKuo0821, 2019-05-15:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
Hi , I trying to use bandwidth 160Mhz with QCA9984 pcie card , it working under 80 Mhz,But it don't work for 160Mhz I see the following errors:

> nl80211: Start radar detection (CAC) 5500 MHz (ht_enabled=1, vht_enabled=1, bandwidth=160 MHz, cf1=5570 MHz, cf2=0 MHz)
>   * freq=5500
>   * vht_enabled=1
>   * ht_enabled=1
>   * bandwidth=160
>   * channel_width=5
>   * center_freq1=5570
>   * center_freq2=0
> nl80211: Failed to start radar detection: -16 (Device or resource busy)
> DFS start_dfs_cac() failed, -1

Software (OS, Firmware version, kernel, driver, etc)
OS : Ubuntu  18.04
Kernel  :  4.18.17
Driver : ath10k-ct 4.19
Firmware :  firmware-5-ct-non-commercial-9.bin
hostpad : hostapd-ct 2.8

Hardware (NIC chipset, platform, etc)
Platform : X86 - Laptop
Chipset : WMX6401(QCA9984 )
https://www.jjplus.com/wmx6401/

Logs (dmesg, maybe supplicant and/or hostap)
https://drive.google.com/open?id=1mS9lfaI9W8uH2YcfVDdOlyvl8kPW1Q31

Can you check it  for me or give some advice? thank you very much.

lantis1008, 2019-08-10:
Hi @greearb i'd like to second this report on OpenWrt 19.07.
Unable to get VHT160 running on any channel.
VHT80P80 works unless either segment is in a DFS area and then it fails with `DFS start_dfs_cac() failed, -1`.
Lower channel widths work in the DFS space fine.

If you need additional information from my system please let me know.

greearb, 2019-08-10:
This is not likely a firmware or driver problem.  Probably you either have regulatory domain issues and/or the mac80211 stack doesn't want to do radar detection (CAC) on your system for some reason.
Either way, 160Mhz is mostly useless since ath10k radios will use 1/2 of the spatial streams in 160Mhz mode as compared to 80Mhz mode, so total throughput is the same unless perhaps you are trying to do a very long range communication where spatial streams won't be used anyway.

lantis1008, 2019-08-11:
https://github.com/greearb/ath10k-ct/blob/f98b6dc4d27ea2d79a1577285d1d5cb0641b3eb4/ath10k-4.19/mac.c#L9260-L9275
vs
https://github.com/greearb/ath10k-ct/blob/f98b6dc4d27ea2d79a1577285d1d5cb0641b3eb4/ath10k-4.19/mac.c#L9297-L9309

If i'm reading this right (and i do defer to your better judgement and experience), 160 and 80+80 aren't enabled as valid DFS widths for the CT firmware?
Why? I'm happy to patch it and test myself but just wondering if there's a better reason before i go down this route.

lantis1008, 2019-08-12:
The following solves this problem on Openwrt. Please consider patching upstream at your next convenience on all applicable kernels moving forward. Thanks.
```diff
--- a/ath10k-4.19/mac.c
+++ b/ath10k-4.19/mac.c
@@ -9392,6 +9392,8 @@ static struct ieee80211_iface_combinatio
 		.radar_detect_widths =	BIT(NL80211_CHAN_WIDTH_20_NOHT) |
 					BIT(NL80211_CHAN_WIDTH_20) |
 					BIT(NL80211_CHAN_WIDTH_40) |
+					BIT(NL80211_CHAN_WIDTH_80P80) |
+					BIT(NL80211_CHAN_WIDTH_160) |
 					BIT(NL80211_CHAN_WIDTH_80),
 #endif
 	},
```

greearb, 2019-08-12:
Thanks for the patch.  Probably I had a mis-merge when porting forward my patches to this kernel.  I'll get these upstream in my kernels/driver in a day or two.

lantis1008, 2019-08-12:
It happens! Thanks for the update :)

greearb, 2019-09-16:
This fix should be upstream now.

closed 2019-09-16 by greearb

julianocss, 2020-11-17:
@greearb 

160 MHz it is not available for my router (QCA9886)
I am using the latest commit from GitHub.
Am i doing something wrong?

Juliano

lantis1008, 2021-07-24:
@greearb This same issue is present in 5.10, 5.11 and 5.12.
https://github.com/greearb/ath10k-ct/blob/master/ath10k-5.10/mac.c#L10780
https://github.com/greearb/ath10k-ct/blob/master/ath10k-5.11/mac.c#L10782
https://github.com/greearb/ath10k-ct/blob/master/ath10k-5.12/mac.c#L10891

greearb, 2021-09-22:
Should be fixed in the commit I just pushed.