greearb/ath10k-ct #89: Fails to receive packets over 1500 bytes in length.

colinw-smartrg, 2019-08-09:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).

Software (OS, Firmware version, kernel, driver, etc)

Hardware (NIC chipset, platform, etc)

Logs (dmesg, maybe supplicant and/or hostap)

colinw-smartrg, 2019-08-09:
When Linux wifi device is configured with an MTU above 1500, e.g. 1600, any packet received above 1500 bytes payload gets dropped. 

Linux 4.14.134, QCA Dakota 4019.
Firmware:
QCA4019-firmware-5-ct-full-htt-mgt-community-12.bin-lede.009

colinw-smartrg, 2019-08-09:
The issue appears to be the ring buffer size defined by:
HTT_RX_BUF_SIZE 1920
The largest MDSU is
#define HTT_RX_MSDU_SIZE (HTT_RX_BUF_SIZE - (int)sizeof(struct htt_rx_desc))

Which is 1560, and with 60 bytes of overhead, the max payload is 1500.

An MSDU larger than 1560 is possible in the air, and will not be received properly by the driver as the firmware doesn't appear to fragment.

greearb, 2019-08-09:
Stock driver/firmware has the same issue, or is this CT only problem?

colinw-smartrg, 2019-08-10:
Just tried the stock driver/firmware and see the same issue.

Changing HTT_RX_BUF_SIZE to 2048 ( 1 cache line more ) fixes the issue.

colinw-smartrg, 2019-08-12:
This seems like a firmware issue in that the firmware should create two fragments since the RX_BUF is not big enough for the packet. But I only get one encrypted buffer.

The driver appears to detect the length and appears to send two buffers for decryption as I see two decrypted buffers returned chained in ath10k_htt_rx_handle_amsdu() but ath10k_htt_rx_h_unchain() fails with rx_drop_decap_non_raw_chained. Maybe that is a driver issue as the comment says FIXME.

greearb, 2019-08-12:
Since this happens in stock firmware/driver, maybe report it to the ath10k mailing list and see if someone will fix the driver?  Please also mention that changing the RX_BUF_SIZE to 2048 works around the problem.  Maybe just making that larger is the correct fix?

greearb, 2019-09-16:
I believe this is now fixed by increasing the RX_BUF_SIZE in ath10k-ct driver.  Changes should be in upstream OpenWRT now.

closed 2019-09-16 by greearb

adrianschmutzler, 2020-04-25:
@greearb This has been brought up again for OpenWrt since it's still not "fixed" in kernel.
In that context, I wonder why the value 2048 has been chosen, which will allow a little more payload, but seems not much less arbitrary than the previous value. What prevents enabling the full max. MTU of 2304 without fragmentation, before something else will break the 2048 limit again?

greearb, 2020-04-25:
I haven't tried larger values, but it should be easy enough for someone to test.