greearb/ath10k-ct #90: Stacktrace in latest ath10k-htt openwrt

jekkos, 2019-08-21:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
Seeing the following stacktrace with ath10k-ct from openwrt master (built on 22/08 22h)

Software (OS, Firmware version, kernel, driver, etc)
Openwrt trunk Linux b1300-2 4.19.66 #0 SMP Tue Aug 20 22:07:00 2019 armv7l GNU/Linux

Hardware (NIC chipset, platform, etc)
gl.inet b1300

Logs (dmesg, maybe supplicant and/or hostap)
```
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.530052] ------------[ cut here ]------------                                                                                                                                                                                            
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.530460] WARNING: CPU: 0 PID: 0 at backports-5.3-rc4-1/net/wireless/util.c:1171 cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211]                                                                                                        
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.534035] invalid rate bw=0, mcs=15, nss=4                                                                                                                                                                                                
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.545858] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async iptable_nat ipt_MASQUERADE ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD rtlwifi pppox ppp_generic nf_nat_ipv4 nf_nat 
nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwm
on crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug                                                                                 
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.591445] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.19.66 #0                                                                                                                                                                            
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.612517] Hardware name: Generic DT based system                                                                                                                                                                                          
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.618814] [<c030e2c4>] (unwind_backtrace) from [<c030ad20>] (show_stack+0x10/0x14)                                                                                                                                                        
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.623399] [<c030ad20>] (show_stack) from [<c0782a30>] (dump_stack+0x88/0x9c)                                                                                                                                                              
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.631493] [<c0782a30>] (dump_stack) from [<c031b898>] (__warn+0xd4/0xf0)                                                                                                                                                                  
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.638522] [<c031b898>] (__warn) from [<c031b8ec>] (warn_slowpath_fmt+0x38/0x48)                                                                                                                                                           
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.646341] [<c031b8ec>] (warn_slowpath_fmt) from [<bf13efbc>] (cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211])                                                                                                                          
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.655112] [<bf13efbc>] (cfg80211_calculate_bitrate [cfg80211]) from [<bf1deafc>] (ieee80211s_update_metric+0x78/0x9c [mac80211])                                                                                                          
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.665393] [<bf1deafc>] (ieee80211s_update_metric [mac80211]) from [<bf1919b4>] (ieee80211_tx_monitor+0x7c4/0xba8 [mac80211])                                                                                                              
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.676991] [<bf1919b4>] (ieee80211_tx_monitor [mac80211]) from [<bf191e20>] (ieee80211_tx_status+0x88/0x90 [mac80211])                                                                                                                     
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.688161] [<bf191e20>] (ieee80211_tx_status [mac80211]) from [<bf2e16e4>] (ath10k_txrx_tx_unref+0x540/0x58c [ath10k_core])                                                                                                                
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.698464] [<bf2e16e4>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf2dadd0>] (ath10k_htt_t2h_msg_handler+0x610/0x1fb8 [ath10k_core])                                                                                                     
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.709231] [<bf2dadd0>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf32c274>] (ath10k_pci_htt_rx_cb+0x1c8/0x218 [ath10k_pci])                                                                                                       
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.721206] [<bf32c274>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf2f7bb8>] (ath10k_ce_per_engine_service+0x80/0xd8 [ath10k_core])                                                                                                       
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.733820] [<bf2f7bb8>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf2f7c7c>] (ath10k_ce_per_engine_service_any+0x6c/0x98 [ath10k_core])                                                                                          
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.745230] [<bf2f7c7c>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf32dcac>] (ath10k_pci_napi_poll+0x64/0x100 [ath10k_pci])                                                                                                  
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.757584] [<bf32dcac>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c067fea8>] (net_rx_action+0x138/0x2fc)                                                                                                                                  
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.769861] [<c067fea8>] (net_rx_action) from [<c03021c8>] (__do_softirq+0xe0/0x268)                                                                                                                                                        
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.779471] [<c03021c8>] (__do_softirq) from [<c031f9e4>] (irq_exit+0x9c/0xfc)                                                                                                                                                              
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.787384] [<c031f9e4>] (irq_exit) from [<c035f11c>] (__handle_domain_irq+0x9c/0xac)                                                                                                                                                       
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.794428] [<c035f11c>] (__handle_domain_irq) from [<c0537490>] (gic_handle_irq+0x5c/0x90)                                                                                                                                                 
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.802299] [<c0537490>] (gic_handle_irq) from [<c0301a0c>] (__irq_svc+0x6c/0x90)                                                                                                                                                           
[... cut ...]
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.853557] [<c0341ad4>] (do_idle) from [<c0341e40>] (cpu_startup_entry+0x18/0x20)                                                                                                                                                          
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.860574] [<c0341e40>] (cpu_startup_entry) from [<c0a00d1c>] (start_kernel+0x428/0x438)                                                                                                                                                   
Wed Aug 21 07:33:08 2019 kern.warn kernel: [ 1920.870813] ---[ end trace 42efa85437038ad5 ]---
```

klukonin, 2019-08-21:
Hmmmm.

As I remember, this bug was fixed not so long time ago.
Could you please provide a commit of your OpenWrt master and dmesg (driver version with firmware version).

This can be a known driver problem.

jekkos, 2019-08-21:
qca4019 hw1.0 target 0x01000000 chip_id 0x003900ff sub 0000:0000
[  246.599312] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[  246.612051] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fH-012-e8020273 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 a47d8012
[  246.627591] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:16 crc32 e2dfaa91
[  246.646066] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1

the commit I built from is
164037983d00cc5b27f970b40c4a697d31277164

jekkos, 2019-08-21:
Also it's running in 802.11s (mesh) using openwrt , with 3 peers and combination of ath9k and ath10k

klukonin, 2019-08-21:
Could you please try this firmware

https://www.candelatech.com/downloads/ath10k-4019-10-4b/firmware-5-ct-full-htt-mgt-community-12.bin-lede.010

klukonin, 2019-08-21:
@greearb 

Hello, Ben.
Possibly the problem is in ath10k_bw_to_mac80211_bw function.

Firmware can return something wrong instead of 0,1,2,3, range and function returns zero value.
mac80211 is not ready for that.

klukonin, 2019-08-21:
Or may be I'm wrong and it's another issue when mac80211 can't handle MCS more than 9  for VHT bitrate. Because the right encoding should be something like 0-9 MCS and 0-4 NSS value.
But firmware returns an old style 0-XX MCS.
This issue was in some previous original ath10k firmwares.

jekkos, 2019-08-21:
Ok this stacktrace is gone now with the new firmware, however i see now this one

```
[  123.189754] ------------[ cut here ]------------
[  123.189936] WARNING: CPU: 0 PID: 0 at /home/jekkos/hacking/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-2019-06-13-f0aa8130/ath10k-4.19/htt_rx.c:909 ath10k_htt_t2h_msg_handler+0xf00/0x1fb8 [ath10k_core]                                     
[  123.193543] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async iptable_nat ipt_MASQUERADE ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD rtlwifi pppox ppp_generic nf_nat_ipv4 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack
_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat nf_log_ipv6 nf_log_comm
on ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug                                                                                                                           
[  123.255732] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.19.66 #0
[  123.277478] Hardware name: Generic DT based system
[  123.283914] [<c030e2c4>] (unwind_backtrace) from [<c030ad20>] (show_stack+0x10/0x14)
[  123.288470] [<c030ad20>] (show_stack) from [<c0782a30>] (dump_stack+0x88/0x9c)
[  123.296401] [<c0782a30>] (dump_stack) from [<c031b898>] (__warn+0xd4/0xf0)
[  123.303419] [<c031b898>] (__warn) from [<c031b984>] (warn_slowpath_null+0x40/0x48)
[  123.311068] [<c031b984>] (warn_slowpath_null) from [<bf2db6c0>] (ath10k_htt_t2h_msg_handler+0xf00/0x1fb8 [ath10k_core])
[  123.319359] [<bf2db6c0>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2db890>] (ath10k_htt_t2h_msg_handler+0x10d0/0x1fb8 [ath10k_core])                                                                                                                                        
[  123.329887] [<bf2db890>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf2dd2f0>] (ath10k_htt_get_rx_ring_size_64+0xb70/0xbc0 [ath10k_core])                                                                                                                                      
[  123.342714] [<bf2dd2f0>] (ath10k_htt_get_rx_ring_size_64 [ath10k_core]) from [<bf2dda78>] (ath10k_htt_txrx_compl_task+0x738/0xbd8 [ath10k_core])                                                                                                                                      
[  123.355281] [<bf2dda78>] (ath10k_htt_txrx_compl_task [ath10k_core]) from [<bf32dcb8>] (ath10k_pci_napi_poll+0x70/0x100 [ath10k_pci])
[  123.367614] [<bf32dcb8>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c067fea8>] (net_rx_action+0x138/0x2fc)
[  123.379286] [<c067fea8>] (net_rx_action) from [<c03021c8>] (__do_softirq+0xe0/0x268)
[  123.388721] [<c03021c8>] (__do_softirq) from [<c031f9e4>] (irq_exit+0x9c/0xfc)
[  123.396636] [<c031f9e4>] (irq_exit) from [<c035f11c>] (__handle_domain_irq+0x9c/0xac)
[  123.403672] [<c035f11c>] (__handle_domain_irq) from [<c0537490>] (gic_handle_irq+0x5c/0x90)
[  123.411551] [<c0537490>] (gic_handle_irq) from [<c0301a0c>] (__irq_svc+0x6c/0x90)
[  123.419607] Exception stack(0xc0b01f30 to 0xc0b01f78)
[  123.427278] 1f20:                                     00000000 005d1054 cfd752c4 c0313a40
[  123.432381] 1f40: ffffe000 c0b04c6c 00000001 c0b04cac c0a3bda8 00000000 00000000 00000000
[  123.440512] 1f60: 0f33b000 c0b01f80 c03082f4 c03082f8 60000013 ffffffff
[  123.448718] [<c0301a0c>] (__irq_svc) from [<c03082f8>] (arch_cpu_idle+0x34/0x38)
[  123.455172] [<c03082f8>] (arch_cpu_idle) from [<c0341ad4>] (do_idle+0xf8/0x208)
[  123.462802] [<c0341ad4>] (do_idle) from [<c0341e40>] (cpu_startup_entry+0x18/0x20)
[  123.469826] [<c0341e40>] (cpu_startup_entry) from [<c0a00d1c>] (start_kernel+0x428/0x438)
[  123.477581] ---[ end trace 0f35daad4dae7454 ]---
[  123.487286] ath10k_ahb a000000.wifi: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[  126.548927] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  126.548982] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[  126.584596] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[  126.585936] ath10k_ahb a800000.wifi: wmi print 'free: 56528 iram: 23752 sram: 32280'
[  126.852596] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
```

jekkos, 2019-08-21:
also for some reason `iw dev wlan0 station dump` gives me a `command failed: No such device (-19)` while the interface is definitely there

`root@b1300-1:~# ifconfig
br-wlan   Link encap:Ethernet  HWaddr E4:95:6E:44:C1:64  
          inet addr:192.168.1.10  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::e695:6eff:fe44:c164/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:2602 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1433 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:224718 (219.4 KiB)  TX bytes:522773 (510.5 KiB)

eth0      Link encap:Ethernet  HWaddr E4:95:6E:44:C1:64  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:3199 errors:0 dropped:16 overruns:0 frame:0
          TX packets:2449 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:309011 (301.7 KiB)  TX bytes:912429 (891.0 KiB)

eth1      Link encap:Ethernet  HWaddr E4:95:6E:44:C1:65  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:22 errors:0 dropped:0 overruns:0 frame:0
          TX packets:22 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:4247 (4.1 KiB)  TX bytes:4247 (4.1 KiB)

wlan0     Link encap:Ethernet  HWaddr E4:95:6E:44:C1:64  
          inet6 addr: fe80::e695:6eff:fe44:c164/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:983 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1313 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:384930 (375.9 KiB)  TX bytes:179478 (175.2 KiB)
```

greearb, 2019-08-21:
Please check your compile system and paste to me the code around that warning
(line 909) so I can see exactly where that is.

Thanks,
Ben

jekkos, 2019-08-21:
```
  37 static struct sk_buff *                                                                                                                                                                                                                                                              
  38 ath10k_htt_rx_find_skb_paddr(struct ath10k *ar, u64 paddr)                                                                                                                                                                                                                           
:909                                                                                                                                                                                                                                                                                      
 891         __cpu_to_le32(RX_ATTENTION_FLAGS_PEER_IDX_INVALID))                                                                                                                                                                                                                          
 892         return NULL;
 893 
 894     if (!(rxd->msdu_end.common.info0 &
 895           __cpu_to_le32(RX_MSDU_END_INFO0_FIRST_MSDU)))
 896         return NULL;
 897 
 898     peer_id = MS(__le32_to_cpu(rxd->mpdu_start.info0),
 899              RX_MPDU_START_INFO0_PEER_IDX);
 900 
 901     peer = ath10k_peer_find_by_id(ar, peer_id);
 902     if (!peer)
 903         return NULL;
 904 
 905     arvif = ath10k_get_arvif(ar, peer->vdev_id);
 906     if (WARN_ON_ONCE(!arvif))
 907         return NULL;
 908 
 909     if (WARN_ON_ONCE(ath10k_mac_vif_chan(arvif->vif, &def)))
 910         return NULL;
 911 
 912     return def.chan;
 913 }
```

greearb, 2019-08-21:
Ahh, that one.

I think I just need to make it go away, I think it is probably not a serious
problem.

Thanks,
Ben

jekkos, 2019-08-21:
Ok thanks for the quick answers here. I'll test a bit further with this latest firmware version to see if everything is stable enough for daily use.

jekkos, 2019-08-23:
Saw another stacktrace today

```
[19466.115558] ------------[ cut here ]------------
[19466.116023] WARNING: CPU: 0 PID: 0 at backports-5.3-rc4-1/net/wireless/util.c:1171 cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211]
[19466.119321] invalid rate bw=0, mcs=15, nss=4
[19466.131180] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async iptable_nat ipt_MASQUERADE ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD rtlwifi pppox ppp_generic nf_nat_ipv4 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug                                                                                                                           
[19466.175851] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W         4.19.66 #0
[19466.198031] Hardware name: Generic DT based system
[19466.205709] [<c030e2c4>] (unwind_backtrace) from [<c030ad20>] (show_stack+0x10/0x14)
[19466.210298] [<c030ad20>] (show_stack) from [<c0782a30>] (dump_stack+0x88/0x9c)
[19466.218198] [<c0782a30>] (dump_stack) from [<c031b898>] (__warn+0xd4/0xf0)
[19466.225227] [<c031b898>] (__warn) from [<c031b8ec>] (warn_slowpath_fmt+0x38/0x48)
[19466.232332] [<c031b8ec>] (warn_slowpath_fmt) from [<bf13efbc>] (cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211])
[19466.240225] [<bf13efbc>] (cfg80211_calculate_bitrate [cfg80211]) from [<bf1deafc>] (ieee80211s_update_metric+0x78/0x9c [mac80211])
[19466.250419] [<bf1deafc>] (ieee80211s_update_metric [mac80211]) from [<bf1919b4>] (ieee80211_tx_monitor+0x7c4/0xba8 [mac80211])
[19466.262106] [<bf1919b4>] (ieee80211_tx_monitor [mac80211]) from [<bf191e20>] (ieee80211_tx_status+0x88/0x90 [mac80211])
[19466.273459] [<bf191e20>] (ieee80211_tx_status [mac80211]) from [<bf2e16e4>] (ath10k_txrx_tx_unref+0x540/0x58c [ath10k_core])
[19466.283989] [<bf2e16e4>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf2dadd0>] (ath10k_htt_t2h_msg_handler+0x610/0x1fb8 [ath10k_core])
[19466.295311] [<bf2dadd0>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf32c274>] (ath10k_pci_htt_rx_cb+0x1c8/0x218 [ath10k_pci])
[19466.307190] [<bf32c274>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf2f7bb8>] (ath10k_ce_per_engine_service+0x80/0xd8 [ath10k_core])
[19466.319476] [<bf2f7bb8>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf2f7c7c>] (ath10k_ce_per_engine_service_any+0x6c/0x98 [ath10k_core])                                                                                                                                    
[19466.331326] [<bf2f7c7c>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf32dcac>] (ath10k_pci_napi_poll+0x64/0x100 [ath10k_pci])                                                                                                                                            
[19466.344187] [<bf32dcac>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c067fea8>] (net_rx_action+0x138/0x2fc)
[19466.356568] [<c067fea8>] (net_rx_action) from [<c03021c8>] (__do_softirq+0xe0/0x268)
[19466.366196] [<c03021c8>] (__do_softirq) from [<c031f9e4>] (irq_exit+0x9c/0xfc)
[19466.374098] [<c031f9e4>] (irq_exit) from [<c035f11c>] (__handle_domain_irq+0x9c/0xac)
[19466.381133] [<c035f11c>] (__handle_domain_irq) from [<c0537490>] (gic_handle_irq+0x5c/0x90)
[19466.389024] [<c0537490>] (gic_handle_irq) from [<c0301a0c>] (__irq_svc+0x6c/0x90)
[19466.397160] Exception stack(0xc0b01f30 to 0xc0b01f78)
[19466.404806] 1f20:                                     00000000 021432f0 cfd752c4 c0313a40
[19466.409855] 1f40: ffffe000 c0b04c6c 00000001 c0b04cac c0a3bda8 00000000 00000000 00000000
[19466.418009] 1f60: 0f33b000 c0b01f80 c03082f4 c03082f8 60000013 ffffffff
[19466.426180] [<c0301a0c>] (__irq_svc) from [<c03082f8>] (arch_cpu_idle+0x34/0x38)
[19466.432613] [<c03082f8>] (arch_cpu_idle) from [<c0341ad4>] (do_idle+0xf8/0x208)
[19466.440248] [<c0341ad4>] (do_idle) from [<c0341e40>] (cpu_startup_entry+0x18/0x20)
[19466.447279] [<c0341e40>] (cpu_startup_entry) from [<c0a00d1c>] (start_kernel+0x428/0x438)
[19466.455036] ---[ end trace f7dc54ea57ea69af ]---
```

greearb, 2019-08-23:
If you are able to patch your ath10k-ct driver, please try running with this patch:

diff --git a/drivers/net/wireless/ath/ath10k/txrx.c b/drivers/net/wireless/ath/ath10k/txrx.c
index ae4b950572c5..617aa5008c02 100644
--- a/drivers/net/wireless/ath/ath10k/txrx.c
+++ b/drivers/net/wireless/ath/ath10k/txrx.c
@@ -117,6 +117,8 @@ static void ath10k_set_tx_rate_status(struct ath10k *ar,
                 rate->flags |= IEEE80211_TX_RC_VHT_MCS;
                 /* TODO-BEN:  Not sure this is correct. */
                 rate->idx = (nss << 4) | hw_rate;
+               WARN_ONCE(((hw_rate > 9) || (nss > 3)), "Invalid VHT rate, nss: %d  hw_rate: %d ratecode: %d\n",
+                         nss, hw_rate, tx_done->tx_rate_code);
         }

         if (tx_done->tx_rate_flags & ATH10K_RC_FLAG_40MHZ)

And let me know if you see more splats from this new WARN or the old one, or both.

Thanks,
Ben

jekkos, 2019-08-24:
Ok I applied the patch you suggested, seeing the following stack now

```
[38867.415235] ------------[ cut here ]------------
[38867.415630] WARNING: CPU: 0 PID: 0 at /home/jekkos/hacking/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/ath10k-ct-2019-06-13-f0aa8130/ath10k-4.19/txrx.c:132 ath10k_txrx_tx_unref+0x454/0x5d4 [ath10k_core]                                      
[38867.419032] Invalid VHT rate, nss: 3  hw_rate: 15 ratecode: 255
[38867.440245] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async iptable_nat ipt_MASQUERADE ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD rtlwifi pppox ppp_generic nf_nat_ipv4 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack
_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat nf_log_ipv6 nf_log_comm
on ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug                                                                                                                           
[38867.486861] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W         4.19.66 #0
[38867.508723] Hardware name: Generic DT based system
[38867.516406] [<c030e2c4>] (unwind_backtrace) from [<c030ad20>] (show_stack+0x10/0x14)
[38867.520995] [<c030ad20>] (show_stack) from [<c0782a30>] (dump_stack+0x88/0x9c)
[38867.528896] [<c0782a30>] (dump_stack) from [<c031b898>] (__warn+0xd4/0xf0)
[38867.535925] [<c031b898>] (__warn) from [<c031b8ec>] (warn_slowpath_fmt+0x38/0x48)
[38867.543431] [<c031b8ec>] (warn_slowpath_fmt) from [<bf2e15f8>] (ath10k_txrx_tx_unref+0x454/0x5d4 [ath10k_core])
[38867.551923] [<bf2e15f8>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf2dadd0>] (ath10k_htt_t2h_msg_handler+0x610/0x1fb8 [ath10k_core])
[38867.561240] [<bf2dadd0>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf32c274>] (ath10k_pci_htt_rx_cb+0x1c8/0x218 [ath10k_pci])
[38867.573368] [<bf32c274>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf2f7c00>] (ath10k_ce_per_engine_service+0x80/0xd8 [ath10k_core])
[38867.586072] [<bf2f7c00>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf2f7cc4>] (ath10k_ce_per_engine_service_any+0x6c/0x98 [ath10k_core])                                                                                                                                    
[38867.597514] [<bf2f7cc4>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf32dcac>] (ath10k_pci_napi_poll+0x64/0x100 [ath10k_pci])
[38867.609866] [<bf32dcac>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c067fea8>] (net_rx_action+0x138/0x2fc)
[38867.622138] [<c067fea8>] (net_rx_action) from [<c03021c8>] (__do_softirq+0xe0/0x268)
[38867.631755] [<c03021c8>] (__do_softirq) from [<c031f9e4>] (irq_exit+0x9c/0xfc)
[38867.639668] [<c031f9e4>] (irq_exit) from [<c035f11c>] (__handle_domain_irq+0x9c/0xac)
[38867.646709] [<c035f11c>] (__handle_domain_irq) from [<c0537490>] (gic_handle_irq+0x5c/0x90)
[38867.654584] [<c0537490>] (gic_handle_irq) from [<c0301a0c>] (__irq_svc+0x6c/0x90)
[38867.662639] Exception stack(0xc0b01f30 to 0xc0b01f78)
[38867.670191] 1f20:                                     00000000 026a6720 cfd752c4 c0313a40
[38867.675241] 1f40: ffffe000 c0b04c6c 00000001 c0b04cac c0a3bda8 00000000 00000000 00000000
[38867.683395] 1f60: 0f33b000 c0b01f80 c03082f4 c03082f8 60000013 ffffffff
[38867.691565] [<c0301a0c>] (__irq_svc) from [<c03082f8>] (arch_cpu_idle+0x34/0x38)
[38867.698204] [<c03082f8>] (arch_cpu_idle) from [<c0341ad4>] (do_idle+0xf8/0x208)
[38867.705838] [<c0341ad4>] (do_idle) from [<c0341e40>] (cpu_startup_entry+0x18/0x20)
[38867.712859] [<c0341e40>] (cpu_startup_entry) from [<c0a00d1c>] (start_kernel+0x428/0x438)
[38867.720571] ---[ end trace 22f7c3e6aa663dd5 ]---
[38867.728964] ------------[ cut here ]------------
```

jekkos, 2019-08-24:
@klukonin I'm still seeing my original stacktrace, even with the newer firmware (firmware-5-ct-full-htt-mgt-community-12.bin-lede.010). Is there anything else I can do here?

```
[38867.728964] ------------[ cut here ]------------
[38867.734430] WARNING: CPU: 0 PID: 0 at backports-5.3-rc4-1/net/wireless/util.c:1171 cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211]                                                                                                                                                  
[38867.737987] invalid rate bw=0, mcs=15, nss=4
[38867.749687] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async iptable_nat ipt_MASQUERADE ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD rtlwifi pppox ppp_generic nf_nat_ipv4 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack
_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat nf_log_ipv6 nf_log_comm
on ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug                                                                                                                           
[38867.794492] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W         4.19.66 #0
[38867.816537] Hardware name: Generic DT based system
[38867.824213] [<c030e2c4>] (unwind_backtrace) from [<c030ad20>] (show_stack+0x10/0x14)                                                                                                                                                                                                  
[38867.828804] [<c030ad20>] (show_stack) from [<c0782a30>] (dump_stack+0x88/0x9c)
[38867.836710] [<c0782a30>] (dump_stack) from [<c031b898>] (__warn+0xd4/0xf0)
[38867.843929] [<c031b898>] (__warn) from [<c031b8ec>] (warn_slowpath_fmt+0x38/0x48)
[38867.851717] [<c031b8ec>] (warn_slowpath_fmt) from [<bf13efbc>] (cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211])
[38867.860527] [<bf13efbc>] (cfg80211_calculate_bitrate [cfg80211]) from [<bf1de8f8>] (ieee80211s_update_metric+0x78/0x9c [mac80211])
[38867.870812] [<bf1de8f8>] (ieee80211s_update_metric [mac80211]) from [<bf191954>] (ieee80211_tx_monitor+0x768/0xb4c [mac80211])
[38867.882398] [<bf191954>] (ieee80211_tx_monitor [mac80211]) from [<bf191dc0>] (ieee80211_tx_status+0x88/0x90 [mac80211])
[38867.893557] [<bf191dc0>] (ieee80211_tx_status [mac80211]) from [<bf2e172c>] (ath10k_txrx_tx_unref+0x588/0x5d4 [ath10k_core])
[38867.903751] [<bf2e172c>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf2dadd0>] (ath10k_htt_t2h_msg_handler+0x610/0x1fb8 [ath10k_core])
[38867.914627] [<bf2dadd0>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf32c274>] (ath10k_pci_htt_rx_cb+0x1c8/0x218 [ath10k_pci])
[38867.926498] [<bf32c274>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf2f7c00>] (ath10k_ce_per_engine_service+0x80/0xd8 [ath10k_core])
[38867.939202] [<bf2f7c00>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf2f7cc4>] (ath10k_ce_per_engine_service_any+0x6c/0x98 [ath10k_core])                                                                                                                                    
[38867.950633] [<bf2f7cc4>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf32dcac>] (ath10k_pci_napi_poll+0x64/0x100 [ath10k_pci])
[38867.962987] [<bf32dcac>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c067fea8>] (net_rx_action+0x138/0x2fc)
[38867.975270] [<c067fea8>] (net_rx_action) from [<c03021c8>] (__do_softirq+0xe0/0x268)
[38867.984880] [<c03021c8>] (__do_softirq) from [<c031f9e4>] (irq_exit+0x9c/0xfc)
[38867.992793] [<c031f9e4>] (irq_exit) from [<c035f11c>] (__handle_domain_irq+0x9c/0xac)
[38867.999833] [<c035f11c>] (__handle_domain_irq) from [<c0537490>] (gic_handle_irq+0x5c/0x90)
[38868.007711] [<c0537490>] (gic_handle_irq) from [<c0301a0c>] (__irq_svc+0x6c/0x90)
[38868.015768] Exception stack(0xc0b01f30 to 0xc0b01f78)
[38868.023434] 1f20:                                     00000000 026a6720 cfd752c4 c0313a40
[38868.028532] 1f40: ffffe000 c0b04c6c 00000001 c0b04cac c0a3bda8 00000000 00000000 00000000
[38868.036671] 1f60: 0f33b000 c0b01f80 c03082f4 c03082f8 60000013 ffffffff
[38868.044875] [<c0301a0c>] (__irq_svc) from [<c03082f8>] (arch_cpu_idle+0x34/0x38)
[38868.051331] [<c03082f8>] (arch_cpu_idle) from [<c0341ad4>] (do_idle+0xf8/0x208)
[38868.058963] [<c0341ad4>] (do_idle) from [<c0341e40>] (cpu_startup_entry+0x18/0x20)
[38868.065984] [<c0341e40>] (cpu_startup_entry) from [<c0a00d1c>] (start_kernel+0x428/0x438)
[38868.073805] ---[ end trace 22f7c3e6aa663dd6 ]---
[38868.086144] ath10k_ahb a800000.wifi: SWBA overrun on vdev 0, skipped old beacon
```

greearb, 2019-08-25:
Please add this patch as well:

[greearb@ben-dt4 ath10k]$ git diff
diff --git a/drivers/net/wireless/ath/ath10k/htt_rx.c b/drivers/net/wireless/ath/ath10k/htt_rx.c
index 9ad21f3bb1d5..489199b74c51 100644
--- a/drivers/net/wireless/ath/ath10k/htt_rx.c
+++ b/drivers/net/wireless/ath/ath10k/htt_rx.c
@@ -2960,7 +2960,7 @@ bool ath10k_htt_t2h_msg_handler(struct ath10k *ar, struct sk_buff *skb)
                 break;
         }
         case HTT_T2H_MSG_TYPE_MGMT_TX_COMPLETION: {
-               struct htt_tx_done tx_done = {};
+               struct htt_tx_done tx_done = {0};
                 int status = __le32_to_cpu(resp->mgmt_tx_completion.status);
                 int info = __le32_to_cpu(resp->mgmt_tx_completion.info);

Thanks,
Ben

jekkos, 2019-08-25:
Ok patch applied, so far no stacktraces. Let's wait and evaluate.

greearb, 2019-08-27:
> Ok got two more stacks here, not sure if it's the same thing as before or not. I just thought they could be helpful?
> |38693.476053] ------------[ cut here ]------------ [38693.477431] WARNING: CPU: 0 PID: 0 at
> /home/jekkos/hacking/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/ath10k-ct-2019-08-14-9e5ab250/ath10k-4.19/txrx.c:132
> ath10k_txrx_tx_unref+0x454/0x5d4 [ath10k_core] [38693.480158] Invalid VHT rate, nss: 3 hw_rate: 15 ratecode: 255 [38693.501176] Modules linked in: rtl8192cu
> xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug [38693.547696] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.19.66 #0 [38693.569648] Hardware
> name: Generic DT based system [38693.575946] [<c030e2c4>] (unwind_backtrace) from [<c030ad20>] (show_stack+0x10/0x14) [38693.580535] [<c030ad20>] (show_stack)
> from [<c0782a30>] (dump_stack+0x88/0x9c) [38693.588627] [<c0782a30>] (dump_stack) from [<c031b898>] (__warn+0xd4/0xf0) [38693.595660] [<c031b898>] (__warn) from
> [<c031b8ec>] (warn_slowpath_fmt+0x38/0x48) [38693.603310] [<c031b8ec>] (warn_slowpath_fmt) from [<bf2e15f8>] (ath10k_txrx_tx_unref+0x454/0x5d4 [ath10k_core])
> [38693.611596] [<bf2e15f8>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf2dadd0>] (ath10k_htt_t2h_msg_handler+0x610/0x1fb8 [ath10k_core]) [38693.620799]
> [<bf2dadd0>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf32c274>] (ath10k_pci_htt_rx_cb+0x1c8/0x218 [ath10k_pci]) [38693.633028] [<bf32c274>]
> (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf2f7c00>] (ath10k_ce_per_engine_service+0x80/0xd8 [ath10k_core]) [38693.645633] [<bf2f7c00>]
> [... cut ...]
> c03082f4 c03082f8 60000013 ffffffff [68497.708206] [<c0301a0c>] (__irq_svc) from [<c03082f8>] (arch_cpu_idle+0x34/0x38) [68497.714662] [<c03082f8>]
> (arch_cpu_idle) from [<c0341ad4>] (do_idle+0xf8/0x208) [68497.722295] [<c0341ad4>] (do_idle) from [<c0341e40>] (cpu_startup_entry+0x18/0x20) [68497.729314]
> [<c0341e40>] (cpu_startup_entry) from [<c0a00d1c>] (start_kernel+0x428/0x438) [68497.737147] ---[ end trace 61e07c5fdd9e0cf4 ]--- |

Is this running this patch below?  If so, then I guess it needs more work, but I think
that patch will fix this problem.

[greearb@ben-dt4 ath10k]$ git diff
diff --git a/drivers/net/wireless/ath/ath10k/htt_rx.c b/drivers/net/wireless/ath/ath10k/htt_rx.c
index 9ad21f3bb1d5..489199b74c51 100644
--- a/drivers/net/wireless/ath/ath10k/htt_rx.c
+++ b/drivers/net/wireless/ath/ath10k/htt_rx.c
@@ -2960,7 +2960,7 @@ bool ath10k_htt_t2h_msg_handler(struct ath10k *ar, struct sk_buff *skb)
                 break;
         }
         case HTT_T2H_MSG_TYPE_MGMT_TX_COMPLETION: {
-               struct htt_tx_done tx_done = {};
+               struct htt_tx_done tx_done = {0};
                 int status = __le32_to_cpu(resp->mgmt_tx_completion.status);
                 int info = __le32_to_cpu(resp->mgmt_tx_completion.info);

Thanks,
Ben

jekkos, 2019-08-27:
Ok got two more stacks here, not sure if it's the same thing as before or not. I just thought they could be helpful?

```
38693.476053] ------------[ cut here ]------------
[38693.477431] WARNING: CPU: 0 PID: 0 at /home/jekkos/hacking/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/ath10k-ct-2019-08-14-9e5ab250/ath10k-4.19/txrx.c:132 ath10k_txrx_tx_unref+0x454/0x5d4 [ath10k_core]                                      
[38693.480158] Invalid VHT rate, nss: 3  hw_rate: 15 ratecode: 255
[38693.501176] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async iptable_nat ipt_MASQUERADE ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD rtlwifi pppox ppp_generic nf_nat_ipv4 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug                                                                                                                           
[38693.547696] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.19.66 #0
[38693.569648] Hardware name: Generic DT based system
[38693.575946] [<c030e2c4>] (unwind_backtrace) from [<c030ad20>] (show_stack+0x10/0x14)
[38693.580535] [<c030ad20>] (show_stack) from [<c0782a30>] (dump_stack+0x88/0x9c)
[38693.588627] [<c0782a30>] (dump_stack) from [<c031b898>] (__warn+0xd4/0xf0)
[38693.595660] [<c031b898>] (__warn) from [<c031b8ec>] (warn_slowpath_fmt+0x38/0x48)
[38693.603310] [<c031b8ec>] (warn_slowpath_fmt) from [<bf2e15f8>] (ath10k_txrx_tx_unref+0x454/0x5d4 [ath10k_core])
[38693.611596] [<bf2e15f8>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf2dadd0>] (ath10k_htt_t2h_msg_handler+0x610/0x1fb8 [ath10k_core])
[38693.620799] [<bf2dadd0>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf32c274>] (ath10k_pci_htt_rx_cb+0x1c8/0x218 [ath10k_pci])
[38693.633028] [<bf32c274>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf2f7c00>] (ath10k_ce_per_engine_service+0x80/0xd8 [ath10k_core])
[38693.645633] [<bf2f7c00>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf2f7cc4>] (ath10k_ce_per_engine_service_any+0x6c/0x98 [ath10k_core])
[38693.657051] [<bf2f7cc4>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf32dcac>] (ath10k_pci_napi_poll+0x64/0x100 [ath10k_pci])
[38693.669404] [<bf32dcac>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c067fea8>] (net_rx_action+0x138/0x2fc)
[38693.681682] [<c067fea8>] (net_rx_action) from [<c03021c8>] (__do_softirq+0xe0/0x268)
[38693.691293] [<c03021c8>] (__do_softirq) from [<c031f9e4>] (irq_exit+0x9c/0xfc)
[38693.699206] [<c031f9e4>] (irq_exit) from [<c035f11c>] (__handle_domain_irq+0x9c/0xac)
[38693.706252] [<c035f11c>] (__handle_domain_irq) from [<c0537490>] (gic_handle_irq+0x5c/0x90)
[38693.714122] [<c0537490>] (gic_handle_irq) from [<c0301a0c>] (__irq_svc+0x6c/0x90)
[38693.722181] Exception stack(0xc0b01f30 to 0xc0b01f78)
[38693.729849] 1f20:                                     00000000 023c0948 cfd752c4 c0313a40
[38693.734949] 1f40: ffffe000 c0b04c6c 00000001 c0b04cac c0a3bda8 00000000 00000000 00000000
[38693.743085] 1f60: 0f33b000 c0b01f80 c03082f4 c03082f8 60000013 ffffffff
[38693.751309] [<c0301a0c>] (__irq_svc) from [<c03082f8>] (arch_cpu_idle+0x34/0x38)
[38693.757745] [<c03082f8>] (arch_cpu_idle) from [<c0341ad4>] (do_idle+0xf8/0x208)
[38693.765377] [<c0341ad4>] (do_idle) from [<c0341e40>] (cpu_startup_entry+0x18/0x20)
[38693.772399] [<c0341e40>] (cpu_startup_entry) from [<c0a00d1c>] (start_kernel+0x428/0x438)
[38693.780203] ---[ end trace 61e07c5fdd9e0cf3 ]---
[68497.397401] ------------[ cut here ]------------
[68497.397817] WARNING: CPU: 0 PID: 0 at backports-5.3-rc4-1/net/wireless/util.c:1171 cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211]
[68497.401181] invalid rate bw=0, mcs=15, nss=4
[68497.413035] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async iptable_nat ipt_MASQUERADE ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD rtlwifi pppox ppp_generic nf_nat_ipv4 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack
_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat nf_log_ipv6 nf_log_comm
on ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug
[68497.457943] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W         4.19.66 #0
[68497.479852] Hardware name: Generic DT based system
[68497.487513] [<c030e2c4>] (unwind_backtrace) from [<c030ad20>] (show_stack+0x10/0x14)
[68497.492317] [<c030ad20>] (show_stack) from [<c0782a30>] (dump_stack+0x88/0x9c)
[68497.500235] [<c0782a30>] (dump_stack) from [<c031b898>] (__warn+0xd4/0xf0)
[68497.507262] [<c031b898>] (__warn) from [<c031b8ec>] (warn_slowpath_fmt+0x38/0x48)
[68497.515059] [<c031b8ec>] (warn_slowpath_fmt) from [<bf13efbc>] (cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211])
[... cut ...]
[68497.691867] 1f40: ffffe000 c0b04c6c 00000001 c0b04cac c0a3bda8 00000000 00000000 00000000
[68497.700003] 1f60: 0f33b000 c0b01f80 c03082f4 c03082f8 60000013 ffffffff
[68497.708206] [<c0301a0c>] (__irq_svc) from [<c03082f8>] (arch_cpu_idle+0x34/0x38)
[68497.714662] [<c03082f8>] (arch_cpu_idle) from [<c0341ad4>] (do_idle+0xf8/0x208)
[68497.722295] [<c0341ad4>] (do_idle) from [<c0341e40>] (cpu_startup_entry+0x18/0x20)
[68497.729314] [<c0341e40>] (cpu_startup_entry) from [<c0a00d1c>] (start_kernel+0x428/0x438)
[68497.737147] ---[ end trace 61e07c5fdd9e0cf4 ]---
```

jekkos, 2019-08-27:
Yeah I do think I patched it. To make sure I'll build again from your latest commit here.

jekkos, 2019-08-28:
Ok I have rebuilt with latest HEAD of this repo + firmwares

still seeing this one appear

```
[44246.150403] ------------[ cut here ]------------
[44246.150779] WARNING: CPU: 0 PID: 0 at backports-5.3-rc4-1/net/wireless/util.c:1171 cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211]
[44246.154145] invalid rate bw=0, mcs=15, nss=4
[44246.166111] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async iptable_nat ipt_MASQUERADE ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD rtlwifi pppox ppp_generic nf_nat_ipv4 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug                                                                                                                           
[44246.210663] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.19.67 #0
[44246.232846] Hardware name: Generic DT based system
[44246.239361] [<c030e2c4>] (unwind_backtrace) from [<c030ad20>] (show_stack+0x10/0x14)
[44246.243930] [<c030ad20>] (show_stack) from [<c07828d0>] (dump_stack+0x88/0x9c)
[44246.251843] [<c07828d0>] (dump_stack) from [<c031b898>] (__warn+0xd4/0xf0)
[44246.258877] [<c031b898>] (__warn) from [<c031b8ec>] (warn_slowpath_fmt+0x38/0x48)
[44246.266665] [<c031b8ec>] (warn_slowpath_fmt) from [<bf13efbc>] (cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211])
[44246.275460] [<bf13efbc>] (cfg80211_calculate_bitrate [cfg80211]) from [<bf1de8f8>] (ieee80211s_update_metric+0x78/0x9c [mac80211])
[44246.285742] [<bf1de8f8>] (ieee80211s_update_metric [mac80211]) from [<bf191954>] (ieee80211_tx_monitor+0x768/0xb4c [mac80211])
[44246.297333] [<bf191954>] (ieee80211_tx_monitor [mac80211]) from [<bf191dc0>] (ieee80211_tx_status+0x88/0x90 [mac80211])
[44246.308514] [<bf191dc0>] (ieee80211_tx_status [mac80211]) from [<bf2e1708>] (ath10k_txrx_tx_unref+0x540/0x58c [ath10k_core])
[44246.318811] [<bf2e1708>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf2dadd0>] (ath10k_htt_t2h_msg_handler+0x610/0x1fdc [ath10k_core])
[44246.329574] [<bf2dadd0>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf32c274>] (ath10k_pci_htt_rx_cb+0x1c8/0x218 [ath10k_pci])
[44246.341551] [<bf32c274>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf2f7bdc>] (ath10k_ce_per_engine_service+0x80/0xd8 [ath10k_core])
[44246.354167] [<bf2f7bdc>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf2f7ca0>] (ath10k_ce_per_engine_service_any+0x6c/0x98 [ath10k_core])                                                                                                                                    
[44246.365591] [<bf2f7ca0>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf32dcac>] (ath10k_pci_napi_poll+0x64/0x100 [ath10k_pci])                                                                                                                                            
[44246.377929] [<bf32dcac>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c067fd48>] (net_rx_action+0x138/0x2fc)
[44246.390205] [<c067fd48>] (net_rx_action) from [<c03021c8>] (__do_softirq+0xe0/0x268)
[44246.399823] [<c03021c8>] (__do_softirq) from [<c031f9e4>] (irq_exit+0x9c/0xfc)
[44246.407736] [<c031f9e4>] (irq_exit) from [<c035f11c>] (__handle_domain_irq+0x9c/0xac)
[44246.414774] [<c035f11c>] (__handle_domain_irq) from [<c0537320>] (gic_handle_irq+0x5c/0x90)
[44246.422649] [<c0537320>] (gic_handle_irq) from [<c0301a0c>] (__irq_svc+0x6c/0x90)
[44246.430708] Exception stack(0xc0b01f30 to 0xc0b01f78)
[44246.438377] 1f20:                                     00000000 0302266c cfd752c4 c0313a40
[44246.443476] 1f40: ffffe000 c0b04c6c 00000001 c0b04cac c0a3bda8 00000000 00000000 00000000
[44246.451611] 1f60: 0f33b000 c0b01f80 c03082f4 c03082f8 60000013 ffffffff
[44246.459824] [<c0301a0c>] (__irq_svc) from [<c03082f8>] (arch_cpu_idle+0x34/0x38)
[44246.466272] [<c03082f8>] (arch_cpu_idle) from [<c0341ad4>] (do_idle+0xf8/0x208)
[44246.473907] [<c0341ad4>] (do_idle) from [<c0341e40>] (cpu_startup_entry+0x18/0x20)
[44246.480923] [<c0341e40>] (cpu_startup_entry) from [<c0a00d1c>] (start_kernel+0x428/0x438)
[44246.488761] ---[ end trace d0af01c37206e51b ]---
```

greearb, 2019-08-28:
Can you check the ath10k-ct source code you compiled to make sure it has the patch
applied?  You may have to change more than just the md5sum in the ath10k-ct openwrt
make file to get it to re-compile the new code.

Thanks,
Ben

jekkos, 2019-08-31:
Yeah the changes are in there, I checked with previous build and then cleared all dirs again to make sure.

greearb, 2019-09-16:
I am pretty sure that this is already fixed, and it has been pushed to openwrt.  If you can reproduce with a clean build of upstream openwrt, please re-open.

jekkos, 2019-09-25:
Ok I'm running latest openwrt but still seeing this issue in fact. I want to make sure I have ath10k-ct loaded properly. Does it show differently in lsmod? I currently just see

``
ath                    24576  1 ath10k_core
ath10k_core           364544  1 ath10k_pci
ath10k_pci             40960  0
cfg80211              258048  4 ath10k_core,ath,rtlwifi,mac80211
hwmon                  16384  1 ath10k_core
```

maybe ath10k-ct presents itself the same way?

Anyway this is the stacktrace on latest master

```
[33983.320247] ------------[ cut here ]------------
[33983.321749] WARNING: CPU: 0 PID: 0 at backports-5.3-rc4-1/net/wireless/util.c:1171 cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211]
[33983.324273] invalid rate bw=0, mcs=15, nss=4
[33983.336131] Modules linked in: rtl8192cu rtl8192c_common rtl_usb pppoe ppp_async iptable_nat ipt_MASQUERADE ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD rtlwifi pppox ppp_generic nf_nat_ipv4 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug                                                                                                                           
[33983.380824] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 4.19.72 #0
[33983.402808] Hardware name: Generic DT based system
[33983.409110] [<c030e2c4>] (unwind_backtrace) from [<c030ad20>] (show_stack+0x10/0x14)
[33983.413697] [<c030ad20>] (show_stack) from [<c0782c10>] (dump_stack+0x88/0x9c)
[33983.421794] [<c0782c10>] (dump_stack) from [<c031b898>] (__warn+0xd4/0xf0)
[33983.428820] [<c031b898>] (__warn) from [<c031b8ec>] (warn_slowpath_fmt+0x38/0x48)
[33983.436626] [<c031b8ec>] (warn_slowpath_fmt) from [<bf13ffbc>] (cfg80211_calculate_bitrate+0x1d8/0x24c [cfg80211])
[33983.445421] [<bf13ffbc>] (cfg80211_calculate_bitrate [cfg80211]) from [<bf1df8e8>] (ieee80211s_update_metric+0x78/0x9c [mac80211])
[33983.455700] [<bf1df8e8>] (ieee80211s_update_metric [mac80211]) from [<bf192954>] (ieee80211_tx_monitor+0x768/0xb4c [mac80211])
[33983.467287] [<bf192954>] (ieee80211_tx_monitor [mac80211]) from [<bf192dc0>] (ieee80211_tx_status+0x88/0x90 [mac80211])
[33983.478463] [<bf192dc0>] (ieee80211_tx_status [mac80211]) from [<bf2e275c>] (ath10k_txrx_tx_unref+0x540/0x58c [ath10k_core])
[33983.488757] [<bf2e275c>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf2dbe24>] (ath10k_htt_t2h_msg_handler+0x610/0x1fdc [ath10k_core])
[33983.499523] [<bf2dbe24>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf32cfb4>] (ath10k_pci_htt_rx_cb+0x1c8/0x218 [ath10k_pci])
[33983.511510] [<bf32cfb4>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf2f8c88>] (ath10k_ce_per_engine_service+0x80/0xd8 [ath10k_core])
[33983.524112] [<bf2f8c88>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf2f8d4c>] (ath10k_ce_per_engine_service_any+0x6c/0x98 [ath10k_core])                                                                                                                                    
[33983.535534] [<bf2f8d4c>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf32ecac>] (ath10k_pci_napi_poll+0x64/0x100 [ath10k_pci])                                                                                                                                            
[33983.547883] [<bf32ecac>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c067fe20>] (net_rx_action+0x138/0x2fc)
[33983.560159] [<c067fe20>] (net_rx_action) from [<c03021c8>] (__do_softirq+0xe0/0x268)
[33983.569767] [<c03021c8>] (__do_softirq) from [<c031f9e4>] (irq_exit+0x9c/0xfc)
[33983.577682] [<c031f9e4>] (irq_exit) from [<c035f130>] (__handle_domain_irq+0x9c/0xac)
[33983.584725] [<c035f130>] (__handle_domain_irq) from [<c0537488>] (gic_handle_irq+0x5c/0x90)
[33983.592597] [<c0537488>] (gic_handle_irq) from [<c0301a0c>] (__irq_svc+0x6c/0x90)
[33983.600656] Exception stack(0xc0b01f30 to 0xc0b01f78)
[33983.608326] 1f20:                                     00000000 0378fdd8 cfd752c4 c0313a40
[33983.613421] 1f40: ffffe000 c0b04c6c 00000001 c0b04cac c0a3bda8 00000000 00000000 00000000
[33983.621557] 1f60: 0f33b000 c0b01f80 c03082f4 c03082f8 60000013 ffffffff
[33983.629772] [<c0301a0c>] (__irq_svc) from [<c03082f8>] (arch_cpu_idle+0x34/0x38)
[33983.636221] [<c03082f8>] (arch_cpu_idle) from [<c0341ad4>] (do_idle+0xf8/0x208)
[33983.643854] [<c0341ad4>] (do_idle) from [<c0341e40>] (cpu_startup_entry+0x18/0x20)
[33983.650874] [<c0341e40>] (cpu_startup_entry) from [<c0a00d18>] (start_kernel+0x424/0x434)
[33983.658772] ---[ end trace 5ad4bea2af79501e ]---
[176815.726938] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 1
[214794.163289] Rekeying PTK for STA 28:5a:eb:d0:20:a4 but driver can't safely do that.
[219325.759616] Rekeying PTK for STA 28:5a:eb:d0:20:a4 but driver can't safely do that.
[220605.762388] Rekeying PTK for STA 28:5a:eb:d0:20:a4 but driver can't safely do that.
[341636.800231] Rekeying PTK for STA d8:1c:79:e3:29:f1 but driver can't safely do that.
[344096.517549] Rekeying PTK for STA d8:1c:79:e3:29:f1 but driver can't safely do that.
[344176.799090] Rekeying PTK for STA d8:1c:79:e3:29:f1 but driver can't safely do that.
[378788.177312] Rekeying PTK for STA d8:1c:79:e3:29:f1 but driver can't safely do that.
[427671.412599] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 2
[474250.300934] ath10k_ahb a000000.wifi: peer-unmap-event: unknown peer id 1
```

greearb, 2019-09-25:
Can you please tar up your ath10k-ct source code from your build tree and attach that or send it to greearb@candelatech.com?

jekkos, 2019-09-27:
Ok you should have it in your mailbox

closed 2020-04-30 by greearb