greearb/ath10k-ct #91: crashes on R7800

timkgh, 2019-08-24:
Netgear R7800 hnyman's build r10829-91c84e87c2
Not sure whether these traces help uncover issues in the `-ct` firmware
[dmesg.txt](https://github.com/greearb/ath10k-ct/files/3537499/dmesg.txt)

timkgh, 2019-10-27:
And another one R7800 hnyman's build r11329-b9d58f7e06
[dmesg.txt](https://github.com/greearb/ath10k-ct/files/3776011/dmesg.txt)

greearb, 2019-10-27:
That is not a firmware crash, and is not a kernel crash.  It is a warning about an invalid rate which should be harmless.  There could of course be problems with your system, but at least there are no obvious serious problems in the dmesg.

timkgh, 2019-10-27:
Thanks for looking. Could the warnings be printed in a less scary way that does not look like a crash?

greearb, 2019-10-29:
It does say WARNING.  Real kernel/driver crashes will say BUG in them, and firmware crashes look different and have register dumps and such.  I also think I have already fixed at least one root cause of this warning, but either way, it should be at least mostly harmless.  Closing this bug, if you have instability issues or see other crashes or kernel splats, please open a new bug with details on that.

closed 2019-10-29 by greearb