greearb/ath10k-ct #92: QCA9980, r7500v2, kernel 4.19, ath10k-ct-htt firmware version 10.4b-ct-9980-fH-012-c7b0f3f9.  ath10k-ct warning

notmyrealhandle, 2019-09-01:
Just a warning.  dmesg output [here](https://gist.github.com/notmyrealhandle/fc1df4d5f6d7344480913b6cf2b48448).  

Also an unrelated question to the warning but possibly related to the ath10k driver/firmware (feel free not to respond if your too busy or uninterested...).  

From your knowledge of the ath10k-ct driver and firmware, could the following change to the dtsi for thermal sensors cause (difficult to reproduce) issues with the ct driver/firmware?

```
--- a/target/linux/ipq806x/files-4.19/arch/arm/boot/dts/qcom-ipq8064.dtsi
+++ b/target/linux/ipq806x/files-4.19/arch/arm/boot/dts/qcom-ipq8064.dtsi
@@ -956,9 +956,9 @@
 			#power-domain-cells = <1>;
 		};
 
-		tsens: thermal-sensor@900000 {
+		tsens: thermal-sensor@904000 {
 			compatible = "qcom,ipq8064-tsens";
-			reg = <0x900000 0x3680>;
+			reg = <0x904000 0x3680>;
 			nvmem-cells = <&tsens_calib>, <&tsens_backup>;
```

So far, this the only way I can get thermal sensors working on kernel 4.19 on my r7500v2 (on one build up 17+ days with no issues).  It works for me; however, initially I observed non reproducible (and therefore possibly unrelated) trouble with wireless.  I also saw 5 GHz wireless go out after 5 days after the 17+ days of no trouble.  

A r7800 user also reported a non booting image using this change.  A non booting image is the number one symptom I get when trying other approaches to get thermal sensors working on 4.19.

notmyrealhandle, 2019-09-02:
I rebuilt an image and I no longer see this warning.  The warning was consistent on startup on the prior image; however, I've had all kinds of build issues the past few days like git silently applied half a patch.  I'll leave this open for a day or two and then close it if doesn't come back.  

I'm still  curious about the other question tho...

greearb, 2019-09-03:
I don't know about the thermal sensor question.

The ath10k radios themselves can report thermal sensor for the rx chains, btw.

notmyrealhandle, 2019-09-03:
Thank you for the response.  I have not seen the warning message after a trying out a few new images so I'm going to attribute it to being self inflicted and close this for now.

closed 2019-09-03 by notmyrealhandle