greearb/ath10k-ct #93: IPQ4019 firmware crashes

matejkupljen, 2019-09-04:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).

We noticed a firmware crash where we have more stations connected to the AP, more then 80 on both radios.
I do not have exact procedure to reproduce it, just when many STAs connect to this AP, we see the crash.
Please note, that I used fwcfg-ahb-xxxxxx.wifi.txt to change the default parameters. The contents is the same for 2.4GHz and for 5GHz radio and it looks like:
> cat /lib/firmware/ath10k/fwcfg-ahb-a800000.wifi.txt
vdevs = 8
peers = 64
active_peers = 64
stations = 128

Also when I enabled debug output trough sysfs, we saw a lot of debug messages in the log, however in that case we couldn't crash the firmware.

Software (OS, Firmware version, kernel, driver, etc)

Custom OpenWRT OS, with the latest ath10k-ct driver (as available in OpenWRT OS, driver from 14th August 2019) and latest CT firmware.

> uname -a
Linux OpenWrt 4.14.71 #0 SMP Fri Aug 23 11:26:36 2019 armv7l GNU/Linux

Hardware (NIC chipset, platform, etc)

Yuncore XD5200 AP with IPQ4019.

Logs (dmesg, maybe supplicant and/or hostap)

[dmesg-xd5200.txt](https://github.com/greearb/ath10k-ct/files/3573081/dmesg-xd5200.txt) output is attached.

Here is the output when the firmware crashed:

[ 2830.546110] ath10k_ahb a000000.wifi: firmware crashed! (guid n/a)
[ 2830.546798] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[ 2830.551432] ath10k_ahb a000000.wifi: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[ 2830.564045] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-012-c7b0f3f98 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 b27a5aa8
[ 2830.578935] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:20 crc32 e2dfaa91
[ 2830.597427] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 128 raw 0 hwcrypto 1
[ 2830.605903] ath10k_ahb a000000.wifi: firmware register dump:
[ 2830.615025] ath10k_ahb a000000.wifi: [00]: 0x0000000B 0x00000000 0x00971534 0x00000000
[ 2830.620667] ath10k_ahb a000000.wifi: [04]: 0x00000000 0x00060124 0x00000000 0x00000000
[ 2830.628371] ath10k_ahb a000000.wifi: [08]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.636286] ath10k_ahb a000000.wifi: [12]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.644186] ath10k_ahb a000000.wifi: [16]: 0x009C1FE3 0x009606B6 0x00960DD2 0x00971534
[ 2830.652086] ath10k_ahb a000000.wifi: [20]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.659983] ath10k_ahb a000000.wifi: [24]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.667866] ath10k_ahb a000000.wifi: [28]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.675784] ath10k_ahb a000000.wifi: [32]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.683682] ath10k_ahb a000000.wifi: [36]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.691583] ath10k_ahb a000000.wifi: [40]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.699478] ath10k_ahb a000000.wifi: [44]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.707362] ath10k_ahb a000000.wifi: [48]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.715279] ath10k_ahb a000000.wifi: [52]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.723177] ath10k_ahb a000000.wifi: [56]: 0x00000000 0x00000000 0x00000000 0x00000000
[ 2830.731077] ath10k_ahb a000000.wifi: Copy Engine register dump:
[ 2830.738962] ath10k_ahb a000000.wifi: [00]: 0x0004a000   2   2   3   3
[ 2830.744792] ath10k_ahb a000000.wifi: [01]: 0x0004a400  20  20 122 123
[ 2830.751390] ath10k_ahb a000000.wifi: [02]: 0x0004a800  26  26  24  26
[ 2830.757798] ath10k_ahb a000000.wifi: [03]: 0x0004ac00   1   1   5   1
[ 2830.764239] ath10k_ahb a000000.wifi: [04]: 0x0004b000 5570 5570  12 231
[ 2830.770660] ath10k_ahb a000000.wifi: [05]: 0x0004b400   2   2   0   2
[ 2830.777068] ath10k_ahb a000000.wifi: [06]: 0x0004b800  21  21  21  21
[ 2830.783683] ath10k_ahb a000000.wifi: [07]: 0x0004bc00   1   1   1   1
[ 2830.790106] ath10k_ahb a000000.wifi: [08]: 0x0004c000   0   0 127   0
[ 2830.796513] ath10k_ahb a000000.wifi: [09]: 0x0004c400   1   1   1   1
[ 2830.802965] ath10k_ahb a000000.wifi: [10]: 0x0004c800   0   0   0   0
[ 2830.809383] ath10k_ahb a000000.wifi: [11]: 0x0004cc00   0   0   0   0
[ 2830.816811] ath10k_ahb a000000.wifi: debug log header, dbuf: 0x418810  dropped: 0
[ 2830.823248] ath10k_ahb a000000.wifi: [0] next: 0x418828 buf: 0x414c20 sz: 1500 len: 0 count: 0 free: 0
[... 151 lines cut ...]
[ 3011.087722] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer daf97200 vdev: 0 addr: 44:d1:fa:5e:02:ef
[ 3011.098665] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dd487200 vdev: 0 addr: 44:d1:fa:5e:02:2b
[ 3011.109589] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dd36ac00 vdev: 0 addr: 44:d1:fa:5e:02:cf
[ 3011.120528] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dd487000 vdev: 0 addr: 44:d1:fa:5e:02:d3
[ 3011.131461] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dd487600 vdev: 0 addr: 44:d1:fa:5e:01:47
[ 3011.142402] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dd487c00 vdev: 0 addr: 44:d1:fa:5e:01:73
[ 3011.153354] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dd486a00 vdev: 0 addr: 44:d1:fa:5e:02:4b
[ 3011.164288] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer db29b600 vdev: 0 addr: 44:d1:fa:5e:02:7b
[ 3011.175232] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer db29aa00 vdev: 0 addr: 44:d1:fa:5e:02:e3
[ 3011.186154] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dd232800 vdev: 0 addr: f4:0f:24:3a:24:9a
[ 3011.197088] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dd54b400 vdev: 0 addr: 88:dc:96:6f:1f:41
[ 3011.208031] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dd54a400 vdev: 0 addr: 88:dc:96:6f:1f:3f
[ 3011.218972] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer de016a00 vdev: 3 addr: 44:d1:fa:5e:02:5f
[ 3011.229904] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dd965a00 vdev: 2 addr: 4e:d1:fa:5e:02:5f
[ 3011.240841] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dd77ec00 vdev: 1 addr: 42:d1:fa:5e:02:5f
[ 3011.251855] ath10k_ahb a800000.wifi: removing peer, cleanup-all, deleting: peer dc669600 vdev: 0 addr: 46:d1:fa:5e:02:5f
[ 3011.270749] ieee80211 phy3: Hardware restart was requested
[ 3012.750684] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 8  peers: 64  tid: 96
[ 3012.751021] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[ 3012.802119] ath10k_ahb a800000.wifi: wmi print 'P 64/64 V 8 K 192 PH 208 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[ 3012.802532] ath10k_ahb a800000.wifi: wmi print 'free: 54344 iram: 23480 sram: 32824'
[ 3013.128930] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[ 3014.191429] ath10k_ahb a800000.wifi: device successfully recovered

If you need anything else, I am happy to provide.

klukonin, 2019-09-04:
Hello.

Possibly you have wring configuration.

Please try
cat /lib/firmware/ath10k/fwcfg-ahb-a800000.wifi.txt
vdevs = 4
peers = 128
active_peers = 64
stations = 127

matejkupljen, 2019-09-04:
Hello,

thanks for the info, but if I use those values the WiFi doesn't come up:

[   14.032621] ath10k_ahb a000000.wifi: fwcfg key: vdevs  val: 4
[   14.032668] ath10k_ahb a000000.wifi: fwcfg key: peers  val: 128
[   14.037384] ath10k_ahb a000000.wifi: fwcfg key: active_peers  val: 64
[   14.043221] ath10k_ahb a000000.wifi: fwcfg key: stations  val: 127
[   14.059483] ath10k_ahb a000000.wifi: Direct firmware load for ath10k/QCA4019/hw1.0/ct-firmware-5.bin failed with error -2
[   14.059530] ath10k_ahb a000000.wifi: Falling back to user helper
[   15.347615] firmware ath10k!QCA4019!hw1.0!ct-firmware-5.bin: firmware_loading_store: map pages failed
[   15.348021] ath10k_ahb a000000.wifi: Direct firmware load for ath10k/QCA4019/hw1.0/ct-firmware-2.bin failed with error -2
[   15.355985] ath10k_ahb a000000.wifi: Falling back to user helper
[   15.420214] firmware ath10k!QCA4019!hw1.0!ct-firmware-2.bin: firmware_loading_store: map pages failed
[   15.420623] ath10k_ahb a000000.wifi: Direct firmware load for ath10k/QCA4019/hw1.0/firmware-6.bin failed with error -2
[   15.428552] ath10k_ahb a000000.wifi: Falling back to user helper
[   15.503510] firmware ath10k!QCA4019!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[   16.217058] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   16.217103] ath10k_ahb a000000.wifi: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[   16.228727] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-012-c7b0f3f98 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-8
[   16.322233] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:20 crc32 e2dfaa91
[   17.632371] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 4  peers: 128  tid: 96
[   17.632417] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[   22.641366] ath10k_ahb a000000.wifi: wmi unified ready event not received
[   22.664295] ath10k_ahb a000000.wifi: could not init core (-110)
[   22.664512] ath10k_ahb a000000.wifi: could not probe fw (-110)
[   22.841174] ath10k_ahb a800000.wifi: fwcfg key: vdevs  val: 4
[   22.841224] ath10k_ahb a800000.wifi: fwcfg key: peers  val: 128
[   22.845915] ath10k_ahb a800000.wifi: fwcfg key: active_peers  val: 64
[   22.851677] ath10k_ahb a800000.wifi: fwcfg key: stations  val: 127
[   22.861897] ath10k_ahb a800000.wifi: Direct firmware load for ath10k/QCA4019/hw1.0/ct-firmware-5.bin failed with error -2
[   22.864327] ath10k_ahb a800000.wifi: Falling back to user helper
[   22.915794] firmware ath10k!QCA4019!hw1.0!ct-firmware-5.bin: firmware_loading_store: map pages failed
[   22.916150] ath10k_ahb a800000.wifi: Direct firmware load for ath10k/QCA4019/hw1.0/ct-firmware-2.bin failed with error -2
[   22.924087] ath10k_ahb a800000.wifi: Falling back to user helper
[   22.989115] firmware ath10k!QCA4019!hw1.0!ct-firmware-2.bin: firmware_loading_store: map pages failed
[   22.989469] ath10k_ahb a800000.wifi: Direct firmware load for ath10k/QCA4019/hw1.0/firmware-6.bin failed with error -2
[   22.997425] ath10k_ahb a800000.wifi: Falling back to user helper
[   23.058769] firmware ath10k!QCA4019!hw1.0!firmware-6.bin: firmware_loading_store: map pages failed
[   23.060376] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   23.066713] ath10k_ahb a800000.wifi: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[   23.080721] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-fW-012-c7b0f3f98 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-8
[   23.135204] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:21 crc32 e2dfaa91
[   24.445598] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 4  peers: 128  tid: 96
[   24.445644] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[   29.521398] ath10k_ahb a800000.wifi: wmi unified ready event not received
[   29.536245] ath10k_ahb a800000.wifi: could not init core (-110)
[   29.536525] ath10k_ahb a800000.wifi: could not probe fw (-110)

As you can see, there is an error "could not init core (-110)".

cat /lib/firmware/ath10k/fwcfg-ahb-a800000.wifi.txt
 
vdevs = 4
peers = 128
active_peers = 64
stations = 127

greearb, 2019-09-04:
> [ 2830.954447] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer dd364c00 vdev: 0 addr: 50:3e:aa:db🆎ca
> [ 2831.009173] ath10k_ahb a000000.wifi: removing peer, cleanup-all, deleting: peer dd334600 vdev: 0 addr: 50:3e:aa:db🆎b2

Hello,

You are hitting a null-ptr, and I need the binary crash dump to debug where.
Most likely you are running out of some resource, maybe tids.

Please try increasing tids to 4x the number of peers.

peers and active peers should always be the same with ath10k-ct firmware.
Stations are typically 2x peers.

Thanks,
Ben

klukonin, 2019-09-05:
@greearb 

Is it right to say that each vdev should create one additional peer?
So if we want to have 64 active peers and 4 vdevs, we need to configure "peers = 68" ?

greearb, 2019-09-05:
I think that might work for AP mode.  Station-mode vdevs need 2 peers each.

Thanks,
Ben

matejkupljen, 2019-09-06:
@greearb, will change the configuration as you suggested and will run it on the AP.
I'll monitor it and if I get the firmware crash again, I'll try to dump binary crash dump and send it too you.

Thanks,
Matej

greearb, 2019-10-29:
I updated the ath10k help page with some fwcfg files that work for 9984.  They most likely will NOT work for 4019, but maybe it will provide some clues.  In case you get a good working example for 4019 (at least minimally tested with the maximum number of configured stations), please let me know and I'll add it to the page.

http://www.candelatech.com/ath10k-10.4.php#config

Closing this bug, I think it was just some problem with fwcfg.

closed 2019-10-29 by greearb