greearb/ath10k-ct #95: qca9980: htt firmware crash (ver 10.4b-ct-9980-fH-012-c7b0f3f9r7500v2, r7500v2 kernel 4.19))

notmyrealhandle, 2019-09-20:
dmesg [here](https://gist.github.com/notmyrealhandle/e1d8c0f5c09d7498ff553eaac6283db8).

this came after 13+ days up with no other issues - the router is still going ATM.  I'm running a DYI build with some patches on kernel 4.19 for cpuidle, thermal sensors (a different patch than I mentioned in a prior bug report), and usb.  I suspect (hope) these are unrelated.

I see an updated ath10k-ct driver and firmware in openwrt master and I plan to build a new image with those this weekend.

Let me know if there is something else you like to see or for me to test...

notmyrealhandle, 2019-09-23:
an update to the next version seems to resolved this.  FWIW, this seemed to originate from one windows 10 client that had previously been stable...

closed 2019-09-23 by notmyrealhandle

greearb, 2019-09-23:
Ok, the crash was in an area I had worked on, so maybe I did fix it.  Please open a new one if you see more issues on the latest firmware.