greearb/ath10k-ct #96: qca9980: htt firmware crash (ver 10.4b-ct-9980-fH-012-54863dff, r7500v2 kernel 4.19)

notmyrealhandle, 2019-09-23:
[dmesg](https://gist.github.com/notmyrealhandle/1733a35be97e50a7ba50c258a3bf5b23)
```
root@OpenWrt:~/local# uname -a
Linux OpenWrt 4.19.74 #0 SMP Sat Sep 21 11:41:55 2019 armv7l GNU/Linux
root@OpenWrt:~/local# uptime
 16:10:53 up 2 days,  4:28,  load average: 0.29, 0.16, 0.10
```

this does seem to correlate with a windows 10 client.  I tested a few times with it over the past couple of days but did not see this until now (its being used more today)

notmyrealhandle, 2019-09-25:
@greearb [wrote](https://gist.github.com/notmyrealhandle/1733a35be97e50a7ba50c258a3bf5b23#gistcomment-3036354):
> Looks like this is the same or similar as bug 77. Please try the attached 9980 firmware and send me the logs.
> 
> [image]
> 
> NOTE: Rename file to *.gz, then gunzip. github will not longer accept .gz images it seems.

k, quick question.  I downloaded unziped, copied to router in tmp and then
```
root@OpenWrt:~# mv /tmp/firmware-5-full-htt-mgt-community /lib/firmware/ath10k/QCA99X0/hw2.0/ct-firmware-5.bin
root@OpenWrt:~# reboot
...
root@OpenWrt:~# md5sum /lib/firmware/ath10k/QCA99X0/hw2.0/ct-firmware-5.bin 
6086fe7bd2d3ccdec73f695736b8b7be  /lib/firmware/ath10k/QCA99X0/hw2.0/ct-firmware-5.bin
```
which matches the md5 of the unziped file I downloaded.  However dmesg after reboot shows
```
[   16.502480] ath10k_pci 0001:01:00.0: firmware ver 10.4b-ct-9980-fH-012-54863dff api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 b847f29f
```
which looks like the same version I was using.  Is this what you want?

notmyrealhandle, 2019-09-25:
didn't take long...

[dmesg](https://gist.github.com/notmyrealhandle/3b9f0b0169182390d0800edb01e30d58)

greearb, 2019-09-25:
Please repeat test with this one and send me logs.  The problem device has MAC ending with :52 as far as I can tell.  That your windows system?
[image]

notmyrealhandle, 2019-09-25:
> The problem device has MAC ending with :52 as far as I can tell. That your windows system?

yes

notmyrealhandle, 2019-09-28:
2+ days up, some usage with the windows client but not heavy.  Seems ok.  I'll report back in about a week as the client will not be used much for the next several days.

I and others have observed the warning below so perhaps its unrelated.

```
root@OpenWrt:~# uptime
 08:24:56 up 2 days, 23:06,  load average: 0.00, 0.05, 0.07

[251156.055311] ------------[ cut here ]------------
[251156.055476] WARNING: CPU: 0 PID: 2173 at backports-5.3-rc4-1/net/wireless/util.c:1171 cfg80211_calculate_bitrate+0x1cc/0x254 [cfg80211]
[251156.059014] invalid rate bw=0, mcs=15, nss=4
[251156.071531] Modules linked in: pppoe ppp_async ath10k_pci ath10k_core ath rndis_host pppox ppp_generic mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_policy xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conncount iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables crc_itu_t crc_ccitt compat sch_cake
[251156.124102]  nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb xfrm6_mode_tunnel xfrm6_mode_transport xfrm6_mode_beet ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet ipcomp esp4 ah4 tunnel6 tunnel4 xfrm_user xfrm_ipcomp af_key xfrm_algo vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_generic md5 echainiv authenc uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom ohci_platform ohci_hcd ahci ehci_platform sd_mod ahci_platform
[251156.195016]  libahci_platform libahci libata scsi_mod ehci_hcd gpio_button_hotplug f2fs ext4 mbcache jbd2 crc32c_generic crc32_generic
[251156.217287] CPU: 0 PID: 2173 Comm: reader#0 Not tainted 4.19.74 #0
[251156.229058] Hardware name: Generic DT based system
[251156.235332] [<c030f144>] (unwind_backtrace) from [<c030b850>] (show_stack+0x14/0x20)
[251156.240193] [<c030b850>] (show_stack) from [<c0829174>] (dump_stack+0x88/0x9c)
[251156.248184] [<c0829174>] (dump_stack) from [<c031d5fc>] (__warn+0xe0/0x10c)
[251156.255293] [<c031d5fc>] (__warn) from [<c031d65c>] (warn_slowpath_fmt+0x34/0x4c)
[251156.262666] [<c031d65c>] (warn_slowpath_fmt) from [<bf6863f0>] (cfg80211_calculate_bitrate+0x1cc/0x254 [cfg80211])
[251156.270199] [<bf6863f0>] (cfg80211_calculate_bitrate [cfg80211]) from [<bf6a56a0>] (nl80211_put_sta_rate+0x54/0x11a4 [cfg80211])
[251156.280457] [<bf6a56a0>] (nl80211_put_sta_rate [cfg80211]) from [<bf6a5ef4>] (nl80211_put_sta_rate+0x8a8/0x11a4 [cfg80211])
[251156.292256] [<bf6a5ef4>] (nl80211_put_sta_rate [cfg80211]) from [<bf6a68d8>] (nl80211_dump_station+0xe8/0x13c [cfg80211])
[251156.303472] [<bf6a68d8>] (nl80211_dump_station [cfg80211]) from [<c073f914>] (genl_lock_dumpit+0x3c/0x60)
[251156.314235] [<c073f914>] (genl_lock_dumpit) from [<c073bda8>] (netlink_dump+0x114/0x2a4)
[251156.323867] [<c073bda8>] (netlink_dump) from [<c073d9c0>] (__netlink_dump_start+0xd8/0x17c)
[251156.332111] [<c073d9c0>] (__netlink_dump_start) from [<c07412a4>] (genl_rcv_msg+0x1ac/0x3cc)
[251156.340620] [<c07412a4>] (genl_rcv_msg) from [<c073f50c>] (netlink_rcv_skb+0xac/0x12c)
[251156.349038] [<c073f50c>] (netlink_rcv_skb) from [<c073fcc0>] (genl_rcv+0x2c/0x48)
[251156.356849] [<c073fcc0>] (genl_rcv) from [<c073ed48>] (netlink_unicast+0x168/0x21c)
[251156.364489] [<c073ed48>] (netlink_unicast) from [<c073f13c>] (netlink_sendmsg+0x340/0x39c)
[251156.372402] [<c073f13c>] (netlink_sendmsg) from [<c06d9090>] (___sys_sendmsg+0x230/0x278)
[251156.380465] [<c06d9090>] (___sys_sendmsg) from [<c06da02c>] (__sys_sendmsg+0x48/0x78)
[251156.388800] [<c06da02c>] (__sys_sendmsg) from [<c0301000>] (ret_fast_syscall+0x0/0x54)
[251156.396685] Exception stack(0xda147fa8 to 0xda147ff0)
[251156.404599] 7fa0:                   00000000 00000000 00000004 b6c57738 00000000 00000000
[251156.409818] 7fc0: 00000000 00000000 011816a0 00000128 0118b2e0 b6c577d8 b6ea8264 b6ea09ec
[251156.418050] 7fe0: b6c576e0 b6c576cc b6f8af0c b6f8a3e4
[251156.426410] ---[ end trace f7f7b669722c4b37 ]---
[251156.431798] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[251156.436274] ath10k_pci 0000:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[251156.444246] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 1, skipped old beacon
[251156.451077] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
```

notmyrealhandle, 2019-10-07:
up almost 12 days and I still don't see this error coming back. 

The client that seems to cause it has been periodically used on the network about the same as in the past but won't be used for the next several days.  I can either continue testing for a few more weeks or close this now and reopen if it comes back.  Let me know if you have a preference.

notmyrealhandle, 2019-10-17:
no errors since my last post.  router has been used but the continous uptime has only been a few days at at time due to power outages or me making changes.

Today I built a new openwrt image and tried to load the second image you posted above and got

```
[   14.128187] ath10k 4.19 driver, optimized for CT firmware, probing pci device: 0x40.
[   14.128914] ath10k_pci 0000:01:00.0: enabling device (0140 -> 0142)
[   14.135159] ath10k_pci 0000:01:00.0: enabling bus mastering
[   14.135742] ath10k_pci 0000:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   14.338585] firmware ath10k!fwcfg-pci-0000:01:00.0.txt: firmware_loading_store: map pages failed
[   14.342280] ath10k_pci 0000:01:00.0: loading /lib/firmware/ath10k/QCA99X0/hw2.0/ct-firmware-5.bin failed with error -22
[   14.377541] firmware ath10k!QCA99X0!hw2.0!ct-firmware-5.bin: firmware_loading_store: map pages failed
[   14.428979] firmware ath10k!QCA99X0!hw2.0!ct-firmware-2.bin: firmware_loading_store: map pages failed
[   14.509072] firmware ath10k!QCA99X0!hw2.0!firmware-6.bin: firmware_loading_store: map pages failed
[   14.568641] firmware ath10k!QCA99X0!hw2.0!firmware-5.bin: firmware_loading_store: map pages failed
[   14.634458] firmware ath10k!QCA99X0!hw2.0!firmware-4.bin: firmware_loading_store: map pages failed
[   14.701205] firmware ath10k!QCA99X0!hw2.0!firmware-3.bin: firmware_loading_store: map pages failed
[   14.763981] firmware ath10k!QCA99X0!hw2.0!firmware-2.bin: firmware_loading_store: map pages failed
[   14.764123] ath10k_pci 0000:01:00.0: Failed to find firmware-N.bin (N between 2 and 6) from ath10k/QCA99X0/hw2.0: -11
[   14.771935] ath10k_pci 0000:01:00.0: could not fetch firmware files (-11)
[   14.782591] ath10k_pci 0000:01:00.0: could not probe fw (-11)
[   14.789534] ath10k_pci 0001:01:00.0: assign IRQ: got 37
[   14.789556] ath10k 4.19 driver, optimized for CT firmware, probing pci device: 0x40.
[   14.798474] ath10k_pci 0001:01:00.0: enabling device (0140 -> 0142)
[   14.802890] ath10k_pci 0001:01:00.0: enabling bus mastering
[   14.803478] ath10k_pci 0001:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   15.026312] firmware ath10k!fwcfg-pci-0001:01:00.0.txt: firmware_loading_store: map pages failed
[   15.028316] ath10k_pci 0001:01:00.0: loading /lib/firmware/ath10k/QCA99X0/hw2.0/ct-firmware-5.bin failed with error -22
[   15.086387] firmware ath10k!QCA99X0!hw2.0!ct-firmware-5.bin: firmware_loading_store: map pages failed
[   15.138159] firmware ath10k!QCA99X0!hw2.0!ct-firmware-2.bin: firmware_loading_store: map pages failed
[   15.214246] firmware ath10k!QCA99X0!hw2.0!firmware-6.bin: firmware_loading_store: map pages failed
[   15.285199] firmware ath10k!QCA99X0!hw2.0!firmware-5.bin: firmware_loading_store: map pages failed
[   15.358167] firmware ath10k!QCA99X0!hw2.0!firmware-4.bin: firmware_loading_store: map pages failed
[   15.422266] firmware ath10k!QCA99X0!hw2.0!firmware-3.bin: firmware_loading_store: map pages failed
[   15.494374] firmware ath10k!QCA99X0!hw2.0!firmware-2.bin: firmware_loading_store: map pages failed
[   15.494542] ath10k_pci 0001:01:00.0: Failed to find firmware-N.bin (N between 2 and 6) from ath10k/QCA99X0/hw2.0: -11
[   15.502334] ath10k_pci 0001:01:00.0: could not fetch firmware files (-11)
[   15.512983] ath10k_pci 0001:01:00.0: could not probe fw (-11)
[   15.521881] kmodloader: done loading kernel modules from /etc/modules.d/*
```

did I do something wrong?

notmyrealhandle, 2019-10-17:
tried it again and this time it works so I must have made a misstep.  

Thankyou.

FWIW

r7500v2 # uname -a
Linux r7500v2 4.19.79 #0 SMP Wed Oct 16 19:54:40 2019 armv7l GNU/Linux
r7500v2 # cat /sys/kernel/debug/ieee80211/phy0/ath10k/firmware_info
directory: ath10k/QCA99X0/hw2.0
firmware:  ct-firmware-5.bin
fwcfg:     fwcfg-pci-0000:01:00.0.txt
bus:       0000:01:00.0
features:  mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT
version:   10.4b-ct-9980-fH-012-54863dff
hw_rev:    99x0
board:     board-2.bin

notmyrealhandle, 2019-10-26:
[dmesg](https://gist.github.com/notmyrealhandle/b296d45e1494fb92dabf4be82af05786) with crash log.

The AP (I no longer use it as a GW/AP) recovered and is still going.

r7500v2 # uptime
 07:45:33 up 8 days, 18:52,  load average: 0.01, 0.02, 0.00

firmware crash occurred:
Fri Oct 25 16:38:26 2019 kern.err kernel: [704707.211275] ath10k_pci 0001:01:00.0: firmware crashed! (guid n/a)

notmyrealhandle, 2019-10-27:
[dmesg](https://gist.github.com/notmyrealhandle/b296d45e1494fb92dabf4be82af05786)

Subsequent crash with the windows client online and an iphone in the middle of a video chat.  About 5 total clients on that wifi radio at the time.  An android phone didn't even notice the outage.

greearb, 2019-10-29:
Upon more debugging, I think this is probably something that the driver needs to prevent from happening and/or we need to debug more in the driver.  Please enable the MAC debugging in ath10k and use your windows device to crash the firmware.  Please attach dmesg output of that.

notmyrealhandle, 2019-10-29:
will do but may take a few days.  I think the reason I see sporadic time lags between crashes is that the device uses the 5 GHz radio much of the time.  Forcing it to 2.4 should do the trick but I have to pick a good time for that. This is an employer provided device which, when it is being used, is used for a good reason.

notmyrealhandle, 2019-10-29:
[dmesg](https://gist.github.com/notmyrealhandle/8c81a2f86d4037a954c700b02fbdf616) with crash at debug level 0x10

i.e.
```
r7500v2 # cat /sys/kernel/debug/ieee80211/phy1/ath10k/debug_level 
Current debug level: 0x10
...
```
If this is not what you want or you would like more, now is a good time to ask.

notmyrealhandle, 2019-10-30:
[dmesg](https://gist.github.com/notmyrealhandle/8c81a2f86d4037a954c700b02fbdf616) with crash at debug level  0xc0000032 (file r7500v2-dmesg-20191029-1.txt)

```
r7500v2 # cat /sys/kernel/debug/ieee80211/phy0/ath10k/debug_level
Current debug level: 0xc0000032
```

greearb, 2019-10-30:
There should be a lot more debugging from the ath10k driver.  Do you have debugging compiled
in?  And, maybe it is phy1 that is crashing, not phy0?  Try setting debug on both of them?

notmyrealhandle, 2019-10-30:
>  Do you have debugging compiled in?
It is a custom build but most _kernel_ debug options I left at the defaults.  Some are on others are off - see the "kernel build options" from my .config below.  I can roll a new image if you need something different.

> And, maybe it is phy1 that is crashing, not phy0?
It is phy1 that is crashing which is why I sent phy1 at debug level 0x10.  I thought you wanted phy0 which is why I sent that - my bad I knew and should have asked.

I'll see if I can't get both tonight.  Specifically I will:
```
echo 0xc0000032 >  /sys/kernel/debug/ieee80211/phy0/ath10k/debug_level
echo 0xc0000032 >  /sys/kernel/debug/ieee80211/phy1/ath10k/debug_level
```
then force the client to connect on 2.4 GHz.  A crash usually follows in 5 to 10 minutes.

It's best to think of me as a "novice with some skills."  Given me some reasonable direction I'll do my best to get you what you want.  I really appreciate the time you spend on this, your responsiveness, and your patience.

```
#
# Kernel build options
#
CONFIG_KERNEL_BUILD_USER=""
CONFIG_KERNEL_BUILD_DOMAIN=""
CONFIG_KERNEL_PRINTK=y
CONFIG_KERNEL_SWAP=y
CONFIG_KERNEL_DEBUG_FS=y
# CONFIG_KERNEL_ARM_PMU is not set
# CONFIG_KERNEL_PERF_EVENTS is not set
# CONFIG_KERNEL_PROFILING is not set
# CONFIG_KERNEL_TASKSTATS is not set
CONFIG_KERNEL_KALLSYMS=y
# CONFIG_KERNEL_FTRACE is not set
CONFIG_KERNEL_DEBUG_KERNEL=y
CONFIG_KERNEL_DEBUG_INFO=y
# CONFIG_KERNEL_DEBUG_LL_UART_NONE is not set
# CONFIG_KERNEL_DEBUG_LL is not set
# CONFIG_KERNEL_DYNAMIC_DEBUG is not set
# CONFIG_KERNEL_EARLY_PRINTK is not set
# CONFIG_KERNEL_KPROBES is not set
CONFIG_KERNEL_AIO=y
CONFIG_KERNEL_FHANDLE=y
CONFIG_KERNEL_FANOTIFY=y
# CONFIG_KERNEL_BLK_DEV_BSG is not set
CONFIG_KERNEL_MAGIC_SYSRQ=y
# CONFIG_KERNEL_DEBUG_PINCTRL is not set
# CONFIG_KERNEL_DEBUG_GPIO is not set
CONFIG_KERNEL_COREDUMP=y
CONFIG_KERNEL_ELF_CORE=y
# CONFIG_KERNEL_PROVE_LOCKING is not set
CONFIG_KERNEL_PRINTK_TIME=y
# CONFIG_KERNEL_SLABINFO is not set
# CONFIG_KERNEL_PROC_PAGE_MONITOR is not set
CONFIG_KERNEL_KEXEC=y
CONFIG_KERNEL_PROC_VMCORE=y
CONFIG_KERNEL_CRASH_DUMP=y
# CONFIG_USE_RFKILL is not set
# CONFIG_USE_SPARSE is not set
# CONFIG_KERNEL_DEVTMPFS is not set
# CONFIG_KERNEL_KEYS is not set
CONFIG_KERNEL_CGROUPS=y
# CONFIG_KERNEL_CGROUP_DEBUG is not set
CONFIG_KERNEL_FREEZER=y
CONFIG_KERNEL_CGROUP_FREEZER=y
CONFIG_KERNEL_CGROUP_DEVICE=y
CONFIG_KERNEL_CGROUP_PIDS=y
CONFIG_KERNEL_CPUSETS=y
# CONFIG_KERNEL_PROC_PID_CPUSET is not set
CONFIG_KERNEL_CGROUP_CPUACCT=y
CONFIG_KERNEL_RESOURCE_COUNTERS=y
CONFIG_KERNEL_MM_OWNER=y
CONFIG_KERNEL_MEMCG=y
# CONFIG_KERNEL_MEMCG_SWAP is not set
CONFIG_KERNEL_MEMCG_KMEM=y
# CONFIG_KERNEL_CGROUP_PERF is not set
CONFIG_KERNEL_CGROUP_SCHED=y
CONFIG_KERNEL_FAIR_GROUP_SCHED=y
# CONFIG_KERNEL_CFS_BANDWIDTH is not set
CONFIG_KERNEL_RT_GROUP_SCHED=y
CONFIG_KERNEL_BLK_CGROUP=y
# CONFIG_KERNEL_CFQ_GROUP_IOSCHED is not set
# CONFIG_KERNEL_BLK_DEV_THROTTLING is not set
# CONFIG_KERNEL_DEBUG_BLK_CGROUP is not set
CONFIG_KERNEL_NET_CLS_CGROUP=y
CONFIG_KERNEL_NETPRIO_CGROUP=y
CONFIG_KERNEL_NAMESPACES=y
CONFIG_KERNEL_UTS_NS=y
CONFIG_KERNEL_IPC_NS=y
CONFIG_KERNEL_USER_NS=y
CONFIG_KERNEL_PID_NS=y
CONFIG_KERNEL_NET_NS=y
CONFIG_KERNEL_DEVPTS_MULTIPLE_INSTANCES=y
# CONFIG_USE_RFKILL is not set
# CONFIG_USE_SPARSE is not set
# CONFIG_KERNEL_DEVTMPFS is not set
# CONFIG_KERNEL_KEYS is not set
CONFIG_KERNEL_CGROUPS=y
# CONFIG_KERNEL_CGROUP_DEBUG is not set
CONFIG_KERNEL_FREEZER=y
CONFIG_KERNEL_CGROUP_FREEZER=y
CONFIG_KERNEL_CGROUP_DEVICE=y
CONFIG_KERNEL_CGROUP_PIDS=y
CONFIG_KERNEL_CPUSETS=y
# CONFIG_KERNEL_PROC_PID_CPUSET is not set
CONFIG_KERNEL_CGROUP_CPUACCT=y
CONFIG_KERNEL_RESOURCE_COUNTERS=y
CONFIG_KERNEL_MM_OWNER=y
CONFIG_KERNEL_MEMCG=y
# CONFIG_KERNEL_MEMCG_SWAP is not set
CONFIG_KERNEL_MEMCG_KMEM=y
# CONFIG_KERNEL_CGROUP_PERF is not set
CONFIG_KERNEL_CGROUP_SCHED=y
CONFIG_KERNEL_FAIR_GROUP_SCHED=y
# CONFIG_KERNEL_CFS_BANDWIDTH is not set
CONFIG_KERNEL_RT_GROUP_SCHED=y
CONFIG_KERNEL_BLK_CGROUP=y
# CONFIG_KERNEL_CFQ_GROUP_IOSCHED is not set
# CONFIG_KERNEL_BLK_DEV_THROTTLING is not set
# CONFIG_KERNEL_DEBUG_BLK_CGROUP is not set
CONFIG_KERNEL_NET_CLS_CGROUP=y
CONFIG_KERNEL_NETPRIO_CGROUP=y
CONFIG_KERNEL_NAMESPACES=y
CONFIG_KERNEL_UTS_NS=y
CONFIG_KERNEL_IPC_NS=y
CONFIG_KERNEL_USER_NS=y
CONFIG_KERNEL_PID_NS=y
CONFIG_KERNEL_NET_NS=y
CONFIG_KERNEL_DEVPTS_MULTIPLE_INSTANCES=y
CONFIG_KERNEL_POSIX_MQUEUE=y
CONFIG_KERNEL_SECCOMP_FILTER=y
CONFIG_KERNEL_SECCOMP=y
CONFIG_KERNEL_IP_MROUTE=y
CONFIG_KERNEL_IPV6=y
CONFIG_KERNEL_IPV6_MULTIPLE_TABLES=y
CONFIG_KERNEL_IPV6_SUBTREES=y
CONFIG_KERNEL_IPV6_MROUTE=y
# CONFIG_KERNEL_IPV6_PIMSM_V2 is not set
# CONFIG_KERNEL_IP_PNP is not set
```

notmyrealhandle, 2019-10-31:
[dmesg](https://gist.github.com/notmyrealhandle/ad8fb22f22e543a135e2f8fc9b65e48c)

I'm not entirely sure PACKAGE_ATH_DEBUG got set as I did a short cut build but there is a lot more output.

the crashes came in pairs

I missed the first two crashes below but caught a warning trace in between near [ 8571.155573]
'''
Wed Oct 30 20:19:23 2019 kern.err kernel: [ 8562.525345] ath10k_pci 0001:01:00.0: firmware crashed! (guid n/a)
Wed Oct 30 20:19:57 2019 kern.err kernel: [ 8596.264590] ath10k_pci 0001:01:00.0: firmware crashed! (guid n/a)
'''

I did get 
'''
Wed Oct 30 20:42:07 2019 kern.err kernel: [ 9926.042516] ath10k_pci 0001:01:00.0: firmware crashed! (guid n/a)
Wed Oct 30 20:42:36 2019 kern.err kernel: [ 9955.608565] ath10k_pci 0001:01:00.0: firmware crashed! (guid n/a)
'''

I can turn on remote logging if you need more.

FWIW
'''
7500v2 # cat /sys/kernel/debug/ieee80211/phy1/ath10k/debug_level
Current debug level: 0xc0000032
r7500v2 # cat /sys/kernel/debug/ieee80211/phy0/ath10k/debug_level
Current debug level: 0xc0000032
'''

greearb, 2019-10-31:
That is excellent debugging output.  I'm going to look over it in detail when I get a chance and try to figure out what is going on.

greearb, 2019-10-31:
Please run with same debugging and this firmware below.  It should still crash, but hopefully show me the root cause this time.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/3796027/firmware-5-full-htt-mgt-community.bin.gz)

notmyrealhandle, 2019-11-01:
[dmesg](https://gist.github.com/notmyrealhandle/be859f90cec627a4989a4eff5c8d7bbc)

Large dump, 4 crashes

For reference:
38bd6dcb9d1dbd21364cf7c1f24c5d5b  /lib/firmware/ath10k/QCA99X0/hw2.0/ct-firmware-5.bin

greearb, 2019-11-01:
Please try this one, I think I found the fix.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/3796502/firmware-5-full-htt-mgt-community.bin.gz)

notmyrealhandle, 2019-11-02:
troublesome client connect to phy1 and so far it seems ok.

I do see
```
Fri Nov  1 21:47:35 2019 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED XX:XX:XX:XX:df:52 static
Fri Nov  1 21:47:36 2019 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED XX:XX:XX:XX:df:52 off
Fri Nov  1 21:47:38 2019 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED XX:XX:XX:XX:df:52 static
Fri Nov  1 21:47:39 2019 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED XX:XX:XX:XX:df:52 off
```
spaming my logs which did not happen before.

was the bug related to dynamic/static smps (Spatial Multiplexing Power Save)?

regardless someone is going to inquire/complain about these (new) hostapd messages...

for reference:
r7500v2 # md5sum /lib/firmware/ath10k/QCA99X0/hw2.0/ct-firmware-5.bin
47ac41502be70a9508bd7954fa10e5bc  /lib/firmware/ath10k/QCA99X0/hw2.0/ct-firmware-5.bin
r7500v2 # uptime
 21:48:22 up  3:00,  load average: 0.04, 0.04, 0.00

greearb, 2019-11-02:
Yes, crash was related to that.  Those messages are coming from hostapd, so I think any fix to make it more quiet would go there as well.  Probably you didn't see them before because the firmware often crashed when the smps changes happened.

greearb, 2019-11-04:
New firmware uploaded and offered to OpenWRT.  Closing this bug.  Thanks for taking time to debug it!

closed 2019-11-04 by greearb