greearb/ath10k-ct #99: successfully compile driver - unable to load it

gsustek, 2019-10-14:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
NAME="Ubuntu"
VERSION="19.10 (Eoan Ermine)"
Installed ubuntu version 19.10, dowloaded kernel source (https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.2.8/) 
git clone  git://git.launchpad.net/~ubuntu-kernel-test/ubuntu/+source/linux/+git/mainline-crack v5.2.8
git reset --hard  d36a8d2fb62c7c9415213bea9cf576d8b1f9071f
cd ~/Downloads/v5.2.8+
patch -p1 < ../0001-*.patch
...
patch -p1 < ../0006-*.patch
cp /boot/config-`uname -r` .config
yes '' | make oldconfig
make clean
make -j 2
sudo make modules_install
sudo make install
###reboot to that kernel
cd ~/Downloads/ath10k-ct
./build_me.sh
~/Downloads/v5.2.8
5.2
####
copy four co files to /usr/lib/modules/5.2.8+/kernel/drivers/net/wireless/ath/ath10k/
sudo modprobe -r ath10k_pci
sudo modprobe ath10k_pci

 error in dmesg -T

Software (OS, Firmware version, kernel, driver, etc)
OS - Linux gs-server 5.2.8+ #2 SMP Mon Oct 14 09:34:19 CEST 2019 x86_64 x86_64 x86_64 GNU/Linux

Firmware_version: firmware-5-ct-non-commercial-htt-mgt-9.bin
-rw-r--r-- 1 root root 537316 lis  13 21:59 /lib/firmware/ath10k/QCA9984/hw1.0/firmware-5.bin

board:  -rw-r--r-- 1 root root 157648 lis  13 19:14 /lib/firmware/ath10k/QCA9984/hw1.0/board-2.bin

ath10k 5.1 driver

Hardware (NIC chipset, platform, etc)
NIC: Compex COMPEX WLE1216V5-20 5GHz
Platform:  KANSUNG Model: K515G6
(https://bit.ly/2IOlSeQ)

Logs (dmesg, maybe supplicant and/or hostap)

[pon lis 14 18:39:51 2019] ath10k 5.1 driver, optimized for CT firmware, probing pci device: 0x46.
[pon lis 14 18:39:51 2019] ath10k_pci 0000:06:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[pon lis 14 18:39:51 2019] ath10k_pci 0000:06:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[pon lis 14 18:39:51 2019] ath10k_pci 0000:06:00.0: kconfig debug 0 debugfs 1 tracing 1 dfs 0 testmode 0
[pon lis 14 18:39:51 2019] ath10k_pci 0000:06:00.0: firmware ver 10.4-ct-9984-xtH-009-dfa0083 api 5 features peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,rxswcrypt-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT crc32 50d1a82f
[pon lis 14 18:39:52 2019] ath10k_pci 0000:06:00.0: board_file api 2 bmi_id 0:31 crc32 e807b522
[pon lis 14 18:39:55 2019] ath10k_pci 0000:06:00.0: unsupported HTC service id: 1536
[pon lis 14 18:39:55 2019] ath10k_pci 0000:06:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[pon lis 14 18:39:55 2019] ath10k_pci 0000:06:00.0: msdu-desc: 2500  skid: 32
[pon lis 14 18:39:55 2019] ath10k_pci 0000:06:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[pon lis 14 18:39:55 2019] ath10k_pci 0000:06:00.0: wmi print 'free: 123116 iram: 13444 sram: 38552'
[pon lis 14 18:39:55 2019] ath10k_pci 0000:06:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[pon lis 14 18:39:55 2019] ath: EEPROM regdomain: 0x0
[pon lis 14 18:39:55 2019] ath: EEPROM indicates default country code should be used
[pon lis 14 18:39:55 2019] ath: doing EEPROM country->regdmn map search
[pon lis 14 18:39:55 2019] ath: country maps to regdmn code: 0x3a
[pon lis 14 18:39:55 2019] ath: Country alpha2 being used: US
[pon lis 14 18:39:55 2019] ath: Regpair used: 0x3a
[pon lis 14 18:39:55 2019] ------------[ cut here ]------------
[pon lis 14 18:39:55 2019] WARNING: CPU: 1 PID: 3323 at net/wireless/core.c:621 wiphy_register+0x432/0x910 [cfg80211]
[pon lis 14 18:39:55 2019] Modules linked in: ath10k_pci(OE) ath10k_core(OE) ath mac80211 binfmt_misc arc4 nls_iso8859_1 snd_hda_codec_hdmi snd_soc_skl snd_soc_hdac_hda snd_hda_ext_core snd_soc_skl_ipc snd_soc_sst_ipc snd_soc_sst_dsp snd_soc_acpi_intel_match snd_soc_acpi intel_rapl snd_soc_core x86_pkg_temp_thermal snd_compress intel_powerclamp ac97_bus coretemp snd_pcm_dmaengine wl(POE) kvm_intel i915 snd_hda_intel kvm snd_hda_codec snd_hda_core irqbypass snd_hwdep snd_pcm crct10dif_pclmul snd_seq_midi crc32_pclmul snd_seq_midi_event ghash_clmulni_intel snd_rawmidi snd_seq aesni_intel aes_x86_64 crypto_simd snd_seq_device cryptd snd_timer glue_helper drm_kms_helper snd intel_cstate cfg80211 drm intel_rapl_perf intel_wmi_thunderbolt mei_me soundcore 8250_dw input_leds joydev fb_sys_fops syscopyarea intel_xhci_usb_role_switch idma64 sysfillrect roles virt_dma sysimgblt mei intel_pch_thermal acpi_pad mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 hid_generic usbhid hid igb ahci
[pon lis 14 18:39:55 2019]  i2c_i801 i2c_algo_bit wmi dca libahci intel_lpss_pci intel_lpss video pinctrl_sunrisepoint pinctrl_intel [last unloaded: ath]
[pon lis 14 18:39:55 2019] CPU: 1 PID: 3323 Comm: kworker/u4:1 Tainted: P        W  OE     5.2.8+ #2
[pon lis 14 18:39:55 2019] Hardware name: Default string Default string/Default string, BIOS 5.12 05/26/2019
[pon lis 14 18:39:55 2019] Workqueue: ath10k_wq ath10k_core_register_work [ath10k_core]
[pon lis 14 18:39:55 2019] RIP: 0010:wiphy_register+0x432/0x910 [cfg80211]
[pon lis 14 18:39:55 2019] Code: 93 99 c7 0f 0b b8 ea ff ff ff e9 15 fd ff ff 40 f6 c7 02 74 46 44 8b 60 14 45 85 e4 74 3d 48 c7 c7 e8 d5 3c c0 e8 71 93 99 c7 <0f> 0b b8 ea ff ff ff e9 ee fc ff ff 66 f7 c7 00 10 74 d2 66 83 3e
[pon lis 14 18:39:55 2019] RSP: 0018:ffffc13c024c7ce0 EFLAGS: 00010246
[pon lis 14 18:39:55 2019] RAX: 0000000000000024 RBX: ffffa079de900300 RCX: 0000000000000006
[pon lis 14 18:39:55 2019] RDX: 0000000000000000 RSI: 0000000000000082 RDI: ffffa079e6b17440
[pon lis 14 18:39:55 2019] RBP: ffffc13c024c7d58 R08: 00000000000003e3 R09: 0000000000000004
[pon lis 14 18:39:55 2019] R10: 0000000000000000 R11: 0000000000000001 R12: 0000000000000001
[pon lis 14 18:39:55 2019] R13: 00000060043dddd1 R14: 000000000000001b R15: 0000000000000001
[pon lis 14 18:39:55 2019] FS:  0000000000000000(0000) GS:ffffa079e6b00000(0000) knlGS:0000000000000000
[pon lis 14 18:39:55 2019] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[pon lis 14 18:39:55 2019] CR2: 000055af0fde0c68 CR3: 000000012f20a003 CR4: 00000000003606e0
[pon lis 14 18:39:55 2019] Call Trace:
[pon lis 14 18:39:55 2019]  ? freq_reg_info_regd.part.19+0xc8/0xf0 [cfg80211]
[pon lis 14 18:39:55 2019]  ? _cond_resched+0x19/0x30
[pon lis 14 18:39:55 2019]  ? __kmalloc+0x177/0x220
[pon lis 14 18:39:55 2019]  ieee80211_register_hw+0x456/0xc60 [mac80211]
[pon lis 14 18:39:55 2019]  ? ath10k_regd_update+0x2b0/0x2b0 [ath10k_core]
[pon lis 14 18:39:55 2019]  ath10k_mac_register+0x8ca/0xe00 [ath10k_core]
[pon lis 14 18:39:55 2019]  ? ath10k_wmi_detach+0x32/0x90 [ath10k_core]
[pon lis 14 18:39:55 2019]  ath10k_core_register_work+0x86f/0xcd0 [ath10k_core]
[pon lis 14 18:39:55 2019]  process_one_work+0x20f/0x410
[pon lis 14 18:39:55 2019]  worker_thread+0x34/0x400
[pon lis 14 18:39:55 2019]  kthread+0x120/0x140
[pon lis 14 18:39:55 2019]  ? process_one_work+0x410/0x410
[pon lis 14 18:39:55 2019]  ? __kthread_parkme+0x70/0x70
[pon lis 14 18:39:55 2019]  ret_from_fork+0x35/0x40
[pon lis 14 18:39:55 2019] ---[ end trace eae2e8ed5f30be10 ]---
[pon lis 14 18:39:55 2019] ath10k_pci 0000:06:00.0: failed to register ieee80211: -22
[pon lis 14 18:39:55 2019] ath10k_pci 0000:06:00.0: could not register to mac80211 (-22)

greearb, 2019-10-14:
Do you have this patch applied?

https://github.com/greearb/linux-ct-5.2/commit/5a18eefe3650c3e31b5c7ff140d8421f8d78f626

gsustek, 2019-10-14:
Thank you for such quick answer. i think i do not that patch applied. So as i understand, you also provide kernel-ct. I must use it? or i can use ubuntu one and patch it with patches from yours kernel-ct? 
Is there somewhere a compilation of all patches that it needs to be applied to my kernel to make it work?

greearb, 2019-10-14:
You at least need that patch, probably others would not hurt but may not be required.

Thanks,
Ben

gsustek, 2019-10-16:
yeah, this patch is missing.

closed 2019-10-16 by gsustek