greearb/ath10k-ct #100: start-radar: ops->start-radar-detection failed: -16

gsustek, 2019-10-16:
lease provide this info. See this link for more info on how to gather debug
info: http://www.candelatech.com/ath10k-bugs.php

###Description of the problem (how to configure, how to reproduce, how often it happens).
NAME="Ubuntu"
VERSION="19.10 (Eoan Ermine)"
###Installed ubuntu version 19.10, downloaded kernel source (https://github.com/greearb/linux-ct-5.2),
git clone https://github.com/greearb/linux-ct-5.2.git
download patches for ubuntu and applied: https://kernel.ubuntu.com/~kernel-ppa/mainline/v5.2.14/
cd ~/Downloads/linux-ct-5.2
patch -p1 < ../0001-.patch
...
patch -p1 < ../0006-.patch
cp /boot/config-uname -r .config
yes '' | make oldconfig
make clean
make -j 2
sudo make modules_install
sudo make install
###reboot to that kernel
cd ~/Downloads/ath10k-ct
./build_me.sh
~/Downloads//linux-ct-5.2
enter:
5.2

###copy four co files to /usr/lib/modules/5.2.14+/kernel/drivers/net/wireless/ath/ath10k/
sudo modprobe -r ath10k_pci
sudo modprobe ath10k_pci

###Software (OS, Firmware version, kernel, driver, etc)
uname -a
Linux gs-server 5.2.14+ #3 SMP Wed Oct 16 00:32:21 CEST 2019 x86_64 x86_64 x86_64 GNU/Linux

###Firmware_version: firmware-5-ct-non-commercial-htt-mgt.bin
-rw-r--r-- 1 root root 543764 lis  16 09:41 /lib/firmware/ath10k/QCA9984/hw1.0/firmware-5.bin

###board: -rw-r--r-- 1 root root 157648 lis 13 19:14 /lib/firmware/ath10k/QCA9984/hw1.0/board-2.bin

###Ath10k driver
Atf10k 5.1 driver
ethtool  -i wlp6s0
driver: ath10k_pci
version: 5.2.14+
firmware-version: 10.4b-ct-9984-xtH-012-e747272a4
expansion-rom-version: 
bus-info: 0000:06:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

###Hardware (NIC chipset, platform, etc)
NIC: Compex COMPEX WLE1216V5-20 5GHz
Platform: KANSUNG Model: K515G6
(https://bit.ly/2IOlSeQ)

###hostapd -v
hostapd v2.9
User space daemon for IEEE 802.11 AP management,

###cat /etc/hostapd/hostapd.conf 
driver=nl80211
logger_syslog=-1
logger_syslog_level=2
logger_stdout=-1
logger_stdout_level=2
hw_mode=a
channel=100
[... 230 lines cut ...]
[sri lis 16 10:50:52 2019] cfg80211: wiphy 04:f0:21:49:44:26: Disabling freq 5865 MHz (regdom-not-set-by-driver)
[sri lis 16 10:50:52 2019] cfg80211: intersect-dfs-region, region1: 1  region2: 2
[sri lis 16 10:50:52 2019] cfg80211: Setting DFS Master region in update_regulatory, was: US unset, new: 98 unset  lr: 00000000c8cd9d30  regdom: 000000003fdf0336
[sri lis 16 10:50:52 2019] cfg80211: wiphy 04:f0:21:49:44:26: Disabling freq 5845 MHz (regdom-not-set-by-driver)
[sri lis 16 10:50:52 2019] cfg80211: wiphy 04:f0:21:49:44:26: Disabling freq 5865 MHz (regdom-not-set-by-driver)
[sri lis 16 10:50:52 2019] ath10k_pci 0000:06:00.0 wlp6s0: renamed from wlan0
[sri lis 16 10:51:04 2019] ath10k_pci 0000:06:00.0: unsupported HTC service id: 1536
[sri lis 16 10:51:04 2019] ath10k_pci 0000:06:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[sri lis 16 10:51:04 2019] ath10k_pci 0000:06:00.0: msdu-desc: 2500  skid: 32
[sri lis 16 10:51:04 2019] ath10k_pci 0000:06:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[sri lis 16 10:51:04 2019] ath10k_pci 0000:06:00.0: wmi print 'free: 128804 iram: 23924 sram: 44084'
[sri lis 16 10:51:04 2019] ath10k_pci 0000:06:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[sri lis 16 10:51:05 2019] ath10k_pci 0000:06:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[sri lis 16 10:51:09 2019] start-radar: dfs_region == DFS_UNSET
[sri lis 16 10:51:09 2019] wlp6s0: do_stop, canceling dfs-cac-timer-work.
[sri lis 16 10:51:09 2019] ath10k_pci 0000:06:00.0: peer-unmap-event: unknown peer id 0

i tried every firmware from beta12:
    firmware-5-ct-full-community.bin  | 9984
    firmware-5-ct-community.bin  | 9984 |
    firmware-5-ct-non-commercial.bin  | 9984 |
    firmware-5-ct-non-commercial-full.bin  | 9984
     firmware-5-ct-full-htt-mgt-community.bin  | 9984
    firmware-5-ct-non-commercial-htt-mgt.bin  | 9984
   firmware-5-ct-non-commercial-full-htt-mgt.bin  | 9984

gsustek, 2019-10-16:
just added some debugging.

[sri lis 16 11:00:41 2019] ath10k_pci 0000:06:00.0: Setting debug-mask to: 0xc0000032  old: 0x10000000
[sri lis 16 11:03:11 2019] wlp6s0: do_stop, canceling dfs-cac-timer-work.
[sri lis 16 11:03:13 2019] ath10k_pci 0000:06:00.0: unsupported HTC service id: 1536
[sri lis 16 11:03:13 2019] ath10k_pci 0000:06:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[sri lis 16 11:03:13 2019] ath10k_pci 0000:06:00.0: msdu-desc: 2500  skid: 32
[sri lis 16 11:03:13 2019] ath10k_pci 0000:06:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[sri lis 16 11:03:13 2019] ath10k_pci 0000:06:00.0: wmi print 'free: 128804 iram: 23924 sram: 44084'
[sri lis 16 11:03:14 2019] ath10k_pci 0000:06:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[sri lis 16 11:03:14 2019] ath10k_pci 0000:06:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[sri lis 16 11:03:14 2019] ath10k: [0000]: 00000992 13FC0007 60002070 00000010 00000030 004E0001 000009DF 17FC3C2A
[sri lis 16 11:03:14 2019] ath10k: [0008]: 61100001 01000004 00010000 14640000 00000001 000009E0 17FC582E 00000003
[sri lis 16 11:03:14 2019] ath10k: [0016]: 0000001E 0000094C 0000094C 00000000 000009E0 17FC582E 00000012 0000001E
[sri lis 16 11:03:14 2019] ath10k: [0024]: 0000094C 0000094C 00000000 000009E0 17FC582E 00000045 0000001E 0000094C
[sri lis 16 11:03:14 2019] ath10k: [0032]: 0000094C 00000000 000009E0 17FC582E 00000067 0000001E 0000094C 0000094C
[sri lis 16 11:03:14 2019] ath10k: [0040]: 00000000 00000A03 17FC3C2A 61100001 01000001 01010000 14640000 00000002
[sri lis 16 11:03:14 2019] ath10k: [0048]: 00000A03 13FC5822 9110B000 00000001 00000001 00000001 00000A04 1400587C
[sri lis 16 11:03:14 2019] ath10k: [0056]: 51100001 000EE260 000003FC 00000001 004505F0 00000A04 1000587A 0042CD10
[sri lis 16 11:03:14 2019] ath10k: [0064]: 004505F0 FF000001 00000000 00000A04 14006403 00000000 01000000 00000000
[sri lis 16 11:03:14 2019] ath10k: [0072]: 00000000 004505F0 00000A04 14006402 71103331 00000000 00000100 00000001
[sri lis 16 11:03:14 2019] ath10k: [0080]: 00000000 00000A04 14006402 71103332 00000000 00000100 0042AA08 00000000
[sri lis 16 11:03:14 2019] ath10k: [0088]: 00000A04 1400587D 51100002 000EE260 00000057 00000000 004505F0 00000A04
[sri lis 16 11:03:14 2019] ath10k: [0096]: 1000587B 0042CD10 004505F0 00000000 0042AA08 00000A24 17FC3C2A 61100001
[sri lis 16 11:03:14 2019] ath10k: [0104]: 01000001 01010000 14640000 00000002 00000A24 13FC5822 9110B000 00000001
[sri lis 16 11:03:14 2019] ath10k: [0112]: 00000001 00000001 00000A44 17FC3C2A 61100001 01000001 01010000 14640000
[sri lis 16 11:03:14 2019] ath10k: [0120]: 00000002 00000A44 13FC5822 9110B000 00000001 00000001 00000001 00000A65
[sri lis 16 11:03:14 2019] ath10k: [0128]: 17FC3C2A 61100001 01000001 01010000 14640000 00000002 00000A65 13FC5822
[sri lis 16 11:03:14 2019] ath10k: [0136]: 9110B000 00000001 00000001 00000001 00000A86 17FC3C2A 61100001 01000001
[sri lis 16 11:03:14 2019] ath10k: [0144]: 01010000 14640000 00000002 00000A86 13FC5822 9110B000 00000001 00000001
[sri lis 16 11:03:14 2019] ath10k: [0152]: 00000001 00000AA6 17FC3C2A 61100001 01000001 01010000 14640000 00000002
[sri lis 16 11:03:14 2019] ath10k: [0160]: 00000AA6 13FC5822 9110B000 00000001 00000001 00000001 00000AC7 17FC3C2A
[sri lis 16 11:03:14 2019] ath10k: [0168]: 61100001 01000001 01010000 14640000 00000002 00000AC7 13FC5822 9110B000
[sri lis 16 11:03:14 2019] ath10k: [0176]: 00000001 00000001 00000001 00000AE8 17FC3C2A 61100001 01000001 01010000
[sri lis 16 11:03:14 2019] ath10k: [0184]: 14640000 00000002 00000AE8 13FC5822 9110B000 00000001 00000001 00000001
[sri lis 16 11:03:14 2019] ath10k: [0192]: 00000B08 17FC3C2A 61100001 01000001 01010000 14640000 00000002 00000B08
[sri lis 16 11:03:14 2019] ath10k: [0200]: 13FC5822 9110B000 00000001 00000001 00000001 00000B28 17FC3C2A 61100001
[sri lis 16 11:03:14 2019] ath10k: [0208]: 01000001 01010000 14640000 00000002 00000B28 13FC5822 9110B000 00000001
[sri lis 16 11:03:14 2019] ath10k: [0216]: 00000001 00000001 00000B49 17FC3C2A 61100001 01000001 01010000 14640000
[sri lis 16 11:03:14 2019] ath10k: [0224]: 00000002 00000B49 13FC5822 9110B000 00000001 00000001 00000001 00000B69
[sri lis 16 11:03:14 2019] ath10k: [0232]: 17FC3C2A 61100001 01000001 01010000 14640000 00000002 00000B69 13FC5822
[sri lis 16 11:03:14 2019] ath10k: [0240]: 9110B000 00000001 00000001 00000001 00000B6B 1400587C 51100001 000EE260
[sri lis 16 11:03:14 2019] ath10k: [0248]: 000003FC 00000001 004505F0 00000B6B 1400587C 51100001 000EE214 00000408
[sri lis 16 11:03:14 2019] ath10k: [0256]: 00000002 004505F0 00000B6B 1000587A 0042CD10 004505F0 FF010000 00000000
[sri lis 16 11:03:14 2019] ath10k: [0264]: 00000B6B 14006403 4921F004 26440000 00000000 00000001 004505F0 00000B6C
[sri lis 16 11:03:14 2019] ath10k: [0272]: 10005881 00003112 0042CD10 00000003 FFFFFFFF 00000B6C 10005881 00003112
[sri lis 16 11:03:14 2019] ath10k: [0280]: 0042CD10 0000000E 00000000
[sri lis 16 11:03:14 2019] ath10k_pci 0000:06:00.0: ATH10K_END
[sri lis 16 11:03:19 2019] 0: iter-comb radar-detect-needed: 0x20  combo->radar_detect_widths: 0x0
[sri lis 16 11:03:19 2019] cfg-comb-check: iter-combinations returned num==0
[sri lis 16 11:03:19 2019] wlp6s0: ieee80211_check_combinations:  cfg80211-check-combinations failed: -16
[sri lis 16 11:03:19 2019] wlp6s0: vif-use-channel:  check-combinations failed: -16
[... cut ...]
[sri lis 16 11:03:22 2019] ath10k: [0248]: 000003FC 00000001 004505F0 00000B66 1000587A 0042CD10 004505F0 FF000001
[sri lis 16 11:03:22 2019] ath10k: [0256]: 00000000 00000B66 14006403 00000000 01000000 00000000 00000000 004505F0
[sri lis 16 11:03:22 2019] ath10k: [0264]: 00000B66 10005881 00003112 0042CD10 00000003 FFFFFFFF 00000B66 10005881
[sri lis 16 11:03:22 2019] ath10k: [0272]: 00003112 0042CD10 0000000E 00000000
[sri lis 16 11:03:22 2019] ath10k_pci 0000:06:00.0: ATH10K_END
gs@gs-server:~$ date
sri, 16.10.2019.  11:03:44 CEST

greearb, 2019-10-16:
If you are trying 160Mhz, I fixed a bug with that a week or so ago, maybe you need to re-pull down the kernel if you cloned earlier?

Otherwise, maybe you have regulatory domain issues that are not specifically related to the driver/firmware.

gsustek, 2019-10-16:
i have the latest linux.ct and driver-ct source. I'm located in Europe, and my WRT3200ACS works on 100 and 36 channel with 160Mhz width.
How to confirm out from logs, that i do not have this 160 Mhz solved bug, or regulatory domain issues?

gsustek, 2019-10-16:
i just do:
git clone git clone https://github.com/greearb/linux-ct-5.2.git
cp /boot/config-`uname -r` .config
yes '' | make oldconfig
make -j 2
sudo make modules_install
sudo make install
reboot to the kernel!
###hostpad:sudo -i hostapd -dd /etc/hostapd/hostapd.conf 
random: getrandom() support available
Configuration file: /etc/hostapd/hostapd.conf
ctrl_interface_group=0
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported cipher 00-0f-ac:13
nl80211: Supported cipher 00-0f-ac:11
nl80211: Supported cipher 00-0f-ac:12
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Supported cipher 00-0f-ac:10
nl80211: Using driver-based off-channel TX
nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 04 00 00 00 00 00 00 40
nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8): 04 00 00 00 00 00 00 40
nl80211: use P2P_DEVICE support
nl80211: interface wlp6s0 in phy phy0
nl80211: Set mode ifindex 7 iftype 3 (AP)
nl80211: Failed to set interface 7 to mode 3: -16 (Device or resource busy)
nl80211: Try mode change after setting interface down
nl80211: Set mode ifindex 7 iftype 3 (AP)
nl80211: Mode change succeeded while interface is down
nl80211: Setup AP(wlp6s0) - device_ap_sme=0 use_monitor=0
nl80211: Subscribe to mgmt frames with AP handle 0x55c14cdd97c0
nl80211: Register frame type=0xb0 (WLAN_FC_STYPE_AUTH) nl_handle=0x55c14cdd97c0 match=
nl80211: Register frame type=0x0 (WLAN_FC_STYPE_ASSOC_REQ) nl_handle=0x55c14cdd97c0 match=
nl80211: Register frame type=0x20 (WLAN_FC_STYPE_REASSOC_REQ) nl_handle=0x55c14cdd97c0 match=
nl80211: Register frame type=0xa0 (WLAN_FC_STYPE_DISASSOC) nl_handle=0x55c14cdd97c0 match=
nl80211: Register frame type=0xc0 (WLAN_FC_STYPE_DEAUTH) nl_handle=0x55c14cdd97c0 match=
nl80211: Register frame type=0x40 (WLAN_FC_STYPE_PROBE_REQ) nl_handle=0x55c14cdd97c0 match=
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c14cdd97c0 match=04
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c14cdd97c0 match=0501
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c14cdd97c0 match=0503
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c14cdd97c0 match=0504
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c14cdd97c0 match=06
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c14cdd97c0 match=08
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c14cdd97c0 match=09
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c14cdd97c0 match=0a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c14cdd97c0 match=11
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c14cdd97c0 match=12
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x55c14cdd97c0 match=7f
rfkill: initial event: idx=0 type=1 op=0 soft=0 hard=0
nl80211: Add own interface ifindex 7 (ifidx_reason -1)
nl80211: if_indices[16]: 7(-1)
phy: phy0
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
wlp6s0: interface state UNINITIALIZED->COUNTRY_UPDATE
Previous country code US, new country code FR 
Continue interface setup after channel list update
ctrl_iface not configured!
RTM_NEWLINK: ifi_index=7 ifname=wlp6s0 operstate=2 linkmode=0 ifi_family=0 ifi_flags=0x1002 ()
nl80211: Ignore interface down event since interface wlp6s0 is up
RTM_NEWLINK: ifi_index=7 ifname=wlp6s0 operstate=2 linkmode=0 ifi_family=0 ifi_flags=0x1003 ([UP])
nl80211: Event message available
nl80211: Drv Event 36 (NL80211_CMD_REG_CHANGE) received for wlp6s0
nl80211: Regulatory domain change
 * initiator=1
 * type=0
 * alpha2=FR
[... 232 lines cut ...]
[sri lis 16 21:35:47 2019] ath10k: [0088]: 00000A04 1400587D 51100002 000EE260 00000057 00000000 004505F0 00000A04
[sri lis 16 21:35:47 2019] ath10k: [0096]: 1000587B 0042CD10 004505F0 00000000 0042AA08 00000A25 17FC3C2A 61100001
[sri lis 16 21:35:47 2019] ath10k: [0104]: 01000001 01010000 14640000 00000002 00000A25 13FC5822 9110B000 00000001
[sri lis 16 21:35:47 2019] ath10k: [0112]: 00000001 00000001 00000A45 17FC3C2A 61100001 01000001 01010000 14640000
[sri lis 16 21:35:47 2019] ath10k: [0120]: 00000002 00000A45 13FC5822 9110B000 00000001 00000001 00000001 00000A66
[sri lis 16 21:35:47 2019] ath10k: [0128]: 17FC3C2A 61100001 01000001 01010000 14640000 00000002 00000A66 13FC5822
[sri lis 16 21:35:47 2019] ath10k: [0136]: 9110B000 00000001 00000001 00000001 00000A85 17FC3C2A 61100001 01000001
[sri lis 16 21:35:47 2019] ath10k: [0144]: 01010000 14640000 00000002 00000A85 13FC5822 9110B000 00000001 00000001
[sri lis 16 21:35:47 2019] ath10k: [0152]: 00000001 00000AA6 17FC3C2A 61100001 01000001 01010000 14640000 00000002
[sri lis 16 21:35:47 2019] ath10k: [0160]: 00000AA6 13FC5822 9110B000 00000001 00000001 00000001 00000AC6 17FC3C2A
[sri lis 16 21:35:47 2019] ath10k: [0168]: 61100001 01000001 01010000 14640000 00000002 00000AC6 13FC5822 9110B000
[sri lis 16 21:35:47 2019] ath10k: [0176]: 00000001 00000001 00000001 00000AE7 17FC3C2A 61100001 01000001 01010000
[sri lis 16 21:35:47 2019] ath10k: [0184]: 14640000 00000002 00000AE7 13FC5822 9110B000 00000001 00000001 00000001
[sri lis 16 21:35:47 2019] ath10k: [0192]: 00000B07 17FC3C2A 61100001 01000001 01010000 14640000 00000002 00000B07
[... cut ...]
[sri lis 16 21:35:47 2019] ath10k: [0240]: 9110B000 00000001 00000001 00000001 00000B6A 1400587C 51100001 000EE260
[sri lis 16 21:35:47 2019] ath10k: [0248]: 000003FC 00000001 004505F0 00000B6A 1000587A 0042CD10 004505F0 FF000001
[sri lis 16 21:35:47 2019] ath10k: [0256]: 00000000 00000B6B 14006403 00000000 01000000 00000000 00000000 004505F0
[sri lis 16 21:35:47 2019] ath10k: [0264]: 00000B6B 10005881 00003112 0042CD10 00000003 FFFFFFFF 00000B6B 10005881
[sri lis 16 21:35:47 2019] ath10k: [0272]: 00003112 0042CD10 0000000E 00000000
[sri lis 16 21:35:47 2019] ath10k_pci 0000:06:00.0: ATH10K_END

gsustek, 2019-10-16:
gs@gs-server:~/Music/linux-ct-5.2$ git rev-parse HEAD
56b6398df85ba4610f33dfa6d0fa8351fb4285e6

gsustek, 2019-10-16:
###VHT80 hostpad - conf

sudo cat /etc/hostapd/hostapd.conf 
# For more options, please visit the following:
#  https://w1.fi/cgit/hostap/plain/hostapd/hostapd.conf
driver=nl80211
logger_syslog=-1
logger_syslog_level=2
logger_stdout=-1
logger_stdout_level=2
# a=5GHz, g=2.4GHz
hw_mode=a
# channel=0 turns on ACS survey
#channel=100
# Please take the following into consideration:
# Country code (ISO/IEC 3166-1). Used to set regulatory domain.
# Set as needed to indicate country in which device is operating.
# This can limit available channels and transmit power.
country_code=AU
# Enable IEEE 802.11d. This advertises the country_code and the set of allowed
# channels and transmit power levels based on the regulatory limits. The
# country_code setting must be configured with the correct country for
# IEEE 802.11d functions.
# (default: 0 = disabled)
ieee80211d=1
# Enable IEEE 802.11h. This enables radar detection and DFS support if
# available. DFS support is required on outdoor 5 GHz channels in most countries
# of the world. This can be used only with ieee80211d=1.
# (default: 0 = disabled)
ieee80211h=1
interface=wlp6s0
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0
disassoc_low_ack=1
preamble=1
wmm_enabled=1
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
# Put a 3 here if you want both WPA/WPA2
wpa=2
wpa_passphrase=Ciganja01!
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP
ssid=test-ssid
ieee80211n=1
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][DSSS_CCK-40]
ieee80211ac=1
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=155
vht_capab=[RXLDPC][SHORT-GI-80][SHORT-GI-160][TX-STBC-2BY1][SU-BEAMFORMER][SU-BEAMFORMEE][MU-BEAMFORMER][MU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC1][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

###VHT80Mhz - DFS: working

sudo -i hostapd -dd /etc/hostapd/hostapd.conf 
random: getrandom() support available
Configuration file: /etc/hostapd/hostapd.conf
ctrl_interface_group=0
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported cipher 00-0f-ac:13
nl80211: Supported cipher 00-0f-ac:11
nl80211: Supported cipher 00-0f-ac:12
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Supported cipher 00-0f-ac:10
[... 673 lines cut ...]
nl80211: TX queue param set: queue=3 aifs=7 cw_min=15 cw_max=1023 burst_time=0 --> res=0
wlp6s0: interface state HT_SCAN->ENABLED
wlp6s0: AP-ENABLED 
wlp6s0: Setup of interface done.
RTM_NEWLINK: ifi_index=7 ifname=wlp6s0 operstate=6 linkmode=0 ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
VLAN: RTM_NEWLINK: ifi_index=7 ifname=wlp6s0 ifi_family=0 ifi_flags=0x11043 ([UP][RUNNING][LOWER_UP])
VLAN: vlan_newlink(wlp6s0)
nl80211: Event message available
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlp6s0
nl80211: MLME event 59 (NL80211_CMD_FRAME) on wlp6s0(04:f0:21:49:44:26) A1=ff:ff:ff:ff:ff:ff A2=f0:18:98:46:ab:db
nl80211: MLME event frame - hexdump(len=97): 40 00 00 00 ff ff ff ff ff ff f0 18 98 46 ab db ff ff ff ff ff ff 70 23 00 00 01 08 0c 12 18 24 30 48 60 6c 2d 1a ef 01 17 ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7f 08 00 00 00 80 00 00 00 40 6b 07 0f ff ff ff ff ff ff bf 0c b2 59 82 0f ea ff 00 00 ea ff 00 00
nl80211: Frame event
nl80211: RX frame da=ff:ff:ff:ff:ff:ff sa=f0:18:98:46:ab:db bssid=ff:ff:ff:ff:ff:ff freq=5180 ssi_signal=-42 fc=0x40 seq_ctrl=0x2370 stype=4 (WLAN_FC_STYPE_PROBE_REQ) len=97
nl80211: send_mlme - da= f0:18:98:46:ab:db noack=1 freq=0 no_cck=0 offchanok=0 wait_time=0 fc=0x50 (WLAN_FC_STYPE_PROBE_RESP) nlmode=3
nl80211: send_mlme -> send_frame
nl80211: send_frame - Use bss->freq=5180
nl80211: send_frame -> send_frame_cmd
nl80211: CMD_FRAME freq=5180 wait=0 no_cck=0 no_ack=1 offchanok=0
CMD_FRAME - hexdump(len=221): 50 00 00 00 f0 18 98 46 ab db 04 f0 21 49 44 26 04 f0 21 49 44 26 00 00 00 00 00 00 00 00 00 00 64 00 31 00 00 09 74 65 73 74 2d 73 73 69 64 01 08 8c 12 98 24 b0 48 60 6c 03 01 24 07 12 41 55 20 24 04 17 34 04 14 64 05 1b 84 04 1b 95 05 24 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 0c 00 3b 02 80 00 2d 1a ef 01 1b ff ff ff ff 00 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 3d 16 24 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7f 08 04 00 00 02 00 00 00 40 bf 0c f2 18 88 33 aa ff 18 06 aa ff 18 06 c0 05 01 2a 00 fc ff c3 04 02 2e 2e 2e dd 18 00 50 f2 02 01 01 01 00 03 a4 00 00 27 a4 00 00 42 43 5e 00 62 32 2f 00
nl80211: Frame TX command accepted (no ACK); cookie 0x0
nl80211: Event message available
nl80211: BSS Event 59 (NL80211_CMD_FRAME) received for wlp6s0
nl80211: MLME event 59 (NL80211_CMD_FRAME) on wlp6s0(04:f0:21:49:44:26) A1=ff:ff:ff:ff:ff:ff A2=f0:18:98:46:ab:db
nl80211: MLME event frame - hexdump(len=97): 40 00 00 00 ff ff ff ff ff ff f0 18 98 46 ab db ff ff ff ff ff ff 80 23 00 00 01 08 0c 12 18 24 30 48 60 6c 2d 1a ef 01 17 ff ff ff 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7f 08 00 00 00 80 00 00 00 40 6b 07 0f ff ff ff ff ff ff bf 0c b2 59 82 0f ea ff 00 00 ea ff 00 00
nl80211: Frame event

greearb, 2019-10-16:
I don't have time to dig into this problem, but I did test 160Mhz 9984 using our firmware and driver recently and it worked OK.  Others have also used 160Mhz mode on openwrt (see recently opened bugs, it has issues with 1x1 160Mhz stations it seems, haven't had time to fix that either)

gsustek, 2019-10-16:
can you give me your working hostapd configuration and iw phy info and iw reg get? i have this output for country_ AU
Frequencies:
			* 5180 MHz [36] (23.0 dBm)
			* 5200 MHz [40] (23.0 dBm)
			* 5220 MHz [44] (23.0 dBm)
			* 5240 MHz [48] (23.0 dBm)
			* 5260 MHz [52] (20.0 dBm) (no IR, radar detection)
			* 5280 MHz [56] (20.0 dBm) (no IR, radar detection)
			* 5300 MHz [60] (20.0 dBm) (no IR, radar detection)
			* 5320 MHz [64] (20.0 dBm) (no IR, radar detection)
			* 5500 MHz [100] (27.0 dBm) (no IR, radar detection)
			* 5520 MHz [104] (27.0 dBm) (no IR, radar detection)
			* 5540 MHz [108] (27.0 dBm) (no IR, radar detection)
			* 5560 MHz [112] (27.0 dBm) (no IR, radar detection)
			* 5580 MHz [116] (27.0 dBm) (no IR, radar detection)
			* 5600 MHz [120] (disabled)
			* 5620 MHz [124] (disabled)
			* 5640 MHz [128] (disabled)
			* 5660 MHz [132] (27.0 dBm) (no IR, radar detection)
			* 5680 MHz [136] (27.0 dBm) (no IR, radar detection)
			* 5700 MHz [140] (27.0 dBm) (no IR, radar detection)
			* 5720 MHz [144] (27.0 dBm) (radar detection)
			* 5745 MHz [149] (30.0 dBm)
			* 5765 MHz [153] (30.0 dBm)
			* 5785 MHz [157] (30.0 dBm)
			* 5805 MHz [161] (30.0 dBm)
			* 5825 MHz [165] (30.0 dBm)
			* 5845 MHz [169] (disabled)
			* 5865 MHz [173] (disabled)
	Supported commands:

gsustek, 2019-10-16:
fixed!
To be able to have 160MHZ, we need to enable DFS support in kernel: 
ap ~ # dmesg
...
[    2.850086] ath10k_pci 0000:04:00.0: debug 1 debugfs 1 tracing 0 dfs 0 testmode 0
...
now i enabled
[    2.850086] ath10k_pci 0000:04:00.0: debug 1 debugfs 1 tracing 0 dfs 1 testmode 0

The output indicates that DFS is disabled (“dfs 0”). To enable DFS you must enable following options:

“Configure standard kernel features (expert users)” under “General Setup”
“cfg80211 certification onus” under “Networking support > Wireless”
“Atheros DFS support for certified platforms” under “Device Drivers > Network device support > Wireless LAN > Atheros Wireless Cards”
Re-compile the kernel and verify if the kernel output indicates DFS support. Then, re-enable DFS and choose a higher channel of your liking.

Thnx to http://pisarenko.net/blog/2015/05/05/updates-to-the-802-dot-11ac-guide/

gsustek, 2019-10-17:
This fix works only with regular driver from linux-ct-5.2/ ath10k drivers, and transfer rate is a disaster. So i'm searching to resolve it with ath10k-c drivers.

closed 2019-10-29 by gsustek