greearb/ath10k-ct #101: IBSS mode doesn't work on QCA9888

ammarahm1, 2019-10-18:
I am trying to construct an ad-hoc network using two COMFAST CF-E313AC units. I flashed them with OpenWRT and configured them to work in IBSS mode, without using any wireless encryption. The two units don't see each other wirelessly, as each one creates a different BSSID.
Here is the dmesg:
```
[    0.000000] Linux version 4.19.78 (ammar@ubuntu) (gcc version 8.3.0 (OpenWrt GCC 8.3.0 r11190-2b1a0f2)) #0 Sat Oct 12 10:09:02 2019
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] MIPS: machine is COMFAST CF-E313AC
[    0.000000] SoC: Qualcomm Atheros QCA9533 ver 2 rev 0
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
[    0.000000] random: get_random_bytes called from start_kernel+0x98/0x4a8 with crng_init=0
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 57580K/65536K available (4273K kernel code, 179K rwdata, 1008K rodata, 1188K init, 206K bss, 7956K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 650.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 5880801374 ns
[    0.000009] sched_clock: 32 bits at 325MHz, resolution 3ns, wraps every 6607641598ns
[    0.008270] Calibrating delay loop... 432.53 BogoMIPS (lpj=2162688)
[    0.074882] pid_max: default: 32768 minimum: 301
[    0.080021] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.086996] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.100195] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.110646] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.117263] pinctrl core: initialized pinctrl subsystem
[    0.124043] NET: Registered protocol family 16
[    0.137157] PCI host bridge /ahb/pcie-controller@180c0000 ranges:
[    0.143684]  MEM 0x0000000010000000..0x0000000013ffffff
[    0.149182]   IO 0x0000000000000000..0x0000000000000000
[    0.181734] PCI host bridge to bus 0000:00
[    0.186165] pci_bus 0000:00: root bus resource [mem 0x10000000-0x13ffffff]
[    0.193404] pci_bus 0000:00: root bus resource [io  0x0000]
[    0.199282] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[    0.206436] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.216286] pci 0000:00:00.0: BAR 0: assigned [mem 0x10000000-0x101fffff 64bit]
[    0.228654] clocksource: Switched to clocksource MIPS
[    0.235601] NET: Registered protocol family 2
[    0.241233] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
[    0.249485] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.256846] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.263586] TCP: Hash tables configured (established 1024 bind 1024)
[    0.270496] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.276680] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.283864] NET: Registered protocol family 1
[    0.292862] Crashlog allocated RAM at address 0x3f00000
[    0.300490] workingset: timestamp_bits=14 max_order=14 bucket_order=0
[    0.315502] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.321733] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.350232] io scheduler noop registered
[    0.354370] io scheduler deadline registered (default)
[    0.362361] pinctrl-single 1804002c.pinmux: 576 pins, size 72
[    0.369718] Serial: 8250/16550 driver, 1 ports, IRQ sharing disabled
[    0.377258] console [ttyS0] disabled
[    0.381153] 18020000.uart: ttyS0 at MMIO 0x18020000 (irq = 9, base_baud = 1562500) is a 16550A
[    0.390272] console [ttyS0] enabled
[    0.390272] console [ttyS0] enabled
[... 123 lines cut ...]
[   46.830758] ath10k_pci 0000:00:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[   49.148690] wlan0: Trigger new scan to find an IBSS to join
[   49.170967] wlan0: Trigger new scan to find an IBSS to join
[   53.148683] wlan0: Trigger new scan to find an IBSS to join
[   53.265202] wlan0: Trigger new scan to find an IBSS to join
[   53.271247] wlan0: Creating new IBSS network, BSSID 52:d1:b1:be:6d:b9
[   53.503086] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   60.896792] ath10k_pci 0000:00:00.0: Invalid peer id 0 or peer stats buffer, peer:   (null)  sta:   (null)

```
Here is the output of ``` cat /sys/kernel/debug/ieee80211/phy0/ath10k/firmware_info ```:
```
directory: ath10k/QCA9888/hw2.0
firmware:  ct-firmware-5.bin
fwcfg:     fwcfg-pci-0000:00:00.0.txt
bus:       0000:00:00.0
features:  mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-T
version:   10.4b-ct-9888-fH-012-54863dff
hw_rev:    9888
board:     board.bin
```
Here is the output of ``` sha256sum /lib/firmware/ath10k/QCA9888/hw2.0/ct-firmware-5.bin ```:
```
dcb1bd826e5e1ef266fd7ee04410b44d4474d59f6eca0cc634e6432aaf326426  /lib/firmware/ath10k/QCA9888/hw2.0/ct-firmware-5.bin
```

Catfriend1, 2019-12-28:
I have the same problem - see bug report:
https://bugs.openwrt.org/index.php?do=details&task_id=2702

Catfriend1, 2020-01-05:
I've also filed it at: https://forum.openwrt.org/t/archer-c7-v2-kernel-warn-comm-wpa-supplicant-not-tainted-4-14-156/51664

Today I noticed despite the 5ghz radio ap+ibss still working - 2.4ghz radio does not work. If o disable the ad-hoc ibss wifi on radio0 5ghz , 2.4ghz starts working again and the crash disappears.