greearb/ath10k-ct #102: different download speed on QCA9984 chipset by different CT firmware

gsustek, 2019-10-29:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).

Openwrt x86 installation on x86 hardware with Compex QCA9984 as wifi.
Testing scenario: 
Sata connected SSD 120G, continuous transfer rate of 220MB/s read. file 10GB.
vsftpd daemon serving file
Two routers equally configured with two same sata drive and same file, same vsftpd conf, default openwrt config:

WRT3200acs : OpenWRT
- 36 channel , 160Mhz channel width, WPA2 - mvebu driver and firmware
 Client: macbook 15" 2018 . - filezilla client
Client <-- WiFi << Router <--SSD 
                           99MB/s

KANSUNG with Compex QCA9984 as wifi: OpenWRT
- 36channel,80Mhz channel width, WPA2 ath10kct driver 4.19 and firmware
Client <-- WiFi << Router <--SSD 
                           72MB/s with firmware-5-ct-non-commercial-full-htt-mgt-11.bin  or 10,9,8
                           52MB/s with any other firmware-5-ct-****

KANSUNG with Compex QCA9984 as wifi: OpenWRT
- 36channel,80Mhz channel width, WPA2 ath10k vanilla driver and kvalo firmware
Client <-- WiFi << Router <--SSD 
                           60MB/s

KANSUNG with Compex QCA9984 as wifi: DD-WRT
- 36channel,80Mhz channel width, WPA2 ath10k vanilla driver and kvalo firmware
Client <-- WiFi << Router <--SSD 
                           99MB/s

Software (OS, Firmware version, kernel, driver, etc)
openwrt- latest trunk

Hardware (NIC chipset, platform, etc)
NIC: Compex COMPEX WLE1216V5-20 5GHz
Platform: KANSUNG Model: K515G6
(https://bit.ly/2IOlSeQ)

Logs (dmesg, maybe supplicant and/or hostap)

Why is this firmware firmware-5-ct-non-commercial-full-htt-mgt-11.bin  the fastest one?

greearb, 2019-10-29:
If the ct fw version 11 is faster than recent ct-fw, then I can build a series of FW for you to bisect to find where the slow-down happened.  Would you like to try that?

gsustek, 2019-10-29:
any version(numeric one) of this type of firmware:  firmware-5-ct-non-commercial-full-htt-mgt-XX.bin is faster then other types of firmware of the same numeric version.
i would like to test for better throughput.

greearb, 2019-10-29:
htt mgt can give some advantages in cases where mgt frames may have trouble getting through or there are lots of stations (and so lots of mgt frames).  I suggest using htt-mgt firmware.

greearb, 2019-10-30:
Just drop the CT firmware in /lib/firmware/ath10k/*/ location (save a copy of the old firmware) and reboot.  Should not require recompiling.