greearb/ath10k-ct #103: Nokia 8.1 frequently disconnects and reconnects to network

huaracheguarache, 2019-10-29:
Quite often my Nokia 8.1 will disconnect from my network and quickly reconnect, i.e around 1 to 6 times an hour. The disconnects do not seem to have any specific pattern as far as I can see. I believe I have experienced similar problems in the past with my Nexus 6P, but I did not get around to write a bug report. Other devices such as a couple of laptops, an iPhone and and iPad do not seem to experience this issue.

I am running the latest OpenWrt master with a patch to use kernel 4.19 on the R7800. The firmware version is listed below:

```
root@telia:~# ethtool -i wlan0
driver: ath10k_pci
version: 4.19.79
firmware-version: 10.4b-ct-9984-fH-012-54863dff2
expansion-rom-version: 
bus-info: 0000:01:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

Here is my network config:

```
root@telia:~# cat /etc/config/wireless 

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11a'
	option path 'soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'VHT80'
	option country 'NO'
	option legacy_rates '0'
	option channel '112'
	option log_level '0'
	option txpower '26'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option encryption 'psk2+ccmp'
	option ssid 'OpenWrt'
	option key 'redacted'
	option ieee80211w '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option channel '11'
	option hwmode '11g'
	option path 'soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0'
	option htmode 'HT20'
	option country 'NO'
	option legacy_rates '0'
	option txpower '20'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option encryption 'psk2+ccmp'
	option ieee80211w '1'
	option ssid 'OpenWrt_'
	option key 'redacted'
```

Here are logs from four disconnect/reconnect events I have recorded:

https://gist.github.com/huaracheguarache/ac3efeaf8906522d98197650d5026ce8

huaracheguarache, 2019-10-29:
Yes, please do build the firmware. I will try to find the bad commit.

greearb, 2019-10-31:
Here are notes on debugging and specifically, on how to bisect:  https://www.candelatech.com/ath10k-bugs.php
Please see this link for a series of all 9984 CT firmware images:
http://www.candelatech.com/downloads/ath10k-9984-10-4b/bisect/

huaracheguarache, 2019-11-04:
Just wanted to give you a quick update: I'm trying out firmwares, but the bug doesn't show up as often as I initially described. The one time I experienced up to 6 disconnects in an hour must have been an outlier, so I'm slowly but surely working my way through the firmwares.

huaracheguarache, 2019-11-04:
Sometimes the bug does not show up for more than an hour, which is an issue when I test older firmware images that have their own bugs since I cannot be sure what triggered a disconnect or a firmware crash. I have tried to find ways to reliably trigger the bug, but I have not found any. Unfortunately, other people who use the network require a stable connection, and I cannot keep testing the images.

So far I have provided you dmesg logs with debug_level 0xc0000032. If setting a different debug_level can help you debug this issue I am open to sending you more logs.

huaracheguarache, 2019-11-09:
Could a different log level help you?

CFSworks, 2020-01-06:
Did a bisect of my own; the latest version not to log a `10005881` message is `firmware-5-full-htt-mgt-community-commit-210-083191268.bin`, while the earliest version that does is `firmware-5-full-htt-mgt-community-commit-259-f41383662.bin`.

I don't think I can test any of the versions between those. 211 and 258 fail to bring up the interface (and I've tried a few other random versions in that range as well with similar results).

huaracheguarache, 2020-01-10:
@greearb, I could try to capture a Wireshark log of block-ack frames between the Nokia 8.1 and the R7800. Would that be something of interest?

huaracheguarache, 2020-01-11:
@CFSworks What does a `10005881` message mean?

CFSworks, 2020-01-11:
@huaracheguarache By that I mean a `ATH10K_DBG_BUFFER` appearing in `dmesg` with `10005881` appearing as the second word (which I believe is the debug message type; the first increases steadily at a rate of about 1024/sec so it looks to be a timestamp or some other counter).

huaracheguarache, 2020-01-11:
But what does `10005881` mean? Do you know that it's relevant to this bug?

CFSworks, 2020-01-11:
> But what does `10005881` mean?
That would be a question for @greearb, but since the ath10k firmware is protected by NDA, I don't know that he is at liberty to answer that question specifically. He did say `The logs show that block-ack setup is failing.`, which is a clue that it may be related to block-ack in some way.

> Do you know that it's relevant to this bug?
Yes, it appears in the logs you posted.

huaracheguarache, 2020-01-11:
Ah, ok! Just to make it clear, I thought that you had some insider knowledge about the driver/firmware that prompted you to look for `10005881` in the dmesg during your bisect. Thanks for elaborating!

huaracheguarache, 2020-01-13:
I haven't heard back from you but I decided to run a capture anyway: https://www.dropbox.com/s/n3z2mcjbot567d7/wireshark_capture.pcapng?dl=0

It took some time before the disconnect/reconnect happened so I commented a frame where it happens. Just click on the small green light (expert info button) in the lower left corner to quickly find the commented frame.

greearb, 2020-01-23:
In frame 65817, the Nokia sends disconnect.  I guess you are running PMF, so I cannot decode any reason code in the deauth message.  In frame 6815, it goes to sleep.  I don't see any obvious problems in the capture.

CFSworks:  When you bisected what bug are you seeing, and what did you use as bisect pass/fail criteria?

huaracheguarache, 2020-01-23:
Yeah, I have PMF enabled. I'll try to make a capture without it enabled.

CFSworks, 2020-01-24:
@greearb Hi!

For me this issue manifests as high packet loss with broadcast/multicast Ethernet traffic (and my devices purge out the IPv6 default route due to missing such a high percentage of router advertisements), and roughly a similar percentage of unicast frames have the retry flag set.

Some other devices on the same WLAN randomly disconnect as well, much as described in the issue report above.

My debug logs greatly resemble the ones posted here, so I bisected until I didn't see `10005881`. The 210 firmware is the latest that doesn't give those logs, doesn't have disconnects and (significant) packet loss, and while it has a few issues of its own, it's at least stable enough for my daily use.

If you'd like I can try running 259 for a day to see if the broadcast packet loss and random disconnects come back.

huaracheguarache, 2020-01-24:
I just ran a capture without PMF enabled. It took quite some time for the disconnect to happen so I commented the relevant frame again: https://www.dropbox.com/s/q6mlbhicrwmgcn4/wireshark_capture2.pcapng?dl=0

Here's also a log from the router with debug level `0xc0000032`: https://gist.github.com/huaracheguarache/e2549058012ad2d9ff796d7147732d1e

greearb, 2020-01-24:
I don't see any obvious reason why the Nokia would decide to disconnect.  The AP is acking frames sent by the Nokia, so it should not think the connection is bad.

CFSworks, please attach a log of one of those early builds with the 10005881 logging output so I can decode it to see what you are talking about.

huaracheguarache, 2020-01-25:
I've been sending you captures of frames between only the Nokia and the R7800. I assume it's not relevant, but would a capture will all the traffic be of interest?

Also, have you ever worked with logs from Android devices to debug connectivity issues? I can capture a log from my Nokia if you would like me to.

-------- Original Message --------

greearb, 2020-01-25:
If you can figure out why the Nokia device thinks it should disconnect, that might help.  Supplicant logs or similar would probably be the best, maybe 'dmesg' output would also help if that is available. I doubt capture of other devices would help.

huaracheguarache, 2020-01-25:
I'm currently testing out a theory that IPv6 might be the cause of this. I searched around and found out that other people with Android phones have been experiencing random WiFi disconnects with reports that disabling IPv6 resolved the issue. 

I captured a log that seems to indicate some issues with an IPv6 route (search for NETWORK_SELECTION_DISABLED_DHCP_FAILURE, and check out a few of the proceeding entries): https://gist.github.com/huaracheguarache/247bebc26797d2f854ced6c53ab42640

From what I understand the people that disabled IPv6 completely disabled it. I'm going to try out something less drastic first by only disabling the 6rd interface while still keeping local IPv6 running. If that fails I'll try to disable IPv6 completely. Either way, I'll need to test things for a few days to make sure that I've found the cause.

CFSworks, 2020-01-26:
@huaracheguarache Based on your theory, I suspect we're actually having the same exact issue: the default IPv6 route goes away. The only difference is in how our devices react: mine silently gives up on IPv6 support, yours abandons the WiFi connection.

For me, the IPv6 route goes away due to high packet loss on router advertisement packets. Actually, I suspect there's high packet loss on **any** packet that the QCA9984 tries to transmit, but the AP expects an ACK on unicast packets and retries until it gets one. IPv6 router advertisements are multicast, however. (Those are "best effort," right?)

Instead of disabling IPv6, would you be willing to try this on the router:
```
root@router:~# uci set dhcp.lan.ra_mininterval=5
root@router:~# uci set dhcp.lan.ra_maxinterval=20
root@router:~# uci commit
root@router:~# /etc/init.d/odhcpd restart
```

This reconfigures IPv6 route advertisements to be sent every 5-20 seconds (the default is every 200-600 seconds, and each received one resets the validity period to 1800 seconds). The idea here is to fight against the packet loss on RAs by making odhcpd very chatty. If the Nokia stops disconnecting, we know it's due to the IPv6 default route expiring.

@greearb I still owe you that log, but I think this problem is due to the PHY silently failing to transmit some frames. I'm working to get set up where I can get some packet captures on a nice quiet channel, and do a bisect there. I'll share both the capture and associated logs once I have them.

huaracheguarache, 2020-01-26:
I tried what you suggested and it causes my Nokia to disconnect every 1.5 minutes. I think there's something seriously wrong with the way Android handles IPv6.

greearb, 2020-01-26:
If you can bisect, here is a new series for 9984:

http://www.candelatech.com/downloads/ath10k-9984-10-4b/bisect/all_builds-9984-H-jan-27-2020.tar.gz

huaracheguarache, 2020-01-26:
Ok, here are some things that I've found out which are worth noting:

- The Nokia doesn't experience any issues while connected to the 2.4GHz network.
- Even with the suggested settings of `ra_mininterval '5'` and `ra_maxinterval '20'`, which made the Nokia disconnect and reconnect every 1.5 minutes on 5GHz, it has a steady connection on 2.4GHz.

I can probably try to bisect, but it will take some time since I will have to test the firmwares at times when I won't be interrupting the work of others in the house.

huaracheguarache, 2020-01-26:
@greearb What's new with the bisect series compared to the last one?

It's a shame that vanilla ath10k doesn't work on the R7800 ([see flyspray bug report](https://bugs.openwrt.org/index.php?do=details&task_id=2480)). That way I could have have tested and compared to see if this is some ath10k-ct specific issue or not.

CFSworks, 2020-01-26:
> I tried what you suggested and it causes my Nokia to disconnect every 1.5 minutes, while all the other devices seem to work just fine on the network. I think there's something seriously wrong with the way Android handles IPv6.

The documentation misled me, sorry about that: you should also set ra_lifetime=1800 because the default lifetime is 3x the max interval (meaning you were using 1 minute - makes sense that it'd make disconnects worse)

The plus side is the deceased lifetime makes detecting the bug a lot easier for a bisect.

huaracheguarache, 2020-01-26:
@CFSworks No worries! I took a look at the documentation and it's indeed incorrect.

I've been using the new suggested settings for about 1.5 hours without any disconnects so far, and I'll be keeping a close eye on the router log to look for disconnects.

huaracheguarache, 2020-01-27:
It seems like your observations about packet loss are true. I ran Wireshark to capture router advertisements on my laptop, and if you compare the capture to the log from the R7800 you can see that a lot of the packets get lost. 

Here's the Wireshark capture: https://www.dropbox.com/s/nllybma5mwgnv4v/packet_loss.pcapng?dl=0

And here's the router log (you can see the Nokia, with MAC 94:ee:9f:06:db:37, disconnect 3 times): https://gist.github.com/huaracheguarache/19bf029ff83045f10f1043edd527d796

I ran this with:
```
option ra_mininterval '5'
option ra_maxinterval '20'
option ra_lifetime '60'
```

These observations were made with 6 clients on the 5GHz network and 2 clients on the 2.4GHz network. When I tried the same thing with only 4 mostly idle clients in total, all on the 5GHz network, I did not experience anywhere near the level of packet loss seen here.

greearb, 2020-01-27:
Can you tell if packets are transmitted by the AP or not?  In other words, is it a failure to transmit, or a failure to receive?

huaracheguarache, 2020-01-27:
No, I can't tell that. Is there a way I can set up a capture or do something else to figure that out?

greearb, 2020-01-27:
If you have a third device to act as sniffer, then sniff on the AP to see when it thinks it is transmitting the frames, and sniff on other device on-air to see if you see the frame on-air, and compare packet captures.

But that said, that is still liable to be tricky to debug, so bisect is probably the best way to find the root cause.  I'm building another set of new images now to help with another bug, will let you know when I have those ready.

CFSworks, 2020-01-28:
@huaracheguarache Would you be willing to repeat your capture of router advertisements with the power save feature of your laptop's WiFi adapter disabled?

On Linux this can be done with `sudo iw dev [wifidevice] set power_save off`, on Windows you can look under the Power Management tab of the adapter's properties page. Barring that, a flood ping (100 pings/sec ought to do it) directed at the router might keep the laptop's WiFi adapter from going into power save.

On WiFi, the AP can immediately transmit multicast/broadcast packets only if there are no clients in power-save mode, otherwise they must be buffered until after the DTIM (a part of every Nth beacon indicating whether there are buffered multicast packets, which power-saving clients use to determine if they should wake up).

My two main hypotheses right now are:
1. The AP is buffering the multicast frame but losing it in the buffer - so it's never transmitted.
2. The AP is transmitting the multicast frame, but timing it incorrectly with respect to the DTIM. (If this is the case, turning off power save on the client should make the packet loss go away - but of course we'd still need to identify the root cause.)

@greearb I'm still willing to do a bisect, but I'm going to need to track down a fast way to determine whether or not a given firmware version is affected before I begin. The issue doesn't seem to occur at all on a quiet channel. Should I be using the series from your comment [here](https://github.com/greearb/ath10k-ct/issues/103#issuecomment-578526050) or wait for the series you're building now? What changed in the past day?

greearb, 2020-01-28:
The new series will contain the fix for #50, and also re-sorts the patches so hopefully the bug that was bisected in #114 happens later in the tree or not at all.  It will be another 24 hours or so before the series is complete.  If you have a fast bisect, you may certainly find the bug in the previous series I uploaded before I complete this series.

CFSworks, 2020-01-28:
@greearb Here's our first look at the actual issue:
[image]

I set a coloring rule to display the TIM multicast bit in green and ran a multicast ping to `ff02::1`. Packets 22811 and 22812 show normal operation. (And everything above that point looks healthy to me.)

With packet 22827, something odd happens: a multicast packet shows up, but the AP just blasts it out right away in spite of sleeping STAs (22823, 22825). The very next beacon, 22828, has the multicast bit set in its TIM, however the DTIM count is 1 (22829 has a DTIM count of 0, but it unset the multicast bit). The only ATH10K_DBG_BUFFER log at that same second is (which I doubt is interesting; pretty much the same log came up every 6 seconds):
```
[71197.285922] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[71197.286000] ath10k: [0000]: 000E5CB0 10005881 00003112 00430C40 0000000A 00000000 000E5CB0 10005881
[71197.291036] ath10k: [0008]: 00003112 00430C40 0000000E 00000000
[71197.299996] ath10k_pci 0000:01:00.0: ATH10K_END
```

Beyond this point, the multicast flag is never set in the TIM; the QCA9984 just continues as if it had no sleeping STAs. What broke it out of this state was when I disconnected one STA and the Mac Mini next exited and reentered power save. Some bug in the code meant to count the STAs in power save, perhaps?

This issue seems to happen only when there are 3+ STAs associated. I was using the 5GHz radio in 802.11N mode, 20MHz bandwidth, WPA/WEP disabled, firmware version 17ba98334.

At this point I know exactly what I'm looking for in a bisect. I'll do that as soon as you have the new series available (unless the above is enough information to track down the problem without it... I hope!)

greearb, 2020-01-28:
Thanks for the detailed debugging.  I think a bisect would be very helpful, this sounds like a tricky bug and it would be good to know what patch broke the behaviour.  The log message is nothing about multicast, just the driver setting some config.  You could probably bisect the existing tarball, or wait another day and I'll have new builds ready.

huaracheguarache, 2020-01-28:
I can confirm that disabling power save makes the packet loss disappear. Here are the Wireshark captures where the capture with power save turned off captured all but two RA's (these were two RA's that were sent right after two clients connected, so I assume that they are unicast packets directed to those clients):

[Power save ON](https://www.dropbox.com/s/t8cyuqpag5f2y42/capture_poweron.pcapng?dl=0)
[Power save OFF](https://www.dropbox.com/s/hlvqfgayewzcaie/capture_poweroff.pcapng?dl=0)

And here's the router log (log_poweron.txt is the one with power save on, and log_poweroff.txt is the one with power save off):

https://gist.github.com/huaracheguarache/2fa31853a6ece194fd28475eef1df745

I won't have a lot of time to debug the issue the next few days, but once I have some time to spare I'll try to bisect, if there will be a need for it.

greearb, 2020-01-29:
If someone can bisect, here is a new series that just finished building:  http://www.candelatech.com/downloads/ath10k-9984-10-4b/bisect/all_builds-9984-H-jan-28-2020.tar.gz

CFSworks, 2020-01-31:
@greearb 

I used the jan-28-2020 archive and the following methodology:

1. Upload the firmware under test to the device, perform a full reboot. (This is easier and less error-prone than reloading just the QCA firmware.)
2. Connect 3 STAs.
3. Use a shell while loop to send a flood ping to `ff02::1` (multicast) for 10 seconds, followed by a 10 second sleep, repeat. (I don't think the intensity of the flood matters, since the goal is just to get the STAs to cycle power saving on and off several times per minute.)
4. If all STAs remain connected and responsive to multicast ping for 10 full minutes, mark the firmware version as good.
5. If all STAs stop receiving multicast (tends to happen within 1-2 minutes), switch to a continuous 1 pkt/sec ping and toggle power saving on one of the STAs several times to verify that the multicast issue is related to power saving, then mark the firmware version as bad.

The **latest good version**: `582-737d67dfa`
The **earliest bad version**: `583-24d181b7a`

Let me know if/when you have a patch you'd like me to test. Normally I'd remain on `582-737d67dfa` and follow up if the issue resurfaces, but it seems to be very unstable under full load, so I'll be remaining on `210-083191268` for the time being.

greearb, 2020-01-31:
Please see if this one works.  It is top-of-tree with patch 583 reverted.  That 583 is tricky tx-scheduler change, evidently it fixed a mesh problem, but maybe the fix is incomplete.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/4137461/firmware-5-full-htt-mgt-community.bin.gz)

CFSworks, 2020-01-31:
I'm not able to cause the multicast problem with the provided firmware, so I'll gladly run it for a few days to see if I run into any other trouble. I'll refrain from calling it "working" for the purposes of this issue report, though, since the original report is about the Nokia 8.1 failing to stay connected - so I'll defer to @huaracheguarache there.

huaracheguarache, 2020-01-31:
Thanks for doing the bisect, @CFSworks! I'll try to run the firmware with the 583 reverted for a few days to make sure that everything works fine.

huaracheguarache, 2020-01-31:
I just ran a packet capture of RA's on my laptop, and comparing it to the router log gives me a packet loss of a bit over 40 percent. It's not as severe as the loss I observed earlier on the firmware with 538, but it's far greater than the approximately 2 percent I observe on the latest non-ct ath10k firmware (https://github.com/kvalo/ath10k-firmware/tree/master/QCA9984/hw1.0/3.10).

greearb, 2020-02-01:
I'm going to build a new series that moves patch 583 to top of the tree.  Seems it is not root cause of huaracheguarache's issues, so that way you can bisect a tree without the questionable patch.

CFSworks, 2020-02-02:
@huaracheguarache Rather than troubleshooting RAs, it might be faster if we switch our approach to pings to `ff02::1` (this is the "all IPv6 nodes" multicast address). We can do those on-demand and quickly see the multicast performance.

On this `10.4b-ct-9984-fH-013-826165c2b`, I am seeing:
```
root@router:~# ping6 -I br-lan ff02::1 | grep -v ff.fe
PING ff02::1 (ff02::1): 56 data bytes
64 bytes from fe80::9d81:5cde:7d70:e865: seq=1 ttl=64 time=1038.887 ms (DUP!)
64 bytes from fe80::2b98:d076:81a2:8ae8: seq=1 ttl=64 time=1166.077 ms (DUP!)
64 bytes from fe80::9d81:5cde:7d70:e865: seq=2 ttl=64 time=2084.333 ms (DUP!)
64 bytes from fe80::2b98:d076:81a2:8ae8: seq=2 ttl=64 time=2185.868 ms (DUP!)
64 bytes from fe80::9d81:5cde:7d70:e865: seq=3 ttl=64 time=1769.250 ms (DUP!)
64 bytes from fe80::9d81:5cde:7d70:e865: seq=4 ttl=64 time=2449.026 ms (DUP!)
64 bytes from fe80::2b98:d076:81a2:8ae8: seq=4 ttl=64 time=2461.011 ms (DUP!)
64 bytes from fe80::9d81:5cde:7d70:e865: seq=8 ttl=64 time=3446.696 ms (DUP!)
64 bytes from fe80::9d81:5cde:7d70:e865: seq=12 ttl=64 time=5017.735 ms (DUP!)
64 bytes from fe80::2b98:d076:81a2:8ae8: seq=13 ttl=64 time=5092.410 ms (DUP!)
64 bytes from fe80::9d81:5cde:7d70:e865: seq=14 ttl=64 time=4187.626 ms (DUP!)
64 bytes from fe80::2b98:d076:81a2:8ae8: seq=14 ttl=64 time=4203.930 ms (DUP!)
64 bytes from fe80::9d81:5cde:7d70:e865: seq=16 ttl=64 time=2797.223 ms (DUP!)
64 bytes from fe80::9d81:5cde:7d70:e865: seq=17 ttl=64 time=1799.286 ms (DUP!)
64 bytes from fe80::9d81:5cde:7d70:e865: seq=18 ttl=64 time=798.719 ms (DUP!)
```

So, there's still packet loss with multicast traffic, but this doesn't seem to be caused by the AP sending the traffic during power save (and as well, turning off power save doesn't change the behavior at all). I'm inclined to call the first regression fixed, but there's this second regression which is still resulting in multicast packet loss (and high latency to boot). I have no idea whether this second regression was caused by reverting 583 or is unrelated.

This is an issue that *seems* to occur only after the AP has been running for a little while (if so that's going to make bisect pretty difficult), although I wonder if it's related to this crash-and-recovery:
<details><summary>Crash report from dmesg</summary>

<p>
[48664.621399] ath10k_pci 0001:01:00.0: firmware crashed! (guid 16fc1f6a-5e47-47ed-ac4d-583f553ff7cf)
[48664.621500] ath10k_pci 0001:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[48664.629322] ath10k_pci 0001:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[48664.644208] ath10k_pci 0001:01:00.0: firmware ver 10.4b-ct-9984-fH-013-826165c2b api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 37171cde
[48664.656172] ath10k_pci 0001:01:00.0: board_file api 2 bmi_id 0:2 crc32 85498734
[48664.677125] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[48664.686201] ath10k_pci 0001:01:00.0: firmware register dump:
[48664.694223] ath10k_pci 0001:01:00.0: [00]: 0x0000000A 0x00000000 0x009B2EAE 0x00000000
[48664.700122] ath10k_pci 0001:01:00.0: [04]: 0x00000000 0x00060024 0x00000000 0x00000000
[48664.707850] ath10k_pci 0001:01:00.0: [08]: 0x00000000 0x00000000 0x00000000 0x00000000
[48664.715747] ath10k_pci 0001:01:00.0: [12]: 0x00000000 0x00000000 0x00000000 0x00000000
[48664.723645] ath10k_pci 0001:01:00.0: [16]: 0x009700BB 0x00991BB9 0x009B1832 0x009B2EAE
[48664.731545] ath10k_pci 0001:01:00.0: [20]: 0x00000000 0x00401C10 0x00000000 0x00000000
[48664.739442] ath10k_pci 0001:01:00.0: [24]: 0x00000000 0x00000000 0x00000000 0x00000000
[48664.747342] ath10k_pci 0001:01:00.0: [28]: 0x00000000 0x00000000 0x00000000 0x00000000
[48664.755242] ath10k_pci 0001:01:00.0: [32]: 0x00000000 0x00000000 0x00000000 0x00000000
[48664.763140] ath10k_pci 0001:01:00.0: [36]: 0x00000000 0x00000000 0x00000000 0x00000000
[48664.771040] ath10k_pci 0001:01:00.0: [40]: 0x00000000 0x00000000 0x00000000 0x00000000
[48664.778940] ath10k_pci 0001:01:00.0: [44]: 0x00000000 0x00000000 0x00000000 0x00000000
[48664.786839] ath10k_pci 0001:01:00.0: [48]: 0x00000000 0x00000000 0x00000000 0x00000000
[48664.794740] ath10k_pci 0001:01:00.0: [52]: 0x00000000 0x00000000 0x00000000 0x00000000
[48664.802638] ath10k_pci 0001:01:00.0: [56]: 0x00000000 0x00000000 0x00000000 0x00000000
[48664.810536] ath10k_pci 0001:01:00.0: Copy Engine register dump:
[48664.818442] ath10k_pci 0001:01:00.0: [00]: 0x0004a000   7   7   3   3
[48664.824257] ath10k_pci 0001:01:00.0: [01]: 0x0004a400   5   5 107 108
[48664.830855] ath10k_pci 0001:01:00.0: [02]: 0x0004a800  49  49  48  49
[48664.837277] ath10k_pci 0001:01:00.0: [03]: 0x0004ac00  26  26  29  26
[48664.843702] ath10k_pci 0001:01:00.0: [04]: 0x0004b000 7378 7378 147 107
[48664.850128] ath10k_pci 0001:01:00.0: [05]: 0x0004b400  14  14 480 494
[48664.856549] ath10k_pci 0001:01:00.0: [06]: 0x0004b800  25  25  25  25
[48664.863145] ath10k_pci 0001:01:00.0: [07]: 0x0004bc00   1   1   1   1
[48664.869570] ath10k_pci 0001:01:00.0: [08]: 0x0004c000   0   0 127   0
[48664.875992] ath10k_pci 0001:01:00.0: [09]: 0x0004c400   1   1   1   1
[48664.882419] ath10k_pci 0001:01:00.0: [10]: 0x0004c800   0   0   0   0
[48664.888839] ath10k_pci 0001:01:00.0: [11]: 0x0004cc00   0   0   0   0
[48664.897296] ath10k_pci 0001:01:00.0: debug log header, dbuf: 0x423018  dropped: 0
[48664.902708] ath10k_pci 0001:01:00.0: [0] next: 0x423000 buf: 0x419630 sz: 1500 len: 28 count: 1 free: 0
[48664.910173] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[48664.918350] ath10k: [0000]: 028D5E4B 14006402 71103332 CF3431D8 0000144A 0042DCC0 00000001
[48664.924597] ath10k_pci 0001:01:00.0: ATH10K_END
</p>
</details>

CFSworks, 2020-02-02:
<details><summary>Here are some ATH10K_DBG logs for the severe multicast delay</summary>
<p>

```
[156180.172228] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156180.172327] ath10k: [0000]: 011E1178 13FC5875 91103779 004569DC 00000010 000008C0
[156180.177354] ath10k_pci 0000:01:00.0: ATH10K_END
[156181.172384] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156181.172480] ath10k: [0000]: 011E1591 13FC5875 91103779 004569DC 00000010 000008C1 011E1662 13FC5875
[156181.177523] ath10k: [0008]: 91103779 004569DC 00000010 000008C1 011E1734 13FC5875 91103779 004569DC
[156181.186892] ath10k: [0016]: 00000010 000008C0
[156181.195977] ath10k_pci 0000:01:00.0: ATH10K_END
[156182.172963] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156182.173013] ath10k: [0000]: 011E19A9 13FC5875 91103779 004569DC 00000010 000008C1 011E1A7B 13FC5875
[156182.178084] ath10k: [0008]: 91103779 004569DC 00000010 000008C1 011E1B4D 13FC5875 91103779 004569DC
[156182.187563] ath10k: [0016]: 00000010 000008C0
[156182.196650] ath10k_pci 0000:01:00.0: ATH10K_END
[156183.172869] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156183.172952] ath10k: [0000]: 011E1DC2 13FC5875 91103779 004569DC 00000010 000008C1 011E1E93 13FC5875
[156183.177978] ath10k: [0008]: 91103779 004569DC 00000010 000008C0 011E1F87 10005881 00003112 004303D0
[156183.187352] ath10k: [0016]: 0000000A 00000000 011E1F87 10005881 00003112 004303D0 0000000E 00000000
[156183.196484] ath10k_pci 0000:01:00.0: ATH10K_END
[156183.390766] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156183.390860] ath10k: [0000]: 068FD833 10005881 00003112 004303D0 0000000A 00000000 068FD833 10005881
[156183.395980] ath10k: [0008]: 00003112 004303D0 0000000E 00000001
[156183.405264] ath10k_pci 0001:01:00.0: ATH10K_END
[156184.173006] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156184.173103] ath10k: [0000]: 011E21DA 13FC5875 91103779 004569DC 00000010 000008C1 011E22AC 13FC5875
[156184.178127] ath10k: [0008]: 91103779 004569DC 00000010 000008C0
[156184.187503] ath10k_pci 0000:01:00.0: ATH10K_END
[156185.173171] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156185.173253] ath10k: [0000]: 011E25F3 13FC5875 91103779 004569DC 00000010 000008C1 011E26C4 13FC5875
[156185.178302] ath10k: [0008]: 91103779 004569DC 00000010 000008C0 011E2796 13FC5875 91103779 004569DC
[156185.187668] ath10k: [0016]: 00000010 000008C1
[156185.196773] ath10k_pci 0000:01:00.0: ATH10K_END
[156186.173674] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156186.173757] ath10k: [0000]: 011E27FF 13FC5875 91103779 004569DC 00000010 000008C0 011E2939 13FC5875
[156186.178818] ath10k: [0008]: 91103779 004569DC 00000010 000008C1 011E2A0B 13FC5875 91103779 004569DC
[156186.188156] ath10k: [0016]: 00000010 000008C1 011E2ADD 13FC5875 91103779 004569DC 00000010 000008C1
[156186.197291] ath10k: [0024]: 011E2BAF 13FC5875 91103779 004569DC 00000010 000008C1
[156186.206380] ath10k_pci 0000:01:00.0: ATH10K_END
[156187.173945] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156187.174044] ath10k: [0000]: 011E2C17 13FC5875 91103779 004569DC 00000010 000008C0 011E2D52 13FC5875
[156187.179054] ath10k: [0008]: 91103779 004569DC 00000010 000008C0 011E2E24 13FC5875 91103779 004569DC
[156187.188447] ath10k: [0016]: 00000010 000008C0 011E2EF5 13FC5875 91103779 004569DC 00000010 000008C0
[156187.197632] ath10k_pci 0000:01:00.0: ATH10K_END
[156188.174142] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156188.174197] ath10k: [0000]: 011E3099 13FC5875 91103779 004569DC 00000010 000008C1 011E316A 13FC5875
[156188.179263] ath10k: [0008]: 91103779 004569DC 00000010 000008C0 011E323C 13FC5875 91103779 004569DC
[156188.188721] ath10k: [0016]: 00000010 000008C1 011E330E 13FC5875 91103779 004569DC 00000010 000008C1
[156188.197838] ath10k: [0024]: 011E33E0 13FC5875 91103779 004569DC 00000010 000008C1
[156188.206914] ath10k_pci 0000:01:00.0: ATH10K_END
[156189.174493] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156189.174581] ath10k: [0000]: 011E34B1 13FC5875 91103779 004569DC 00000010 000008C1 011E3583 13FC5875
[156189.179624] ath10k: [0008]: 91103779 004569DC 00000010 000008C0 011E3655 13FC5875 91103779 004569DC
[156189.189008] ath10k: [0016]: 00000010 000008C1 011E3726 13FC5875 91103779 004569DC 00000010 000008C1
[156189.198107] ath10k: [0024]: 011E3789 10005881 00003112 004303D0 0000000A 00000000 011E3789 10005881
[156189.207203] ath10k: [0032]: 00003112 004303D0 0000000E 00000000
[156189.216381] ath10k_pci 0000:01:00.0: ATH10K_END
[156189.391560] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[156189.391653] ath10k: [0000]: 068FF035 10005881 00003112 004303D0 0000000A 00000000 068FF035 10005881
[156189.396690] ath10k: [0008]: 00003112 004303D0 0000000E 00000001
[156189.406115] ath10k_pci 0001:01:00.0: ATH10K_END
```
</p></details>

greearb, 2020-02-02:
Here is a new series to bisect.  The very last commit is the problematic 583 that I previously reverted for testing.
http://www.candelatech.com/downloads/ath10k-9984-10-4b/bisect/all_builds-9984-H-feb-2-2020.tar.gz
I only built from (the new) 583 on.

huaracheguarache, 2020-02-02:
First of all, what is considered to be normal latency in a multicast ping? I just ran this multicast ping with the last firmware (the one with the problematic 583) in the provided series and got these results:

```

```

huaracheguarache, 2020-02-02:
Sorry about that! I zoned out and pressed the wrong button. 

A few moments ago I got these results with the last firmware in that series (the one with the problematic 583 patch): 

```
root@telia:~# ping ff02::1
PING ff02::1 (ff02::1): 56 data bytes
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=0 ttl=64 time=0.244 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=0 ttl=64 time=63.023 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=0 ttl=64 time=63.363 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=0 ttl=64 time=135.064 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=1 ttl=64 time=0.208 ms
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=1 ttl=64 time=83.910 ms (DUP!)
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=1 ttl=64 time=84.250 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=1 ttl=64 time=153.545 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=2 ttl=64 time=0.240 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=2 ttl=64 time=107.339 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=2 ttl=64 time=107.679 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=2 ttl=64 time=177.066 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=3 ttl=64 time=0.208 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=3 ttl=64 time=130.401 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=3 ttl=64 time=130.867 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=3 ttl=64 time=200.825 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=4 ttl=64 time=0.213 ms
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=4 ttl=64 time=153.456 ms (DUP!)
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=4 ttl=64 time=153.794 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=4 ttl=64 time=218.587 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=5 ttl=64 time=0.211 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=5 ttl=64 time=176.046 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=5 ttl=64 time=176.333 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=5 ttl=64 time=240.865 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=6 ttl=64 time=0.204 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=6 ttl=64 time=198.409 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=6 ttl=64 time=198.841 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=6 ttl=64 time=263.453 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=7 ttl=64 time=0.226 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=7 ttl=64 time=15.744 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=7 ttl=64 time=16.181 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=7 ttl=64 time=86.068 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=8 ttl=64 time=0.219 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=8 ttl=64 time=37.617 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=8 ttl=64 time=38.052 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=8 ttl=64 time=102.968 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=9 ttl=64 time=0.208 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=9 ttl=64 time=59.794 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=9 ttl=64 time=60.172 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=9 ttl=64 time=125.808 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=10 ttl=64 time=0.219 ms
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=10 ttl=64 time=81.356 ms (DUP!)
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=10 ttl=64 time=81.728 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=10 ttl=64 time=96.692 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=11 ttl=64 time=0.214 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=11 ttl=64 time=104.689 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=11 ttl=64 time=104.946 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=11 ttl=64 time=175.507 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=12 ttl=64 time=0.221 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=12 ttl=64 time=127.509 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=12 ttl=64 time=127.945 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=12 ttl=64 time=199.211 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=13 ttl=64 time=0.210 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=13 ttl=64 time=151.208 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=13 ttl=64 time=151.703 ms (DUP!)
64 bytes from fe80::183d:2f01:1827:e6cb: seq=13 ttl=64 time=154.049 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=13 ttl=64 time=221.899 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=14 ttl=64 time=0.215 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=14 ttl=64 time=174.791 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=14 ttl=64 time=175.038 ms (DUP!)
64 bytes from fe80::183d:2f01:1827:e6cb: seq=14 ttl=64 time=177.719 ms (DUP!)
64 bytes from fe80::96ee:9fff:fe06:db37: seq=14 ttl=64 time=189.623 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=15 ttl=64 time=0.202 ms
[... 293 lines cut ...]
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=96 ttl=64 time=11.494 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=97 ttl=64 time=0.223 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=97 ttl=64 time=1.282 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=97 ttl=64 time=2.017 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=98 ttl=64 time=0.214 ms
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=98 ttl=64 time=1.376 ms (DUP!)
[... cut ...]
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=102 ttl=64 time=1.313 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=102 ttl=64 time=1.870 ms (DUP!)
64 bytes from fe80::a240:a0ff:fe7c:cbb7: seq=103 ttl=64 time=0.215 ms
64 bytes from fe80::4ad6:d5ff:fe00:581e: seq=103 ttl=64 time=1.057 ms (DUP!)
64 bytes from fe80::4bdb:471a:24e5:beb6: seq=103 ttl=64 time=1.472 ms (DUP!)
```

The values I got with the very first firmware in that series are closer to the topmost example I provided. That makes things very tricky to bisect since I don't really know what to look for.

CFSworks, 2020-02-04:
@huaracheguarache Hopefully the "teach a man to fish" approach will help you track this down more quickly - here's a quick primer on some of the technical aspects involved:

Under the default AP settings (beacon interval 100, DTIM period 2), and when there are any WiFi clients in power saving mode, multicast packets may only be transmitted during a "wake up" window that occurs every 204.8 ms. This is why your first block of ping responses shows an increase in delay of 20ms for each ping - that's the "beat" between pings being sent every 1000ms and then being buffered until the next multiple of 204.8ms; think of it as a network [Moiré pattern](https://en.wikipedia.org/wiki/Moir%C3%A9_pattern) of sorts.

So, I can see that your wireless clients are `fe80::4ad6:d5ff:fe00:581e`, `fe80::4bdb:471a:24e5:beb6`, and `fe80::96ee:9fff:fe06:db37`, because they all follow this pattern. (`fe80::96ee:9fff:fe06:db37` looks like it's on a different wireless interface than the other 2 - presumably you have 2 on 5GHz and 1 on 2.4GHz, or vice versa - hence why its delay pattern is out of sync.) The thing responding in under 0.25 ms is most likely the R7800 itself.

After a few moments, the regression from 583 appears - the QCA firmware appears to believe (mistakenly) that there are no clients in power saving mode (that is, all clients are receiving 100% of the time), which is the only circumstance where it's acceptable to send multicast frames immediately without buffering them until the "wake up" window. Unfortunately there are still clients in power-saving mode, so this only serves to prevent the power-saving clients from receiving multicast traffic. (Including IPv6 RAs.)

In your second block of ping responses, there's no delay on the multicast traffic w/ wireless clients, so you can see that the AP is *not* buffering. As well, power-saving clients aren't receiving the pings:
- `fe80::4bdb:471a:24e5:beb6` responds intermittently, this is probably because it's entering and exiting power saving mode regularly as it uses its interface for other (unrelated) traffic.
- `fe80::96ee:9fff:fe06:db37` just disappears altogether, which suggests that it's largely idle (and has no reason to exit power saving mode).
- `fe80::4ad6:d5ff:fe00:581e` is regularly receiving the multicast traffic, so it's probably configured not to go into (or does not support) power saving mode.

So, the tell-tale signs of the 583 regression are:
- Multicast traffic stops getting delayed, even when there are power-saving clients
- Idle devices stop receiving multicast traffic altogether (or receive only intermittently)
- The problem goes away for clients when you disable their `power_save` setting, and returns when that setting is reenabled.

As for this [second regression](https://github.com/greearb/ath10k-ct/issues/103#issuecomment-580908788) (if there is one? we still need to confirm it), we need to figure out if reverting 583 introduced it or simply allowed it to be found. I presume you're currently running `firmware-5-full-htt-mgt-community-commit-1137-83d76a491.bin` (from that most recent series), so once you've gotten familiar with how to recognize this problem, see how the behavior of 1136 differs?

greearb, 2020-02-09:
Can you try the 'y3' build from bug 114?  It has a mcast related fix found by bisect in that bug, and is missing the problem patch found in this bug.

huaracheguarache, 2020-02-09:
@CFSworks I would like to thank you for your patience and your very detailed analysis and tutorial. It's been very instructive!

@greearb First things first: the y3 build appears to be running fine. The RA packet loss seems to be at 5-7 percent (what's actually considered to be normal?), which is much less than the almost 60 percent I experienced on the firmware with 583 reverted. In addition, multicast pings do not display the extreme high latency and packet loss previously experienced.

On to the second thing: I have been trying to bisect the February series to locate the commit that causes the extreme latency and packet loss in multicast pings. I have gotten to the point that 855 seems to be the earliest instance where I see the ping issue, but due to limited time I have not been able to finish the bisect. If this bug is related to #114 then the issue likely presents itself much earlier in the series. If you want me to I can continue to bisect.

greearb, 2020-02-19:
Can you please test the y6 and  z7 image in bug 114?  Hopefully y6 works at least as good as y3 did for you.  If z7 acts poorly, then I would like to ask your help with debugging that further.

huaracheguarache, 2020-02-20:
I'm unfortunately unable to test them right now, but I'll try them out during the weekend.

huaracheguarache, 2020-02-23:
I tested y6 and z7, and I also decided to retest y3. I have no idea how the previous tests I ran with y3 had such good results (perhaps I inadvertently disabled power management). I tested the RA packet loss:

| Firmware | Power Management ON | Power Management OFF |
| ------------ | -------------------------------- | ---------------------------------- |
| y3 | 41% | 4% |
| y6 | 60% | 19% |
| z7 | 89% | 6 % |

greearb, 2020-03-05:
In case someone is interested in bisecting, I put up a new series of 9984 firmware images.  This fixes some bugs found in earlier bisect attempts.  I suspect there are bugs in the last 4 commits as well.
https://www.candelatech.com/downloads/ath10k-9984-10-4b/bisect/all_builds-9984-H-march-5-2020.tar.gz

CFSworks, 2020-03-09:
I don't know that I can be of much help; `firmware-5-full-htt-mgt-community-commit-1140-b0d0aff2a.bin` has the same multicast-not-respecting-PS regression and `firmware-5-full-htt-mgt-community-commit-1139-ca8927cee.bin` doesn't.

However, after about an hour, the latter begins throwing `ath10k_pci 0001:01:00.0: failed to increase tx pending count: -16, dropping` every few dozen seconds. I restarted the interface and got a firmware crash in the process.

<details>

<summary>Firmware crash</summary>

```
[ 9315.581152] ath10k_pci 0000:01:00.0: firmware crashed! (guid d72af22b-9ad0-4b90-8de0-c761b4d1911d)
[ 9315.581187] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[ 9315.589071] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[ 9315.600937] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fH-013-ca8927cee api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 34c12f39
[ 9315.615230] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 85498734
[ 9315.636891] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[ 9315.655892] ath10k_pci 0000:01:00.0: failed to get memcpy hi address for firmware address 4: -16
[ 9315.655919] ath10k_pci 0000:01:00.0: failed to read firmware dump area: -16
[ 9315.663732] ath10k_pci 0000:01:00.0: in crash-regs-harder
[ 9316.387114] ath10k_pci 0000:01:00.0: in crash-regs-harder, firmware did not provide indicator: 0xdeadbeef
[ 9316.387143] ath10k_pci 0000:01:00.0: Copy Engine register dump:
[ 9316.395713] ath10k_pci 0000:01:00.0: [00]: 0x0004a000 3735928559 3735928559 3735928559 3735928559
[ 9316.401386] ath10k_pci 0000:01:00.0: [01]: 0x0004a400 3735928559 3735928559 3735928559 3735928559
[ 9316.410488] ath10k_pci 0000:01:00.0: [02]: 0x0004a800 3735928559 3735928559 3735928559 3735928559
[ 9316.419331] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00 3735928559 3735928559 3735928559 3735928559
[ 9316.428170] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 3735928559 3735928559 3735928559 3735928559
[ 9316.437043] ath10k_pci 0000:01:00.0: [05]: 0x0004b400 3735928559 3735928559 3735928559 3735928559
[ 9316.445884] ath10k_pci 0000:01:00.0: [06]: 0x0004b800 3735928559 3735928559 3735928559 3735928559
[ 9316.454685] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00 3735928559 3735928559 3735928559 3735928559
[ 9316.463596] ath10k_pci 0000:01:00.0: [08]: 0x0004c000 3735928559 3735928559 3735928559 3735928559
[ 9316.472437] ath10k_pci 0000:01:00.0: [09]: 0x0004c400 3735928559 3735928559 3735928559 3735928559
[ 9316.481311] ath10k_pci 0000:01:00.0: [10]: 0x0004c800 3735928559 3735928559 3735928559 3735928559
[ 9316.490151] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00 3735928559 3735928559 3735928559 3735928559
[ 9316.499027] ath10k_pci 0000:01:00.0: failed to get memcpy hi address for firmware address 8: -28
[ 9316.507860] ath10k_pci 0000:01:00.0: failed to dump debug log area: -28
[ 9316.516737] ath10k_pci 0000:01:00.0: failed to get memcpy hi address for firmware address 72: -28
[ 9316.523001] ath10k_pci 0000:01:00.0: failed to get memcpy hi address for firmware address 72: -28
[ 9316.532073] ath10k_pci 0000:01:00.0: failed to get memcpy hi address for firmware address 76: -28
[ 9316.540948] ath10k_pci 0000:01:00.0: failed to get memcpy hi address for firmware address 76: -28
[ 9316.549789] ath10k_pci 0000:01:00.0: failed to read firmware RAM BSS memory from 4291056 (46768 B): -28
[ 9316.558659] ath10k_pci 0000:01:00.0: failed to read firmware ROM BSS memory from 4197376 (12552 B): -28
[ 9316.568300] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 9316.577285] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 9316.584532] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 9316.591847] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
```

</details>

greearb, 2020-03-11:
That crash is because the firmware quit communicating with the driver and/or died horribly for some reason.  Unfortunately, I cannot debug it since the driver was unable to get any debug info out of the firmware.

I am attaching a new image.  It has the 1140 commit pulled before the other three questionable commits.  Please test to see if it exhibits the previous bad behaviour you saw with 1140.
[firmware-5-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/4319309/firmware-5-full-htt-mgt-community.bin.gz)

CFSworks, 2020-03-15:
I've been running this for about 36 hours so far with no trouble. I'll report back if I have problems, but otherwise assume no news is good news.

CFSworks, 2020-03-16:
<details>
<summary>Firmware crash</summary>

```
[102353.380236] ath10k_pci 0001:01:00.0: failed to increase tx pending count: -16, dropping
[104384.049280] ath10k_pci 0001:01:00.0: firmware crashed! (guid 0ff30f30-e767-4d29-a9c1-2d0917cf2078)
[104384.049471] ath10k_pci 0001:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[104384.057222] ath10k_pci 0001:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[104384.070983] ath10k_pci 0001:01:00.0: firmware ver 10.4b-ct-9984-fH-013-ca8927cee api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 34c12f39
[104384.083976] ath10k_pci 0001:01:00.0: board_file api 2 bmi_id 0:2 crc32 85498734
[104384.105255] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[104384.114761] ath10k_pci 0001:01:00.0: firmware register dump:
[104384.122867] ath10k_pci 0001:01:00.0: [00]: 0x0000000A 0x00000000 0x009B2EEA 0x00000000
[104384.128512] ath10k_pci 0001:01:00.0: [04]: 0x00000000 0x00060024 0x00000000 0x00000000
[104384.136323] ath10k_pci 0001:01:00.0: [08]: 0x00000000 0x00000000 0x00000000 0x00000000
[104384.144309] ath10k_pci 0001:01:00.0: [12]: 0x00000000 0x00000000 0x00000000 0x00000000
[104384.152296] ath10k_pci 0001:01:00.0: [16]: 0x009700BB 0x00991BA8 0x00963B15 0x009B2EEA
[104384.160282] ath10k_pci 0001:01:00.0: [20]: 0x00000000 0x00401C10 0x00000000 0x00000000
[104384.168269] ath10k_pci 0001:01:00.0: [24]: 0x00000000 0x00000000 0x00000000 0x00000000
[104384.176255] ath10k_pci 0001:01:00.0: [28]: 0x00000000 0x00000000 0x00000000 0x00000000
[104384.184240] ath10k_pci 0001:01:00.0: [32]: 0x00000000 0x00000000 0x00000000 0x00000000
[104384.192227] ath10k_pci 0001:01:00.0: [36]: 0x00000000 0x00000000 0x00000000 0x00000000
[104384.200212] ath10k_pci 0001:01:00.0: [40]: 0x00000000 0x00000000 0x00000000 0x00000000
[104384.208199] ath10k_pci 0001:01:00.0: [44]: 0x00000000 0x00000000 0x00000000 0x00000000
[104384.216184] ath10k_pci 0001:01:00.0: [48]: 0x00000000 0x00000000 0x00000000 0x00000000
[104384.224171] ath10k_pci 0001:01:00.0: [52]: 0x00000000 0x00000000 0x00000000 0x00000000
[104384.232157] ath10k_pci 0001:01:00.0: [56]: 0x00000000 0x00000000 0x00000000 0x00000000
[104384.240142] ath10k_pci 0001:01:00.0: Copy Engine register dump:
[104384.248135] ath10k_pci 0001:01:00.0: [00]: 0x0004a000   7   7   3   3
[104384.254384] ath10k_pci 0001:01:00.0: [01]: 0x0004a400  13  13 115 116
[104384.260720] ath10k_pci 0001:01:00.0: [02]: 0x0004a800  45  45  44  45
[104384.267230] ath10k_pci 0001:01:00.0: [03]: 0x0004ac00  20  20  23  20
[104384.273740] ath10k_pci 0001:01:00.0: [04]: 0x0004b000  74  74  79  39
[104384.280251] ath10k_pci 0001:01:00.0: [05]: 0x0004b400   5   5  24  37
[104384.286762] ath10k_pci 0001:01:00.0: [06]: 0x0004b800  13  13  13  13
[104384.293271] ath10k_pci 0001:01:00.0: [07]: 0x0004bc00   1   1   1   1
[104384.299782] ath10k_pci 0001:01:00.0: [08]: 0x0004c000   0   0 127   0
[104384.306292] ath10k_pci 0001:01:00.0: [09]: 0x0004c400   1   1   1   1
[104384.312803] ath10k_pci 0001:01:00.0: [10]: 0x0004c800   0   0   0   0
[104384.319314] ath10k_pci 0001:01:00.0: [11]: 0x0004cc00   0   0   0   0
[104384.327850] ath10k_pci 0001:01:00.0: debug log header, dbuf: 0x423000  dropped: 0
[104384.333351] ath10k_pci 0001:01:00.0: [0] next: 0x423018 buf: 0x419040 sz: 1500 len: 28 count: 1 free: 0
[104384.340902] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[104384.349518] ath10k: [0000]: 0172512D 14006402 71103332 CF3431D8 0000144A 0042DCC0 00000001
[104384.355505] ath10k_pci 0001:01:00.0: ATH10K_END
[104384.364854] ath10k_pci 0001:01:00.0: [1] next: 0x423000 buf: 0x419630 sz: 1500 len: 0 count: 0 free: 0
```

</details>

[3 more replies]

closed 2020-04-30 by greearb