greearb/ath10k-ct #105: unable to set Transmit Packet Steering "xps_cpus"

gsustek, 2019-11-04:
system: ubuntu eaon
driver ath10k-ct , any firmware,board-bin

root@gs-Default-string:/home/gs# ls -lrt /sys/class/net/wlp6s0/queues/tx-0/xps_cpus
-rw-r--r-- 1 root root 4096 stu   4 17:03 /sys/class/net/wlp6s0/queues/tx-0/xps_cpus
root@gs-Default-string:/home/gs# cat /sys/class/net/wlp6s0/queues/tx-0/xps_cpus
cat: /sys/class/net/wlp6s0/queues/tx-0/xps_cpus: No such file or directory
root@gs-Default-string:/home/gs# 

root@gs-Default-string:/home/gs# ls -lrt /sys/class/net/
total 0
lrwxrwxrwx 1 root root 0 stu   4 16:50 lo -> ../../devices/virtual/net/lo
lrwxrwxrwx 1 root root 0 stu   4 16:50 enp5s0 -> ../../devices/pci0000:00/0000:00:1c.4/0000:05:00.0/net/enp5s0
lrwxrwxrwx 1 root root 0 stu   4 16:50 enp4s0 -> ../../devices/pci0000:00/0000:00:1c.3/0000:04:00.0/net/enp4s0
lrwxrwxrwx 1 root root 0 stu   4 16:50 enp3s0 -> ../../devices/pci0000:00/0000:00:1c.2/0000:03:00.0/net/enp3s0
lrwxrwxrwx 1 root root 0 stu   4 16:50 enp2s0 -> ../../devices/pci0000:00/0000:00:1c.1/0000:02:00.0/net/enp2s0
lrwxrwxrwx 1 root root 0 stu   4 16:50 enp1s0 -> ../../devices/pci0000:00/0000:00:1c.0/0000:01:00.0/net/enp1s0
lrwxrwxrwx 1 root root 0 stu   4 16:50 wlp6s0 -> ../../devices/pci0000:00/0000:00:1d.0/0000:06:00.0/net/wlp6s0
root@gs-Default-string:/home/gs# 

for other interfaces are able to set, read,

greearb, 2019-11-04:
Does this work with upstream driver?

What about upstream driver plus ct firmware?

gsustek, 2019-11-04:
"What about upstream driver plus ct firmware? " not working, how is this possible? is it hardware issue? by compex? board bin or firmware?

greearb, 2019-11-04:
Does it work with plain upstream driver plus upstream firmware?

gsustek, 2019-11-05:
No, so in openwrt and ubuntu, you can not view or  change, queues/tx-0/xps_cpus. 
I wonder why is that, was hoping that it is a firmware or driver issue, is it hardwer issue?

greearb, 2019-11-05:
Well, whatever is the problem, it is very unlikely to be an issue with the firmware, and it is at least not a regression in the driver, so closing this bug.  Maybe ask about the feature on the ath10k mailing list.

closed 2019-11-05 by greearb