greearb/ath10k-ct #107: .sta_add support missing

MartB, 2019-12-03:
The ath10k-ct driver is currently missing the `.sta_add` implementation, this is needed to get 802.11r (FT) working as hostapd uses it.
```
Tue Dec  3 22:51:23 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:ff:ff:ff IEEE 802.11: binding station to interface 'wlan0'
Tue Dec  3 22:51:23 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:ff:ff:ff IEEE 802.11: authentication OK (FT)
Tue Dec  3 22:51:23 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:ff:ff:ff MLME: MLME-AUTHENTICATE.indication(90:2b:d2:ff:ff:ff, FT)
Tue Dec  3 22:51:23 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:ff:ff:ff IEEE 802.11: association OK (aid 1)
Tue Dec  3 22:51:23 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:ff:ff:ff IEEE 802.11: Could not set STA to kernel driver
```

for reference see ath9k with exposed support: https://github.com/kvalo/ath10k/blob/master/drivers/net/wireless/ath/ath9k/main.c#L2362

PS: Im new to wireless stuff in the linux kernel so if this report is invalid please tell me where i went wrong please.

greearb, 2019-12-03:
Can you run hostapd with verbose logging enabled and also attached 'dmesg' output from when
this problem happens?

We've tested .11r in AP and STA mode with ath10k-ct and it has worked well, so I'm not
sure what problem you are hitting.

Thanks,
Ben

-- 
Ben Greear <greearb@candelatech.com>
Candela Technologies Inc  http://www.candelatech.com

MartB, 2019-12-03:
@greearb In order to reproduce this i need to walk around the house which is gonna be an issue right now. I will tackle this tomorrow and provide all the output you requested.

Is there any additional step required besides setting the hostapd log_level to 0 for it to output verbose?

jeolives, 2019-12-04:
Just to clarify, do you have 802.11w enabled at the same time as 802.11r? 

Have you also tried testing FT-over-the-Air instead of FT-over-the-DS?

MartB, 2019-12-04:
@jeolives 802.11w is enabled as optional, just like in the issue i linked.
Im tried both FTA and FTDS, both result in the same behavior.

jeolives, 2019-12-04:
There seems to be a few fixes with 802.11r FT + 802.11w PMF that didn't make it into the v2.9 tag... 

https://w1.fi/cgit/hostap/commit/?id=ded56f2fafb00090118f82a135aa63b393cc66cf
https://w1.fi/cgit/hostap/commit/?id=528f263c4a4bc0255a429e502336534b0a52fe35

But you should still wait for Jouni Malinen or someone else for their input.

MartB, 2019-12-04:
In case its relevant it happened again, but this time its completely unrelated to 802.11r

```
Wed Dec  4 16:01:24 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: authentication OK (open system)
Wed Dec  4 16:01:24 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX WPA: event 0 notification
Wed Dec  4 16:01:24 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX MLME: MLME-AUTHENTICATE.indication(90:2b:d2:XX:XX:XX, OPEN_SYSTEM)
Wed Dec  4 16:01:24 2019 daemon.info hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: authenticated
Wed Dec  4 16:01:24 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association SA Query attempt 1
Wed Dec  4 16:01:24 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association SA Query attempt 2
Wed Dec  4 16:01:25 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association SA Query attempt 3
Wed Dec  4 16:01:25 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association SA Query attempt 4
Wed Dec  4 16:01:25 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association SA Query attempt 5
Wed Dec  4 16:01:25 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association SA Query timed out
Wed Dec  4 16:01:25 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association OK (aid 1)
Wed Dec  4 16:01:25 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: Could not set STA to kernel driver
Wed Dec  4 16:01:28 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: authentication OK (open system)
Wed Dec  4 16:01:28 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX WPA: event 0 notification
Wed Dec  4 16:01:28 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX MLME: MLME-AUTHENTICATE.indication(90:2b:d2:XX:XX:XX, OPEN_SYSTEM)
Wed Dec  4 16:01:28 2019 daemon.info hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: authenticated
Wed Dec  4 16:01:28 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association OK (aid 1)
Wed Dec  4 16:01:28 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: Could not set STA to kernel driver
Wed Dec  4 16:01:28 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association OK (aid 1)
Wed Dec  4 16:01:28 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: Could not set STA to kernel driver
Wed Dec  4 16:01:29 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association OK (aid 1)
Wed Dec  4 16:01:29 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: Could not set STA to kernel driver
Wed Dec  4 16:01:29 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association OK (aid 1)
Wed Dec  4 16:01:29 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: Could not set STA to kernel driver
Wed Dec  4 16:01:29 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association OK (aid 1)
Wed Dec  4 16:01:29 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: Could not set STA to kernel driver
Wed Dec  4 16:01:32 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: authentication OK (open system)
Wed Dec  4 16:01:32 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX WPA: event 0 notification
Wed Dec  4 16:01:32 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX MLME: MLME-AUTHENTICATE.indication(90:2b:d2:XX:XX:XX, OPEN_SYSTEM)
Wed Dec  4 16:01:32 2019 daemon.info hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: authenticated
Wed Dec  4 16:01:32 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association OK (aid 1)
Wed Dec  4 16:01:32 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: Could not set STA to kernel driver
Wed Dec  4 16:01:32 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association OK (aid 1)
Wed Dec  4 16:01:32 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: Could not set STA to kernel driver
Wed Dec  4 16:01:33 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association OK (aid 1)
Wed Dec  4 16:01:33 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: Could not set STA to kernel driver
Wed Dec  4 16:01:33 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association OK (aid 1)
Wed Dec  4 16:01:33 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: Could not set STA to kernel driver
Wed Dec  4 16:01:33 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: association OK (aid 1)
Wed Dec  4 16:01:33 2019 daemon.notice hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: Could not set STA to kernel driver
Wed Dec  4 16:01:40 2019 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 90:2b:d2:XX:XX:XX
Wed Dec  4 16:01:40 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.11: authentication OK (open system)
Wed Dec  4 16:01:40 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX WPA: event 0 notification
Wed Dec  4 16:01:40 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX MLME: MLME-AUTHENTICATE.indication(90:2b:d2:XX:XX:XX, OPEN_SYSTEM)
Wed Dec  4 16:01:40 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX MLME: MLME-DELETEKEYS.request(90:2b:d2:XX:XX:XX)
[... cut ...]
Wed Dec  4 16:01:45 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 90:2b:d2:XX:XX:XX
Wed Dec  4 16:01:45 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.1X: authorizing port
Wed Dec  4 16:01:45 2019 daemon.info hostapd: wlan1: STA 90:2b:d2:XX:XX:XX RADIUS: starting accounting session XXXXXXXXXXXXXX
Wed Dec  4 16:01:45 2019 daemon.info hostapd: wlan1: STA 90:2b:d2:XX:XX:XX IEEE 802.1X: authenticated - EAP type: 25 (PEAP)
Wed Dec  4 16:01:45 2019 daemon.debug hostapd: wlan1: STA 90:2b:d2:XX:XX:XX WPA: Added PMKSA cache entry (IEEE 802.1X)
```

greearb, 2019-12-04:
If this is not specific to ath10k, please report the bug elsewhere so it has better chance of being fixed.

MartB, 2019-12-04:
> If this is not specific to ath10k, please report the bug elsewhere so it has better chance of being fixed.

yeah i already wrote to the hostapd mailing list, waiting for the reply will close this for now.

closed 2019-12-04 by MartB