greearb/ath10k-ct #108: upstream-ath10k pcie DMA discussion

MartB, 2019-12-18:
Hey there found the following discussion while browsing patches: 
https://patchwork.kernel.org/patch/11204029/

Looks interesting too me.

greearb, 2019-12-18:
That patch did not go upstream yet as far as I can tell, and Felix's original patch claims to fix problems as well.  So, maybe this needs more testing or investigation before it is worth applying.

nicklowe, 2020-06-30:
The final upstream commit can be found at: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=a1769bb68a850508a492e3674ab1e5e479b11254

_"For QCA9984/QCA9888, the DMA_BURST_SIZE register controls the AXI burst size
of the RD/WR access to the HOST MEM.
0 - No split , RAW read/write transfer size from MAC is put out on bus
    as burst length
1 - Split at 256 byte boundary
2,3 - Reserved

With PCIe protocol analyzer, we can see DMA Read crossing 4KB boundary when
issue happened. It broke PCIe spec and caused PCIe stuck. So revert
the default value from 0 to 1."_

The issue seems to be that a value of 0 means different things depending on the specific chip.

This subsequently has only been changed for the 10_4 target and not others:

**TARGET_10_4_DMA_BURST_SIZE		1**

greearb, 2020-06-30:
I applied this back to 4.16 ath10k-ct drivers (not sure anyone uses anything older).  Please pull in latest and test if you can.

closed 2020-06-30 by greearb