greearb/ath10k-ct #111: IPQ4019: 5.4 version crashes when registerign wiphy

robimarko, 2020-01-02:
Hi, Ben.
I am working on a Gentoo based embedded system and ath10k-ct version 5.4 will always crash when trying to register the wiphy.
It probes both radios and finds the firmware and board files but crashes on registration.

I am running this under Gentoo snapshot from December 31 and kernel 5.4.7.

Logs (dmesg, maybe supplicant and/or hostap)

```
root@localhost / # modprobe ath10k_pci
root@localhost / # [  140.339443] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[  140.339492] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 0 testmode 1
[  140.350186] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[  140.351029] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-012-6acc9b999 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc4
[  140.355820] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 0 testmode 1
[  140.396969] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-fW-012-6acc9b999 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc4
[  140.415937] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:18 crc32 e2dfaa91
[  140.452741] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:25 crc32 e2dfaa91
[  141.738927] ath10k_ahb a000000.wifi: unsupported HTC service id: 1536
[  141.739907] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  141.744367] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[  141.760253] ath10k_ahb a800000.wifi: unsupported HTC service id: 1536
[  141.761213] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[  141.765682] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[  141.792412] ath10k_ahb a000000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[  141.793409] ath10k_ahb a000000.wifi: wmi print 'free: 56528 iram: 23400 sram: 32520'
[  141.814215] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[  141.815191] ath10k_ahb a800000.wifi: wmi print 'free: 56528 iram: 23400 sram: 32520'
[  141.929880] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[  141.952720] ath10k_ahb a800000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[  142.089293] ------------[ cut here ]------------
[  142.090592] WARNING: CPU: 0 PID: 35 at net/wireless/core.c:619 wiphy_register+0x598/0xbc8 [cfg80211]
[  142.093172] Modules linked in: ath10k_pci ath10k_core ath mac80211 libarc4 cfg80211 autofs4 [last unloaded: ath10k_pci]
[  142.102874] CPU: 0 PID: 35 Comm: kworker/u8:1 Tainted: G        W         5.4.7 #1
[  142.112796] Hardware name: Generic DT based system
[  142.121263] Workqueue: ath10k_wq ath10k_core_register_work [ath10k_core]
[  142.125426] [<c030e3d8>] (unwind_backtrace) from [<c030ab4c>] (show_stack+0x10/0x14)
[  142.125648] [<c030ab4c>] (show_stack) from [<c0921b08>] (dump_stack+0x90/0xa4)
[  142.139941] [<c0921b08>] (dump_stack) from [<c031bf98>] (__warn+0xbc/0xd8)
[  142.140182] [<c031bf98>] (__warn) from [<c031c004>] (warn_slowpath_fmt+0x50/0x94)
[  142.154781] [<c031c004>] (warn_slowpath_fmt) from [<bf04f144>] (wiphy_register+0x598/0xbc8 [cfg80211])
[  142.156891] [<bf04f144>] (wiphy_register [cfg80211]) from [<bf0940a8>] (ieee80211_register_hw+0x480/0xbf0 [mac80211])
[  142.161676] ------------[ cut here ]------------
[  142.172261] [<bf0940a8>] (ieee80211_register_hw [mac80211]) from [<bf109fac>] (ath10k_mac_register+0x768/0xb34 [ath10k_core])
[  142.173670] [<bf109fac>] (ath10k_mac_register [ath10k_core]) from [<bf115d70>] (ath10k_core_register_work+0x81c/0xbb8 [ath10k_core])
[  142.182338] WARNING: CPU: 3 PID: 161 at net/wireless/core.c:619 wiphy_register+0x598/0xbc8 [cfg80211]
[  142.186619] [<bf115d70>] (ath10k_core_register_work [ath10k_core]) from [<c0333398>] (process_one_work+0x1e8/0x434)
[  142.197201] Modules linked in: ath10k_pci ath10k_core ath mac80211 libarc4 cfg80211 autofs4 [last unloaded: ath10k_pci]
[  142.209228] [<c0333398>] (process_one_work) from [<c0333630>] (worker_thread+0x4c/0x564)
[  142.209458] [<c0333630>] (worker_thread) from [<c0339140>] (kthread+0x124/0x150)
[  142.247558] [<c0339140>] (kthread) from [<c03010e8>] (ret_from_fork+0x14/0x2c)
[  142.254826] Exception stack(0xcf9b7fb0 to 0xcf9b7ff8)
[  142.261890] 7fa0:                                     00000000 00000000 00000000 00000000
[  142.267064] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  142.275203] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  142.283382] CPU: 3 PID: 161 Comm: kworker/u8:2 Tainted: G        W         5.4.7 #1
[  142.284000] ---[ end trace fcf39019cc0a712c ]---
[  142.289701] Hardware name: Generic DT based system
[  142.290563] Workqueue: ath10k_wq ath10k_core_register_work [ath10k_core]
[  142.297554] ath10k_ahb a000000.wifi: failed to register ieee80211: -22
[  142.302349] [<c030e3d8>] (unwind_backtrace) from [<c030ab4c>] (show_stack+0x10/0x14)
[  142.302537] [<c030ab4c>] (show_stack) from [<c0921b08>] (dump_stack+0x90/0xa4)
[  142.302776] [<c0921b08>] (dump_stack) from [<c031bf98>] (__warn+0xbc/0xd8)
[  142.306905] ath10k_ahb a000000.wifi: could not register to mac80211 (-22)
[  142.313804] [<c031bf98>] (__warn) from [<c031c004>] (warn_slowpath_fmt+0x50/0x94)
[  142.314936] [<c031c004>] (warn_slowpath_fmt) from [<bf04f144>] (wiphy_register+0x598/0xbc8 [cfg80211])
[  142.317042] [<bf04f144>] (wiphy_register [cfg80211]) from [<bf0940a8>] (ieee80211_register_hw+0x480/0xbf0 [mac80211])
[  142.365425] [<bf0940a8>] (ieee80211_register_hw [mac80211]) from [<bf109fac>] (ath10k_mac_register+0x768/0xb34 [ath10k_core])
[  142.375989] [<bf109fac>] (ath10k_mac_register [ath10k_core]) from [<bf115d70>] (ath10k_core_register_work+0x81c/0xbb8 [ath10k_core])
[  142.387233] [<bf115d70>] (ath10k_core_register_work [ath10k_core]) from [<c0333398>] (process_one_work+0x1e8/0x434)
[  142.399167] [<c0333398>] (process_one_work) from [<c0333630>] (worker_thread+0x4c/0x564)
[  142.409318] [<c0333630>] (worker_thread) from [<c0339140>] (kthread+0x124/0x150)
[  142.417647] [<c0339140>] (kthread) from [<c03010e8>] (ret_from_fork+0x14/0x2c)
[  142.425018] Exception stack(0xceeb3fb0 to 0xceeb3ff8)
[  142.432051] 3fa0:                                     00000000 00000000 00000000 00000000
[  142.437177] 3fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  142.445335] 3fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  142.453541] ---[ end trace fcf39019cc0a712d ]---
[  142.459942] ath10k_ahb a800000.wifi: failed to register ieee80211: -22
[  142.464780] ath10k_ahb a800000.wifi: could not register to mac80211 (-22)
```

greearb, 2020-01-02:
You need to patch mac80211, I forget the exact patch, but I think it may be related to the beacon interval or something like that.  My kernels on github will have the patch in them, and openwrt has it as well.

closed 2020-01-02 by greearb

robimarko, 2020-01-02:
Ill try to track it down, but as far as I know 5.2 and 5.4 ones dont also work under OpenWrt.