greearb/ath10k-ct #112: Crash during startup: cannot restart a device that hasn't been started

fantom-x, 2020-01-13:
OpenWrt master branch 
Netgear R7800 / QCA9984

Sometimes (quite rarely) the wifi does no fully come up and this time I looked at the log and saw this crash report. The logs show a crash three minutes after a reboot: it is because the router always comes up with wifi disabled and my script starts wifi only after WAN is fully operational.
I have no way of reproducing it reliably, though.

```
Mon Jan 13 12:58:06 2020 kern.warn kernel: [  170.849514] ath10k_pci 0001:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
Mon Jan 13 12:58:06 2020 kern.warn kernel: [  170.849541] ath10k_pci 0001:01:00.0: msdu-desc: 2500  skid: 32
Mon Jan 13 12:58:06 2020 kern.info kernel: [  170.933454] ath10k_pci 0001:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
Mon Jan 13 12:58:06 2020 kern.info kernel: [  170.934311] ath10k_pci 0001:01:00.0: wmi print 'free: 81784 iram: 23204 sram: 14440'
Mon Jan 13 12:58:06 2020 kern.warn kernel: [  171.417137] ath10k_pci 0001:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
Mon Jan 13 12:58:07 2020 kern.warn kernel: [  171.833438] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
Mon Jan 13 12:58:07 2020 kern.warn kernel: [  171.839242] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
Mon Jan 13 12:58:07 2020 kern.err kernel: [  171.855053] ath10k_pci 0000:01:00.0: firmware crashed! (guid n/a)
Mon Jan 13 12:58:07 2020 kern.info kernel: [  171.855085] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
Mon Jan 13 12:58:07 2020 kern.info kernel: [  171.860144] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
Mon Jan 13 12:58:07 2020 kern.info kernel: [  171.872348] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fW-012-6acc9b999 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 2f261949
Mon Jan 13 12:58:07 2020 kern.info kernel: [  171.885232] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 85498734
Mon Jan 13 12:58:07 2020 kern.info kernel: [  171.906885] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
Mon Jan 13 12:58:07 2020 kern.warn kernel: [  171.926036] ath10k_pci 0000:01:00.0: failed to get memcpy hi address for firmware address 4: -16
Mon Jan 13 12:58:07 2020 kern.err kernel: [  171.926063] ath10k_pci 0000:01:00.0: failed to read firmware dump area: -16
Mon Jan 13 12:58:07 2020 kern.warn kernel: [  171.933934] ath10k_pci 0000:01:00.0: in crash-regs-harder
Mon Jan 13 12:58:08 2020 kern.warn kernel: [  172.660560] ath10k_pci 0000:01:00.0: in crash-regs-harder, firmware did not provide indicator: 0xdeadbeef
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.660591] ath10k_pci 0000:01:00.0: Copy Engine register dump:
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.669102] ath10k_pci 0000:01:00.0: [00]: 0x0004a000 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.674889] ath10k_pci 0000:01:00.0: [01]: 0x0004a400 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.683917] ath10k_pci 0000:01:00.0: [02]: 0x0004a800 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.692789] ath10k_pci 0000:01:00.0: [03]: 0x0004ac00 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.701630] ath10k_pci 0000:01:00.0: [04]: 0x0004b000 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.710471] ath10k_pci 0000:01:00.0: [05]: 0x0004b400 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.719276] ath10k_pci 0000:01:00.0: [06]: 0x0004b800 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.728186] ath10k_pci 0000:01:00.0: [07]: 0x0004bc00 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.737056] ath10k_pci 0000:01:00.0: [08]: 0x0004c000 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.745897] ath10k_pci 0000:01:00.0: [09]: 0x0004c400 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.754769] ath10k_pci 0000:01:00.0: [10]: 0x0004c800 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.763610] ath10k_pci 0000:01:00.0: [11]: 0x0004cc00 3735928559 3735928559 3735928559 3735928559
Mon Jan 13 12:58:08 2020 kern.warn kernel: [  172.772457] ath10k_pci 0000:01:00.0: failed to get memcpy hi address for firmware address 8: -28
Mon Jan 13 12:58:08 2020 kern.err kernel: [  172.781320] ath10k_pci 0000:01:00.0: failed to dump debug log area: -28
Mon Jan 13 12:58:08 2020 kern.warn kernel: [  172.790854] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jan 13 12:58:08 2020 kern.warn kernel: [  172.796562] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jan 13 12:58:08 2020 kern.warn kernel: [  172.803893] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jan 13 12:58:08 2020 kern.warn kernel: [  172.811170] ath10k_pci 0001:01:00.0: SWBA overrun on vdev 0, skipped old beacon
Mon Jan 13 12:58:08 2020 kern.warn kernel: [  172.833246] ath10k_pci 0000:01:00.0: cannot restart a device that hasn't been started
Mon Jan 13 12:58:13 2020 kern.warn kernel: [  177.638593] ath10k_pci 0001:01:00.0: Invalid peer id 1 or peer stats buffer, peer: 0cc9be2b  sta:   (null)

```

greearb, 2020-01-14:
Unfortunately the driver was unable to read any debugging info from the firmware when it crashed:

Mon Jan 13 12:58:07 2020 kern.err kernel: [  171.926063] ath10k_pci 0000:01:00.0: failed to read firmware dump area: -16
Mon Jan 13 12:58:07 2020 kern.warn kernel: [  171.933934] ath10k_pci 0000:01:00.0: in crash-regs-harder
Mon Jan 13 12:58:08 2020 kern.warn kernel: [  172.660560] ath10k_pci 0000:01:00.0: in crash-regs-harder, firmware did not provide indicator: 0xdeadbeef

So, I cannot debug that.

greearb, 2020-01-14:
I guess some driver work could be done to try to restart things again, but maybe that is best handled by removing driver and reloading it entirely, by some external process.

fantom-x, 2020-01-22:
I figured as much. Thx for looking. I am now trying the htt variant, which seems to be behaving better so far.