greearb/ath10k-ct #113: Lost ping packets on CT firmware when not much traffic

oldium, 2020-01-14:
#### Please provide this info.  See this link for more info on how to gather debug info:  http://www.candelatech.com/ath10k-bugs.php
I used recommended value `0xc0000032`.

#### Description of the problem (how to configure, how to reproduce, how often it happens).
After some time, the router becomes laggy. Sometimes the packets are lost even when the traffic is rather low. Pingings are without reply. I enabled debug log and tried to reproduce it - there is something in the dmesg, I dumped it couple of seconds after I saw lost ping. There was really not much traffic going on.

I know there are more similar/same issues, but I cannot tell whether the issue is the same.

#### Software (OS, Firmware version, kernel, driver, etc)
OpenWrt, latest Git master

#### Hardware (NIC chipset, platform, etc)
TP-Link Archer C7 v4

#### Logs (dmesg, maybe supplicant and/or hostap)

<details>
<summary>hostapd-phy0.conf</summary>
<pre>
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
logger_stdout_level=2
country_code=CZ
ieee80211d=1
ieee80211h=1
hw_mode=a
beacon_int=100
channel=36

tx_queue_data2_burst=2.0
ieee80211n=1
ht_coex=0
ht_capab=[HT40+][LDPC][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935][DSSS_CCK-40]
vht_oper_chwidth=1
vht_oper_centr_freq_seg0_idx=42
ieee80211ac=1
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN][RX-STBC-1][MAX-MPDU-11454][MAX-A-MPDU-LEN-EXP7]

interface=wlan0-lan
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
chan_util_avg_period=600
disassoc_low_ack=0
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
multi_ap=0
tdls_prohibit=1
nas_identifier=xxxxxxxx-0.home
wpa_passphrase=xxxxxxxx
auth_algs=1
wpa=3
wpa_pairwise=CCMP
ssid=domaci-soukroma
bridge=br-lan
mobility_domain=xxx3
ft_psk_generate_local=1
ft_over_ds=1
reassociation_deadline=1000
wpa_disable_eapol_key_retries=1
wpa_key_mgmt=WPA-PSK FT-PSK
okc=0
disable_pmksa_caching=1
config_id=33xxx2f
bssid=b0:xx:xx:xx:xx:aa
[... 550 lines cut ...]
[86077.333378] ath10k: [0000]: CA044105 E84F0008 00010001 00000000 CA044105 E84F0008 00030000 00000000
[86077.342740] ath10k: [0008]: CA044105 E84F0008 00010001 04000000 CA044105 E84F0008 00030000 04000000
[86077.352088] ath10k: [0016]: CA044105 E84F0008 00010001 06000000 CA044105 E84F0008 00030000 06000000
[86077.361440] ath10k: [0024]: CA044105 E84F0008 00010001 07000000 CA044105 E84F0008 00030000 07000000
[86077.370786] ath10k: [0032]: CA044105 E84F0008 00010001 11000000 CA044105 E84F0008 00030000 11000000
[86077.380131] ath10k: [0040]: CA044105 E84F0008 00010001 10000000 CA044105 E84F0008 00030000 10000000
[86077.389483] ath10k_pci 0000:00:00.0: ATH10K_END
[86081.327281] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[86081.333685] ath10k: [0000]: DD174105 E84F0008 00020301 10000000 DD174105 E84F0008 00000300 10000000
[86081.343054] ath10k: [0008]: DD174105 E84F0008 00020301 00000000 DD174105 E84F0008 00000300 00000000
[86081.352405] ath10k: [0016]: DD174105 E86B0008 00000401 00000300 DD174105 E86B0008 00020400 01000300
[86081.361762] ath10k: [0024]: DD174105 E84F0008 00020301 04000000 DD174105 E84F0008 00000300 04000000
[86081.371113] ath10k: [0032]: DD174105 E84F0008 00020301 06000000 DD174105 E84F0008 00000300 06000000
[86081.380466] ath10k: [0040]: DD174105 E84F0008 00020301 07000000 DD174105 E84F0008 00000300 07000000
[86081.389812] ath10k: [0048]: DD174105 E84F0008 00020301 11000000 DD174105 E84F0008 00000300 11000000
[86081.399165] ath10k: [0056]: DD174105 E86B0008 00010201 01000300 DD174105 E86B0008 00030200 01000300
[86081.408510] ath10k: [0064]: DD174105 E86B0008 00030301 01000300 DD174105 E86B0008 00040300 00000300
[86081.417856] ath10k: [0072]: ED174105 E84F0008 00010001 00000000 ED174105 E84F0008 00030000 00000000
[86081.427201] ath10k: [0080]: ED174105 E84F0008 00010001 04000000 ED174105 E84F0008 00030000 04000000
[86081.436546] ath10k: [0088]: ED174105 E84F0008 00010001 06000000 ED174105 E84F0008 00030000 06000000
[86081.445889] ath10k: [0096]: ED174105 E84F0008 00010001 07000000 ED174105 E84F0008 00030000 07000000
[86081.455237] ath10k: [0104]: ED174105 E84F0008 00010001 11000000 ED174105 E84F0008 00030000 11000000
[86081.464582] ath10k: [0112]: ED174105 E84F0008 00010001 10000000 ED174105 E84F0008 00030000 10000000
[86081.473915] ath10k_pci 0000:00:00.0: ATH10K_END
</pre></details>

oldium, 2020-01-14:
I also use 2,4GHz (ath9k) at the same time as 5GHz (ath10k). There is really nothing much going on the router, CPU load is very low (~2%).

fantom-x, 2020-01-22:
I had a similar issue (different branch, different router) and the htt version of the firmware seems to have fixed most of the issues. Unfortunately the non-CT firmware works even worse in master.

The package name would look like **ath10k-firmware-qca9984-ct-htt**.

oldium, 2020-01-22:
That is your experience, I have read about that elsewhere - that non-CT might be bad on some routers and CT works fine. But this is not my case, non-CT works just fine on my Archer C7 v4. I would like to have the CT firmware fixed, because it is the OpenWrt default, which is very unfortunate for me (others _might_ have different experience).

fantom-x, 2020-01-22:
Yes, but there are two CT firmwares: did you try the other one?

greearb, 2020-01-22:
When you can retest, can you attach new dmesg output?  The DBGLOG info seems partially corrupted, or you are managing to hit some logging event I have not seen before and which is producing some garbage values.

oldium, 2020-01-22:
According to my build history, I built OpenWrt image with CT+HTT on 29.11.2019 and 8.12.2019. I do  not remember the details, but because I use non-CT now, I guess that it behave the same. But I can test it if it helps in analysing the problem, otherwise I will keep non-CT.

oldium, 2020-01-22:
Ok, so I will retest with CT+HTT and attach the dmesg output. Last time the issue appeared next day, so we need to give it time.

oldium, 2020-01-22:
> When you can retest, can you attach new dmesg output? The DBGLOG info seems partially corrupted, or you are managing to hit some logging event I have not seen before and which is producing some garbage values.

If you meant to retest with CT and not CT+HTT, please let me know. I am going to test with CT+HTT.

greearb, 2020-01-22:
I suggest CT+HTT.

oldium, 2020-01-22:
Hm. I am now experiencing the same issue with non-CT firmware (lost pings), but that is after removing the small-buffer patches for non-CT firmware in https://github.com/openwrt/openwrt/commit/1e27befe63ff4c69f110c7310316f4af75ee63e9

So this might be related... Recently OpenWrt got smallbuffer package for CT, so I think it might be interesting to give it a try.

First, I will try the regular CT+HTT, try to reproduce the issue. Then I will do the same with the small-buffers CT+HTT version.

oldium, 2020-01-22:
No, sorry, forget it, the `dmesg` shows that I am using the CT version.

greearb, 2020-01-22:
There is CT firmware and also CT driver.  Please make sure you are checking correctly.

oldium, 2020-01-22:
Yes, I am always using the matching driver - non-CT driver with non-CT firmware and CT driver with CT/+HTT firmware.

fantom-x, 2020-01-22:
For my education: is **board.bin** a driver and **ct-firmware-5.bin** a firmware?

oldium, 2020-01-22:
Here you go, fresh master with CT driver and CT-HTT firmware.

<details>
  <summary>dmesg</summary>
  <pre>
[    0.000000] Linux version 4.19.97 (dev@dev-Ubuntu1804) (gcc version 8.3.0 (OpenWrt GCC 8.3.0 r11618-416d2cc71e)) #0 Wed Jan 22 16:18:55 2020
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link Archer C7 v4
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] random: get_random_bytes called from start_kernel+0x98/0x4a8 with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 122420K/131072K available (4298K kernel code, 177K rwdata, 1016K rodata, 1220K init, 206K bss, 8652K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000007] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008224] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.074771] pid_max: default: 32768 minimum: 301
[    0.079825] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.086813] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.098753] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.109157] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.115716] pinctrl core: initialized pinctrl subsystem
[    0.122105] NET: Registered protocol family 16
[    0.133424] PCI host bridge /ahb/pcie-controller@18250000 ranges:
[    0.139853]  MEM 0x0000000012000000..0x0000000013ffffff
[    0.145397]   IO 0x0000000000000000..0x0000000000000000
[    0.170030] PCI host bridge to bus 0000:00
[    0.174356] pci_bus 0000:00: root bus resource [mem 0x12000000-0x13ffffff]
[    0.181652] pci_bus 0000:00: root bus resource [io  0x0000]
[    0.187521] pci_bus 0000:00: root bus resource [??? 0x00000000 flags 0x0]
[    0.194682] pci_bus 0000:00: No busn resource found for root bus, will use [bus 00-ff]
[    0.203055] pci 0000:00:00.0: [168c:003c] type 00 class 0x028000
[    0.203110] pci 0000:00:00.0: reg 0x10: [mem 0x00000000-0x001fffff 64bit]
[    0.203162] pci 0000:00:00.0: reg 0x30: [mem 0x00000000-0x0000ffff pref]
[    0.203234] pci 0000:00:00.0: supports D1 D2
[    0.204343] pci_bus 0000:00: busn_res: [bus 00-ff] end is updated to 00
[    0.204369] pci 0000:00:00.0: BAR 0: assigned [mem 0x12000000-0x121fffff 64bit]
[    0.212130] pci 0000:00:00.0: BAR 6: assigned [mem 0x12200000-0x1220ffff pref]
[    0.222859] clocksource: Switched to clocksource MIPS
[    0.229256] NET: Registered protocol family 2
[    0.234646] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes)
[    0.242771] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.250167] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.256885] TCP: Hash tables configured (established 1024 bind 1024)
[    0.263709] UDP hash table entries: 256 (order: 0, 4096 bytes)
[... 924 lines cut ...]
[  793.395432] ath10k: [0104]: DE7A0C00 E86B0008 00010201 01000300 DE7A0C00 E86B0008 00030200 01000300
[  793.404786] ath10k: [0112]: DE7A0C00 E86B0008 00030301 01000300 DE7A0C00 E86B0008 00040300 00000300
[  793.414141] ath10k: [0120]: E87A0C00 E86B0008 00000401 00000300 E87A0C00 E86B0008 00020400 01000300
[  793.423499] ath10k: [0128]: E87A0C00 E86B0008 00010201 01000300 E87A0C00 E86B0008 00030200 01000300
[  793.432851] ath10k: [0136]: E87A0C00 E86B0008 00030301 01000300 E87A0C00 E86B0008 00040300 00000300
[  793.442201] ath10k_pci 0000:00:00.0: ATH10K_END
[  794.267468] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  794.273882] ath10k: [0000]: 577F0C00 E86B0008 00000401 00000300 577F0C00 E86B0008 00020400 01000300
[  794.283249] ath10k: [0008]: 587F0C00 E86B0008 00010201 01000300 587F0C00 E86B0008 00030200 01000300
[  794.292601] ath10k: [0016]: 587F0C00 E86B0008 00030301 01000300 587F0C00 E86B0008 00040300 00000300
[... cut ...]
[  796.283405] ath10k: [0008]: 4B860C00 E86B0008 00010201 01000300 4B860C00 E86B0008 00030200 01000300
[  796.292756] ath10k: [0016]: 4B860C00 E86B0008 00030301 01000300 4B860C00 E86B0008 00040300 00000300
[  796.302120] ath10k: [0024]: 7C870C00 E86B0008 00000401 00000300 7C870C00 E86B0008 00020400 01000300
[  796.311478] ath10k: [0032]: 7C870C00 E86B0008 00010201 01000300 7C870C00 E86B0008 00030200 01000300
[  796.320830] ath10k: [0040]: 7C870C00 E86B0008 00030301 01000300 7C870C00 E86B0008 00040300 00000300
[  796.330178] ath10k_pci 0000:00:00.0: ATH10K_END
</pre></details>

oldium, 2020-01-22:
Another one with long ping (379ms) and lost packet afterwards. The delayed ping should be approximately 25 seconds before the lost one, so around 1468 (if SWBA overrun at 1493 is the lost packet).

<details><summary>ping output</summary><pre>
Reply from 10.84.1.5: bytes=32 time=379ms TTL=64
Reply from 10.84.1.5: bytes=32 time<1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=4ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=2ms TTL=64
Reply from 10.84.1.5: bytes=32 time=2ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=3ms TTL=64
Reply from 10.84.1.5: bytes=32 time=2ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time<1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Request timed out.
</pre></details>

<details><summary>dmesg output (few seconds after timeout)</summary><pre>
[ 1388.328040] ath10k: [0008]: 7FC61500 E86B0008 00010201 01000300 7FC61500 E86B0008 00030200 01000300
[ 1388.337386] ath10k: [0016]: 7FC61500 E86B0008 00030301 01000300 7FC61500 E86B0008 00040300 00000300
[ 1388.346733] ath10k: [0024]: 3AC71500 E86B0008 00000401 00000300 3AC71500 E86B0008 00020400 01000300
[ 1388.356078] ath10k: [0032]: 3BC71500 E86B0008 00010201 01000300 3BC71500 E86B0008 00030200 01000300
[ 1388.365422] ath10k: [0040]: 3BC71500 E86B0008 00030301 01000300 3BC71500 E86B0008 00040300 00000300
[ 1388.374768] ath10k: [0048]: 94C71500 E86B0008 00000401 00000300 94C71500 E86B0008 00020400 01000300
[ 1388.384112] ath10k: [0056]: 94C71500 E86B0008 00010201 01000300 94C71500 E86B0008 00030200 01000300
[ 1388.393458] ath10k: [0064]: 94C71500 E86B0008 00030301 01000300 94C71500 E86B0008 00040300 00000300
[ 1388.402803] ath10k: [0072]: 95C71500 E86B0008 00000401 00000300 95C71500 E86B0008 00020400 01000300
[ 1388.412149] ath10k: [0080]: 95C71500 E86B0008 00010201 01000300 95C71500 E86B0008 00030200 01000300
[ 1388.421485] ath10k: [0088]: 95C71500 E86B0008 00030301 01000300 95C71500 E86B0008 00040300 00000300
[ 1388.430831] ath10k: [0096]: 9AC71500 E86B0008 00000401 00000300 9AC71500 E86B0008 00020400 01000300
[ 1388.440176] ath10k: [0104]: 9AC71500 E86B0008 00010201 01000300 9AC71500 E86B0008 00030200 01000300
[ 1388.449521] ath10k: [0112]: 9AC71500 E86B0008 00030301 01000300 9AC71500 E86B0008 00040300 00000300
[ 1388.458866] ath10k: [0120]: 30C81500 E86B0008 00000401 00000300 30C81500 E86B0008 00020400 01000300
[ 1388.468212] ath10k: [0128]: 30C81500 E86B0008 00010201 01000300 30C81500 E86B0008 00030200 01000300
[ 1388.477554] ath10k: [0136]: 30C81500 E86B0008 00000301 01000300 30C81500 E86B0008 00030301 02000300
[ 1388.486903] ath10k: [0144]: 30C81500 E86B0008 00020300 01000300 31C81500 E86B0008 00010201 01000300
[ 1388.496248] ath10k: [0152]: 31C81500 E86B0008 00030200 01000300 31C81500 E86B0008 00030301 01000300
[ 1388.505593] ath10k: [0160]: 31C81500 E86B0008 00040300 00000300 66C81500 E86B0008 00000401 00000300
[ 1388.514937] ath10k: [0168]: 66C81500 E86B0008 00020400 01000300 66C81500 E86B0008 00010201 01000300
[ 1388.524283] ath10k: [0176]: 66C81500 E86B0008 00030200 01000300 66C81500 E86B0008 00030301 01000300
[ 1388.533627] ath10k: [0184]: 66C81500 E86B0008 00040300 00000300
[ 1388.539737] ath10k_pci 0000:00:00.0: ATH10K_END
[ 1389.312325] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 1389.318739] ath10k: [0000]: 4ACC1500 E86B0008 00000401 00000300 4ACC1500 E86B0008 00020400 01000300
[ 1389.328111] ath10k: [0008]: 4ACC1500 E86B0008 00010201 01000300 4ACC1500 E86B0008 00030200 01000300
[ 1389.337457] ath10k: [0016]: 4ACC1500 E86B0008 00030301 01000300 4ACC1500 E86B0008 00040300 00000300
[ 1389.346805] ath10k_pci 0000:00:00.0: ATH10K_END
[ 1390.312397] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 1390.318805] ath10k: [0000]: 5BD01500 E86B0008 00000401 00000300 5BD01500 E86B0008 00020400 01000300
[ 1390.328172] ath10k: [0008]: 5BD01500 E86B0008 00010201 01000300 5BD01500 E86B0008 00030200 01000300
[ 1390.337524] ath10k: [0016]: 5BD01500 E86B0008 00030301 01000300 5BD01500 E86B0008 00040300 00000300
[ 1390.346876] ath10k_pci 0000:00:00.0: ATH10K_END
[ 1391.312481] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 1391.318883] ath10k: [0000]: A1D11500 E86B0008 00000401 00000300 A1D11500 E86B0008 00020400 01000300
[ 1391.328255] ath10k: [0008]: A1D11500 E86B0008 00010201 01000300 A1D11500 E86B0008 00030200 01000300
[ 1391.337607] ath10k: [0016]: A1D11500 E86B0008 00030301 01000300 A1D11500 E86B0008 00040300 00000300
[ 1391.346957] ath10k: [0024]: D1D11500 E86B0008 00000401 00000300 D1D11500 E86B0008 00020400 01000300
[... 1194 lines cut ...]
[ 1496.345916] ath10k: [0016]: 22761700 E86B0008 00030301 01000300 22761700 E86B0008 00040300 00000300
[ 1496.355266] ath10k: [0024]: 43771700 E86B0008 00000401 00000300 43771700 E86B0008 00020400 01000300
[... cut ...]
[ 1498.336707] ath10k: [0008]: 557F1700 E86B0008 00010201 01000300 557F1700 E86B0008 00030200 01000300
[ 1498.346061] ath10k: [0016]: 557F1700 E86B0008 00030301 01000300 557F1700 E86B0008 00040300 00000300
[ 1498.355418] ath10k: [0024]: CF7F1700 E86B0008 00000401 00000300 CF7F1700 E86B0008 00020400 01000300
[ 1498.364773] ath10k: [0032]: D07F1700 E86B0008 00010201 01000300 D07F1700 E86B0008 00030200 01000300
[ 1498.374127] ath10k: [0040]: D07F1700 E86B0008 00030301 01000300 D07F1700 E86B0008 00040300 00000300
[ 1498.383475] ath10k_pci 0000:00:00.0: ATH10K_END
</pre></details>

greearb, 2020-01-22:
The debugging output is still weird.  Can you enable the debugging, then assoc/disassoc a station and send me that output?  I want to see if all debugging is weird or just the debugging that shows up when you detect the problem.

greearb, 2020-01-22:
Nevermind, I found it a few comments up.  Anyone else have this platform with ath10k-ct on it?  If so, plz post dmesg of startup with DBGLOG logging enabled, I am curious if this logging is weird on other systems.

fantom-x, 2020-01-23:
https://git.openwrt.org/?p=openwrt/openwrt.git;a=commitdiff;h=c07f6e8659ea1348c75c04dac2924616f0042293

Have not tried this yet, but could this be the solution?

oldium, 2020-01-23:
I saw it, I will update to latest Git master and try it in the evening (CET time) from home.

oldium, 2020-01-23:
Is the debugging able to slow the device down a lot? I saw `93% sirq` in `top` in the first line and at that time ping was in hundreds of milliseconds, router was unusably slow on both bands - even on 2,4GHz. Unfortunately I had to switch back to non-CT firmware (and forgot `dmesg` log on `/tmp`), my wife needs to do more serious work... So no more tests today.

greearb, 2020-01-23:
Debugging should not seriously slow down the system.  You can also test with ath10k-ct driver and stock firmware to see if it is something to do with the ath10k-ct driver?

oldium, 2020-01-23:
Ok, will test tomorrow.

oldium, 2020-01-29:
Tested with fresh Git master of OpenWrt and CT driver with non-CT firmware. I can see the same issue with slowness as seen last time with CT firmware, debug messages are turned on from the startup of router (`echo` from init script).

<details><summary>dmesg</summary>
<pre>
[ 2119.712520] ath10k: [0192]: 28352100 2F04FC0B 80000000 00000000 28352100 2F04FC0B 00000000 00000000
[ 2119.721877] ath10k: [0200]: 28352100 2E04FC03 4B352100 2904FC0B 57900000 1C000000 4B352100 2D04FC0B
[ 2119.731221] ath10k: [0208]: 57900000 1C000000 4B352100 2F04FC0B 02000000 08010000 4B352100 2F04FC0B
[ 2119.740574] ath10k: [0216]: 00404E05 00000000 4B352100 2F04FC0B 80000000 00000000 4B352100 2F04FC0B
[ 2119.749921] ath10k: [0224]: 00000000 00000000 4B352100 2E04FC03 6E352100 2904FC0B 57900000 1C000000
[ 2119.759286] ath10k: [0232]: 6E352100 2D04FC0B 57900000 1C000000 6E352100 2F04FC0B 00000000 0B010000
[ 2119.768642] ath10k: [0240]: 6E352100 2F04FC0B 0090BA05 00000000 6E352100 2F04FC0B 80000000 00000000
[ 2119.777991] ath10k: [0248]: 6E352100 2F04FC0B 00000000 00000000 6E352100 2E04FC03 91352100 2904FC0B
[ 2119.787347] ath10k: [0256]: 57900000 1C000000 91352100 2D04FC0B 57900000 1C000000 91352100 2F04FC0B
[ 2119.796703] ath10k: [0264]: 01000000 02010000 91352100 2F04FC0B 00104A05 00000000 91352100 2F04FC0B
[ 2119.806052] ath10k: [0272]: 80000000 01000000 91352100 2F04FC0B 00000000 00000000 91352100 2E04FC03
[ 2119.815400] ath10k: [0280]: B4352100 2904FC0B 57900000 1C000000 B4352100 2D04FC0B 57900000 1C000000
[ 2119.824755] ath10k: [0288]: B4352100 2F04FC0B 02000000 08010000 B4352100 2F04FC0B 00404E05 00000000
[ 2119.834110] ath10k: [0296]: B4352100 2F04FC0B 80000000 01000000 B4352100 2F04FC0B 00000000 00000000
[ 2119.843458] ath10k: [0304]: B4352100 2E04FC03 D7352100 2904FC0B 57900000 1C000000 D7352100 2D04FC0B
[ 2119.852808] ath10k: [0312]: 57900000 1C000000 D7352100 2F04FC0B 00000000 0B010000 D7352100 2F04FC0B
[ 2119.862164] ath10k: [0320]: 0090BA05 00000000 D7352100 2F04FC0B 80000000 01000000 D7352100 2F04FC0B
[ 2119.871508] ath10k: [0328]: 00000000 00000000 D7352100 2E04FC03 F9352100 2904FC0B 57900000 1C000000
[ 2119.880860] ath10k: [0336]: F9352100 2D04FC0B 57900000 1C000000 F9352100 2F04FC0B 01000000 02010000
[ 2119.890207] ath10k: [0344]: F9352100 2F04FC0B 00104A05 00000000 F9352100 2F04FC0B 80000000 00000000
[ 2119.899563] ath10k: [0352]: F9352100 2F04FC0B 00000000 00000000 F9352100 2E04FC03 1C362100 2904FC0B
[ 2119.908914] ath10k: [0360]: 57900000 1C000000 1C362100 2D04FC0B 57900000 1C000000 1C362100 2F04FC0B
[ 2119.918258] ath10k: [0368]: 02000000 08010000
[ 2119.922764] ath10k_pci 0000:00:00.0: ATH10K_END
[ 2119.930527] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 2, skipped old beacon
[ 2119.938169] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 2119.945770] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 1, skipped old beacon
[ 2119.953365] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 2, skipped old beacon
[ 2119.960933] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 2119.967339] ath10k: [0000]: 1C362100 2F04FC0B 00404E05 00000000 1C362100 2F04FC0B 80000000 00000000
[ 2119.976693] ath10k: [0008]: 1C362100 2F04FC0B 00000000 00000000 1C362100 2E04FC03
[ 2119.984429] ath10k_pci 0000:00:00.0: ATH10K_END
[ 2119.989127] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 2119.996723] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 1, skipped old beacon
[ 2120.004332] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 2, skipped old beacon
[ 2120.011930] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon
[ 2120.024548] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 2, skipped old beacon
[ 2120.266747] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 2120.273186] ath10k: [0000]: E8372100 2904FC0B 57900000 1C000000 E8372100 2D04FC0B 57900000 1C000000
[ 2120.282547] ath10k: [0008]: E8372100 2F04FC0B 00000000 0B010000 E8372100 2F04FC0B 0090BA05 00000000
[ 2120.291899] ath10k: [0016]: E8372100 2F04FC0B 80000000 00000000 E8372100 2F04FC0B 00000000 00000000
[ 2120.301239] ath10k: [0024]: E8372100 2E04FC03 E8372100 2904FC0B 57900000 1C000000 E8372100 2D04FC0B
[ 2120.310590] ath10k: [0032]: 57900000 1C000000 E8372100 2F04FC0B 01000000 02010000 E8372100 2F04FC0B
[ 2120.319935] ath10k: [0040]: 00104A05 00000000 E8372100 2F04FC0B 80000000 01000000 E8372100 2F04FC0B
[ 2120.329279] ath10k: [0048]: 00000000 00000000 E8372100 2E04FC03 48382100 2904FC0B 57900000 1C000000
[ 2120.338624] ath10k: [0056]: 48382100 2D04FC0B 57900000 1C000000 48382100 2F04FC0B 00000000 0B010000
[ 2120.347979] ath10k: [0064]: 48382100 2F04FC0B 0090BA05 00000000 48382100 2F04FC0B 80000000 00000000
[ 2120.357326] ath10k: [0072]: 48382100 2F04FC0B 00000000 00000000 48382100 2E04FC03 48382100 2904FC0B
[ 2120.366669] ath10k: [0080]: 57900000 1C000000 48382100 2D04FC0B 57900000 1C000000 48382100 2F04FC0B
[ 2120.376015] ath10k: [0088]: 01000000 02010000 48382100 2F04FC0B 00104A05 00000000 48382100 2F04FC0B
[ 2120.385359] ath10k: [0096]: 80000000 00000000 48382100 2F04FC0B 00000000 00000000 48382100 2E04FC03
[ 2120.394705] ath10k: [0104]: 51382100 0064FC13 49170000 E2030000 02000000 00000000 51382100 0D64FC0B
[ 2120.404050] ath10k: [0112]: 03000000 02000000 53382100 2904FC0B 57900000 1C000000 53382100 2D04FC0B
[ 2120.413393] ath10k: [0120]: 57900000 1C000000 53382100 2F04FC0B 01000000 02010000 53382100 2F04FC0B
[... cut ...]
 1106     1 root     S     2196   2%   0% /usr/sbin/wpa_supplicant -s -n phy1 -g /var/run/wpa_supplicant-phy1/global
 1165     1 root     S     2072   2%   0% /sbin/rpcd
 1408     1 root     S     1724   1%   0% /sbin/netifd
    1     0 root     S     1612   1%   0% /sbin/procd
  582     1 root     S     1264   1%   0% /sbin/ubusd
 2937  1337 root     S     1240   1%   0% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 2915  2906 root     S     1232   1%   0% -ash
^C938  2937 root     S     1232   1%   0% -ash
</pre></details>

oldium, 2020-01-29:
Router ended-up restarting itself. I removed the debug mode from init script (mask `0xc0000032`) and router looks good (sirq is low). I am trying to ping it. Still combination CT driver with non-CT firmware. I think I should repeat the test with CT driver and CT firmware without the debug mode.

This is how the `top` looks like without debugging:
<details><summary>top</summary><pre>
Mem: 60752K used, 62888K free, 76K shrd, 5956K buff, 16040K cached
CPU:   0% usr   0% sys   0% nic  98% idle   0% io   0% irq   0% sirq
Load average: 0.00 0.06 0.07 1/52 2886
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
 2877     2 root     IW       0   0%   1% [kworker/0:2-eve]
 1103     1 root     S     2292   2%   0% /usr/sbin/hostapd -s -n phy1 -g /var/run/hostapd-phy1/global
 1101     1 root     S     2264   2%   0% /usr/sbin/hostapd -s -n phy0 -g /var/run/hostapd-phy0/global
 1408     1 root     S     1724   1%   0% /sbin/netifd
 2865  1337 root     S     1224   1%   0% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
 2886  2879 root     R     1224   1%   0% top
    7     2 root     SW       0   0%   0% [ksoftirqd/0]
 2875     2 root     IW       0   0%   0% [kworker/u2:0-at]
 1480     1 root     S     4848   4%   0% /usr/sbin/lighttpd -D -f /etc/lighttpd/lighttpd.conf
 2876  2866 root     S     4624   4%   0% mc
 1301     1 dnsmasq  S     2492   2%   0% /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf.cfg01411c -k -x /var/run/dnsmasq/dnsmasq.cfg01411
 1102     1 root     S     2196   2%   0% /usr/sbin/wpa_supplicant -s -n phy0 -g /var/run/wpa_supplicant-phy0/global
 1104     1 root     S     2196   2%   0% /usr/sbin/wpa_supplicant -s -n phy1 -g /var/run/wpa_supplicant-phy1/global
 1167     1 root     S     2004   2%   0% /sbin/rpcd
    1     0 root     S     1612   1%   0% /sbin/procd
 1135     1 root     S     1260   1%   0% /sbin/logd -S 64
  582     1 root     S     1252   1%   0% /sbin/ubusd
 2866  2865 root     S     1232   1%   0% -ash
 1868  1647 root     S     1232   1%   0% -ash
 2517     1 root     S<    1228   1%   0% /usr/sbin/ntpd -n -N -S /usr/sbin/ntpd-hotplug -p 10.84.1.1
 1647  1337 root     S     1224   1%   0% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300 -T 3
^C245  1868 root     S     1224   1%   0% top
</pre></details>

greearb, 2020-01-29:
If your debugging is sent to serial console, then it will slow down your system.  Normally the DBGLOG messages are only found in 'dmesg', at least on the systems I have looked at.  Those should not significantly slow down the system.  Probably this bug is related to power-save.

oldium, 2020-01-29:
Ok, I checked `/proc/cmdline` and there is `console=ttyS0,115200n8 rootfstype=squashfs,jffs2`. I will try to disable the `console` part (not enabled by me) and see how the debugging experience changes.

greearb, 2020-01-29:
That is now the way to fix the problem.  You need to instead somehow adjust the verbosity of what gets thrown on the serial console.  Maybe someone on openwrt mailing list or forum can help with this.

oldium, 2020-01-30:
`sysctl "kernel.printk=0 4 1 7"` did the trick, now enabling debug does not increase the `sirq`. It really was console output, maybe please note this to the http://www.candelatech.com/ath10k-bugs.php page with bold letters, because having the console enabled looks like to be the default for OpenWrt.

Now back to reproducing the issue. I am back at the beginning - testing the CT driver + CT firmware combination.

oldium, 2020-01-30:
Fresh lost packet. I do not see time 4372 in logs, which plus minus corresponds to the lost packet. I stopped the `ping` command and made `dmesg` with few seconds difference. Are the logs from `dmesg` now better without the console slow-down?

<details><summary>dmesg</summary><pre>
[ 4321.138825] ath10k: [0056]: E3984300 E86B0008 00010201 01000300 E3984300 E86B0008 00030200 01000300
[ 4321.138834] ath10k: [0064]: E3984300 E86B0008 00030301 01000300 E3984300 E86B0008 00040300 00000300
[ 4321.138843] ath10k: [0072]: 699B4300 E86B0008 00000401 00000300 699B4300 E86B0008 00020400 01000300
[ 4321.138853] ath10k: [0080]: 6A9B4300 E86B0008 00010201 01000300 6A9B4300 E86B0008 00030200 01000300
[ 4321.138862] ath10k: [0088]: 6A9B4300 E86B0008 00030301 01000300 6A9B4300 E86B0008 00040300 00000300
[ 4321.138868] ath10k_pci 0000:00:00.0: ATH10K_END
[ 4322.138806] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 4322.138825] ath10k: [0000]: FE9C4300 E86B0008 00000401 00000300 FE9C4300 E86B0008 00020400 01000300
[ 4322.138834] ath10k: [0008]: FF9C4300 E86B0008 00010201 01000300 FF9C4300 E86B0008 00030200 01000300
[ 4322.138844] ath10k: [0016]: FF9C4300 E86B0008 00030301 01000300 FF9C4300 E86B0008 00040300 00000300
[ 4322.138854] ath10k: [0024]: 799F4300 E86B0008 00000401 00000300 799F4300 E86B0008 00020400 01000300
[ 4322.138863] ath10k: [0032]: 799F4300 E86B0008 00010201 01000300 799F4300 E86B0008 00030200 01000300
[ 4322.138873] ath10k: [0040]: 799F4300 E86B0008 00030301 01000300 799F4300 E86B0008 00040300 00000300
[ 4322.138882] ath10k: [0048]: A89F4300 E86B0008 00000401 00000300 A89F4300 E86B0008 00020400 01000300
[ 4322.138892] ath10k: [0056]: A89F4300 E86B0008 00010201 01000300 A89F4300 E86B0008 00030200 01000300
[ 4322.138901] ath10k: [0064]: A89F4300 E86B0008 00030301 01000300 A89F4300 E86B0008 00040300 00000300
[ 4322.138907] ath10k_pci 0000:00:00.0: ATH10K_END
[ 4323.138892] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 4323.138911] ath10k: [0000]: E8A04300 E86B0008 00000401 00000300 E8A04300 E86B0008 00020400 01000300
[ 4323.138921] ath10k: [0008]: E8A04300 E86B0008 00010201 01000300 E8A04300 E86B0008 00030200 01000300
[ 4323.138931] ath10k: [0016]: E8A04300 E86B0008 00030301 01000300 E8A04300 E86B0008 00040300 00000300
[ 4323.138940] ath10k: [0024]: 23A14300 E86B0008 00000401 00000300 23A14300 E86B0008 00020400 01000300
[ 4323.138949] ath10k: [0032]: 23A14300 E86B0008 00010201 01000300 23A14300 E86B0008 00030200 01000300
[ 4323.138958] ath10k: [0040]: 23A14300 E86B0008 00030301 01000300 23A14300 E86B0008 00040300 00000300
[ 4323.138967] ath10k: [0048]: 9EA14300 E86B0008 00000401 00000300 9EA14300 E86B0008 00020400 01000300
[ 4323.138977] ath10k: [0056]: 9EA14300 E86B0008 00010201 01000300 9EA14300 E86B0008 00030200 01000300
[ 4323.138986] ath10k: [0064]: 9EA14300 E86B0008 00000301 01000300 9EA14300 E86B0008 00030301 02000300
[ 4323.138995] ath10k: [0072]: 9EA14300 E86B0008 00020300 01000300 9EA14300 E86B0008 00010201 01000300
[ 4323.139004] ath10k: [0080]: 9EA14300 E86B0008 00030200 01000300 9EA14300 E86B0008 00030301 01000300
[ 4323.139013] ath10k: [0088]: 9EA14300 E86B0008 00040300 00000300 A4A14300 E86B0008 00000401 00000300
[ 4323.139023] ath10k: [0096]: A4A14300 E86B0008 00020400 01000300 A5A14300 E86B0008 00010201 01000300
[ 4323.139032] ath10k: [0104]: A5A14300 E86B0008 00030200 01000300 A5A14300 E86B0008 00030301 01000300
[ 4323.139041] ath10k: [0112]: A5A14300 E86B0008 00040300 00000300 ABA14300 E86B0008 00000401 00000300
[ 4323.139050] ath10k: [0120]: ABA14300 E86B0008 00020400 01000300 ABA14300 E86B0008 00010201 01000300
[ 4323.139059] ath10k: [0128]: ABA14300 E86B0008 00030200 01000300 ABA14300 E86B0008 00030301 01000300
[ 4323.139068] ath10k: [0136]: ABA14300 E86B0008 00040300 00000300 B2A14300 E86B0008 00000401 00000300
[ 4323.139078] ath10k: [0144]: B2A14300 E86B0008 00020400 01000300 B3A14300 E86B0008 00010201 01000300
[ 4323.139087] ath10k: [0152]: B3A14300 E86B0008 00030200 01000300 B3A14300 E86B0008 00030301 01000300
[ 4323.139094] ath10k: [0160]: B3A14300 E86B0008 00040300 00000300
[ 4323.139100] ath10k_pci 0000:00:00.0: ATH10K_END
[ 4324.138963] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 4324.138982] ath10k: [0000]: 41A54300 E86B0008 00000401 00000300 41A54300 E86B0008 00020400 01000300
[ 4324.138992] ath10k: [0008]: 42A54300 E86B0008 00010201 01000300 42A54300 E86B0008 00030200 01000300
[ 4324.139002] ath10k: [0016]: 42A54300 E86B0008 00030301 01000300 42A54300 E86B0008 00040300 00000300
[ 4324.139011] ath10k: [0024]: B3A54300 E86B0008 00000401 00000300 B3A54300 E86B0008 00020400 01000300
[ 4324.139021] ath10k: [0032]: B3A54300 E86B0008 00010201 01000300 B3A54300 E86B0008 00030200 01000300
[ 4324.139030] ath10k: [0040]: B3A54300 E86B0008 00030301 01000300 B3A54300 E86B0008 00040300 00000300
[ 4324.139039] ath10k: [0048]: 7BA74300 E86B0008 00000401 00000300 7BA74300 E86B0008 00020400 01000300
[ 4324.139049] ath10k: [0056]: 7CA74300 E86B0008 00010201 01000300 7CA74300 E86B0008 00030200 01000300
[ 4324.139058] ath10k: [0064]: 7CA74300 E86B0008 00030301 01000300 7CA74300 E86B0008 00040300 00000300
[ 4324.139065] ath10k_pci 0000:00:00.0: ATH10K_END
[ 4325.139038] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 4325.139058] ath10k: [0000]: 5EA94300 E86B0008 00000401 00000300 5EA94300 E86B0008 00020400 01000300
[ 4325.139067] ath10k: [0008]: 5EA94300 E86B0008 00010201 01000300 5EA94300 E86B0008 00030200 01000300
[... cut ...]
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=2ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=2ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=2ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=2ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
Reply from 10.84.1.5: bytes=32 time=1ms TTL=64
</pre></details>

greearb, 2020-02-09:
Can you try the 'y3' build in bug 114, found in comments from a few days ago?

oldium, 2020-02-09:
y3 does not look like the right firmware for my hardware. I copied it to /lib/firmware/ath10k/QCA988X/hw2.0/ct-firmware-5.bin and loading failed:

```
[   16.086227] ath10k 4.19 driver, optimized for CT firmware, probing pci device: 0x3c.
[   16.103647] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   16.110264] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   16.698759] firmware ath10k!fwcfg-pci-0000:00:00.0.txt: firmware_loading_store: map pages failed
[   17.168922] firmware ath10k!pre-cal-pci-0000:00:00.0.bin: firmware_loading_store: map pages failed
[   18.938101] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[   18.947676] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   18.967466] ath10k_pci 0000:00:00.0: firmware ver 10.4b-ct-9984-fH-013-a91922785 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 21a671fb
[   22.042830] ath10k_pci 0000:00:00.0: unable to read from the device
[   22.049307] ath10k_pci 0000:00:00.0: could not execute otp for board id check: -145
[   22.057222] ath10k_pci 0000:00:00.0: failed to get board id from otp: -145
[   22.064505] ath10k_pci 0000:00:00.0: could not probe fw (-145)
```

greearb, 2020-02-19:
Yeah, the y3 was for wave-2, I was confused about what hardware you have.  I am also confused by your dmesg output though.  I cannot figure out why the debug logs do not decode to anything useful.  It doesn't appear to be corrupted, just that it is not something that my firmware would create.  Can you do a test where you enable the DBGLOG debugging, reboot, allow a station to connect, and then send me dmesg?  I am hoping to see some DBGLOG messages that I can decode as well as firmware bootup logs.

oldium, 2020-02-20:
Fine, I will try to modify kernel command line in the OpenWrt image - to enable DBGLOG during boot. Hopefully `make kernel_menuconfig` does the trick.

oldium, 2020-02-21:
Finally ready. Is this one better? 
[dmesg.txt](https://github.com/greearb/ath10k-ct/files/4237984/dmesg.txt)

greearb, 2020-03-19:
I think I just fixed a problem in wave-1 firmware with power-save.  Can you see if this firmware improves the problem?
[firmware-2-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/4356570/firmware-2-full-htt-mgt-community.bin.gz)

oldium, 2020-03-19:
Yes, give me few days to test this.

oldium, 2020-03-25:
After some days using this firmware, I have to say that this one is definitely better - no lost pings. I can see some delays now and then, but it is definitely usable.

greearb, 2020-03-25:
ok, thanks for the patience and testing!  I just sent email to openwrt to ask them to pull in the latest firmware.  Closing this bug, please re-open or open new one if you see more problems after upgrading to that firmware when it hits OpenWrt.

closed 2020-03-25 by greearb