greearb/ath10k-ct #115: Setting the MCS fails when NSS=3 and MCS<4 (WLE900VX)

Warxo, 2020-01-28:
Hello,

I'm using 2 WLE900VX  (Qualcomm Atheros QCA986x/988x 802.11ac Wireless Network Adapter), one is an access point and the other is a client that is connected to the first one. The bandwidth is 80MHz and the Guard Interval is 800 ns long.

When i try to change the MCS value with "iw dev wlan0 set bitrates vht-mcs-5 3:0" (3 NSS, MCS 0), the transmission bitrate as reported by iw is 6 Mbit/s while it should be something like 87.8 Mbit/s. I observe the same behavior for MCS 1, 2 and 3 (but only with NSS set to 3). Setting the NSS to 3 and the MCS to 4 gives the expected outcome, as well as setting the NSS to 1 or 2 with any MCS.

Current debug level: 0xc0000032

Software (OS, Firmware version, kernel, driver, etc)
- OS : OpenWrt
- Firmware version : firmware ver 10.1-ct-8x-__fW-022-0e382b4a
- kernel : 4.19.86 (custom build)
- driver : ath10k-ct

Hardware (NIC chipset, platform, etc)
- Qualcomm Atheros QCA986x/988x 802.11ac Wireless Network Adapter
- WLE900VX
- Platform: PC Engines APU4C4

Logs (on the client):

```
root@OpenWrt:~# iw dev
phy#0
        Interface wlan0
                ifindex 7
                wdev 0x2
                addr 04:f0:21:54:37:39
                ssid OpenWrt
                type managed
                channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 23.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
root@OpenWrt:~# iperf3 -c10.0.0.1
Connecting to host 10.0.0.1, port 5201
[  5] local 10.0.0.203 port 42792 connected to 10.0.0.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  67.7 MBytes   568 Mbits/sec    0    533 KBytes       
[  5]   1.00-2.00   sec  68.2 MBytes   572 Mbits/sec    0    584 KBytes       
[  5]   2.00-3.00   sec  68.9 MBytes   578 Mbits/sec    0    615 KBytes       
[  5]   3.00-4.00   sec  68.0 MBytes   571 Mbits/sec    0    615 KBytes       
[  5]   4.00-5.00   sec  67.3 MBytes   565 Mbits/sec    0    615 KBytes       
[  5]   5.00-6.00   sec  68.0 MBytes   571 Mbits/sec    0    615 KBytes       
[  5]   6.00-7.00   sec  67.7 MBytes   568 Mbits/sec    0    615 KBytes       
[  5]   7.00-8.00   sec  69.3 MBytes   581 Mbits/sec    0    645 KBytes       
[  5]   8.00-9.00   sec  68.2 MBytes   572 Mbits/sec    0    645 KBytes       
[  5]   9.00-10.00  sec  68.0 MBytes   571 Mbits/sec    0    645 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   681 MBytes   572 Mbits/sec    0             sender
[  5]   0.00-10.04  sec   680 MBytes   568 Mbits/sec                  receiver

iperf Done.
root@OpenWrt:~# iw dev wlan0 set bitrates vht-mcs-5 3:2
root@OpenWrt:~# iw dev wlan0 station dump|grep "tx bitrate"
        tx bitrate:     6.0 MBit/s
root@OpenWrt:~# iperf3 -c10.0.0.1
Connecting to host 10.0.0.1, port 5201
[  5] local 10.0.0.203 port 42796 connected to 10.0.0.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   704 KBytes  5.77 Mbits/sec    0   38.2 KBytes       
[  5]   1.00-2.00   sec   574 KBytes  4.70 Mbits/sec    0   38.2 KBytes
[  5]   2.00-3.00   sec   607 KBytes  4.97 Mbits/sec    0   45.2 KBytes       
^C[  5]   3.00-3.28   sec   139 KBytes  4.10 Mbits/sec    0   45.2 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-3.28   sec  1.96 MBytes  5.00 Mbits/sec    0             sender
[  5]   0.00-3.28   sec  0.00 Bytes  0.00 bits/sec                  receiver
iperf3: interrupt - the client has terminated
root@OpenWrt:~# 
root@OpenWrt:~# iw dev wlan0 set bitrates vht-mcs-5 3:4
root@OpenWrt:~# iw dev wlan0 station dump|grep "tx bitrate"
        tx bitrate:     526.5 MBit/s VHT-MCS 4 80MHz VHT-NSS 3
root@OpenWrt:~# iperf3 -c10.0.0.1
Connecting to host 10.0.0.1, port 5201
[  5] local 10.0.0.203 port 42800 connected to 10.0.0.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  43.7 MBytes   366 Mbits/sec    0    632 KBytes       
[  5]   1.00-2.00   sec  42.5 MBytes   357 Mbits/sec    0    663 KBytes       
[  5]   2.00-3.00   sec  43.4 MBytes   364 Mbits/sec    0    696 KBytes       
[  5]   3.00-4.00   sec  42.6 MBytes   358 Mbits/sec    0    696 KBytes       
[  5]   4.00-5.00   sec  43.3 MBytes   363 Mbits/sec    0    734 KBytes       
[  5]   5.00-6.00   sec  43.5 MBytes   365 Mbits/sec    0    734 KBytes       
[  5]   6.00-7.00   sec  43.2 MBytes   362 Mbits/sec    0    734 KBytes       
[  5]   7.00-8.00   sec  42.9 MBytes   360 Mbits/sec    0    734 KBytes       
[  5]   8.00-9.00   sec  42.7 MBytes   358 Mbits/sec    0    734 KBytes       
[  5]   9.00-10.00  sec  43.7 MBytes   367 Mbits/sec    0    734 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   431 MBytes   362 Mbits/sec    0             sender
[  5]   0.00-10.05  sec   429 MBytes   358 Mbits/sec                  receiver

iperf Done.
root@OpenWrt:~# uname -a
Linux OpenWrt 4.19.86 #0 SMP Sun Dec 8 18:53:11 2019 i686 GNU/Linux
root@OpenWrt:~# grep "model name" /proc/cpuinfo 
model name      : AMD GX-412TC SOC

```

[dmesg.txt](https://github.com/greearb/ath10k-ct/files/4121778/dmesg.txt)

ps: the issue https://github.com/greearb/ath10k-ct/issues/1 seems similar

Warxo, 2020-01-28:
It's me again, 
I edit my issue by adding that the problem is also for values of NSS 2 and MCS 0, 1 and 2.

greearb, 2020-01-28:
Please fix debug level so that DBGLOG messages are shown, you should NOT see this in your dmesg output:

[   23.972485] ath10k_pci 0000:05:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000

http://www.candelatech.com/ath10k-bugs.php

Warxo, 2020-01-29:
Hello,
I fix the debug message

```
root@OpenWrt:~# iw dev
phy#0
        Interface wlan0
                ifindex 7
                wdev 0x2
                addr 04:f0:21:54:37:39
                type managed
                txpower 0.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytestx-packets
                        0       0       0       0       0       0       0       0       0
root@OpenWrt:~# iperf3 -c10.0.0.1
Connecting to host 10.0.0.1, port 5201
[  5] local 10.0.0.203 port 35984 connected to 10.0.0.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  56.1 MBytes   470 Mbits/sec    0    615 KBytes
[  5]   1.00-2.00   sec  63.5 MBytes   533 Mbits/sec    0    643 KBytes
[  5]   2.00-3.00   sec  62.1 MBytes   521 Mbits/sec    0    683 KBytes
[  5]   3.00-4.00   sec  63.0 MBytes   529 Mbits/sec    0    683 KBytes
[  5]   4.00-5.00   sec  61.8 MBytes   519 Mbits/sec    0    683 KBytes
[  5]   5.00-6.00   sec  62.6 MBytes   525 Mbits/sec    0    683 KBytes
[  5]   6.00-7.00   sec  62.5 MBytes   524 Mbits/sec    0    683 KBytes
[  5]   7.00-8.00   sec  63.9 MBytes   536 Mbits/sec    0    683 KBytes
[  5]   8.00-9.00   sec  62.5 MBytes   524 Mbits/sec    0    683 KBytes
[  5]   9.00-10.00  sec  61.3 MBytes   515 Mbits/sec    0    683 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   619 MBytes   519 Mbits/sec    0             sender
[  5]   0.00-10.05  sec   618 MBytes   516 Mbits/sec                  receiver

iperf Done.
root@OpenWrt:~# iw dev wlan0 set bitrates vht-mcs-5 3:2
root@OpenWrt:~# iw dev wlan0 station dump|grep "tx bitrate"
        tx bitrate:     6.0 MBit/s
root@OpenWrt:~# iperf3 -c10.0.0.1
Connecting to host 10.0.0.1, port 5201
[  5] local 10.0.0.203 port 35988 connected to 10.0.0.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   694 KBytes  5.69 Mbits/sec    0   45.2 KBytes
[  5]   1.00-2.00   sec   509 KBytes  4.17 Mbits/sec    0   45.2 KBytes
[  5]   2.00-3.00   sec   573 KBytes  4.69 Mbits/sec    0   45.2 KBytes
^C[  5]   3.00-3.45   sec   266 KBytes  4.88 Mbits/sec    0   45.2 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-3.45   sec  1.99 MBytes  4.85 Mbits/sec    0             sender
[  5]   0.00-3.45   sec  0.00 Bytes  0.00 bits/sec                  receiver
iperf3: interrupt - the client has terminated
root@OpenWrt:~# iw dev wlan0 set bitrates vht-mcs-5 3:4
root@OpenWrt:~# iw dev wlan0 station dump|grep "tx bitrate"
        tx bitrate:     526.5 MBit/s VHT-MCS 4 80MHz VHT-NSS 3
root@OpenWrt:~# iperf3 -c10.0.0.1
Connecting to host 10.0.0.1, port 5201
[  5] local 10.0.0.203 port 35992 connected to 10.0.0.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  42.0 MBytes   352 Mbits/sec    0    649 KBytes
[  5]   1.00-2.00   sec  41.4 MBytes   348 Mbits/sec    0    649 KBytes
[  5]   2.00-3.00   sec  42.3 MBytes   354 Mbits/sec    0    649 KBytes
[  5]   3.00-4.00   sec  42.0 MBytes   352 Mbits/sec    0    649 KBytes
[  5]   4.00-5.00   sec  42.5 MBytes   357 Mbits/sec    0    649 KBytes
[  5]   5.00-6.00   sec  42.9 MBytes   360 Mbits/sec    0    679 KBytes
[  5]   6.00-7.00   sec  41.5 MBytes   348 Mbits/sec    0    679 KBytes
[  5]   7.00-8.00   sec  41.8 MBytes   351 Mbits/sec    0    679 KBytes
[  5]   8.00-9.00   sec  43.6 MBytes   365 Mbits/sec    0    711 KBytes
[  5]   9.00-10.00  sec  42.6 MBytes   357 Mbits/sec    0    745 KBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   423 MBytes   355 Mbits/sec    0             sender
[  5]   0.00-10.04  sec   421 MBytes   352 Mbits/sec                  receiver

iperf Done.

```
[dmesg.txt](https://github.com/greearb/ath10k-ct/files/4129075/dmesg.txt)

greearb, 2020-01-29:
I don't see any indication in your dmesg.txt that the firmware was properly configured.  Maybe also enable 'mac' debugging in the ath10k-ct driver?  Here is how my logs look when I tested mcs-2 NSS 1:
82702.709390] ath10k_pci 0000:03:00.0: set bitrate mask, vid: 0  band: 0
[82702.709392] ath10k_pci 0000:03:00.0: mask: band: 0  legacy: 0xfff  gi: 0
[82702.709394] ath10k_pci 0000:03:00.0: ht_mcs: 00 00 00 00 00 00 00 00 00 00
[82702.709395] ath10k_pci 0000:03:00.0: vht_mcs: 0004 0000 0000 0000 0000 0000 0000 0000
[82702.709396] ath10k_pci 0000:03:00.0: mask: band: 1  legacy: 0xff  gi: 0
[82702.709397] ath10k_pci 0000:03:00.0: ht_mcs: 00 00 00 00 00 00 00 00 00 00
[82702.709398] ath10k_pci 0000:03:00.0: vht_mcs: 0004 0000 0000 0000 0000 0000 0000 0000
[82702.709400] ath10k_pci 0000:03:00.0: mac set fixed rate params vdev 0 rate 0xff nss 1 sgi 0
[82702.709501] ath10k_pci 0000:03:00.0: apply-special-rates: ratemask: 0xfff mcast: 0x40  bcast: 0x40  mgmt: 0x43 band: 0
[82702.709614] ath10k_pci 0000:03:00.0: set bitrate mask, vid: 0  band: 0
[82702.709615] ath10k_pci 0000:03:00.0: mask: band: 0  legacy: 0xfff  gi: 0
[82702.709616] ath10k_pci 0000:03:00.0: ht_mcs: 00 00 00 00 00 00 00 00 00 00
[82702.709617] ath10k_pci 0000:03:00.0: vht_mcs: 0004 0000 0000 0000 0000 0000 0000 0000
[82702.709618] ath10k_pci 0000:03:00.0: mask: band: 1  legacy: 0xff  gi: 0
[82702.709619] ath10k_pci 0000:03:00.0: ht_mcs: 00 00 00 00 00 00 00 00 00 00
[82702.709620] ath10k_pci 0000:03:00.0: vht_mcs: 0004 0000 0000 0000 0000 0000 0000 0000
[82702.709621] ath10k_pci 0000:03:00.0: mac set fixed rate params vdev 0 rate 0xff nss 1 sgi 0
[82702.709676] ath10k_pci 0000:03:00.0: apply-special-rates: ratemask: 0xfff mcast: 0x40  bcast: 0x40  mgmt: 0x43 band: 0
[82702.709778] wlan0: 04:f0:21:0f:3b:c3 authenticate with 04:f0:21:c2:fd:b0
[82702.709788] ath10k_pci 0000:03:00.0: mac chanctx add freq 5180 width 3 ptr 0000000091cf2da2
[82702.709789] ath10k_pci 0000:03:00.0: mac monitor recalc started? 0 needed? 0 allowed? 1
[82702.709790] ath10k_pci 0000:03:00.0: mac chanctx assign ptr 0000000091cf2da2 vdev_id 0
[82702.709792] ath10k_pci 0000:03:00.0: mac vdev 0 start center_freq 5180 phymode 11ac-vht80 beacon-int: 0
[82702.715790] ath10k_pci 0000:03:00.0: mac vdev 0 psmode disable
[82702.715795] ath10k_pci 0000:03:00.0: mac vdev 0 cts_protection 0
[82702.715847] ath10k_pci 0000:03:00.0: mac vdev_id 0 txpower 23
[82702.715847] ath10k_pci 0000:03:00.0: mac txpower 23
[82702.715938] ath10k_pci 0000:03:00.0: mac chanctx change freq 5180 width 3 ptr 0000000091cf2da2 changed 10
[82702.715941] ath10k_pci 0000:03:00.0: mac vdev 0 beacon_interval 240
[82702.716123] ath10k_pci 0000:03:00.0: mac vdev 0 peer create 04:f0:21:c2:fd:b0 (new sta) sta 1 / 127 peer 1 / 128
[82702.716406] wlan0: send auth to 04:f0:21:c2:fd:b0 (try 1/5)
[82702.718069] wlan0: authenticated
[82702.718258] ath10k_pci 0000:03:00.0: set bitrate mask, vid: 0  band: 1
[82702.718260] ath10k_pci 0000:03:00.0: mask: band: 0  legacy: 0xfff  gi: 0
[82702.718262] ath10k_pci 0000:03:00.0: ht_mcs: 00 00 00 00 00 00 00 00 00 00
[82702.718263] ath10k_pci 0000:03:00.0: vht_mcs: 0004 0000 0000 0000 0000 0000 0000 0000
[82702.718265] ath10k_pci 0000:03:00.0: mask: band: 1  legacy: 0xff  gi: 0
[82702.718266] ath10k_pci 0000:03:00.0: ht_mcs: 00 00 00 00 00 00 00 00 00 00
[82702.718268] ath10k_pci 0000:03:00.0: vht_mcs: 0004 0000 0000 0000 0000 0000 0000 0000
[82702.718273] ath10k_pci 0000:03:00.0: mac set fixed rate params vdev 0 rate 0xff nss 1 sgi 0
[82702.718336] ath10k_pci 0000:03:00.0: apply-special-rates: ratemask: 0xff mcast: 0x3  bcast: 0x3  mgmt: 0x3 band: 1
[82702.718431] ath10k_pci 0000:03:00.0: mac update sta 04:f0:21:c2:fd:b0 supp rates, bandwidth: 0
[82702.718437] ath10k_pci 0000:03:00.0: mac peer 04:f0:21:c2:fd:b0 phymode 11a  legacy-supp-rates: 0xff  arvif-legacy-rates: 0xff vht-supp: 0
[82702.718438] ath10k_pci 0000:03:00.0: mac peer 04:f0:21:c2:fd:b0 qos 0
[82702.718442] ath10k_pci 0000:03:00.0: apply-special-rates: ratemask: 0xff mcast: 0x3  bcast: 0x3  mgmt: 0x3 band: 1
[82702.718524] ath10k_pci 0000:03:00.0: set bitrate mask, vid: 0  band: 1
[82702.718526] ath10k_pci 0000:03:00.0: mask: band: 0  legacy: 0xfff  gi: 0
[82702.718527] ath10k_pci 0000:03:00.0: ht_mcs: 00 00 00 00 00 00 00 00 00 00
[82702.718528] ath10k_pci 0000:03:00.0: vht_mcs: 0004 0000 0000 0000 0000 0000 0000 0000
[82702.718529] ath10k_pci 0000:03:00.0: mask: band: 1  legacy: 0xff  gi: 0
[82702.718530] ath10k_pci 0000:03:00.0: ht_mcs: 00 00 00 00 00 00 00 00 00 00
[82702.718531] ath10k_pci 0000:03:00.0: vht_mcs: 0004 0000 0000 0000 0000 0000 0000 0000
[82702.718536] ath10k_pci 0000:03:00.0: mac set fixed rate params vdev 0 rate 0xff nss 1 sgi 0
[82702.718741] ath10k_pci 0000:03:00.0: apply-special-rates: ratemask: 0xff mcast: 0x3  bcast: 0x3  mgmt: 0x3 band: 1
[82702.718806] ath10k_pci 0000:03:00.0: mac update sta 04:f0:21:c2:fd:b0 supp rates, bandwidth: 0
[82702.718809] ath10k_pci 0000:03:00.0: mac peer 04:f0:21:c2:fd:b0 phymode 11a  legacy-supp-rates: 0xff  arvif-legacy-rates: 0xff vht-supp: 0
[82702.718810] ath10k_pci 0000:03:00.0: mac peer 04:f0:21:c2:fd:b0 qos 0
[82702.718812] ath10k_pci 0000:03:00.0: apply-special-rates: ratemask: 0xff mcast: 0x3  bcast: 0x3  mgmt: 0x3 band: 1
[82702.719871] wlan0: associate with 04:f0:21:c2:fd:b0 (try 1/3)
[82702.721594] wlan0: RX AssocResp from 04:f0:21:c2:fd:b0 (capab=0x1 status=0 aid=2)
[82702.722448] ath10k_pci 0000:03:00.0: mac vdev 0 slot_time 2
[... cut ...]
[82702.724049] ath10k_pci 0000:03:00.0: mac peer 04:f0:21:c2:fd:b0 qos 1
[82702.724051] ath10k_pci 0000:03:00.0: apply-special-rates: ratemask: 0xff mcast: 0x3  bcast: 0x3  mgmt: 0x3 band: 1
[82702.724145] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[82702.724195] ath10k_pci 0000:03:00.0: mac monitor recalc started? 0 needed? 0 allowed? 1

I use modifications to supplicant and mac80211 to set the advertised and used bitrates, so possible there is some issue with using 'iw' to set the rates.

greearb, 2020-01-29:
Ahh, I mis-read your bug report.  I do appear to see a problem with MCS-1 and NSS 3.  I'll look closer.

greearb, 2020-01-29:
Ok, here is root cause.  The firmware disables certain ratesets, with comments about this helping to optimize things.

Please try this patch to ath10k-ct driver:

[greearb@ben-dt4 ath10k]$ git diff
diff --git a/drivers/net/wireless/ath/ath10k/wmi.c b/drivers/net/wireless/ath/ath10k/wmi.c
index 9e48c8d64684..8c2ff9e06374 100644
--- a/drivers/net/wireless/ath/ath10k/wmi.c
+++ b/drivers/net/wireless/ath/ath10k/wmi.c
@@ -8411,6 +8411,7 @@ ath10k_wmi_peer_assoc_fill(struct ath10k *ar, void *buf,
                        int i;
                        int opver = ar->running_fw->fw_file.wmi_op_version;
                        ext_flags |= PEER_ASSOC_EXT_USE_OVERRIDES;
+                       ext_flags |= PEER_ASSOC_EXT_IGNORE_MCS_4_NSS_MASK;
                        ext_flags |= PEER_ASSOC_EXT_LEN_32;
 
                        ath10k_dbg(ar, ATH10K_DBG_WMI,
diff --git a/drivers/net/wireless/ath/ath10k/wmi.h b/drivers/net/wireless/ath/ath10k/wmi.h
index c03208111a98..bc66cb441d16 100644
--- a/drivers/net/wireless/ath/ath10k/wmi.h
+++ b/drivers/net/wireless/ath/ath10k/wmi.h
@@ -6821,6 +6821,19 @@ struct wmi_10_1_peer_assoc_complete_cmd {
 struct wmi_ct_assoc_overrides {
        /* CT firmware ver 15+ only */
 #define PEER_ASSOC_EXT_USE_OVERRIDES (1<<0)
+       /* wave-1 ath10k-ct firmware, at least, has logic to ignore certain rates.
+        * setting this flag below will disable that logic and give host full control
+        * of the rate-set to use.
+        * Rates disabled in firmware are:
+        * 2x2 11ac rate table does not include:
+        *   MCS 1x1: 5, 6, 7, 8, 9
+        *       2x2: 0, 1, 2
+         * 3x3 11ac rate table does not include:
+        *   MCS 1x1: 3, 4, 5, 6, 7, 8, 9
+        *       2x2: 0, 1, 5, 6, 7, 8, 9
+        *       3x3: 0, 1, 2, 3
+        * As of Jan 29, 2020, this flag is ignored on wave-2 ath10k-ct firmware
+        */
 #define PEER_ASSOC_EXT_IGNORE_MCS_4_NSS_MASK (1<<1)
 #define PEER_ASSOC_EXT_LEN_32        (1<<2) /* Has 32-override bytes */
        __le32 ext_flags;

greearb, 2020-01-29:
I have pushed kernel changes to my ath10k-ct driver repo that allow fwcfg to enable this option (allow_all_mcs = 1)
Keeping it disabled by default.
wave-2 firmware did not appear to have this limitation anyway, so no changes to it.
Thanks for the bug report, I'm closing it.  Please re-open if you still see troubles.

closed 2020-01-29 by greearb