greearb/ath10k-ct #116: ath10k-ct broken on with 5.2 and 5.4

Ansuel, 2020-01-31:
I'm testing ath10k on ipq806x with kernel 5.4

I tested both variant 5.2 and 5.4 and they both crash with...
Variant 4.19 patched to compile on kernel 5.4 works fine... 
```
[   36.788177] WARNING: CPU: 1 PID: 23 at backports-5.4-rc8-1/net/wireless/core.c:626 wiphy_register+0x5c8/0xc18 [cfg80211]
[   36.791692] Modules linked in: ath10k_pci(+) ath10k_core ath pppox ppp_generic mac80211 lz4 iptable_nat ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack_netlink nf_conncount lz4_decompress lz4_compress iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat fuse sch_cake nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport xt_set ip_set_list_set
[   36.792139]  ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sit tunnel4 ip_tunnel tun nls_utf8 zram zsmalloc crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha1_generic seqiv md5 md4 ghash_generic gf128mul gcm ecb ctr cmac ccm authenc usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 uhci_hcd ohci_platform ohci_hcd ahci fsl_mph_dr_of ehci_platform ehci_fsl sd_mod ahci_platform libahci_platform libahci libata scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 btrfs xor xor_neon raid6_pq libcrc32c crc32c_generic
[   36.946502] CPU: 1 PID: 23 Comm: kworker/u4:1 Tainted: G        W         5.4.14 #0
[   36.968598] Hardware name: Generic DT based system
[   36.976199] Workqueue: ath10k_wq ath10k_core_register_work [ath10k_core]
[   36.980959] [<c030f8cc>] (unwind_backtrace) from [<c030b898>] (show_stack+0x14/0x20)
[   36.987810] [<c030b898>] (show_stack) from [<c086fa40>] (dump_stack+0x90/0xa4)
[   36.995540] [<c086fa40>] (dump_stack) from [<c031dfdc>] (__warn+0xb4/0xd0)
[   37.002556] [<c031dfdc>] (__warn) from [<c031e048>] (warn_slowpath_fmt+0x50/0x90)
[   37.009511] [<c031e048>] (warn_slowpath_fmt) from [<bf6a219c>] (wiphy_register+0x5c8/0xc18 [cfg80211])
[   37.017232] [<bf6a219c>] (wiphy_register [cfg80211]) from [<bf702f18>] (ieee80211_register_hw+0x47c/0xbdc [mac80211])
[   37.026369] [<bf702f18>] (ieee80211_register_hw [mac80211]) from [<bf7b37d8>] (ath10k_mac_register+0x7c0/0xbf4 [ath10k_core])
[   37.037000] [<bf7b37d8>] (ath10k_mac_register [ath10k_core]) from [<bf7c05d8>] (ath10k_core_register_work+0x848/0xbc0 [ath10k_core])
[   37.048227] [<bf7c05d8>] (ath10k_core_register_work [ath10k_core]) from [<c0336840>] (process_one_work+0x204/0x464)
[   37.060115] [<c0336840>] (process_one_work) from [<c0336af4>] (worker_thread+0x54/0x588)
[   37.070271] [<c0336af4>] (worker_thread) from [<c033c8f8>] (kthread+0x11c/0x148)
[   37.078595] [<c033c8f8>] (kthread) from [<c03010e8>] (ret_from_fork+0x14/0x2c)
[   37.085960] Exception stack(0xdd59bfb0 to 0xdd59bff8)
[   37.093009] bfa0:                                     00000000 00000000 00000000 00000000
[   37.098149] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   37.106301] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   37.114571] ---[ end trace 192719ad7b76affd ]---
[   37.120966] ath10k_pci 0001:01:00.0: failed to register ieee80211: -22
[   37.125726] ath10k_pci 0001:01:00.0: could not register to mac80211 (-22)
```

robimarko, 2020-02-03:
It appears that this patch resolves this issue.
https://git.openwrt.org/?p=openwrt/staging/hauke.git;a=commit;h=4f617fc72926924281b0cfe3de3ec4a037aed88d

greearb, 2020-02-03:
Yes, thanks for looking up the link.  I have been unable to convince the maintainer to accept this patch upstream, so ath10k-ct users will need to apply it to their kernels one way or another.

closed 2020-02-03 by greearb