greearb/ath10k-ct #117: Warning INVALID WHT

Ansuel, 2020-02-04:
```
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.434623] ------------[ cut here ]------------
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.434810] WARNING: CPU: 0 PID: 9 at target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ath10k-ct-regular/ath10k-ct-2020-01-29-b7297ab4/ath10k-5.4/txrx.c:120 ath10k_txrx_tx_unref+0x57c/0x6a0 [ath10k_core]
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.438380] Invalid VHT rate, nss: 3  hw_rate: 15 ratecode: 255
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.457574] Modules linked in: pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_generic mac80211 lz4 iptable_nat ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_recent xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_helper xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TCPMSS xt_REDIRECT xt_MASQUERADE xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda slhc nf_reject_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack_netlink nf_conncount lz4_decompress lz4_compress iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat fuse sch_cake nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport xt_set
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.457750]  ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb sit tunnel4 ip_tunnel tun nls_utf8 zram zsmalloc crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_generic sha1_generic seqiv md5 md4 ghash_generic gf128mul gcm ecb ctr cmac ccm authenc usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom uhci_hcd ohci_platform ohci_hcd ahci fsl_mph_dr_of ehci_platform ehci_fsl sd_mod ahci_platform libahci_platform libahci libata scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 btrfs xor xor_neon raid6_pq libcrc32c crc32c_generic
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.609402] CPU: 0 PID: 9 Comm: ksoftirqd/0 Not tainted 5.4.14 #0
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.631544] Hardware name: Generic DT based system
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.637809] [<c030f8cc>] (unwind_backtrace) from [<c030b898>] (show_stack+0x14/0x20)
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.642502] [<c030b898>] (show_stack) from [<c08705c0>] (dump_stack+0x90/0xa4)
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.650398] [<c08705c0>] (dump_stack) from [<c031dfdc>] (__warn+0xb4/0xd0)
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.657421] [<c031dfdc>] (__warn) from [<c031e078>] (warn_slowpath_fmt+0x80/0x90)
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.664318] [<c031e078>] (warn_slowpath_fmt) from [<bf7d759c>] (ath10k_txrx_tx_unref+0x57c/0x6a0 [ath10k_core])
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.671965] [<bf7d759c>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf7d146c>] (ath10k_htt_t2h_msg_handler+0xbac/0x114c [ath10k_core])
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.681776] [<bf7d146c>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf6f1810>] (ath10k_pci_htt_rx_cb+0x16c/0x22c [ath10k_pci])
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.693956] [<bf6f1810>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf7f1414>] (ath10k_ce_per_engine_service+0xa0/0x114 [ath10k_core])
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.706101] [<bf7f1414>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf7f1518>] (ath10k_ce_per_engine_service_any+0x90/0xc4 [ath10k_core])
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.718052] [<bf7f1518>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf6f31a4>] (ath10k_pci_napi_poll+0x54/0x160 [ath10k_pci])
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.731141] [<bf6f31a4>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c07316dc>] (net_rx_action+0x118/0x370)
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.743541] [<c07316dc>] (net_rx_action) from [<c030227c>] (__do_softirq+0x114/0x298)
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.753172] [<c030227c>] (__do_softirq) from [<c0321d28>] (run_ksoftirqd+0x38/0x4c)
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.761069] [<c0321d28>] (run_ksoftirqd) from [<c033fb0c>] (smpboot_thread_fn+0xe0/0x1b0)
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.768537] [<c033fb0c>] (smpboot_thread_fn) from [<c033c8f8>] (kthread+0x11c/0x148)
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.776865] [<c033c8f8>] (kthread) from [<c03010e8>] (ret_from_fork+0x14/0x2c)
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.784672] Exception stack(0xdd461fb0 to 0xdd461ff8)
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.791708] 1fa0:                                     00000000 00000000 00000000 00000000
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.796841] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.804996] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
Tue Feb  4 15:43:47 2020 kern.warn kernel: [  811.813248] ---[ end trace f8cd237f7e44aca1 ]---
```

ynezz, 2020-03-16:
Same here on ipq4018/nbg6617, using 5.4.24 and ath10k-ct-2020-02-18-bed49a5f

```
[  135.533866] ath10k_ahb a000000.wifi: Invalid peer id 1 or peer stats buffer, peer: ec208681  sta: a82eade7
[  155.083079] ath10k_ahb a000000.wifi: Invalid VHT mcs 15 peer stats
[  162.877348] ------------[ cut here ]------------
[  162.877678] WARNING: CPU: 0 PID: 0 at target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/ath10k-ct-regular/ath10k-ct-2020-02-18-bed49a5f/ath10k-5.4/txrx.c:121 ath10k_txrx_tx_unref+0x55c/0x67c [ath10k_core]
[  162.881115] Invalid VHT rate, nss: 3  hw_rate: 15 ratecode: 255
[  162.900395] Modules linked in: pppoe ppp_async iptable_nat ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables hwmon crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom gpio_button_hotplug
[  162.943980] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.24 #0
[  162.966142] Hardware name: Generic DT based system
[  162.972008] [<c030e5c4>] (unwind_backtrace) from [<c030b034>] (show_stack+0x10/0x14)
[  162.976856] [<c030b034>] (show_stack) from [<c086f3b4>] (dump_stack+0x94/0xa8)
[  162.984751] [<c086f3b4>] (dump_stack) from [<c03211b0>] (__warn+0xbc/0xd8)
[  162.991776] [<c03211b0>] (__warn) from [<c0321250>] (warn_slowpath_fmt+0x84/0x94)
[  162.998832] [<c0321250>] (warn_slowpath_fmt) from [<bf2ce420>] (ath10k_txrx_tx_unref+0x55c/0x67c [ath10k_core])
[  163.006523] [<bf2ce420>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf2c872c>] (ath10k_htt_t2h_msg_handler+0xb84/0x114c [ath10k_core])
[  163.016272] [<bf2c872c>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf321544>] (ath10k_pci_htt_rx_cb+0x164/0x20c [ath10k_pci])
[  163.028441] [<bf321544>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf2e68d0>] (ath10k_ce_per_engine_service+0x80/0xd8 [ath10k_core])
[  163.040672] [<bf2e68d0>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf2e6998>] (ath10k_ce_per_engine_service_any+0x70/0xb0 [ath10k_core])
[  163.052542] [<bf2e6998>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf323544>] (ath10k_pci_napi_poll+0x44/0x13c [ath10k_pci])
[  163.065438] [<bf323544>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c0728ba8>] (net_rx_action+0x10c/0x35c)
[  163.077821] [<c0728ba8>] (net_rx_action) from [<c0302288>] (__do_softirq+0x120/0x2b0)
[  163.087448] [<c0302288>] (__do_softirq) from [<c0325150>] (irq_exit+0xb0/0xd4)
[  163.095343] [<c0325150>] (irq_exit) from [<c0369a30>] (__handle_domain_irq+0x60/0xb4)
[  163.102467] [<c0369a30>] (__handle_domain_irq) from [<c05a826c>] (gic_handle_irq+0x4c/0x90)
[  163.110363] [<c05a826c>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[  163.118496] Exception stack(0xc0c01f28 to 0xc0c01f70)
[  163.126147] 1f20:                   00000000 00dd25bc cfd771f4 c0313c80 ffffe000 c0c04e68
[  163.131193] 1f40: c0c04ea8 00000001 00000000 c0b3ede8 c0c0dcf4 00000000 0f237000 c0c01f78
[  163.139337] 1f60: c0308470 c0308474 60000013 ffffffff
[  163.147516] [<c0301a8c>] (__irq_svc) from [<c0308474>] (arch_cpu_idle+0x38/0x3c)
[  163.152554] [<c0308474>] (arch_cpu_idle) from [<c03496d0>] (do_idle+0x1e4/0x21c)
[  163.160014] [<c03496d0>] (do_idle) from [<c0349998>] (cpu_startup_entry+0x18/0x1c)
[  163.167393] [<c0349998>] (cpu_startup_entry) from [<c0b00dfc>] (start_kernel+0x490/0x49c)
[  163.174805] ---[ end trace a839a6d0ba14a6c6 ]---
```

Noltari, 2020-03-19:
Same here on ipq806x/r7800, using 5.4.24 and ath10k-ct-2020-02-18-bed49a5f
`[  697.568886] ------------[ cut here ]------------
[  697.568999] WARNING: CPU: 0 PID: 0 at target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ath10k-ct-regular/ath10k-ct-2020-02-18-bed49a5f/ath10k-5.4/txrx.c:121 ath10k_txrx_tx_unref+0x57c/0x6a0 [ath10k_core]
[  697.572676] Invalid VHT rate, nss: 3  hw_rate: 15 ratecode: 255
[  697.591848] Modules linked in: pppoe ppp_async iptable_nat ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG wireguard slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ip6_udp_tunnel udp_tunnel leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom ohci_platform ohci_hcd phy_qcom_dwc3 ahci fsl_mph_dr_of ehci_platform ehci_fsl sd_mod ahci_platform libahci_platform libahci libata scsi_mod ehci_hcd gpio_button_hotplug
[  697.652247] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.24 #0
[  697.674392] Hardware name: Generic DT based system
[  697.680227] [<c030f9d4>] (unwind_backtrace) from [<c030b9f0>] (show_stack+0x14/0x20)
[  697.685086] [<c030b9f0>] (show_stack) from [<c08a15fc>] (dump_stack+0x94/0xa8)
[  697.692989] [<c08a15fc>] (dump_stack) from [<c031e64c>] (__warn+0xb4/0xd0)
[  697.700012] [<c031e64c>] (__warn) from [<c031e6e8>] (warn_slowpath_fmt+0x80/0x90)
[  697.706905] [<c031e6e8>] (warn_slowpath_fmt) from [<bf3a0580>] (ath10k_txrx_tx_unref+0x57c/0x6a0 [ath10k_core])
[  697.714523] [<bf3a0580>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf39a488>] (ath10k_htt_t2h_msg_handler+0xba0/0x1138 [ath10k_core])
[  697.724368] [<bf39a488>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf3f3768>] (ath10k_pci_htt_rx_cb+0x178/0x230 [ath10k_pci])
[  697.736525] [<bf3f3768>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf3ba5d4>] (ath10k_ce_per_engine_service+0x9c/0x10c [ath10k_core])
[  697.748696] [<bf3ba5d4>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf3ba6c4>] (ath10k_ce_per_engine_service_any+0x80/0xd8 [ath10k_core])
[  697.760648] [<bf3ba6c4>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf3f5118>] (ath10k_pci_napi_poll+0x54/0x15c [ath10k_pci])
[  697.773723] [<bf3f5118>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c075fdbc>] (net_rx_action+0x118/0x374)
[  697.786129] [<c075fdbc>] (net_rx_action) from [<c0302298>] (__do_softirq+0x130/0x2d4)
[  697.795760] [<c0302298>] (__do_softirq) from [<c0322a2c>] (irq_exit+0xbc/0xe0)
[  697.803657] [<c0322a2c>] (irq_exit) from [<c036cc5c>] (__handle_domain_irq+0x6c/0xd0)
[  697.810782] [<c036cc5c>] (__handle_domain_irq) from [<c05b4c38>] (gic_handle_irq+0x5c/0xb8)
[  697.818677] [<c05b4c38>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[  697.826827] Exception stack(0xc0c01ee0 to 0xc0c01f28)
[  697.834480] 1ee0: 00000000 000000a2 1ce53000 dd991a00 dcc29400 00000000 dd990df0 000000a2
[  697.839519] 1f00: 000000a2 00000000 6a4e0ea0 6a4b3e00 00000015 c0c01f30 c0703428 c070342c
[  697.847666] 1f20: 00000013 ffffffff
[  697.855835] [<c0301a8c>] (__irq_svc) from [<c070342c>] (cpuidle_enter_state+0x94/0x4a8)
[  697.859132] [<c070342c>] (cpuidle_enter_state) from [<c0703884>] (cpuidle_enter+0x30/0x4c)
[  697.867116] [<c0703884>] (cpuidle_enter) from [<c034a434>] (do_idle+0x1d8/0x240)
[  697.875443] [<c034a434>] (do_idle) from [<c034a744>] (cpu_startup_entry+0x1c/0x20)
[  697.883001] [<c034a744>] (cpu_startup_entry) from [<c0b00e5c>] (start_kernel+0x4dc/0x4e8)
[  697.890451] ---[ end trace 59ab6e43b03f2eb1 ]---`

Noltari, 2020-03-23:
> Same here on ipq806x/r7800, using 5.4.24 and ath10k-ct-2020-02-18-bed49a5f
>
> ```
> [  697.568886] ------------[ cut here ]------------
> [  697.568999] WARNING: CPU: 0 PID: 0 at [...]

This also happens on latest ath10k-ct master (8df0a7f80a8bb2c49d3a16fe70f9a619d7bedd5a).

ynezz, 2020-03-28:
Still occurring on 5.4.28 with firmware 10.4b-ct-4019-fW-013-b63cea875 and ath10k-ct-2020-03-25-3d173a47 driver.

ptpt52, 2020-04-03:
same here ipq40xx kernel 5.4 etc..
```
[10612.189578] WARNING: CPU: 0 PID: 0 at target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/ath10k-ct-regular/ath10k-ct-2020-03-25-3d173a47/ath10k-5.4/txrx.c:134 ath10k_txrx_tx_unref+0x550/0x714 [ath10k_core]
[10612.193127] Invalid VHT rate, nss: 3  hw_rate: 15 ratecode: 255
[10612.212333] Modules linked in: qcserial pppoe ppp_async option l2tp_ppp cdc_mbim ath10k_pci ath10k_core ath usb_wwan sierra_net sierra rndis_host qmi_wwan pptp pppox ppp_mppe ppp_generic mac80211 ipt_REJECT huawei_cdc_ncm cfg80211 cdc_ncm cdc_ether xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_ipp2p xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_NETMAP xt_MASQUERADE xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY wireguard usbserial usbnet usblp ts_fsm ts_bm slhc sch_cake r8152 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_redir nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_ct nft_counter nf_tproxy_ipv6 nf_tproxy_ipv4
[10612.212561]  nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_rtsp nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_dup_netdev nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtsp nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount macvlan iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables hwmon crc_ccitt compat_xtables compat cdc_wdm br_netfilter asn1_decoder natflow natcap fuse tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac
[10612.283291]  ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 msdos ip6_gre ip_gre gre ifb sit l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ip6_tunnel tunnel6 tunnel4 ip_tunnel tun vfat fat autofs4 nls_utf8 nls_iso8859_1 nls_cp437 sha1_generic sha1_arm_neon sha1_arm arc4 uas usb_storage leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom ohci_platform ohci_hcd fsl_mph_dr_of ehci_platform ehci_fsl sd_mod scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 exfat mii crc32c_generic
[10612.445022] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.28 #0
[10612.467242] Hardware name: Generic DT based system
[10612.473201] [<c030e524>] (unwind_backtrace) from [<c030afa8>] (show_stack+0x10/0x14)
[10612.478044] [<c030afa8>] (show_stack) from [<c0850bb8>] (dump_stack+0x94/0xa8)
[10612.485937] [<c0850bb8>] (dump_stack) from [<c03210ac>] (__warn+0xbc/0xd8)
[10612.492964] [<c03210ac>] (__warn) from [<c032114c>] (warn_slowpath_fmt+0x84/0x94)
[10612.500704] [<c032114c>] (warn_slowpath_fmt) from [<bf94c57c>] (ath10k_txrx_tx_unref+0x550/0x714 [ath10k_core])
[10612.509273] [<bf94c57c>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf946a50>] (ath10k_htt_t2h_msg_handler+0xe0c/0x11f0 [ath10k_core])
[10612.518188] [<bf946a50>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf99c4d4>] (ath10k_pci_htt_rx_cb+0x164/0x20c [ath10k_pci])
[10612.530822] [<bf99c4d4>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf964410>] (ath10k_ce_per_engine_service+0x80/0xd8 [ath10k_core])
[10612.544125] [<bf964410>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf9644d8>] (ath10k_ce_per_engine_service_any+0x70/0xb0 [ath10k_core])
[10612.554471] [<bf9644d8>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf99dcec>] (ath10k_pci_napi_poll+0x44/0x13c [ath10k_pci])
[10612.566639] [<bf99dcec>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c071faa0>] (net_rx_action+0x10c/0x35c)
[10612.579015] [<c071faa0>] (net_rx_action) from [<c0302288>] (__do_softirq+0x120/0x2b0)
[10612.588635] [<c0302288>] (__do_softirq) from [<c0325040>] (irq_exit+0xb0/0xd4)
[10612.596530] [<c0325040>] (irq_exit) from [<c0369858>] (__handle_domain_irq+0x60/0xb4)
[10612.603871] [<c0369858>] (__handle_domain_irq) from [<c059e970>] (gic_handle_irq+0x4c/0x90)
[10612.611746] [<c059e970>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[10612.619784] Exception stack(0xc0c01f28 to 0xc0c01f70)
[10612.628618] 1f20:                   00000000 0113e764 cfd781f4 c0313be0 ffffe000 c0c04e68
[10612.632656] 1f40: c0c04ea8 00000001 00000000 c0b3ede8 c0c0dc14 00000000 0f238000 c0c01f78
[10612.640770] 1f60: c03083e8 c03083ec 60000013 ffffffff
[10612.648982] [<c0301a8c>] (__irq_svc) from [<c03083ec>] (arch_cpu_idle+0x38/0x3c)
[10612.654867] [<c03083ec>] (arch_cpu_idle) from [<c0349428>] (do_idle+0x1e4/0x21c)
[10612.661300] [<c0349428>] (do_idle) from [<c03496f0>] (cpu_startup_entry+0x18/0x1c)
[10612.668678] [<c03496f0>] (cpu_startup_entry) from [<c0b00df8>] (start_kernel+0x48c/0x498)
[10612.676103] ---[ end trace 58ddea154dc252cc ]---

```

PR #129 (2020-04-29): ath10k-ct: workaround TX rate code firmware bug
It seems, that we get a bad tx/rx rate from firmware, it is not a real
problem so just check for invalid rate (0xff) and force it to be zero
instead.

Fixes: #117
Ref: https://bugs.openwrt.org/index.php?do=details&task_id=3055
commit c1b6fa64 (2020-04-29): ath10k-ct: workaround TX rate code firmware bug

It seems, that we get a bad tx/rx rate from firmware, it is not a real
problem so just check for invalid rate (0xff) and force it to be zero
instead.

Fixes: #117
Ref: https://bugs.openwrt.org/index.php?do=details&task_id=3055
closed 2020-04-29 by greearb