greearb/ath10k-ct #120: QCA9884 MU-MIMO in 2.4 Ghz only 2x2

pgawlowicz, 2020-02-19:
Hi,

In our research project, we investigate the performance of MU-MIMO. 
We have setup a small testbed with one AP and 4 stations, all using QCA9884 in the 5Ghz band. We connected stations using RF cables (one STA to one antenna port of the AP) to get ideal channel conditions and manage to get a 4x increase in the throughput as measured with iperf. Therefore, we believe that 4x4 MU-MIMO works correctly in the 5GHz band.

Then we bought another set of QCA9884 but for the 2.4GHz band. It turned out that using standard kernel and firmware, it is not possible to enable MU-MIMO in 2.4 GHz. Fortunately, when using kernel and firmware provided by Candela it works. However, we managed to achieve only a 2x increase in the throughput as measured with iperf (still STAs connected over the cable with AP). It looks like only 2x2 MU-MIMO is available.

I would like to ask whether it is the firmware (or the board file) that was compiled with only 2x2 MU-MIMO capabilities or is it possible to modify ath10k driver to enable 4x4 MU-MIMO also in 2.4GHz band?

I found two functions in ath10k/mac.c file, namely:
- ath10k_mac_get_vht_cap_bf_sts
- ath10k_mac_get_vht_cap_bf_sound_dim

that return the number of space-time streams and sounding dimensions as reported by the firmware. I did not try it yet, but I want to overwrite those numbers before function returns. The question is whether vht_capabilities, as provided by the firmware, mask out the real capability and it will work after we overwrite those numbers or the firmware simply does not support 4x4 MU-MIMO and it will crush after we overwrite it?

@greearb What do you think?

Best Regards,
Piotr

dengqf6, 2020-02-19:
AFAIK MU-MIMO is not a 802.11n standard.

pgawlowicz, 2020-02-19:
Yes, I know. But we are looking for a platform that will can use in our research. And QCA9884 with Candela kernel and firmware seems to be a fit if we manage to enable 4x4 MU-MIMO instead of 2x2 in 2.4GHz band.

greearb, 2020-02-19:
I have not tried MU-MIMO for 2.4Ghz band, but we have tested 4x4 MIMO 40Mhz /AC mode on 2.4 and that works fine, so the firmware and hardware is enabling all of the 4 chains on 2.4Ghz.  Maybe see if you can get 4x4 SU-MIMO working first?