greearb/ath10k-ct #121: Ath10k-ct OR Ath10k-ct-htt

LeeMenHin, 2020-02-29:
I feel very confused about it ,comparing which one is better for my router ,since no difference at all , but also what is the meaning of "HTT”.

Which one  should be perfom better ?

My chip is ipq4019+qca9984+9984

greearb, 2020-03-05:
Use HTT mgt firmware, it puts managements frames over the normal data frame transport (HTT protocol in the driver), and this lets management frames scale better, especially in cases where there are lots of connections.

timkgh, 2020-04-25:
There are 3 flavors here:
https://www.candelatech.com/downloads/ath10k-9984-10-4b/?C=M;O=D
What is the difference between `full`, `full-htt-mgt` and `htt-mgt`? In particular would one want to use `full-htt-mgt` or `htt-mgt`?

greearb, 2020-04-25:
Use the htt-mgt, it has more RAM available to support more stations.  It disables some firmware features that I think no one will miss or is using, but possibly someone is, so that is why the full
version is still there.

Fail-Safe, 2020-04-27:
@greearb First of all, thank you so much for your work on continuing development for ath10k!

Out of curiosity, what is the benefit of disabling those firmware features, even if someone is not using them?

greearb, 2020-04-27:
It allows around 50% more stations to be supported per radio, or more vdevs or tids or other useful things.

Fail-Safe, 2020-04-27:
Oh fantastic! Thanks for that detail. I will switch over to the `htt-mgt` from `full-htt-mgt` on my Netgear R7800.

greearb, 2020-04-27:
You have to mess with fwcfg to actually utilize the additional resources, by the way.  I have not specifically tuned a 9984 fwcfg for AP stations recently, but will do so sometime soon.  The info on this page is still relevant though:
http://www.candelatech.com/ath10k-10.4.php#config

I am hoping someone will improve OpenWrt to utilize the fwcfg api, but I think for now it does not.

timkgh, 2020-09-17:
@greearb Build 020 offers another flavor `qcache`. What is the difference and which one should be the preferred one to run?
https://www.candelatech.com/downloads/ath10k-9984-10-4b/?C=M;O=D