greearb/ath10k-ct #122: Feature (CSI): Add firmware option to ignore 'not sounding bit' on RX packets - return CSI info for all RX, not just sounding packets.

raplin, 2020-03-04:
Lots of people are interested in using Channel State Information, and I see @pgawlowicz  has contributed a patch to feed CSI to userspace. 

However, as I understand it, the QCA firmware only reports CSI on sounding packets; Null Data Packet (NDP) with the "not sounding" bit zeroed.  

It appears to be possible to report CSI on _all_ incoming HT packets; as the process is based on sampling the standard HT-LTF preamble  (info from here: http://www.hitchhikersguidetolearning.com/2017/09/17/channel-sounding-for-transmit-beamforming/ )
It would be great to have a diagnostic option to enable CSI reports on all/most incoming packets, I believe the (commonly used) CSI-modded fw for Intel 5300 works like this.

This obviously requires (hopefully trivial) mods to the QCA firmware, plus adding another option switch to the drivers etc. 

Thanks for considering this!

pgawlowicz, 2020-03-05:
Hi,
Last time when I played with QCA firmware provided by @greearb the CSI was delivered only for the ACK frame send after Probe Response. 
So even, if the AP was configured in 80MHz mode, the ACK frame was sent with 20MHz bandwidth.
It would be nice to make CSI reporting functionality configurable by applying some filters (e.g. source MAC address in the received frame).
I also tried to decode the CSI message but the results were not really satisfactory, see an example plot below:
[image]

I found a note from Ben somewhere in the code that the format is unknown. I tried decoding it trying different combinations. The example above was created in the following way. I decoded the CSI report as an array of uint8 values. It looks like the even values are amplitude (blue) and odd values are phase (red). But I am not quite sure, as they might be also I and Q values encoded in some unknown way.

Best,
Piotr

greearb, 2020-03-05:
I am unlikely to work on this anytime soon...it does not appear to be at all simple to enable more CSI reporting, or at least I don't know the magic incantations in the firmware to do so.

raplin, 2020-03-05:
Thanks both for your prompt replies. 

I see the ESP32 can do this (single antenna); 
API: 
https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#wi-fi-channel-state-information

discussion of usage, sample graphs etc: 
https://github.com/espressif/esp-idf/issues/2012

ESP32 is handy and cheap but it's 1T1R; I would imagine an Ath10k with multiple synchronous RX antennas and superior RF construction to a $4 module would give significantly better data to play with, as well as being available conveniently packaged inside commercial routers.

Thanks for the info I will report back if I make any progress.

raplin, 2020-03-05:
Interestingly I found a csi matrix in the Realtek RTL8812 kernel driver... 
https://github.com/aircrack-ng/rtl8812au/blob/945d6ed6505c32f0993b1dba576388e92e78101b/hal/phydm/phydm_beamforming.h#L145