greearb/ath10k-ct #123: Firmware Crash

Noltari, 2020-03-23:
TP-Link Archer C7 v2 using OpenWrt master (4.19.108) and ath10k-ct 8df0a7f8 with HTT firmware.

```
[255398.211104] ath10k_pci 0000:00:00.0: firmware crashed! (guid n/a)
[255398.217451] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[255398.226914] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[255398.239478] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fH-022-b8a7ba1b api 2 features wmi-10.x,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 9fc30654
[255398.269158] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[255398.276863] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[255398.286889] ath10k_pci 0000:00:00.0: firmware register dump:
[255398.292849] ath10k_pci 0000:00:00.0: [00]: 0x4100016C 0x000015B3 0x00996491 0x00955B31
[255398.300970] ath10k_pci 0000:00:00.0: [04]: 0x00996491 0x00060330 0x00000005 0x00000051
[255398.309196] ath10k_pci 0000:00:00.0: [08]: 0x00000000 0x009BFC9C 0x00426A48 0x00459108
[255398.317345] ath10k_pci 0000:00:00.0: [12]: 0x00000009 0x00000000 0x0095808C 0x009580A2
[255398.325650] ath10k_pci 0000:00:00.0: [16]: 0x00958080 0x0094085D 0x00000000 0x00000000
[255398.334035] ath10k_pci 0000:00:00.0: [20]: 0x40996491 0x0040AC54 0x00000000 0x00000000
[255398.342155] ath10k_pci 0000:00:00.0: [24]: 0x80996D29 0x0040ACB4 0x004266AC 0xC0996491
[255398.350339] ath10k_pci 0000:00:00.0: [28]: 0x80997930 0x0040AD24 0x0041F194 0x00426A48
[255398.358556] ath10k_pci 0000:00:00.0: [32]: 0x80999C45 0x0040ADC4 0x0041F194 0x00426A48
[255398.366904] ath10k_pci 0000:00:00.0: [36]: 0x809B72B9 0x0040AEE4 0x00426450 0x00000002
[255398.375273] ath10k_pci 0000:00:00.0: [40]: 0x80940F18 0x0040AF14 0x00000004 0x004039D0
[255398.383459] ath10k_pci 0000:00:00.0: [44]: 0x80940EEA 0x0040AF44 0x00400000 0x00000000
[255398.391575] ath10k_pci 0000:00:00.0: [48]: 0x80940F31 0x0040AF64 0x00401C10 0x00400600
[255398.399724] ath10k_pci 0000:00:00.0: [52]: 0x40940024 0x0040AF84 0x004068E8 0x004068E8
[255398.407973] ath10k_pci 0000:00:00.0: [56]: 0x00000000 0x0040AFA4 0x009BB001 0x00040020
[255398.416313] ath10k_pci 0000:00:00.0: Copy Engine register dump:
[255398.422477] ath10k_pci 0000:00:00.0: [00]: 0x00057400  14  14   3   3
[255398.429103] ath10k_pci 0000:00:00.0: [01]: 0x00057800  23  23 409 410
[255398.435752] ath10k_pci 0000:00:00.0: [02]: 0x00057c00  53  53  52  53
[255398.442393] ath10k_pci 0000:00:00.0: [03]: 0x00058000  18  18  22  18
[255398.449023] ath10k_pci 0000:00:00.0: [04]: 0x00058400 6547 6547 243 203
[255398.455847] ath10k_pci 0000:00:00.0: [05]: 0x00058800   3   3  66  67
[255398.462484] ath10k_pci 0000:00:00.0: [06]: 0x00058c00  17  17  17  17
[255398.469108] ath10k_pci 0000:00:00.0: [07]: 0x00059000   0   0   0   0
[255398.476064] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x412708  dropped: 0
[255398.484020] ath10k_pci 0000:00:00.0: [0] next: 0x412720 buf: 0x41056c sz: 1500 len: 80 count: 3 free: 0
[255398.493914] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[255398.500321] ath10k: [0000]: CCAB9607 514CFC17 41841091 00000000 00000000 00004E00 08914500 CCAB9607
[255398.509717] ath10k: [0008]: 514CFC13 42841091 00000000 00000000 00000000 CCAB9607 0100FC17 B3150000
[255398.519080] ath10k: [0016]: B3150000 44AB4000 6C010041 00090000
[255398.525210] ath10k_pci 0000:00:00.0: ATH10K_END
[255398.529971] ath10k_pci 0000:00:00.0: [1] next: 0x412708 buf: 0x410b5c sz: 1500 len: 0 count: 0 free: 0
[255398.543779] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 02c47576 vdev: 0 addr: 50:bc:96:c5:d9:47 
[255398.555043] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8115c18c vdev: 0 addr: 14:cc:20:51:3a:35 
[255398.697840] ieee80211 phy0: Hardware restart was requested
[255399.723678] ath10k_pci 0000:00:00.0: unsupported HTC service id: 1536
[255399.730369] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[255399.748019] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[255399.754478] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[255399.762744] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 20920 iram: 26216'
[255399.831650] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[255399.839105] ath10k_pci 0000:00:00.0: set-coverage-class, phyclk: 88  value: 0
[255400.850090] ath10k_pci 0000:00:00.0: Invalid state: 3 in ath10k_htt_tx_32, warning will not be repeated.
[255400.859853] ------------[ cut here ]------------
[255400.864714] WARNING: CPU: 0 PID: 5941 at target-mips_74kc+dsp2_musl/linux-ath79_generic/ath10k-ct-regular/ath10k-ct-2020-02-18-bed49a5f/ath10k-5.4/htt_tx.c:1427 ieee80211_is_robust_mgmt_frame_tx+0xb60/0xbdc [ath10k_core]
[255400.884642] Modules linked in: ath9k ath9k_common pppoe ppp_async iptable_nat ipt_MASQUERADE ath9k_hw ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_FLOWOFFLOAD pppox ppp_generic nf_nat_ipv4 nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ehci_platform ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[255400.949289] CPU: 0 PID: 5941 Comm: kworker/0:2 Not tainted 4.19.108 #0
[255400.956122] Workqueue: events_freezable ieee80211_alloc_hw_nm [mac80211]
[255400.963023] Stack : 8056f1f0 870ab924 870e16a4 00000593 85eb8638 800bc2e4 87d60eb4 00000000
[255400.971584]         00000017 85b089b8 8056f1d8 871a19b4 805d0000 00000001 871a1988 208787ee
[255400.980156]         00000000 00000000 805df030 0000012f 3231315d 00000000 6c6f635f 68775f6e
[255400.988725]         0000012f a2baf404 00000000 65657a61 00000000 00000009 00000000 870ab924
[255400.997298]         870e16a4 00000593 85eb8638 873e0da8 00000000 802e6024 00000000 80720000
[255401.005869]         ...
[255401.008432] Call Trace:
[255401.011014] [<8006b1f0>] show_stack+0x30/0x100
[255401.015638] [<80087270>] __warn+0xe4/0x13c
[255401.019880] [<80087310>] warn_slowpath_null+0x48/0x68
[255401.025153] [<870ab924>] ieee80211_is_robust_mgmt_frame_tx+0xb60/0xbdc [ath10k_core]
[255401.033134] [<870818e8>] 0x870818e8
[255401.036771] ---[ end trace 22c5996a800c771e ]---
[255401.041547] ath10k_pci 0000:00:00.0: failed to transmit packet, dropping: -19
[255401.048892] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[255401.055076] ath10k_pci 0000:00:00.0: failed to transmit frame: -19
[255401.061438] ath10k_pci 0000:00:00.0: failed to transmit packet, dropping: -19
[255401.068783] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[255401.074974] ath10k_pci 0000:00:00.0: failed to transmit frame: -19
[255401.081926] ath10k_pci 0000:00:00.0: failed to transmit packet, dropping: -19
[255401.089306] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[255401.095498] ath10k_pci 0000:00:00.0: failed to push frame: -19
[255401.101805] ath10k_pci 0000:00:00.0: device successfully recovered
```

greearb, 2020-03-23:
This bug is similar to what 'stintel' reported some time back.  The tx-callback code in the firmware has a null-pointer for one of the callback objects.  I did get a bit of useful debug out of this crash (bad address is 0x0), but still no idea why it is happening.  For now, I'm going to ignore this bug and see if anyone manages to reproduce it.  Please let me know if you see more crashes, possibly you have a system that can reproduce this and then I might can make some more progress.

Noltari, 2020-03-28:
Hello @greearb,

It happened again with latest firmware and driver updates:
```[164612.867067] ath10k_pci 0000:00:00.0: failed to set cts protection for vdev 0: -143
[164612.874914] ath10k_pci 0000:00:00.0: failed to set preamble for vdev 0: -143
[164612.882179] ath10k_pci 0000:00:00.0: failed to set mgmt tx rate -143
[164612.888794] ath10k_pci 0000:00:00.0: firmware crashed! (guid n/a)
[164612.895202] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[164612.904731] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[164612.917308] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fH-022-cc5313da api 2 features wmi-10.x,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 84cbd09d
[164612.946887] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[164612.954532] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[164612.964412] ath10k_pci 0000:00:00.0: firmware register dump:
[164612.970243] ath10k_pci 0000:00:00.0: [00]: 0x4100016C 0x000015B3 0x009964D9 0x00955B31
[164612.978392] ath10k_pci 0000:00:00.0: [04]: 0x009964D9 0x00060330 0x00000005 0x00000051
[164612.986544] ath10k_pci 0000:00:00.0: [08]: 0x00000000 0x009BF86C 0x004271C4 0x00459108
[164612.994676] ath10k_pci 0000:00:00.0: [12]: 0x00000009 0x00000000 0x0095808C 0x009580A2
[164613.002862] ath10k_pci 0000:00:00.0: [16]: 0x00958080 0x0094085D 0x00000000 0x00000000
[164613.011011] ath10k_pci 0000:00:00.0: [20]: 0x409964D9 0x0040AC54 0x00000002 0x00000000
[164613.019148] ath10k_pci 0000:00:00.0: [24]: 0x80996CC4 0x0040ACB4 0x00000000 0xC09964D9
[164613.027316] ath10k_pci 0000:00:00.0: [28]: 0x80997988 0x0040AD24 0x0041F184 0x004271C4
[164613.035478] ath10k_pci 0000:00:00.0: [32]: 0x80999C9D 0x0040ADC4 0x0041F184 0x004271C4
[164613.043622] ath10k_pci 0000:00:00.0: [36]: 0x809B738D 0x0040AEE4 0x00426440 0x00000002
[164613.051845] ath10k_pci 0000:00:00.0: [40]: 0x80940F18 0x0040AF14 0x00000005 0x004039E4
[164613.060054] ath10k_pci 0000:00:00.0: [44]: 0x80940EEA 0x0040AF44 0x00400000 0x00000000
[164613.068261] ath10k_pci 0000:00:00.0: [48]: 0x80940F31 0x0040AF64 0x00401C10 0x00400600
[164613.076415] ath10k_pci 0000:00:00.0: [52]: 0x40940024 0x0040AF84 0x004068E8 0x004068E8
[164613.084550] ath10k_pci 0000:00:00.0: [56]: 0x00000000 0x0040AFA4 0x009BB001 0x00040020
[164613.092684] ath10k_pci 0000:00:00.0: Copy Engine register dump:
[164613.098797] ath10k_pci 0000:00:00.0: [00]: 0x00057400   1   1   3   3
[164613.105444] ath10k_pci 0000:00:00.0: [01]: 0x00057800  31  31 384 386
[164613.112067] ath10k_pci 0000:00:00.0: [02]: 0x00057c00  45  45 108 109
[164613.118706] ath10k_pci 0000:00:00.0: [03]: 0x00058000  21  21  22  21
[164613.125340] ath10k_pci 0000:00:00.0: [04]: 0x00058400 6045 6045 248 208
[164613.132139] ath10k_pci 0000:00:00.0: [05]: 0x00058800  20  20  19  20
[164613.138778] ath10k_pci 0000:00:00.0: [06]: 0x00058c00  28  28  28  28
[164613.145413] ath10k_pci 0000:00:00.0: [07]: 0x00059000   0   0   0   0
[164613.152273] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x412710  dropped: 0
[164613.160118] ath10k_pci 0000:00:00.0: [0] next: 0x4126f8 buf: 0x410b4c sz: 1500 len: 264 count: 10 free: 0
[164613.170041] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[164613.176420] ath10k: [0000]: 35280C02 025C0014 04024400 ACBE9B00 4098AD17 97970000 01000000 35280C02
[164613.185706] ath10k: [0008]: 574C0010 88991071 D89D4300 6CF89B00 11000000 35280C02 574C0014 D89D4300
[164613.194988] ath10k: [0016]: B8DC4300 00000000 0C000000 8E1D0000 35280C02 574C0014 D89D4300 B4F89B00
[164613.204350] ath10k: [0024]: 13000000 00000000 223D0000 35280C02 574C0014 D89D4300 FCF89B00 06000000
[164613.213698] ath10k: [0032]: 0C000000 0E3D0000 35280C02 574C0010 88991071 D89D4300 14FC9B00 10000000
[164613.222993] ath10k: [0040]: 35280C02 4B4C0010 20513A35 000014CC 3A350000 14CC2051 35280C02 514CFC17
[164613.232288] ath10k: [0048]: 41841091 00000000 00000000 00004B00 08914500 35280C02 514CFC13 42841091
[164613.241601] ath10k: [0056]: 00000000 00000000 02000000 35280C02 0100FC17 B3150000 B3150000 44AB4000
[164613.250930] ath10k: [0064]: 6C010041 50D99B00
[164613.255511] ath10k_pci 0000:00:00.0: ATH10K_END
[164613.260294] ath10k_pci 0000:00:00.0: [1] next: 0x412710 buf: 0x41055c sz: 1500 len: 0 count: 0 free: 0
[164613.276685] ath10k_pci 0000:00:00.0: failed to send pdev bss chan info request: -143
[164613.285534] ath10k_pci 0000:00:00.0: failed to set beacon mode for vdev 0: -143
[164613.293126] ath10k_pci 0000:00:00.0: failed to set dtim period for vdev 0: -143
[164613.301098] ath10k_pci 0000:00:00.0: failed to recalculate rts/cts prot for vdev 0: -143
[164613.309543] ath10k_pci 0000:00:00.0: failed to set cts protection for vdev 0: -143
[164613.317401] ath10k_pci 0000:00:00.0: failed to set preamble for vdev 0: -143
[164613.324686] ath10k_pci 0000:00:00.0: failed to set mgmt tx rate -143
[164613.332395] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 55120b34 vdev: 0 addr: 14:cc:20:51:3a:35 
[164613.477634] ieee80211 phy0: Hardware restart was requested
[164614.505427] ath10k_pci 0000:00:00.0: unsupported HTC service id: 1536
[164614.512115] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[164614.529707] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[164614.536155] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[164614.544404] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 21000 iram: 25992'
[164614.613195] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[164614.620567] ath10k_pci 0000:00:00.0: set-coverage-class, phyclk: 88  value: 0
[164615.630620] ath10k_pci 0000:00:00.0: device successfully recovered```

Noltari, 2020-03-30:
And again (x3):
```
[263944.472298] ath10k_pci 0000:00:00.0: firmware crashed! (guid n/a)
[263944.478637] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[263944.488192] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[263944.500764] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fH-022-cc5313da api 2 features wmi-10.x,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 84cbd09d
[263944.530547] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[263944.538232] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[263944.548120] ath10k_pci 0000:00:00.0: firmware register dump:
[263944.553982] ath10k_pci 0000:00:00.0: [00]: 0x4100016C 0x000015B3 0x009964D9 0x00955B31
[263944.562120] ath10k_pci 0000:00:00.0: [04]: 0x009964D9 0x00060330 0x00000005 0x00000051
[263944.570246] ath10k_pci 0000:00:00.0: [08]: 0x00000000 0x009BDDFC 0x0042783C 0x00459108
[263944.578379] ath10k_pci 0000:00:00.0: [12]: 0x00000009 0x00000000 0x0095808C 0x009580A2
[263944.586573] ath10k_pci 0000:00:00.0: [16]: 0x00958080 0x0094085D 0x00000000 0x00000000
[263944.594733] ath10k_pci 0000:00:00.0: [20]: 0x409964D9 0x0040AC54 0x00000002 0x00000000
[263944.602864] ath10k_pci 0000:00:00.0: [24]: 0x80996CC4 0x0040ACB4 0x00000000 0xC09964D9
[263944.611066] ath10k_pci 0000:00:00.0: [28]: 0x80997988 0x0040AD24 0x0041F184 0x0042783C
[263944.619278] ath10k_pci 0000:00:00.0: [32]: 0x80999C9D 0x0040ADC4 0x0041F184 0x0042783C
[263944.627480] ath10k_pci 0000:00:00.0: [36]: 0x809B738D 0x0040AEE4 0x00426440 0x00000002
[263944.635644] ath10k_pci 0000:00:00.0: [40]: 0x80940F18 0x0040AF14 0x00000005 0x004039E4
[263944.643836] ath10k_pci 0000:00:00.0: [44]: 0x80940EEA 0x0040AF44 0x00400000 0x00000000
[263944.652007] ath10k_pci 0000:00:00.0: [48]: 0x80940F31 0x0040AF64 0x00401C10 0x00400600
[263944.660137] ath10k_pci 0000:00:00.0: [52]: 0x40940024 0x0040AF84 0x004068E8 0x004068E8
[263944.668283] ath10k_pci 0000:00:00.0: [56]: 0x00000000 0x0040AFA4 0x009BB001 0x00040020
[263944.676406] ath10k_pci 0000:00:00.0: Copy Engine register dump:
[263944.682526] ath10k_pci 0000:00:00.0: [00]: 0x00057400   1   1   3   3
[263944.689155] ath10k_pci 0000:00:00.0: [01]: 0x00057800   9   9 363 364
[263944.695805] ath10k_pci 0000:00:00.0: [02]: 0x00057c00   8   8  71  72
[263944.702447] ath10k_pci 0000:00:00.0: [03]: 0x00058000  14  14  16  14
[263944.709082] ath10k_pci 0000:00:00.0: [04]: 0x00058400  11  11  47   7
[263944.715717] ath10k_pci 0000:00:00.0: [05]: 0x00058800  24  24 503 504
[263944.722352] ath10k_pci 0000:00:00.0: [06]: 0x00058c00   9   9   9   9
[263944.728975] ath10k_pci 0000:00:00.0: [07]: 0x00059000   0   0   0   0
[263944.735965] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x412710  dropped: 0
[263944.743951] ath10k_pci 0000:00:00.0: [0] next: 0x4126f8 buf: 0x410b4c sz: 1500 len: 264 count: 10 free: 0
[263944.753928] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[263944.760306] ath10k: [0000]: CF321C00 025C0014 04024400 78BE9B00 50BC96C5 D9470000 01000000 CF321C00
[263944.769589] ath10k: [0008]: 574C0010 88991071 A88E4300 FCDD9B00 11000000 CF321C00 574C0014 A88E4300
[263944.778875] ath10k: [0016]: 20D54300 00000000 0C000000 8E1D0000 CF321C00 574C0014 A88E4300 44DE9B00
[263944.788192] ath10k: [0024]: 13000000 00000000 223D0000 CF321C00 574C0014 A88E4300 8CDE9B00 06000000
[263944.797534] ath10k: [0032]: 0C000000 0E3D0000 CF321C00 574C0010 88991071 A88E4300 D4DE9B00 10000000
[263944.806820] ath10k: [0040]: CF321C00 4B4C0010 20513A35 000014CC 3A350000 14CC2051 CF321C00 514CFC17
[263944.816184] ath10k: [0048]: 41841091 00000000 00000000 28004B00 08914500 CF321C00 514CFC13 42841091
[263944.825538] ath10k: [0056]: 00000000 00000000 02000000 CF321C00 0100FC17 B3150000 B3150000 44AB4000
[263944.834829] ath10k: [0064]: 6C010041 50D99B00
[263944.839338] ath10k_pci 0000:00:00.0: ATH10K_END
[263944.844110] ath10k_pci 0000:00:00.0: [1] next: 0x412710 buf: 0x41055c sz: 1500 len: 0 count: 0 free: 0
[263944.856669] ath10k_pci 0000:00:00.0: failed to set preamble for vdev 0: -143
[263944.863999] ath10k_pci 0000:00:00.0: failed to set mgmt tx rate -143
[263944.873571] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 2d96394f vdev: 0 addr: 14:cc:20:XX:XX:XX 
[263945.017751] ieee80211 phy0: Hardware restart was requested
[263946.044661] ath10k_pci 0000:00:00.0: unsupported HTC service id: 1536
[263946.051367] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[263946.068933] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[263946.075433] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[263946.083689] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 21000 iram: 25992'
[263946.152842] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[263946.176353] ath10k_pci 0000:00:00.0: device successfully recovered
```

```
[273354.842408] ath10k_pci 0000:00:00.0: firmware crashed! (guid n/a)
[273354.848721] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[273354.858279] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[... cut ...]
[273535.023954] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[273535.032206] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 21000 iram: 25992'
[273535.101785] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[273535.125842] ath10k_pci 0000:00:00.0: failed to transmit packet, dropping: -19
[273535.133224] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[273535.139400] ath10k_pci 0000:00:00.0: failed to transmit frame: -19
[273535.145896] ath10k_pci 0000:00:00.0: device successfully recovered
```

Noltari, 2020-05-09:
Still happening on latest firmware:
```
[252937.194987] ath10k_pci 0000:00:00.0: firmware crashed! (guid n/a)
[252937.201423] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[252937.210900] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[252937.223534] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fH-022-538f0906 api 2 features wmi-10.x,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 d870ee1d
[252937.253142] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[252937.260683] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[252937.270556] ath10k_pci 0000:00:00.0: firmware register dump:
[252937.276412] ath10k_pci 0000:00:00.0: [00]: 0x4100016C 0x000015B3 0x009964D9 0x00955B31
[252937.284638] ath10k_pci 0000:00:00.0: [04]: 0x009964D9 0x00060330 0x00000005 0x00000051
[252937.292802] ath10k_pci 0000:00:00.0: [08]: 0x00000000 0x009BFD0C 0x00427500 0x00410224
[252937.300941] ath10k_pci 0000:00:00.0: [12]: 0x00000009 0x00000000 0x0095808C 0x009580A2
[252937.309135] ath10k_pci 0000:00:00.0: [16]: 0x00958080 0x0094085D 0x00000000 0x00000000
[252937.317338] ath10k_pci 0000:00:00.0: [20]: 0x409964D9 0x0040AC54 0x00000002 0xDEADC0DE
[252937.325486] ath10k_pci 0000:00:00.0: [24]: 0x80996CC4 0x0040ACB4 0x00000000 0xC09964D9
[252937.333704] ath10k_pci 0000:00:00.0: [28]: 0x80997988 0x0040AD24 0x0041F184 0x00427500
[252937.341894] ath10k_pci 0000:00:00.0: [32]: 0x80999C9D 0x0040ADC4 0x0041F184 0x00427500
[252937.350101] ath10k_pci 0000:00:00.0: [36]: 0x809B73B9 0x0040AEE4 0x00426440 0x00000002
[252937.358306] ath10k_pci 0000:00:00.0: [40]: 0x80940F18 0x0040AF14 0x00000005 0x004039E4
[252937.366450] ath10k_pci 0000:00:00.0: [44]: 0x80940EEA 0x0040AF44 0x00400000 0x00000000
[252937.374651] ath10k_pci 0000:00:00.0: [48]: 0x80940F31 0x0040AF64 0x00401C10 0x00400600
[252937.382813] ath10k_pci 0000:00:00.0: [52]: 0x40940024 0x0040AF84 0x004068E8 0x004068E8
[252937.390946] ath10k_pci 0000:00:00.0: [56]: 0x00000000 0x0040AFA4 0x009BB001 0x00040020
[252937.399082] ath10k_pci 0000:00:00.0: Copy Engine register dump:
[252937.405200] ath10k_pci 0000:00:00.0: [00]: 0x00057400   0   0   3   3
[252937.411860] ath10k_pci 0000:00:00.0: [01]: 0x00057800  25  25 379 380
[252937.418504] ath10k_pci 0000:00:00.0: [02]: 0x00057c00  42  42 105 106
[252937.425151] ath10k_pci 0000:00:00.0: [03]: 0x00058000  27  27  29  27
[252937.431790] ath10k_pci 0000:00:00.0: [04]: 0x00058400 3899 3899 199 159
[252937.438601] ath10k_pci 0000:00:00.0: [05]: 0x00058800  27  27 154 155
[252937.445235] ath10k_pci 0000:00:00.0: [06]: 0x00058c00   8   8   8   8
[252937.451871] ath10k_pci 0000:00:00.0: [07]: 0x00059000   0   0   0   0
[252937.458803] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x4126f8  dropped: 0
[252937.466652] ath10k_pci 0000:00:00.0: [0] next: 0x412710 buf: 0x41055c sz: 1500 len: 264 count: 10 free: 0
[252937.476551] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[252937.482940] ath10k: [0000]: F5337007 025C0014 04024400 98BE9B00 50BC96C5 D9470000 01000000 F5337007
[252937.492271] ath10k: [0008]: 574C0010 88991071 889F4300 0CFD9B00 11000000 F5337007 574C0014 889F4300
[252937.501575] ath10k: [0016]: 90DD4300 00000000 0C000000 8E1D0000 F5337007 574C0014 889F4300 54FD9B00
[252937.511096] ath10k: [0024]: 13000000 00000000 223D0000 F5337007 574C0014 889F4300 9CFD9B00 06000000
[252937.520507] ath10k: [0032]: 0C000000 0E3D0000 F5337007 574C0010 88991071 889F4300 E4FD9B00 10000000
[252937.529814] ath10k: [0040]: F5337007 4B4C0010 20513A35 000014CC 3A350000 14CC2051 F5337007 514CFC17
[252937.539150] ath10k: [0048]: 41841091 DEC0ADDE DEC0ADDE 28004800 24024100 F5337007 514CFC13 42841091
[252937.548477] ath10k: [0056]: FEFE0000 FEFE0000 02000000 F5337007 0100FC17 B3150000 B3150000 44AB4000
[252937.557815] ath10k: [0064]: 6C010041 70D99B00
[252937.562414] ath10k_pci 0000:00:00.0: ATH10K_END
[252937.567200] ath10k_pci 0000:00:00.0: [1] next: 0x4126f8 buf: 0x410b4c sz: 1500 len: 0 count: 0 free: 0
[252937.580179] ath10k_pci 0000:00:00.0: failed to set preamble for vdev 0: -143
[252937.587622] ath10k_pci 0000:00:00.0: failed to set mgmt tx rate -143
[252937.598140] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 6660785a vdev: 0 addr: 14:cc:20:51:3a:35 
[252937.713382] ieee80211 phy0: Hardware restart was requested
[252938.707860] ath10k_pci 0000:00:00.0: unsupported HTC service id: 1536
[252938.714591] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[252938.732396] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[252938.738856] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[252938.747105] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 21000 iram: 25960'
[252938.815832] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[252938.823262] ath10k_pci 0000:00:00.0: set-coverage-class, phyclk: 88  value: 0
[252938.839802] ath10k_pci 0000:00:00.0: device successfully recovered
```

```
[501427.539192] ath10k_pci 0000:00:00.0: firmware crashed! (guid n/a)
[501427.545535] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[501427.555093] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[... cut ...]
[501429.056530] ath10k_pci 0000:00:00.0: unsupported HTC service id: 1536
[501429.063318] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[501429.081166] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[501429.087626] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[501429.095894] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 21000 iram: 25960'
[501429.164579] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[501429.187939] ath10k_pci 0000:00:00.0: device successfully recovered
```

greearb, 2020-07-02:
Here is a new image with more debugging in this area.
[firmware-2-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/4866250/firmware-2-full-htt-mgt-community.bin.gz)

Noltari, 2020-07-27:
> Here is a new image with more debugging in this area.
> [firmware-2-full-htt-mgt-community.bin.gz](https://github.com/greearb/ath10k-ct/files/4866250/firmware-2-full-htt-mgt-community.bin.gz)

Crash log with new firmware:
```
[119994.483046] ath10k_pci 0000:00:00.0: firmware crashed! (guid n/a)
[119994.489382] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[119994.499019] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[119994.511742] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fH-022-c443383b api 2 features wmi-10.x,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 a255444c
[119994.541337] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[119994.548998] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[119994.559004] ath10k_pci 0000:00:00.0: firmware register dump:
[119994.564948] ath10k_pci 0000:00:00.0: [00]: 0x4100016C 0x000015B3 0x0099652F 0x00955B31
[119994.573139] ath10k_pci 0000:00:00.0: [04]: 0x0099652F 0x00060330 0x00000005 0x00000051
[119994.581296] ath10k_pci 0000:00:00.0: [08]: 0x00000000 0x009BF384 0x00426E88 0xDEADC0DE
[119994.589433] ath10k_pci 0000:00:00.0: [12]: 0x00000009 0x00000000 0x00996491 0x0099649E
[119994.597710] ath10k_pci 0000:00:00.0: [16]: 0x00958080 0x0094085D 0x00000000 0x00000000
[119994.605974] ath10k_pci 0000:00:00.0: [20]: 0x4099652F 0x0040AC44 0x00000002 0x0041022C
[119994.614144] ath10k_pci 0000:00:00.0: [24]: 0x80996D08 0x0040ACA4 0x0040EF24 0xC099652F
[119994.622287] ath10k_pci 0000:00:00.0: [28]: 0x809979CC 0x0040AD24 0x0041F184 0x00426E88
[119994.630504] ath10k_pci 0000:00:00.0: [32]: 0x80999CE1 0x0040ADC4 0x0041F184 0x00426E88
[119994.638819] ath10k_pci 0000:00:00.0: [36]: 0x809B73FD 0x0040AEE4 0x00426440 0x00000002
[119994.647174] ath10k_pci 0000:00:00.0: [40]: 0x80940F18 0x0040AF14 0x00000005 0x004039E4
[119994.655329] ath10k_pci 0000:00:00.0: [44]: 0x80940EEA 0x0040AF44 0x00400000 0x00000000
[119994.663547] ath10k_pci 0000:00:00.0: [48]: 0x80940F31 0x0040AF64 0x00401C10 0x00400600
[119994.671733] ath10k_pci 0000:00:00.0: [52]: 0x40940024 0x0040AF84 0x004068E8 0x004068E8
[119994.680001] ath10k_pci 0000:00:00.0: [56]: 0x00000000 0x0040AFA4 0x009BB001 0x00040020
[119994.688340] ath10k_pci 0000:00:00.0: Copy Engine register dump:
[119994.694483] ath10k_pci 0000:00:00.0: [00]: 0x00057400   0   0   3   3
[119994.701135] ath10k_pci 0000:00:00.0: [01]: 0x00057800  20  20 470 471
[119994.707775] ath10k_pci 0000:00:00.0: [02]: 0x00057c00  44  44 107 108
[119994.714408] ath10k_pci 0000:00:00.0: [03]: 0x00058000  11  11  13  11
[119994.721043] ath10k_pci 0000:00:00.0: [04]: 0x00058400 5147 5147  55  15
[119994.727854] ath10k_pci 0000:00:00.0: [05]: 0x00058800   0   0 383 384
[119994.734489] ath10k_pci 0000:00:00.0: [06]: 0x00058c00  13  13  13  13
[119994.741124] ath10k_pci 0000:00:00.0: [07]: 0x00059000   0   0   0   0
[119994.748085] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x4126f8  dropped: 0
[119994.756067] ath10k_pci 0000:00:00.0: [0] next: 0x412710 buf: 0x41055c sz: 1500 len: 292 count: 11 free: 0
[119994.766191] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[119994.772708] ath10k: [0000]: 06FF5207 025C0014 04024400 1CBF9B00 4098AD17 97970000 01000000 07FF5207
[119994.782067] ath10k: [0008]: 574C0010 88991071 E8994300 84F39B00 11000000 07FF5207 574C0014 E8994300
[119994.792274] ath10k: [0016]: C0DA4300 00000000 0C000000 8E1D0000 07FF5207 574C0014 E8994300 CCF39B00
[119994.801683] ath10k: [0024]: 13000000 00000000 223D0000 07FF5207 574C0014 E8994300 14F49B00 06000000
[119994.811137] ath10k: [0032]: 0C000000 023D0000 07FF5207 574C0010 88991071 E8994300 5CF49B00 10000000
[119994.820459] ath10k: [0040]: 07FF5207 4B4C0010 20513A35 000014CC 3A350000 14CC2051 07FF5207 514CFC17
[119994.829745] ath10k: [0048]: 43841091 28004F00 64E24200 28004F07 00000000 07FF5207 514CFC17 41841091
[119994.839027] ath10k: [0056]: DEC0ADDE DEC0ADDE 28004F00 2C024100 07FF5207 514CFC13 42841091 FEFE0000
[119994.848373] ath10k: [0064]: FEFE0000 02000000 07FF5207 0100FC17 B3150000 B3150000 34AB4000 6C010041
[119994.857679] ath10k: [0072]: 57000000
[119994.861409] ath10k_pci 0000:00:00.0: ATH10K_END
[119994.866423] ath10k_pci 0000:00:00.0: [1] next: 0x4126f8 buf: 0x410b4c sz: 1500 len: 0 count: 0 free: 0
[119994.879261] ath10k_pci 0000:00:00.0: failed to set cts protection for vdev 0: -143
[119994.887243] ath10k_pci 0000:00:00.0: failed to set preamble for vdev 0: -143
[119994.894690] ath10k_pci 0000:00:00.0: failed to set mgmt tx rate -143
[119994.903637] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 0b7283f2 vdev: 0 addr: 14:cc:20:51:3a:35 
[119995.016101] ieee80211 phy0: Hardware restart was requested
[119996.012444] ath10k_pci 0000:00:00.0: unsupported HTC service id: 1536
[119996.019189] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[119996.036911] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[119996.043472] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[119996.051750] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 21000 iram: 25880'
[119996.120420] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[119996.128139] ath10k_pci 0000:00:00.0: set-coverage-class, phyclk: 88  value: 0
[119997.138797] ath10k_pci 0000:00:00.0: device successfully recovered
```

Noltari, 2020-07-28:
@greearb and another crash with the firmware you provided:
```
[280859.997665] ath10k_pci 0000:00:00.0: firmware crashed! (guid n/a)
[280860.004006] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[280860.013584] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[280860.026164] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fH-022-c443383b api 2 features wmi-10.x,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 a255444c
[280860.055699] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[280860.063244] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[280860.073151] ath10k_pci 0000:00:00.0: firmware register dump:
[280860.079016] ath10k_pci 0000:00:00.0: [00]: 0x4100016C 0x000015B3 0x0099652F 0x00955B31
[280860.087209] ath10k_pci 0000:00:00.0: [04]: 0x0099652F 0x00060330 0x00000005 0x00000051
[280860.095351] ath10k_pci 0000:00:00.0: [08]: 0x00000000 0x009BEDE4 0x00427614 0xDEADC0DE
[280860.103500] ath10k_pci 0000:00:00.0: [12]: 0x00000009 0x00000000 0x00996491 0x0099649E
[280860.111644] ath10k_pci 0000:00:00.0: [16]: 0x00958080 0x0094085D 0x00000000 0x00000000
[280860.119834] ath10k_pci 0000:00:00.0: [20]: 0x4099652F 0x0040AC44 0x00000002 0x0041022C
[280860.128045] ath10k_pci 0000:00:00.0: [24]: 0x80996D08 0x0040ACA4 0xFFFFFFFF 0xC099652F
[280860.136191] ath10k_pci 0000:00:00.0: [28]: 0x809979CC 0x0040AD24 0x0041F184 0x00427614
[280860.144369] ath10k_pci 0000:00:00.0: [32]: 0x80999CE1 0x0040ADC4 0x0041F184 0x00427614
[280860.152548] ath10k_pci 0000:00:00.0: [36]: 0x809B73FD 0x0040AEE4 0x00426440 0x00000002
[280860.160701] ath10k_pci 0000:00:00.0: [40]: 0x80940F18 0x0040AF14 0x00000005 0x004039E4
[280860.168881] ath10k_pci 0000:00:00.0: [44]: 0x80940EEA 0x0040AF44 0x00400000 0x00000000
[280860.177033] ath10k_pci 0000:00:00.0: [48]: 0x80940F31 0x0040AF64 0x00401C10 0x00400600
[280860.185159] ath10k_pci 0000:00:00.0: [52]: 0x40940024 0x0040AF84 0x004068E8 0x004068E8
[280860.193297] ath10k_pci 0000:00:00.0: [56]: 0x00000000 0x0040AFA4 0x009BB001 0x00040020
[280860.201434] ath10k_pci 0000:00:00.0: Copy Engine register dump:
[280860.207544] ath10k_pci 0000:00:00.0: [00]: 0x00057400   0   0   3   3
[280860.214184] ath10k_pci 0000:00:00.0: [01]: 0x00057800   4   4 134 135
[280860.220821] ath10k_pci 0000:00:00.0: [02]: 0x00057c00  35  35  34  35
[280860.227454] ath10k_pci 0000:00:00.0: [03]: 0x00058000   7   7   9   7
[280860.234090] ath10k_pci 0000:00:00.0: [04]: 0x00058400 2167 2167 101  61
[280860.240901] ath10k_pci 0000:00:00.0: [05]: 0x00058800  25  25 504 505
[280860.247536] ath10k_pci 0000:00:00.0: [06]: 0x00058c00   9   9   9   9
[280860.254175] ath10k_pci 0000:00:00.0: [07]: 0x00059000   0   0   0   0
[280860.261254] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x4126f8  dropped: 0
[280860.269150] ath10k_pci 0000:00:00.0: [0] next: 0x412710 buf: 0x41055c sz: 1500 len: 292 count: 11 free: 0
[280860.279127] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[280860.285516] ath10k: [0000]: 5E7E2401 025C0014 04024400 E8BE9B00 50BC96C5 D9470000 01000000 5E7E2401
[280860.294876] ath10k: [0008]: 574C0010 88991071 A8974300 E4ED9B00 11000000 5E7E2401 574C0014 A8974300
[280860.304221] ath10k: [0016]: A0D94300 00000000 0C000000 8E1D0000 5E7E2401 574C0014 A8974300 2CEE9B00
[280860.313513] ath10k: [0024]: 13000000 00000000 223D0000 5E7E2401 574C0014 A8974300 74EE9B00 06000000
[280860.322834] ath10k: [0032]: 0C000000 0E3D0000 5E7E2401 574C0010 88991071 A8974300 BCEE9B00 10000000
[280860.332261] ath10k: [0040]: 5E7E2401 4B4C0010 20513A35 000014CC 3A350000 14CC2051 5E7E2401 514CFC17
[280860.341714] ath10k: [0048]: 43841091 00004F00 D4E44200 00004F07 00000000 5E7E2401 514CFC17 41841091
[280860.351013] ath10k: [0056]: DEC0ADDE DEC0ADDE 00004F00 2C024100 5E7E2401 514CFC13 42841091 FEFE0000
[280860.360463] ath10k: [0064]: FEFE0000 02000000 5E7E2401 0100FC17 B3150000 B3150000 34AB4000 6C010041
[280860.369927] ath10k: [0072]: 57000000
[280860.373684] ath10k_pci 0000:00:00.0: ATH10K_END
[280860.378533] ath10k_pci 0000:00:00.0: [1] next: 0x4126f8 buf: 0x410b4c sz: 1500 len: 0 count: 0 free: 0
[280860.391490] ath10k_pci 0000:00:00.0: failed to set preamble for vdev 0: -143
[280860.398806] ath10k_pci 0000:00:00.0: failed to set mgmt tx rate -143
[280860.408459] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 986698ab vdev: 0 addr: 14:cc:20:51:3a:35 
[280860.523961] ieee80211 phy0: Hardware restart was requested
[280861.519398] ath10k_pci 0000:00:00.0: unsupported HTC service id: 1536
[280861.526106] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[280861.544042] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[280861.550482] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[280861.558722] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 21000 iram: 25880'
[280861.627694] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[280862.644634] ath10k_pci 0000:00:00.0: device successfully recovered
```

svanheule, 2020-09-15:
For now I've just noticed these crashes in the logs. Last week (same FW) I had some issues connecting with some devices, while others were still able to connect. I rebooted before I downloaded the logs however.

```
[   17.531359] ath10k 5.7 driver, optimized for CT firmware, probing pci device: 0x3c.
[   17.553278] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   17.559954] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   19.106985] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[   19.116574] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   19.129036] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fH-022-538f0906 api 2 features wmi-10.x,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,txrate-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 d870ee1d
[   19.461536] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[   20.443491] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[   20.460428] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[   20.466883] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[   20.475415] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 21000 iram: 25960'
[   20.541216] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[   20.551376] ath10k_pci 0000:00:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
```

```
[353530.494765] ath10k_pci 0000:00:00.0: firmware crashed! (guid 529d8720-e66a-4295-8e80-ae3a2bc42873)
[353530.504169] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[353530.513803] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[353530.526356] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fH-022-538f0906 api 2 features wmi-10.x,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 d870ee1d
[353530.556398] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[353530.564051] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[353530.574076] ath10k_pci 0000:00:00.0: firmware register dump:
[353530.580029] ath10k_pci 0000:00:00.0: [00]: 0x4100016C 0x00000000 0x00996468 0x2A150855
[353530.588317] ath10k_pci 0000:00:00.0: [04]: 0x00996468 0x00060130 0x00000000 0x00000006
[353530.596604] ath10k_pci 0000:00:00.0: [08]: 0x00000000 0x009BEE24 0x00426924 0x0042E5F4
[353530.604882] ath10k_pci 0000:00:00.0: [12]: 0x00000009 0x00000000 0x0095808C 0x009580A2
[353530.613160] ath10k_pci 0000:00:00.0: [16]: 0x00958080 0x0094085D 0x00000000 0x00000000
[353530.621441] ath10k_pci 0000:00:00.0: [20]: 0x40996468 0x0040AC54 0x00000000 0x2A150845
[353530.629712] ath10k_pci 0000:00:00.0: [24]: 0x80996D71 0x0040ACB4 0x009BEE24 0xC0996468
[353530.638029] ath10k_pci 0000:00:00.0: [28]: 0x80997988 0x0040AD24 0x0041F184 0x00426924
[353530.646440] ath10k_pci 0000:00:00.0: [32]: 0x80999C9D 0x0040ADC4 0x0041F184 0x00426924
[353530.654742] ath10k_pci 0000:00:00.0: [36]: 0x809B73B9 0x0040AEE4 0x00426440 0x00000002
[353530.663114] ath10k_pci 0000:00:00.0: [40]: 0x80940F18 0x0040AF14 0x00000004 0x004039D0
[353530.671426] ath10k_pci 0000:00:00.0: [44]: 0x80940EEA 0x0040AF44 0x00400000 0x00000000
[353530.679700] ath10k_pci 0000:00:00.0: [48]: 0x80940F31 0x0040AF64 0x00401C10 0x00400600
[353530.687994] ath10k_pci 0000:00:00.0: [52]: 0x40940024 0x0040AF84 0x004068E8 0x004068E8
[353530.696376] ath10k_pci 0000:00:00.0: [56]: 0x00000000 0x0040AFA4 0x009BB001 0x00040020
[353530.704672] ath10k_pci 0000:00:00.0: Copy Engine register dump:
[353530.710937] ath10k_pci 0000:00:00.0: [00]: 0x00057400   0   0   3   3
[353530.717699] ath10k_pci 0000:00:00.0: [01]: 0x00057800   3   3 453 454
[353530.724465] ath10k_pci 0000:00:00.0: [02]: 0x00057c00  58  58  57  58
[353530.731222] ath10k_pci 0000:00:00.0: [03]: 0x00058000  21  21  25  21
[353530.737977] ath10k_pci 0000:00:00.0: [04]: 0x00058400 7317 7317 116  76
[353530.744913] ath10k_pci 0000:00:00.0: [05]: 0x00058800  21  21 500 501
[353530.751673] ath10k_pci 0000:00:00.0: [06]: 0x00058c00   9   9   9   9
[353530.758426] ath10k_pci 0000:00:00.0: [07]: 0x00059000   0   0   0   0
[353530.765450] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x412710  dropped: 0
[353530.773387] ath10k_pci 0000:00:00.0: [0] next: 0x4126f8 buf: 0x410b4c sz: 1500 len: 28 count: 1 free: 0
[353530.783258] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[353530.789761] ath10k: [0000]: EAEA9305 0100FC17 00000000 85100000 44AB4000 6C010041 14AC4000
[353530.798410] ath10k_pci 0000:00:00.0: ATH10K_END
[353530.803267] ath10k_pci 0000:00:00.0: [1] next: 0x412710 buf: 0x41055c sz: 1500 len: 0 count: 0 free: 0
[353530.818492] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 960d4af5 vdev: 0 addr: 4c:57:ca:b5:22:01 
[353530.829959] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 1cdb7468 vdev: 0 addr: 0c:2f:b0:c8:d6:c0 
[353530.841395] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 6779fc4d vdev: 1 addr: 52:c7:bf:4f:e0:a7 
[... cut ...]
[541283.771314] ath10k_pci 0000:00:00.0: failed to transmit packet, dropping: -19
[541283.778796] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[541283.785105] ath10k_pci 0000:00:00.0: failed to transmit frame: -19
[541283.791629] ath10k_pci 0000:00:00.0: failed to transmit packet, dropping: -19
[541283.799124] ath10k_pci 0000:00:00.0: failed to submit frame: -19
[541283.805425] ath10k_pci 0000:00:00.0: failed to push frame: -19
[541283.811610] ath10k_pci 0000:00:00.0: device successfully recovered
```

Noltari, 2020-12-04:
More crashes on latest firmware (.022):
```
[202554.276993] ath10k_pci 0000:00:00.0: firmware crashed! (guid 71668940-6fa7-4302-a9c4-368590387095)
[202554.286262] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[202554.295804] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[202554.308196] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fH-022-ecad3248 api 2 features wmi-10.x,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 1b2a161c
[202554.337708] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[202554.345271] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[202554.355212] ath10k_pci 0000:00:00.0: firmware register dump:
[202554.361093] ath10k_pci 0000:00:00.0: [00]: 0x4100016C 0x000015B3 0x009948FA 0x00955B31
[202554.369303] ath10k_pci 0000:00:00.0: [04]: 0x009948FA 0x00060330 0x00000005 0x00000051
[202554.377516] ath10k_pci 0000:00:00.0: [08]: 0x00000000 0x009BB33C 0x00426138 0x00000000
[202554.385679] ath10k_pci 0000:00:00.0: [12]: 0x00000009 0x00000000 0x0099485C 0x00994869
[202554.393876] ath10k_pci 0000:00:00.0: [16]: 0x00958080 0x0094085D 0x00000000 0x00000000
[202554.402031] ath10k_pci 0000:00:00.0: [20]: 0x409948FA 0x0040AC34 0x00000000 0x00430020
[202554.410224] ath10k_pci 0000:00:00.0: [24]: 0x809950AF 0x0040AC94 0x00000000 0xC09948FA
[202554.418368] ath10k_pci 0000:00:00.0: [28]: 0x809960A0 0x0040AD14 0x0041E524 0x00426138
[202554.426509] ath10k_pci 0000:00:00.0: [32]: 0x80997EC9 0x0040ADB4 0x0041E524 0x00426138
[202554.434650] ath10k_pci 0000:00:00.0: [36]: 0x809AF249 0x0040AEE4 0x00425918 0x00000002
[202554.442778] ath10k_pci 0000:00:00.0: [40]: 0x80940F18 0x0040AF14 0x00000004 0x004039D0
[202554.450942] ath10k_pci 0000:00:00.0: [44]: 0x80940EEA 0x0040AF44 0x00400000 0x00000000
[202554.459162] ath10k_pci 0000:00:00.0: [48]: 0x80940F31 0x0040AF64 0x00401C10 0x00400600
[202554.467333] ath10k_pci 0000:00:00.0: [52]: 0x40940024 0x0040AF84 0x004068E8 0x004068E8
[202554.475482] ath10k_pci 0000:00:00.0: [56]: 0x00000000 0x0040AFA4 0x009BB001 0x00040020
[202554.483663] ath10k_pci 0000:00:00.0: Copy Engine register dump:
[202554.489866] ath10k_pci 0000:00:00.0: [00]: 0x00057400   7   7   3   3
[202554.496527] ath10k_pci 0000:00:00.0: [01]: 0x00057800   8   8 106 107
[202554.503168] ath10k_pci 0000:00:00.0: [02]: 0x00057c00  17  17  16  17
[202554.509800] ath10k_pci 0000:00:00.0: [03]: 0x00058000   4   4   8   4
[202554.516436] ath10k_pci 0000:00:00.0: [04]: 0x00058400 7387 7387 151 111
[202554.523246] ath10k_pci 0000:00:00.0: [05]: 0x00058800   1   1  64  65
[202554.529881] ath10k_pci 0000:00:00.0: [06]: 0x00058c00  13  13  13  13
[202554.536517] ath10k_pci 0000:00:00.0: [07]: 0x00059000   0   0   0   0
[202554.543337] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x411a98  dropped: 0
[202554.551157] ath10k_pci 0000:00:00.0: [0] next: 0x411ab0 buf: 0x40f8fc sz: 1500 len: 108 count: 4 free: 0
[202554.561142] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[202554.567702] ath10k: [0000]: 9CFC5C04 514CFC17 43841091 00004900 4CA04200 00006907 00000000 9CFC5C04
[202554.577008] ath10k: [0008]: 514CFC17 41841091 00000000 00000000 00004900 20004300 9CFC5C04 514CFC13
[202554.586350] ath10k: [0016]: 42841091 00000000 00000000 00000000 9CFC5C04 0100FC17 B3150000 B3150000
[202554.595705] ath10k: [0024]: 24AB4000 6C010041 CB000000
[202554.601089] ath10k_pci 0000:00:00.0: ATH10K_END
[202554.605874] ath10k_pci 0000:00:00.0: [1] next: 0x411a98 buf: 0x40feec sz: 1500 len: 0 count: 0 free: 0
[202554.620987] ath10k_pci 0000:00:00.0: failed to recalculate rts/cts prot for vdev 0: -143
[202554.629400] ath10k_pci 0000:00:00.0: failed to set cts protection for vdev 0: -143
[202554.637201] ath10k_pci 0000:00:00.0: failed to set preamble for vdev 0: -143
[202554.644477] ath10k_pci 0000:00:00.0: failed to set mgmt tx rate -143
[202554.651480] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer ae82a751 vdev: 0 addr: e2:49:48:9a:ac:17 
[202554.662747] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer f8b12b68 vdev: 0 addr: 14:cc:20:51:3a:35 
[202554.777639] ieee80211 phy0: Hardware restart was requested
[202555.717444] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[202555.734871] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[202555.741318] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[202555.749568] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 25000 iram: 38944'
[202555.818139] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[202555.825623] ath10k_pci 0000:00:00.0: set-coverage-class, phyclk: 88  value: 0
[202556.836550] ath10k_pci 0000:00:00.0: Invalid state: 3 in ath10k_htt_tx_32, warning will not be repeated.
[202556.846290] ------------[ cut here ]------------
[202556.851156] WARNING: CPU: 0 PID: 3928 at target-mips_74kc+dsp2_musl/linux-ath79_generic/ath10k-ct-regular/ath10k-ct-2020-10-08-1d28d176/ath10k-5.8/htt_tx.c:1490 ieee80211_is_robust_mgmt_frame_tx+0xb74/0xbec [ath10k_core]
[202556.871078] Modules linked in: ath9k ath9k_common pppoe ppp_async iptable_nat ath9k_hw ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD pppox ppp_generic nf_n [... cut ...]
[283266.981976] ath10k_pci 0000:00:00.0: failed to set dtim period for vdev 0: -143
[283267.925867] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[283267.943300] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
[283267.949742] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
[283267.957993] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 25000 iram: 38944'
[283268.026758] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
[283269.043504] ath10k_pci 0000:00:00.0: device successfully recovered
```

nariox, 2020-12-16:
Hi @greearb, at this point, would more people posting their kernel logs help at all? I'm also seeing the same crashes using the ct-htt firmware.

greearb, 2020-12-16:
sure, please attach some.  I'll dig through these sometime soon.

nariox, 2020-12-16:
Should I use the firmware you posted here? I was using the one that came with openwrt 19.07.5 (2020-07-02-1)

greearb, 2020-12-16:
Either is fine with me, does one firmware work obviously different than the other for you?

nariox, 2020-12-16:
I haven't tested the one you posted here. But comparing the openwrt
supplied ones, the ct-htt firmware gives me these errors, whereas the
regular ct one does not (or at least I haven't encountered it). I don't
think it matters, but I have 802.11w set to optional and am running a WPA3
and a WPA2 multi-SSID network on the 5GHz network.