greearb/ath10k-ct #124: Poor link speed and signal strength on Qualcomm QCA 9377 on linux

abcool, 2020-04-20:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
Getting 1 mbps link speed and average signal strength even on standing next to router on every linux distro whereas windows 10 gives 150 mbps link speeds and excellent signal strength even standing quite far off from router. Tested on Fedora 30/31 and ubuntu 18.04. 
Turning off wireless power management : sudo iwconfig wlan0 power off doesn't make a difference.
Turning off all power saving options using TLP also doesn't make a difference. 
Software (OS, Firmware version, kernel, driver, etc)
Tested on Fedora 30/31 and ubuntu 18.04 LTS
Wifi card: Qualcomm QCA 9377
Device: Dell Vostro 3468 (shipped with ubuntu 16.04)

Hardware (NIC chipset, platform, etc)
Qualcomm QCA 9377

Logs (dmesg, maybe supplicant and/or hostap)

greearb, 2020-04-20:
ath10k-ct firmware does not support the 9377 chipset, so this problem should be reported to the upstream ath10k mailing list and/or linux-wireless mailing list.

closed 2020-04-20 by greearb

abcool, 2020-04-24:
As per your instruction, I tried sending mail but it fails. Please see the attached images.
[image]
[image]

greearb, 2020-04-24:
Please learn to send email in non-html format and try again, nothing I can do to help with the 9377 chipset.