greearb/ath10k-ct #126: Stack trace on latest firmware (QCA9984)

huaracheguarache, 2020-04-24:
Description of the problem (how to configure, how to reproduce, how often it happens).

I got a stack trace with the latest firmware on my R7800. There have been no noticeable issues other than the stack trace.

Hardware: R7800 (QCA9984)
OS: `OpenWrt SNAPSHOT r13046-df27e949fb`
Kernel: `5.4.34`
Firmware: `firmware ver 10.4b-ct-9984-fH-013-b63cea875`

```
[25546.083234] ------------[ cut here ]------------
[25546.083326] WARNING: CPU: 0 PID: 0 at target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ath10k-ct-regular/ath10k-ct-2020-03-25-3d173a47/ath10k-5.4/txrx.c:134 ath10k_txrx_tx_unref+0x574/0x738 [ath10k_core]
[25546.086933] Invalid VHT rate, nss: 3  hw_rate: 15 ratecode: 255
[25546.106172] Modules linked in: xt_connlimit pppoe ppp_async nf_conncount iptable_nat ath10k_pci ath10k_core ath xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_recent xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY wireguard slhc sch_cake nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_raw iptable_mangle iptable_filter ipt_ECN ip_tables crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred ledtrig_usbport ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet
[25546.106268]  ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 ifb ip6_udp_tunnel udp_tunnel sit tunnel4 ip_tunnel leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom ohci_platform ohci_hcd phy_qcom_dwc3 ahci fsl_mph_dr_of ehci_platform ehci_fsl sd_mod ahci_platform libahci_platform libahci libata scsi_mod ehci_hcd gpio_button_hotplug
[25546.224762] CPU: 0 PID: 0 Comm: swapper/0 Not tainted 5.4.34 #0
[25546.246978] Hardware name: Generic DT based system
[25546.252893] [<c030f954>] (unwind_backtrace) from [<c030b96c>] (show_stack+0x14/0x20)
[25546.257756] [<c030b96c>] (show_stack) from [<c08a82a0>] (dump_stack+0x94/0xa8)
[25546.265655] [<c08a82a0>] (dump_stack) from [<c031e59c>] (__warn+0xb4/0xd0)
[25546.272680] [<c031e59c>] (__warn) from [<c031e638>] (warn_slowpath_fmt+0x80/0x90)
[25546.279560] [<c031e638>] (warn_slowpath_fmt) from [<bf563718>] (ath10k_txrx_tx_unref+0x574/0x738 [ath10k_core])
[25546.287173] [<bf563718>] (ath10k_txrx_tx_unref [ath10k_core]) from [<bf55d7d4>] (ath10k_htt_t2h_msg_handler+0xe08/0x11dc [ath10k_core])
[25546.297018] [<bf55d7d4>] (ath10k_htt_t2h_msg_handler [ath10k_core]) from [<bf5b5768>] (ath10k_pci_htt_rx_cb+0x178/0x230 [ath10k_pci])
[25546.309180] [<bf5b5768>] (ath10k_pci_htt_rx_cb [ath10k_pci]) from [<bf57d164>] (ath10k_ce_per_engine_service+0x9c/0x10c [ath10k_core])
[25546.321337] [<bf57d164>] (ath10k_ce_per_engine_service [ath10k_core]) from [<bf57d254>] (ath10k_ce_per_engine_service_any+0x80/0xd8 [ath10k_core])
[25546.333299] [<bf57d254>] (ath10k_ce_per_engine_service_any [ath10k_core]) from [<bf5b7104>] (ath10k_pci_napi_poll+0x54/0x15c [ath10k_pci])
[25546.346389] [<bf5b7104>] (ath10k_pci_napi_poll [ath10k_pci]) from [<c0761ad4>] (net_rx_action+0x118/0x374)
[25546.358797] [<c0761ad4>] (net_rx_action) from [<c0302298>] (__do_softirq+0x130/0x2d4)
[25546.368427] [<c0302298>] (__do_softirq) from [<c0322980>] (irq_exit+0xbc/0xe0)
[25546.376327] [<c0322980>] (irq_exit) from [<c036cc80>] (__handle_domain_irq+0x6c/0xd0)
[25546.383449] [<c036cc80>] (__handle_domain_irq) from [<c05b502c>] (gic_handle_irq+0x5c/0xb8)
[25546.391345] [<c05b502c>] (gic_handle_irq) from [<c0301a8c>] (__irq_svc+0x6c/0x90)
[25546.399499] Exception stack(0xc0c01ee0 to 0xc0c01f28)
[25546.407145] 1ee0: 00000000 0000173b 1ce53000 dd991a00 dcc29400 00000000 dd990df0 0000173b
[25546.412184] 1f00: 0000173b 00000000 e9370460 e935b600 00000015 c0c01f30 c0704d20 c0704d24
[25546.420335] 1f20: a0000013 ffffffff
[25546.428491] [<c0301a8c>] (__irq_svc) from [<c0704d24>] (cpuidle_enter_state+0x94/0x498)
[25546.431795] [<c0704d24>] (cpuidle_enter_state) from [<c070516c>] (cpuidle_enter+0x30/0x4c)
[25546.439782] [<c070516c>] (cpuidle_enter) from [<c034a3ac>] (do_idle+0x1d8/0x240)
[25546.448114] [<c034a3ac>] (do_idle) from [<c034a6bc>] (cpu_startup_entry+0x1c/0x20)
[25546.455672] [<c034a6bc>] (cpu_startup_entry) from [<c0b00e5c>] (start_kernel+0x4dc/0x4e8)
[25546.463105] ---[ end trace 5a28d411d60ef855 ]---
```

ynezz, 2020-04-28:
Possibly duplicate of #117

greearb, 2020-04-30:
The fix is pushed to openwrt now.

closed 2020-04-30 by greearb