greearb/ath10k-ct #131: fixing invalid VHT TX rate code 0xff

ynezz, 2020-04-30:
So far got following fixes for `firmware ver 10.4b-ct-4019-fW-013-d81f62d97` and 5.4.36 with latest ath10k-ct driver:

```
[43567.677687] ath10k_ahb a000000.wifi: wmi: fixing invalid VHT TX rate code 0xff
[50946.792820] ath10k_ahb a000000.wifi: htt tx: fixing invalid VHT TX rate code 0xff
```

timkgh, 2020-05-02:
What do these mean?

notmyrealhandle, 2020-05-03:
After ~ 4 hrs up on the firmware below I'm unable to (re) connect via wireless.  Wired connections work and there is one wifi client that has stayed connected and is functional.  There is a kernel warning (see dmesg) generated about an hour after the ath10k mesgs but as far as I can tell it's not generated by the ath10k system.

r7500v2, ath10k-ct-htt, firmware 10.4b-ct-9980-tH-013-d81f62d97, kernel 5.4.36, run as AP only.

dmesg [here](https://gist.github.com/notmyrealhandle/afb216cda4831e4e6f4cf5df992ebaac).

I'm still looking through the dmesg myself.  At this time, this does not "feel" like only an ath10k issue as other non wifi related functions are misbehaving.

I'm going to reset, the router and see if this happens again.

HTH

notmyrealhandle, 2020-05-04:
Quoting @greearb from the dmesg gist [here](https://gist.github.com/notmyrealhandle/afb216cda4831e4e6f4cf5df992ebaac#gistcomment-3286699):

> I'm pretty sure this is not ath10k related. Seems some mem corruption or something like that?

Thank you for taking a look.

Given how the unit behaved (random things not working), that seems possible.  So far the symptoms have not reoccurred.

@ynezz is pushing 5.4 kernel updates for openwrt of late; however for ipq806x, @ansuel and others have been keeping it going.  If/when this happens again, I'll post it to openwrt.

slicingmelon, 2020-05-11:
Hi guys,
I want to confirm that I have the exact problem on kernel 5.4.39
`htt tx: fixing invalid VHT TX rate code 0xff`

I am looking forward to the next fix, even though the AP seem to behave normally, wlan timeouts randomly for 3-4 minutes and basically there is no internet. Not sure if this particular error correlates with root cause of the actual problem.

shelterx, 2020-09-15:
This is still an issue, however I don't notice any real connection issues.
I always get them once after boot,

Using:
```
Netgear R7800
OpenWrt SNAPSHOT, r14470-51ec51871f
Linux version 5.4.65

firmware ver 10.4b-ct-9984-tH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7d16c2a5

board_file api 2 bmi_id 0:2 crc32 854987
```

shelterx, 2020-09-15:
I'm not sure you can compare firmwares like this but I don't seeem to be getting these messages when using the official qca firmware, firmware-5.bin_10.4-3.9.0.2-00124