greearb/ath10k-ct #138: Low download throughput on Netgear r7800 (qca9984) with ath10k-ct firmware

rickkdotnet, 2020-06-09:
## Description of the problem (how to configure, how to reproduce, how often it happens).
Opening a separate issue for https://github.com/greearb/ath10k-ct/issues/136#issuecomment-641095620:

A Netgear R7800 running OpenWRT with -ct firmware consistently consistenly gives significantly lower download throughput on 5Ghz VHT80 than the same setup with mainline firmware. 

An iperf *single-stream* TCP download from the AP or a host connected to ethernet show an unstable 150-300 mb/s while mainline firmware hovers around 400mb/s. Upload speeds on both firmwares are comparable and well over 300 mb/s in the same circumstances. 

*Multiple streams* such as the Flent RRUL tests show more dramatic differences of over 5x lower throughput, these are typical: 
[image]
[image]

In real world use the connection feels "laggy" at times but usable - no disconnects.
Bitrate as reported by the OpenWRT GUI is constantly changing. 

## Software (OS, Firmware version, kernel, driver, etc):
I've tried at least the following combinations: 

### OpenWRT 19.07.03 (Hnyman build from his dropbox) 
* Kernel: 4.19.x
* kmod: kmod-ath10k-ct - 4.14.180+2019-09-09-5e8cd86f-1
* Firmware: ath10k-firmware-qca9984-ct - 2019-10-03-d622d160-1

### OpenWRT r13399-559b338466 (Hnyman build from his dropbox) 
* Kernel: 5.4.43 
* kmod: kmod-ath10k-ct - 5.4.43+2020-04-29-3637be6f-1
* Firmware: firmware ver 10.4b-ct-9984-fW-013-d81f62d97 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 46b728ef

### OpenWRT r13517-6fca1646dd (own build based on Hnyman scripts) 
* Kernel:  5.4.43
* kmod kmod-ath10k - 5.4.43+5.7-rc3-1-2 (mainline kmod) 
* Firmware: firmware ver 10.4b-ct-9984-fW-013-4ab470999 api 5 features mfp,peer-flow-ctrl crc32 4485a840 

I've also tried both the full and the full-hgt-mgt variants, and a version 012 of the firmware but my logs are incomplete. 

All tests were done in ETSI NL 5Ghz against a MacBook Pro (15-inch, 2018) with an  
AirPort Extreme  (0x14E4, 0x7BF) , Feb 28 2020 15:24:56 version 9.30.357.35.32.5.47 FWID 01-9ce4adf3, 802.11 a/b/g/n/ac card. I'll try a different client soon. 

Channels, regulatory domain and distance to the AP  do not seem to be factors. 

## Hardware (NIC chipset, platform, etc)
Recently bought *Netgear r7800* with *qca9984*.  

## Logs (dmesg, maybe supplicant and/or hostap)
I've attached a dmesg of  OpenWRT r13399-559b338466 (Hnyman build from his dropbox).
[dmesg.boot-r13399-559b338466-10.4b-ct-9984-fW-013-d81f62d97.txt](https://github.com/greearb/ath10k-ct/files/4752890/dmesg.boot-r13399-559b338466-10.4b-ct-9984-fW-013-d81f62d97.txt)

In use I occasionally see ath10k_pci 0000:01:00.0: Invalid VHT mcs 15 peer stats.

As a user I'm currently happy with the mainline firmware, but seeing all the work that went/goes into ath10k-ct it would be nice if this can be fixed.

greearb, 2020-07-16:
Hello,  are you willing to test a series of builds to bisect this problem?  I can build you the image from my FW that is very close to an upstream QCA FW, and if that does not exhibit the problem, then you can bisect to find where the problem was introduced....

Fail-Safe, 2020-09-13:
@rickkdotnet Curious how this is going for you now. I have similar experiences with the -ct firmware and would like to know if you and @greearb made headway on it.

xupefei, 2020-11-18:
Any updates yo this issue? @greearb I can try to bisect firmware if @rickkdotnet cannot make it.

I'm using a Macbook Pro 15', with Wifi upload speed 700mbps and download 150-300mbps.

gsustek, 2020-11-19:
Please try with this firmware:
https://www.candelatech.com/downloads/ath10k-9984-10-4/firmware-5-ct-non-commercial-full-htt-mgt-11.bin

xupefei, 2020-11-26:
@gsustek Thank you! I tried this firmware and see a small improvement over the default OpenWrt firmware. The speed is quite unstable but the peak can be 500mbps and average ~350mbps.

greearb, 2020-12-07:
I uploaded a new set of binaries that you can bisect with.

http://www.candelatech.com/downloads/ath10k-9984-10-4b/bisect/all_builds-9984-H-dec-7-2020.tar.gz

xupefei, 2020-12-19:
Hi @greearb,
I tried a few firmwares in your package but unable to pin down the first bad version due to unstable TX throughput. Here is my record: https://www.icloud.com/numbers/0Y5PrlO8nDED_1kZ7ulZf4aTw#atk10k_bisect

My setup:
 - Hnyman's `R7800-master-r15237-fca0eb2d92-20201218-1651-sysupgrade.bin`
 - Reboot the router after applying a new firmware.
 - Iperf3 server on NAS (wired), client is my MacBook (wireless).
 - WiFi: [image]

My observations:

- One reason of the slow download is due to IRQ32 handling by CPU0. Move it to CPU1 will improve the throughput.
- All candidate firmwares are equally unstable in terms of TX. The speed range from 200mbps to 500mbps, causing overall slowness.

For comparison, I flashed Hnyman's non-CT image `R7800-master-r15207-c29f6121fd-20201213-2136-sysupgrade.bin`, move IRQ32 to CPU1, and can see s stable >500mbps TX speed.

Any suggestion for me to move forward?

ahart241, 2020-12-29:
Hello @xupefei @greearb @gsustek @rickkdotnet

I've been following this thread as I too was experiencing similar issues (high latency, low transmit throughput 150~300Mbps) on a EA8500 running OpenWRT 19.07.03.

I am happy to say that I am seeing much improved throughput and lower latency after updating hostapd 2019-08-08-ca8c2bd2-3 » 2019-08-08-ca8c2bd2-4.  Based on this and what @greearb mentioned about the ath10k-ct firmware versions being the same between 19.07.01->19.07.03, I am inclined to think that this may not be a firmware bug, but rather a regression in hostapd.  Is anyone else able to validate this?   Perhaps the next steps to move forward is test this against different versions of hostapd to see if/when this regression may have been reintroduced and report an issue with hostapd maintainer?

My current configuration:
#OpenWRT 19.07.03
* kernel  4.14.180-1-0894164cab0effc42201a29fec8ce33f
* kmod-mac80211 4.14.180+4.19.120-1-1
* kmod-ath 4.14.180+4.19.120-1-1
* kmod-ath10k-ct 4.14.180+2019-09-09-5e8cd86f-1
* ath10k-firmware-qca99x0-ct-htt2020-07-02-1 (appears to be 10.4b-ct-9980-fH-13-10af6a005 looking a bin file)
* hostapd-common 2019-08-08-ca8c2bd2-4
* wpad-basic 2019-08-08-ca8c2bd2-3

xupefei, 2021-01-09:
> I am happy to say that I am seeing much improved throughput and lower latency after updating hostapd 2019-08-08-ca8c2bd2-3 » 2019-08-08-ca8c2bd2-4

@ahart241 Could you elaborate? the `2019-08-08-ca8c2bd2-4` version has been there for a while. What is the throughput you get before and after?

ahart241, 2021-01-11:
Previous throughput was between 150-300Mbps, after upgrading from hostapd 2019-08-08-ca8c2bd2-3 » 2019-08-08-ca8c2bd2-4 and with Software Flow Offloading enabled I am achieving download speeds topping out around 480Mbps.   Testing using with iperf3, between my phone (Samsung Galaxy S10) and the EA8500 I am getting 470-520Mbps RX and 270-290Mbps TX from perspective of the router.  Also, this ping response time is seemingly better after upgrading hostapd.

igiannakas, 2021-01-12:
experiencing the same issue with my R7800.

1. Router configured as dumb access point
2. Installed release 19.07.5
3. Configured 5Ghz band as 80mhz, channel 32 (no conflicting AP's on this)
4. Run iPERF3 test between my MacBook Pro Touch Bar 2016 and my home server wired to the AP. Download speed while close to the AP was ~700mbps and upload ~550-600mbps
5. Run iPERF3 test between my MacBook Pro Touch Bar 2019 and my home server. Upload speed ~550-600mbps. But the download speed starts at ~350-400mbps and drops to ~150mbps.
6. Reverted firmware to stock latest Netgear. Same channel config. Both laptops now download at full speed (~700-750mbps).

greearb, 2021-01-12:
Comparing against netgear firmware doesn't help me much, there could be huge number of other factors that affect performance vs openwrt.  Comparing against owrt with upstream qca firmware instead of ath10k-ct firmware (and/or upstream driver vs ath10k-ct driver) may be more helpful.

timkgh, 2021-02-15:
I just ran an `iperf3` TCP test using hnyman's build `master-r15759-ce4cb8e51d-20210214` which comes with the CT driver and firmware `10.4b-ct-9984-fW-13-5ae337bb1`. One side is a wired Linux server, the other side a 2019 MacBook Pro on 5GHz radio. The only difference between the tests is the firmware, otherwise the setup was exactly the same, same location.

Mainline firmware `10.4-3.9.0.2-00139`:
```
[ ID] Interval           Transfer     Bitrate         Retr
[  7]   0.00-10.01  sec   571 MBytes   478 Mbits/sec  110             sender
[  7]   0.00-10.00  sec   568 MBytes   476 Mbits/sec                  receiver
```

CT `10.4b-ct-9984-fW-13-5ae337bb1`:
```
[ ID] Interval           Transfer     Bitrate         Retr
[  7]   0.00-10.00  sec   205 MBytes   172 Mbits/sec  250             sender
[  7]   0.00-10.00  sec   202 MBytes   170 Mbits/sec                  receiver
```
The performance difference is significant.