greearb/ath10k-ct #142: OpenWrt 19.07.3 Kernel Trace - 4.14.180 - ZyXEL NBG6817

evilbulgarian, 2020-06-24:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
pretty consistently

Software (OS, Firmware version, kernel, driver, etc)

        "kernel": "4.14.180",
        "hostname": "wifi01",
        "system": "ARMv7 Processor rev 0 (v7l)",
        "model": "ZyXEL NBG6817",
        "board_name": "zyxel,nbg6817",
        "release": {
                "distribution": "OpenWrt",
                "version": "19.07.3",
                "revision": "r11063-85e04e9f46",
                "target": "ipq806x/generic",
                "description": "OpenWrt 19.07.3 r11063-85e04e9f46"
        }

Hardware (NIC chipset, platform, etc)
        "system": "ARMv7 Processor rev 0 (v7l)",
        "model": "ZyXEL NBG6817",

Logs (dmesg, maybe supplicant and/or hostap)
[ 7436.548918] ------------[ cut here ]------------
[ 7436.548975] WARNING: CPU: 0 PID: 2215 at backports-4.19.120-1/net/wireless/util.c:1147 0xbf260d88 [cfg80211@bf25c000+0x37000]
[ 7436.552773] invalid rate bw=4, mcs=15, nss=4
[ 7436.563976] Modules linked in: pppoe ppp_async ath10k_pci ath10k_core ath pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD xt_CT slhc nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_of_simple ohci_platform ohci_hcd phy_qcom_dwc3 ahci ehci_platform
[ 7436.616620]  sd_mod ahci_platform libahci_platform libahci libata scsi_mod ehci_hcd gpio_button_hotplug ext4 mbcache jbd2 crc32c_generic
[ 7436.638844] CPU: 0 PID: 2215 Comm: hostapd Not tainted 4.14.180 #0
[ 7436.651047] Hardware name: Generic DT based system
[ 7436.657044] Function entered at [<c030f1c4>] from [<c030b390>]
[ 7436.661814] Function entered at [<c030b390>] from [<c07c09c4>]
[ 7436.667630] Function entered at [<c07c09c4>] from [<c031f878>]
[ 7436.673444] Function entered at [<c031f878>] from [<c031f8d8>]
[ 7436.679260] Function entered at [<c031f8d8>] from [<bf260d88>]
[ 7436.685118] Function entered at [<bf260d88>] from [<bf26e83c>]
[ 7436.690897] Function entered at [<bf26e83c>] from [<bf27b654>]
[ 7436.696711] Function entered at [<bf27b654>] from [<bf27c04c>]
[ 7436.702526] Function entered at [<bf27c04c>] from [<c06e1a1c>]
[ 7436.708341] Function entered at [<c06e1a1c>] from [<c06e0238>]
[ 7436.714155] Function entered at [<c06e0238>] from [<c06e0a10>]
[ 7436.719972] Function entered at [<c06e0a10>] from [<c06dfa48>]
[ 7436.725787] Function entered at [<c06dfa48>] from [<c06dfe64>]
[ 7436.731605] Function entered at [<c06dfe64>] from [<c0689d5c>]
[ 7436.737421] Function entered at [<c0689d5c>] from [<c068a5c4>]
[ 7436.743235] Function entered at [<c068a5c4>] from [<c0307b60>]
[ 7436.749172] ---[ end trace 907dd917a7094ad8 ]---

evilbulgarian, 2020-06-24:
ath10k-firmware-qca9984-ct - 2019-10-03-d622d160-1

greearb, 2020-07-16:
This warning should go away in more recent versions.  It should be harmless either way.

closed 2020-07-16 by greearb