greearb/ath10k-ct #144: Wave2 - slow, unstable, DFS false positives...

tj90241, 2020-07-13:
I'll gather proper data for a bug report shortly.  Wanted to raise awareness that there looks to be a major regression in either Wave2/`qca998x` or the CT `ath10k-5.7` module.  Disclaimer: I've never tried CT variant of driver/firmware before... so maybe the issues have always been there - though I would imagine that Wave2 is fairly stable at this point.

**Description of the problem (how to configure, how to reproduce, how often it happens)**:
`qca988x` is configured as a 5ghz AP (under `hostapd`).  Regardless of channel width (40/80), channel, or firmware used, I observed that all clients could pull no more than 5mbps down from the AP whereas the upstream `ath10k` driver/firmware remains rock solid stable and easily dishes >100mbps to clients in the same environment and under the same configuration.  At the same time, clients could TX at about 50mbps to the AP with the CT firmware - strangely much more performant in that direction (though again... I routinely observe clients doing >100mbps in this direction with the upstream `ath10k` as well).

Moreover, the AP tended to crash within < 1min when running CT drivers/firmware and when `iperf` testing with clients.  Tried several CT versions and variants with the exact same behavior.  With a very limited number of clients, have not observed similar instability with upstream `ath10k`.

Lastly, (as also seen in the upstream firmware/driver - hence me prompting to try CT...) DFS consistently produces false positives when the airspace is dead silent and the first client turns on and attempts to auth.  I tried all 80mhz channels and several 40mhz channels with the same result -- CAC _consistently_ succeeds and the AP does not detect radar... until the moment a client powers on and attempt to auth with the AP.

**Software (OS, Firmware version, kernel, driver, etc)**:
Compiled kernel 5.7.8 (current upstream release) patched with `0001-wireless-Relax-beacon_int_min_gcd-and-ADHOC-check.patch`.  Some slight modifications to the `ath` driver so that it can emit IR on 5ghz US/FCC-approved channels.  Same kernel with vanilla ath10k and `10.2.4-1.0-00047` firmware is spectacularly stable on non-DFS channels.

**Hardware (NIC chipset, platform, etc)**:
Compex WLE900VX (qca998x) in a Netgate RCC-DFF-2220 (Intel Atom C2338 SoC) running Debian Linux ("buster").
SMA pigtails off the mPCIe card to qty. (3) run of mill omni antennas off Amazon -- probably easily single digit dBi.

**Logs (dmesg, maybe supplicant and/or hostap)**:
Will gather tomorrow as I use the router as my daily driver and need to revert to `ath10k` upstream for the time being.  Please let me know of anything useful outside of what's documented as part of a standard bug request.

greearb, 2020-07-13:
I have found that my wave-1 firmware does not have as good of RF characteristics as upstream 10.2 firmware.  I don't know why at this point, and I am mostly focused on wave-2 at this point.  If wave-1 upstream works for you, then please use that.  For what its worth, we see good performance on the WLE900VX, often near 900+Mbps in good RF conditions.  In previous testing with rate-vs range on a 9882 (2x2 9880 basically), we got good numbers, but not quite as good as 10.2 upstream firmware.

tj90241, 2020-07-13:
OK, sounds good!  Thankfully, it looks like Compex has some very affordable drop in replacements so I may very well do that.

closed 2020-07-13 by tj90241

greearb, 2020-07-13:
We have had nice luck with the Compex 9984 single-band NICs.  ath10k-ct firmware does not support the new dual-band 9984 NICs at this time....