greearb/ath10k-ct #145: Openwrt on Archer C7: non-CT version better performance and stability

ulrich-mayer, 2020-07-22:
Because I read about DAWN I switched from 19.07 to current snapshots. 
When I observed  kernel fails I filed this bug:
https://bugs.openwrt.org/index.php?do=details&task_id=3228&order=dateopened&sort=desc

After I switched to the non-CT version of the ath10k driver stability and performance improved significantly, as described in a forum post here:
https://forum.openwrt.org/t/dawn-on-archer-c7-minor-problems/69809
Even though the non-CT driver also has issues (see end of `dmesg`) the user experience is much better (as of July 2020)

In the forum discussion @greearb  recommended to open an issue here and asked for more detail on the chipset of the failing device:
From  `dmesg` (with the non-CT driver):
`[   15.550417] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000`

Regards
Uli

Full `dmesg`:
```
[    0.000000] Linux version 5.4.52 (builder@buildhost) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r13888-d5a148f5c8)) #0 Sun Jul 19 22:58:30 2020
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019750 (MIPS 74Kc)
[    0.000000] MIPS: machine is TP-Link Archer C7 v5
[    0.000000] SoC: Qualcomm Atheros QCA956X ver 1 rev 0
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000007ffffff]
[    0.000000] On node 0 totalpages: 32768
[    0.000000]   Normal zone: 288 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 32768 pages, LIFO batch:7
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 32480
[    0.000000] Kernel command line: console=ttyS0,115200n8 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 121928K/131072K available (5062K kernel code, 186K rwdata, 1168K rodata, 1192K init, 205K bss, 9144K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS: 51
[    0.000000] random: get_random_bytes called from start_kernel+0x32c/0x520 with crng_init=0
[    0.000000] CPU clock: 775.000 MHz
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4932285024 ns
[    0.000007] sched_clock: 32 bits at 387MHz, resolution 2ns, wraps every 5541893118ns
[    0.008220] Calibrating delay loop... 385.02 BogoMIPS (lpj=770048)
[    0.046714] pid_max: default: 32768 minimum: 301
[    0.051713] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.059424] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.071655] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.081943] futex hash table entries: 256 (order: -1, 3072 bytes, linear)
[    0.089228] pinctrl core: initialized pinctrl subsystem
[    0.097073] NET: Registered protocol family 16
[    0.131383] clocksource: Switched to clocksource MIPS
[    0.137808] NET: Registered protocol family 2
[    0.143260] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 4096 bytes, linear)
[    0.152141] TCP established hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.160234] TCP bind hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.167670] TCP: Hash tables configured (established 1024 bind 1024)
[    0.174496] UDP hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.181431] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes, linear)
[    0.189104] NET: Registered protocol family 1
[    0.193754] PCI: CLS 0 bytes, default 32
[    0.201290] workingset: timestamp_bits=14 max_order=15 bucket_order=1
[... 169 lines cut ...]
[   37.472798] br-lan: port 4(wlan1) entered blocking state
[   37.478325] br-lan: port 4(wlan1) entered forwarding state
[32112.212863] ------------[ cut here ]------------
[32112.217849] WARNING: CPU: 0 PID: 1194 at backports-5.7-rc3-1/net/mac80211/sta_info.c:1929 ieee80211_sta_update_pending_airtime+0x20c/0x214 [mac80211]
[32112.231699] STA ea:46:3d:86:c5:e8 AC 2 txq pending airtime underflow: 4294967100, 196
[32112.231702] Modules linked in: ath9k ath9k_common pppoe ppp_async iptable_nat ath9k_hw ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT pppox ppp_generic nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_rtcache nf_conntrack mac80211 ipt_REJECT cfg80211 xt_time xt_tcpudp xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TCPMSS xt_LOG slhc nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_filter ip_tables crc_ccitt compat ledtrig_usbport ledtrig_heartbeat nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 fsl_mph_dr_of ehci_platform ehci_fsl ehci_hcd gpio_button_hotplug usbcore nls_base usb_common
[32112.306945] CPU: 0 PID: 1194 Comm: hostapd Not tainted 5.4.52 #0
[32112.313147] Stack : 806a0000 80643a24 00000000 00000000 80642bf4 87c0bc7c 87e86ee0 8067dd83
[32112.321782]         805e0c98 000004aa 807d32d8 00000842 40000010 00000001 87c0bc30 ff466b54
[32112.330419]         00000000 00000000 80800000 000000e5 00000030 00000000 6420352e 342e3532
[32112.339056]         000000e5 8a11d732 00000000 0004af01 80000000 00000009 00000000 8770862c
[32112.347689]         00000009 00000842 40000010 87212460 00000003 80307718 00000000 807d0000
[32112.356322]         ...
[32112.358851] Call Trace:
[32112.361389] [<80069944>] show_stack+0x30/0x100
[32112.366010] [<8007e818>] __warn+0xc0/0x10c
[32112.370249] [<8007e8f0>] warn_slowpath_fmt+0x8c/0xac
[32112.375493] [<8770862c>] ieee80211_sta_update_pending_airtime+0x20c/0x214 [mac80211]
[32112.383589] [<877038ec>] ieee80211_tx_monitor+0xf0c/0x120c [mac80211]
[32112.390333] ---[ end trace f6eb2110714d04d3 ]---
[38517.634668] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 2 is up
[39549.825221] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 2 is down
[124957.584432] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 2 is up
[126262.164136] Atheros AR8216/AR8236/AR8316 mdio.0:00: Port 2 is down
```

ulrich-mayer, 2020-08-02:
I found
```
root@aps:/lib/firmware/ath10k/QCA988X# opkg list *firmware*qca988x*
ath10k-firmware-qca988x - 2019-10-03-d622d160-1 - ath10k firmware for QCA988x devices
ath10k-firmware-qca988x-ct - 2020-04-24-2 - Alternative ath10k firmware for QCA988X from Candela Technologies.
 Enables IBSS and other features.  See:
 http://www.candelatech.com/ath10k-10.1.php
 This firmware will NOT be used unless the standard ath10k-firmware-qca988x
 is un-selected since the driver will try to load firmware-5.bin before
 firmware-2.bin
ath10k-firmware-qca988x-ct-full-htt - 2020-04-24-2 - Alternative ath10k firmware for QCA988X from Candela Technologies.
 Uses normal HTT TX data path for management frames, which improves
 stability in busy networks and fixes .11r authentication.
 Enables IBSS and other features.  See:
 http://www.candelatech.com/ath10k-10.1.php
 This firmware selects and requires the ath10k-ct driver.
```
Note the  **_fixes .11r authentication_** on the full-htt version.

I guess that the reason for the bad user experience when I tried last time was the lack of 802.11r support in the default ath10k-ct firmware on openwrt, Since I use dawn and therefore 802.11rkv.

I gave the ath10k-ct driver another try, and now I think the user experience is on par with the non-CT driver/FW.

I still I hit a lot of
```
Sun Aug  2 16:52:21 2020 daemon.err dawn[2508]: not enough memory (4019496580 @ 104)
Sun Aug  2 16:52:21 2020 daemon.err dawn[2508]: not enough memory (4084566110 @ 104)
Sun Aug  2 16:52:21 2020 daemon.err dawn[2508]: not enough memory (1561658278 @ 104)
Sun Aug  2 16:52:21 2020 daemon.err dawn[2508]: not enough memory (3292162862 @ 104)
```
when roaming while transferring big files.

Since I saw various FW versions on http://www.candelatech.com/ath10k-10.1.php I tried the one for machines with tight memory: 
`firmware-2-ct-full-nrcc-community.bin` (after setting `/lib/firmware/ath10k/fwcfg-pci-0000:00:00.0.txt` according to what I found on the bottom of `http://www.candelatech.com/ath10k-10.4.php#config`. It loads properly).
As expected, the DAWN not enough memory messages are still there.

@greearb I'll have an eye on the kernel messages and report if I hit another one with `firmware-2-ct-full-nrcc-community.bin`.
Also, I'd recommend to rephrase the description on http://www.candelatech.com/ath10k-10.1.php a little.
It took me a while to realize that the "compiled out" in  `swbmiss, beacon filtering, roaming code, descriptor-mgt compiled out` does not just refer to "descriptor management", but it might as well be just me...

greearb, 2020-08-03:
Hello,

I'd expect any of the different variants of ath10k-ct firmware to support 11kvr.

Can you test again to see if changing just the firmware variant causes some reproducible
problem?

Thanks,
Ben

ulrich-mayer, 2020-08-17:
Hey @greearb, sorry for the delay....
yesterday I picked a new snapshot because I saw a number of promising commits:
OpenWrt SNAPSHOT r14181-1e696c6ced / LuCI Master git-20.226.17214-f2e9031
which includes 
`commit eff8c76aa02275a7b325e0fa93cc349380299fde`
`commit b7727a8005635a46255518bdf19eb016e160278a`

all I did initially was switching to smallbuffers:
```
opkg update
opkg remove wpad-basic kmod-ath10k-ct
opkg install avahi-daemon-service-http avahi-nodbus-daemon \
             wpad-openssl  \
             kmod-ath10k-ct-smallbuffers
opkg install luci luci-app-dawn
```
With that I quickly hit a firmware issue again:
some 
  `kern.warn kernel: [ 1375.187294] ath10k_pci 0000:00:00.0: SWBA overrun on vdev 0, skipped old beacon`
and eventually 
  `kern.err kernel: [ 1378.051356] ath10k_pci 0000:00:00.0: Cannot communicate with firmware, previous wmi cmds: 36904:268747 36954:268721 36904:268715 36904:268715, jiffies: 269504, attempting to fake crash and restart firmware, dev-flags: 0x42`
[apw-kernel.log](https://github.com/greearb/ath10k-ct/files/5082794/apw-kernel.log)

Then I changed to the htt version of the firmware:
   `ath10k-firmware-qca988x-ct-full-htt`
I don't know whether it's related to the htt version, but there I hit another familiar FW issue:
  `kern.err kernel: [25474.893049] ath10k_pci 0000:00:00.0: firmware crashed! (guid 218d97e0-f6d9-4ed5-b942-5d1b908ecbf5)`
[aps-reconnect.log](https://github.com/greearb/ath10k-ct/files/5082803/aps-reconnect.log)

Please note that there seems to be a significant difference between Archer C7 v2 and v5
NO fails observed so far with CT on v2, it is still running with the CT htt version:
  `SoC: Qualcomm Atheros QCA9558 ver 1 rev 0`

I see the fails on Archer C7 v5, these I have switched back to the non-CT version:
  `SoC: Qualcomm Atheros QCA956X ver 1 rev 0`

The FW identifies the same wifi HW on both Archer C7 v2 and v5:
  `ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000`

greearb, 2020-08-17:
The 'aps-reconnect.log' crash is same or similar as bug 123, dereference some 0xdeadbeff ptr.  I don't know why that would happen, but looks pretty serious, so I think crashing and restarting firmware is best option.  Could be board-file issues on the v5 perhaps?

real-t0mg, 2020-08-18:
Hi, I'm having the same issue I believe. 
Hardware is TP-Link Archer C7 v5 with CT  ath10k-firmware-qca988x-ct - 2019-10-03-d622d160-1. I've opened a bug in the OpenWRT tracker too ( [bug report in openwrt](https://bugs.openwrt.org/index.php?do=details&task_id=3293) ). 

Let me know what I may do to help.

[ath10k_firmware_crash.txt](https://github.com/greearb/ath10k-ct/files/5092403/ath10k_firmware_crash.txt)

ulrich-mayer, 2020-08-20:
@greearb I've diffed board.bin from CT and non-CT packages: they are the same.