greearb/ath10k-ct #146: Is qca9377 supposed to work?

skinkie, 2020-08-02:
I have tried multiple CT firmware images on the QCA9377 wireless interface, following the guide, keeping board.bin. I have noticed the following output on Linux kernel 5.7.8-gentoo. The obvious question: should this have worked?

```
[ 6790.015542] ath10k_pci 0000:04:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[ 6790.473138] ath10k_pci 0000:04:00.0: Unknown FW IE: 30
[ 6790.473146] ath10k_pci 0000:04:00.0: qca9377 hw1.1 target 0x05020001 chip_id 0x003821ff sub 17aa:0901
[ 6790.473149] ath10k_pci 0000:04:00.0: kconfig debug 1 debugfs 1 tracing 1 dfs 1 testmode 0
[ 6790.473710] ath10k_pci 0000:04:00.0: firmware ver 10.4-ct-9980-fW-011-33ec52c8 api 5 features mfp,peer-flow-ctrl crc32 bcedb774
[ 6793.481548] ath10k_pci 0000:04:00.0: unable to write to the device
[ 6793.481555] ath10k_pci 0000:04:00.0: could not write otp for board id check: -110
[ 6793.481557] ath10k_pci 0000:04:00.0: failed to get board id from otp: -110
[ 6793.481605] ath10k_pci 0000:04:00.0: could not probe fw (-110)
```

04:00.0 Network controller: Qualcomm Atheros QCA9377 802.11ac Wireless Network Adapter (rev 31)

greearb, 2020-08-03:
Hello,

There is no ath10k-ct support for that chipset, I do not have access to the firmware source.

Thanks,
Ben

closed 2020-08-04 by greearb

skinkie, 2020-08-04:
@greearb thanks for the quick follow up, it is appreciated. I hope this firmware type will eventually be at your disposal. I am quite surprised how much better the wifi ability of a Nokia 3.1 phone is, over this adapter.