greearb/ath10k-ct #147: increasing the number of virtual AP interfaces

SeungHyunJeong, 2020-08-05:
Hi, I'm trying to use as many virtual APs as possible with my archer c7 ac1750 v2 (qca9880).
It seems that CT ath10k firmware on openwrt supports up to 8 virtual APs. 9th interface does not operate.
(openwrt with ath10k-firmware-qca988x-ct package)
Any suggestions for using more vaps? Can I use more vaps with brand new AP or hostapd on linux?
Thanks in advance.

DanielRIOT, 2020-08-06:
you might be bale to set it according to this :  http://www.candelatech.com/ath10k-10.4.php#config

SeungHyunJeong, 2020-08-07:
Thanks for the reply. I tried to set fwcfg, but the file does not exist.
Here is the firmware_info from my device.

root@OpenWrt:~# cat /sys/kernel/debug/ieee80211/phy0/ath10k/firmware_info
directory: ath10k/QCA988X/hw2.0
firmware:  firmware-2.bin
fwcfg:     fwcfg-pci-0000:00:00.0.txt
bus:       0000:00:00.0
features:  wmi-10.x,has-wmi-mgmt-tx,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT
version:   10.1-ct-8x-__fW-022-b0e1b7cd
hw_rev:    988x
board:     board.bin

I changed board.bin and firmware-2.bin, but fwcfg-pci-0000:00:00.0.txt is still not found.

SeungHyunJeong, 2020-08-07:
It seems that initially there's no text and I'm supposed to write the text.
When I wrote text to the correct directory and rebooted, the device read the parameters written to the text.
The device is not working well yet, so I would not close the issue right now.

SeungHyunJeong, 2020-08-08:
Configured as below, archer c7 ac1750 v2 is working well with up to 16 virtual APs, but enabling 17th virtual AP makes it quite unstable.

> vdevs = 32
> peers = 64
> active_peers = 64
> stations = 63
> rate_ctrl_objs = 8
> regdom = 840
> fwname = firmware-2.bin
> fwver = 2
> nohwcrypt = 1
> ct_sta_mode = 0
> tx_desc = 1496
> #max_nss = 3
> tids = 256
> skid_limit = 360
> max_amsdus = 3
> allow_all_mcs = 0

greearb, 2020-08-10:
Hello, I had to make changes to mac80211 to support more than 32 vdevs, and possibly there were other limitations that I forgot about.  You can always try one of our kernels with all patches applied from greearb github account.  Kernel 5.4 is probably best choice at this point, 5.7 has not had much testing.

SeungHyunJeong, 2020-08-12:
Thanks for the comment. 32 vdevs are enough for now, and I'll try the kernel.