greearb/ath10k-ct #148: 19.07.3 vs Master and qca9984

DanielRIOT, 2020-08-06:
I'm not sure if this is due to differences in kernel 4.14 and 5.x, PCI issues of ath10k.

when I use the master build the system seems to fail after loading firmware to the PCIe cards ( qca9984  ) , they seem OK when using a 19.07.3 build and manually installing ath10k-ct and the ct-htt firmware 

OpenWrt master (  OpenWrt SNAPSHOT, r13545-efde716d14 ) vs 19.07.3 

Platform : mvebu : [ Solidrun clearfog A1 rev 2.1 ( pro ) :](https://openwrt.org/toh/hwdata/solidrun/solidrun_clearfog_pro) 
NIC : Compex WLE1216V5-20, Compex WLE1216V2-20 ( qca9984 )
Driver :  kmod-ath10k-ct
Firmware : ath10k-firmware-qca9984-ct-htt

dmesg : 
 happy 
Using 19.07.3 release from https://downloads.openwrt.org/releases/19.07.3/targets/mvebu/cortexa9/ : .
```
[   11.965570] ath10k_pci 0000:02:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[   11.975526] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   11.987062] ath10k_pci 0000:02:00.0: firmware ver 10.4b-ct-9984-fH-012-17ba98334 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 15d04c8a
[   13.292127] ath10k_pci 0000:02:00.0: board_file api 2 bmi_id 0:30 crc32 85498734
[   15.780638] ath10k_pci 0000:02:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   15.788165] ath10k_pci 0000:02:00.0: msdu-desc: 2500  skid: 32
[   15.867644] ath10k_pci 0000:02:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   15.878564] ath10k_pci 0000:02:00.0: wmi print 'free: 81800 iram: 23476 sram: 14440'
[   16.130412] ath10k_pci 0000:02:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
```
unhappy
built from Master  : 
```
modprobe ath10k_pci
[ 4290.810379] ath10k 5.1 driver, optimized for CT firmware, probing pci device: 0x46.
[ 4290.818520] ath10k_pci 0000:02:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[ 4291.003734] ath10k_pci 0000:02:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[ 4291.013706] ath10k_pci 0000:02:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[ 4291.025032] ath10k_pci 0000:02:00.0: firmware ver 10.4b-ct-9984-tH-013-d81f62d97 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7d7b454d
[ 4292.331284] ath10k_pci 0000:02:00.0: board_file api 2 bmi_id 0:30 crc32 85498734
[ 4295.756999] ath10k_pci 0000:02:00.0: failed to receive control response completion, polling..
[ 4296.796895] ath10k_pci 0000:02:00.0: Service connect timeout
[ 4296.802570] ath10k_pci 0000:02:00.0: failed to connect htt (-110)
[ 4296.897742] ath10k_pci 0000:02:00.0: could not init core (-110)
[ 4296.903756] ath10k_pci 0000:02:00.0: could not probe fw (-110)
```

-Daniel

greearb, 2020-08-11:
Are there any /lib/firmware/ath10k/fwcfg* files on your system that fails?

DanielRIOT, 2020-11-03:
I suspect this is an issue with the Solidrun A1 PCIe implementation between the older and newer kernel version.
 We've moved to use the Compex WPQ865 and x86 , I think its best to close this for now until I can get back to testing the Solidrun boards at some later date

closed 2020-11-03 by DanielRIOT