greearb/ath10k-ct #149: Excessive latency on R7800 / OpenWrt with ath10k-ct-smallbuffers driver and ath10k-ct beta firmware

Fail-Safe, 2020-08-12:
**Description of the problem (how to configure, how to reproduce, how often it happens).**
Multiple 5Ghz connected Apple devices exhibit an issue with excessive latency and packet loss. The issue is most visible when said devices are "idle" in the sense that they are not involved in a consistent data flow (such as an audio/video stream or active download).

Tested & Affected Devices (within my possession):
2019 iMac
2019 MacBook Pro 16"
2019 MacBook Pro 15"
iPhone 11 Pro
iPhone X
iPhone 8
iPad Mini 5

Example Scenarios:
* iPhone 11 Pro screen is powered off and device is set down. A ping is initiated from the R7800 to which the iPhone 11 Pro is connected:
```
root@OpenWrt:~# ping -c60 192.168.45.193
PING 192.168.45.193 (192.168.45.193): 56 data bytes
64 bytes from 192.168.45.193: seq=0 ttl=64 time=99.479 ms
64 bytes from 192.168.45.193: seq=1 ttl=64 time=19.742 ms
64 bytes from 192.168.45.193: seq=2 ttl=64 time=41.425 ms
64 bytes from 192.168.45.193: seq=3 ttl=64 time=65.785 ms
64 bytes from 192.168.45.193: seq=4 ttl=64 time=89.470 ms
64 bytes from 192.168.45.193: seq=37 ttl=64 time=1024.032 ms
64 bytes from 192.168.45.193: seq=38 ttl=64 time=23.681 ms
64 bytes from 192.168.45.193: seq=39 ttl=64 time=4.495 ms
64 bytes from 192.168.45.193: seq=40 ttl=64 time=4.267 ms
64 bytes from 192.168.45.193: seq=41 ttl=64 time=14.181 ms
64 bytes from 192.168.45.193: seq=42 ttl=64 time=39.829 ms
64 bytes from 192.168.45.193: seq=43 ttl=64 time=63.829 ms
64 bytes from 192.168.45.193: seq=44 ttl=64 time=87.284 ms
64 bytes from 192.168.45.193: seq=45 ttl=64 time=110.995 ms
64 bytes from 192.168.45.193: seq=46 ttl=64 time=32.362 ms
64 bytes from 192.168.45.193: seq=47 ttl=64 time=57.043 ms
64 bytes from 192.168.45.193: seq=48 ttl=64 time=80.219 ms
64 bytes from 192.168.45.193: seq=49 ttl=64 time=100.813 ms
64 bytes from 192.168.45.193: seq=50 ttl=64 time=23.303 ms
64 bytes from 192.168.45.193: seq=51 ttl=64 time=48.281 ms
64 bytes from 192.168.45.193: seq=52 ttl=64 time=71.831 ms
64 bytes from 192.168.45.193: seq=53 ttl=64 time=96.371 ms
64 bytes from 192.168.45.193: seq=54 ttl=64 time=4.866 ms
64 bytes from 192.168.45.193: seq=55 ttl=64 time=39.018 ms
64 bytes from 192.168.45.193: seq=56 ttl=64 time=3.588 ms

--- 192.168.45.193 ping statistics ---
60 packets transmitted, 25 packets received, 58% packet loss
round-trip min/avg/max = 3.588/89.847/1024.032 ms
```

* 2019 MacBook Pro 15" is powered on, but not actively being used. No active streams or downloads. A ping is initiated from the R7800 to which the MacBook Pro is connected:
```
root@OpenWrt:~# ping -c60 192.168.9.83
PING 192.168.9.83 (192.168.9.83): 56 data bytes
64 bytes from 192.168.9.83: seq=0 ttl=64 time=142.484 ms
64 bytes from 192.168.9.83: seq=2 ttl=64 time=4017.804 ms
64 bytes from 192.168.9.83: seq=3 ttl=64 time=3017.449 ms
64 bytes from 192.168.9.83: seq=4 ttl=64 time=2016.106 ms
64 bytes from 192.168.9.83: seq=5 ttl=64 time=1015.450 ms
64 bytes from 192.168.9.83: seq=6 ttl=64 time=15.043 ms
64 bytes from 192.168.9.83: seq=7 ttl=64 time=2172.278 ms
64 bytes from 192.168.9.83: seq=8 ttl=64 time=1171.835 ms
64 bytes from 192.168.9.83: seq=9 ttl=64 time=171.398 ms
64 bytes from 192.168.9.83: seq=10 ttl=64 time=268.165 ms
64 bytes from 192.168.9.83: seq=11 ttl=64 time=2.214 ms
64 bytes from 192.168.9.83: seq=12 ttl=64 time=1.776 ms
64 bytes from 192.168.9.83: seq=16 ttl=64 time=3167.838 ms
64 bytes from 192.168.9.83: seq=17 ttl=64 time=2165.365 ms
64 bytes from 192.168.9.83: seq=18 ttl=64 time=1164.100 ms
64 bytes from 192.168.9.83: seq=19 ttl=64 time=161.673 ms
64 bytes from 192.168.9.83: seq=20 ttl=64 time=3489.115 ms
64 bytes from 192.168.9.83: seq=21 ttl=64 time=2488.641 ms
[... 598 lines cut ...]
[   50.605847] br-GUEST: port 2(wlan1-1) entered forwarding state
[   50.643426] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1-1: link becomes ready
[   50.645308] br-IOT: port 2(wlan1-2) entered blocking state
[   50.648956] br-IOT: port 2(wlan1-2) entered disabled state
[   50.654575] device wlan1-2 entered promiscuous mode
[   50.662838] br-IOT: port 2(wlan1-2) entered blocking state
[   50.664557] br-IOT: port 2(wlan1-2) entered forwarding state
[   50.695457] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1-2: link becomes ready
[   51.197509] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   51.197638] br-lan: port 3(wlan0) entered blocking state
[   51.202995] br-lan: port 3(wlan0) entered forwarding state
[   51.210054] br-GUEST: port 3(wlan0-1) entered blocking state
[   51.213756] br-GUEST: port 3(wlan0-1) entered disabled state
[   51.219687] device wlan0-1 entered promiscuous mode
[   51.237783] br-GUEST: port 3(wlan0-1) entered blocking state
[   51.237808] br-GUEST: port 3(wlan0-1) entered forwarding state
[   51.282510] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-1: link becomes ready
[   51.284426] br-IOT: port 3(wlan0-2) entered blocking state
[   51.287929] br-IOT: port 3(wlan0-2) entered disabled state
[   51.293661] device wlan0-2 entered promiscuous mode
[   51.307239] br-IOT: port 3(wlan0-2) entered blocking state
[   51.307265] br-IOT: port 3(wlan0-2) entered forwarding state
[   51.353564] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0-2: link becomes ready
[   53.705310] ath10k_pci 0001:01:00.0: Invalid peer id 0 or peer stats buffer, peer: 7968fa07  sta: 00000000
```

Fail-Safe, 2020-08-12:
I initially believed this to be related to https://github.com/greearb/ath10k-ct/issues/139#issue-636233839. I posted additional information here: https://github.com/greearb/ath10k-ct/issues/139#issuecomment-672118572

With `firmware ver 10.4b-ct-9984-tH-13-795874169`, latency from the R7800 to one of the affected devices is improved in the cases where the MacBooks/iMac are engaged in a consistent data stream, whether audio/video, download, or simply a constant ping to another device on the network (from the affected device).

I loaded the 1158 firmware referenced by @graysky2 here https://github.com/greearb/ath10k-ct/issues/139#issuecomment-660458870 and found a marked improvement. Unfortunately at the present moment I do not have data captured to show this improvement, but will update this comment with that data ASAP.

Fail-Safe, 2020-08-12:
```
root@OpenWrt:~# dmesg | grep firmware
[    2.768381] qcom_rpm 108000.rpm: RPM firmware 3.0.16777364
[   14.093346] ath10k 5.4 driver, optimized for CT firmware, probing pci device: 0x46.
[   14.604608] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-tH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7d16c2a5
[   23.105076] ath10k 5.4 driver, optimized for CT firmware, probing pci device: 0x46.
[   23.682505] ath10k_pci 0001:01:00.0: firmware ver 10.4b-ct-9984-tH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7d16c2a5
```

```
root@OpenWrt:~# ping 192.168.9.83 -c60
PING 192.168.9.83 (192.168.9.83): 56 data bytes
64 bytes from 192.168.9.83: seq=3 ttl=64 time=3495.858 ms
64 bytes from 192.168.9.83: seq=4 ttl=64 time=2482.313 ms
64 bytes from 192.168.9.83: seq=5 ttl=64 time=1481.175 ms
64 bytes from 192.168.9.83: seq=6 ttl=64 time=476.783 ms
64 bytes from 192.168.9.83: seq=7 ttl=64 time=1074.036 ms
64 bytes from 192.168.9.83: seq=8 ttl=64 time=65.571 ms
64 bytes from 192.168.9.83: seq=9 ttl=64 time=605.466 ms
64 bytes from 192.168.9.83: seq=10 ttl=64 time=3017.754 ms
64 bytes from 192.168.9.83: seq=11 ttl=64 time=2015.649 ms
64 bytes from 192.168.9.83: seq=12 ttl=64 time=1008.733 ms
64 bytes from 192.168.9.83: seq=13 ttl=64 time=8.278 ms
64 bytes from 192.168.9.83: seq=14 ttl=64 time=254.260 ms
64 bytes from 192.168.9.83: seq=15 ttl=64 time=251.606 ms
64 bytes from 192.168.9.83: seq=16 ttl=64 time=507.151 ms
64 bytes from 192.168.9.83: seq=17 ttl=64 time=918.400 ms
64 bytes from 192.168.9.83: seq=18 ttl=64 time=3.019 ms
64 bytes from 192.168.9.83: seq=19 ttl=64 time=2.164 ms
64 bytes from 192.168.9.83: seq=20 ttl=64 time=3.814 ms
64 bytes from 192.168.9.83: seq=21 ttl=64 time=1263.659 ms
64 bytes from 192.168.9.83: seq=22 ttl=64 time=261.962 ms
64 bytes from 192.168.9.83: seq=23 ttl=64 time=2.991 ms
64 bytes from 192.168.9.83: seq=27 ttl=64 time=2265.876 ms
64 bytes from 192.168.9.83: seq=28 ttl=64 time=1265.576 ms
64 bytes from 192.168.9.83: seq=29 ttl=64 time=258.511 ms
64 bytes from 192.168.9.83: seq=33 ttl=64 time=2859.909 ms
64 bytes from 192.168.9.83: seq=34 ttl=64 time=1859.704 ms
64 bytes from 192.168.9.83: seq=35 ttl=64 time=848.043 ms
64 bytes from 192.168.9.83: seq=36 ttl=64 time=5.654 ms
64 bytes from 192.168.9.83: seq=37 ttl=64 time=2.373 ms
64 bytes from 192.168.9.83: seq=38 ttl=64 time=92.876 ms
64 bytes from 192.168.9.83: seq=39 ttl=64 time=1562.413 ms
64 bytes from 192.168.9.83: seq=40 ttl=64 time=561.023 ms
64 bytes from 192.168.9.83: seq=41 ttl=64 time=4.121 ms
64 bytes from 192.168.9.83: seq=42 ttl=64 time=834.278 ms
64 bytes from 192.168.9.83: seq=43 ttl=64 time=3.875 ms
64 bytes from 192.168.9.83: seq=44 ttl=64 time=2392.530 ms
64 bytes from 192.168.9.83: seq=45 ttl=64 time=1391.948 ms
64 bytes from 192.168.9.83: seq=46 ttl=64 time=389.467 ms
64 bytes from 192.168.9.83: seq=50 ttl=64 time=2315.519 ms
64 bytes from 192.168.9.83: seq=51 ttl=64 time=1312.532 ms
64 bytes from 192.168.9.83: seq=52 ttl=64 time=312.304 ms
64 bytes from 192.168.9.83: seq=53 ttl=64 time=3.531 ms
64 bytes from 192.168.9.83: seq=54 ttl=64 time=3.155 ms

--- 192.168.9.83 ping statistics ---
60 packets transmitted, 43 packets received, 28% packet loss
round-trip min/avg/max = 2.164/924.322/3495.858 ms
```

```
root@OpenWrt:~# ping 192.168.9.83 -c60
PING 192.168.9.83 (192.168.9.83): 56 data bytes
64 bytes from 192.168.9.83: seq=0 ttl=64 time=3281.805 ms
64 bytes from 192.168.9.83: seq=1 ttl=64 time=2281.926 ms
64 bytes from 192.168.9.83: seq=2 ttl=64 time=1280.053 ms
64 bytes from 192.168.9.83: seq=3 ttl=64 time=278.165 ms
64 bytes from 192.168.9.83: seq=4 ttl=64 time=2711.885 ms
64 bytes from 192.168.9.83: seq=5 ttl=64 time=1710.061 ms
[... 24 lines cut ...]
64 bytes from 192.168.9.83: seq=34 ttl=64 time=1449.175 ms
64 bytes from 192.168.9.83: seq=35 ttl=64 time=446.299 ms
64 bytes from 192.168.9.83: seq=36 ttl=64 time=3.173 ms
64 bytes from 192.168.9.83: seq=37 ttl=64 time=3.864 ms
64 bytes from 192.168.9.83: seq=38 ttl=64 time=681.381 ms
64 bytes from 192.168.9.83: seq=39 ttl=64 time=4.531 ms
64 bytes from 192.168.9.83: seq=43 ttl=64 time=4439.561 ms
64 bytes from 192.168.9.83: seq=44 ttl=64 time=3438.621 ms
64 bytes from 192.168.9.83: seq=45 ttl=64 time=2438.185 ms
64 bytes from 192.168.9.83: seq=46 ttl=64 time=1437.258 ms
64 bytes from 192.168.9.83: seq=47 ttl=64 time=425.822 ms
64 bytes from 192.168.9.83: seq=51 ttl=64 time=3559.122 ms
64 bytes from 192.168.9.83: seq=52 ttl=64 time=2553.035 ms
64 bytes from 192.168.9.83: seq=53 ttl=64 time=1551.441 ms
64 bytes from 192.168.9.83: seq=54 ttl=64 time=540.198 ms
64 bytes from 192.168.9.83: seq=55 ttl=64 time=3.755 ms
64 bytes from 192.168.9.83: seq=56 ttl=64 time=1590.468 ms
64 bytes from 192.168.9.83: seq=57 ttl=64 time=587.417 ms
64 bytes from 192.168.9.83: seq=58 ttl=64 time=475.636 ms
64 bytes from 192.168.9.83: seq=59 ttl=64 time=1802.190 ms

--- 192.168.9.83 ping statistics ---
60 packets transmitted, 50 packets received, 16% packet loss
round-trip min/avg/max = 1.626/1369.552/4439.561 ms
```

Fail-Safe, 2020-08-12:
This is a difficult issue to fully scope because it affects multiple devices, but not consistently. Further, it seems to get better for some time after a reboot (or rmmod/modprobe), but the affected devices may differ even after the issue presents itself again.

graysky2, 2020-08-13:
When you are experiencing the long pings with some packet loss, you mentioned the laptop is "idle."  

1. Do you experience any lag when you google something in a browser?  The lag to which I refer would between the time when you hit "Enter" on the keyboard, and when the search results comes up?
2. When are you streaming video on the laptop, do the ping times tighten up?

graysky2, 2020-08-13:
@Fail-Safe - When you are experiencing the long pings with some packet loss, you mentioned the laptop is "idle."  

1. Do you experience any lag when you google something in a browser on an affected device?  The lag to which I refer is the time between the time when you hit "Enter" on the keyboard, and when the search results comes up?
2. When are you streaming video on the laptop, do the ping times tighten up?

Also, post your `/etc/config/wireless` here (be sure to redact your actual passwords to protect the privacy of your networks).

Fail-Safe, 2020-08-13:
@graysky2 

> 1. Do you experience any lag when you google something in a browser on an affected device?  The lag to which I refer is the time between the time when you hit "Enter" on the keyboard, and when the search results comes up?

I do occasionally, yes. I notice this on my iPhones, especially as distance increases from my R7800 (another complicating factor, I know). But I do see this on my MacBooks occasionally as well.

> 2. When are you streaming video on the laptop, do the ping times tighten up?

Thanks for bringing this up--I actually meant to elaborate more on this in my OP. The answer is yes, absolutely. When this issue is presenting itself with one of my devices, once I get the wireless radio on the device busy with active data flow the ping times actually drop to a consistent <10ms on average. Typically 3-5ms.

> Also, post your `/etc/config/wireless` here (be sure to redact your actual passwords to protect the privacy of your networks).

```
root@OpenWrt:~# cat /etc/config/wireless

config wifi-device 'radio0'
	option type 'mac80211'
	option hwmode '11a'
	option path 'soc/1b500000.pci/pci0000:00/0000:00:00.0/0000:01:00.0'
	option htmode 'VHT80'
	option legacy_rates '0'
	option channel '153'
	option txpower '30'
	option country 'US'
	option distance '450'
	option rts '980'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option key '...'
	option ieee80211w '1'
	option ssid 'LAN'
	option encryption 'psk2+ccmp'

config wifi-device 'radio1'
	option type 'mac80211'
	option hwmode '11g'
	option path 'soc/1b700000.pci/pci0001:00/0001:00:00.0/0001:01:00.0'
	option htmode 'HT20'
	option legacy_rates '0'
	option channel '11'
	option txpower '20'
	option country 'US'
	option distance '450'
	option rts '980'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option key '...'
	option ieee80211w '1'
	option macfilter 'allow'
	option encryption 'psk2+ccmp'
	list maclist '...'
	list maclist '...'
	list maclist '...'
	list maclist '...'
	list maclist '...'
	option ssid 'LAN'

config wifi-iface 'wifinet2'
	option network 'GUEST'
	option device 'radio1'
	option mode 'ap'
	option ieee80211w '1'
	option key '...'
	option ssid 'GUEST'
	option encryption 'psk2+ccmp'

config wifi-iface 'wifinet3'
	option network 'IOT'
	option device 'radio1'
	option mode 'ap'
	option ieee80211w '1'
	option key '...'
	option ssid 'IOT'
	option encryption 'psk2+ccmp'

config wifi-iface 'wifinet4'
	option network 'GUEST'
	option ssid 'GUEST'
	option device 'radio0'
	option mode 'ap'
	option ieee80211w '1'
	option key '...'
	option encryption 'psk2+ccmp'

config wifi-iface 'wifinet5'
	option network 'IOT'
	option ssid 'IOT'
	option device 'radio0'
	option mode 'ap'
	option ieee80211w '1'
	option key '...'
	option encryption 'psk2+ccmp'
```

Fail-Safe, 2020-08-13:
FWIW, I created a new snapshot build last night with the following three patches added:
https://git.openwrt.org/?p=openwrt/staging/nbd.git;a=commit;h=3b26a8b2563a416338c71fc36c1cf8c313457d7c

Updated OS/firmware info:
```
root@OpenWrt:~# dmesg | grep firmware
[    2.779054] qcom_rpm 108000.rpm: RPM firmware 3.0.16777364
[   13.566261] ath10k 5.4 driver, optimized for CT firmware, probing pci device: 0x46.
[   14.059286] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-tH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7d16c2a5
[   22.582070] ath10k 5.4 driver, optimized for CT firmware, probing pci device: 0x46.
[   23.168275] ath10k_pci 0001:01:00.0: firmware ver 10.4b-ct-9984-tH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7d16c2a5

root@OpenWrt:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="SNAPSHOT"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r14154-2d9db93397"
OPENWRT_BOARD="ipq806x/generic"
OPENWRT_ARCH="arm_cortex-a15_neon-vfpv4"
OPENWRT_TAINTS="no-all busybox"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt SNAPSHOT r14154-2d9db93397"

root@OpenWrt:~# opkg list | grep kmod-ath10k
kmod-ath10k-ct-smallbuffers - 5.4.52+2020-06-30-edfbf916-1

root@OpenWrt:~# uname -a
Linux OpenWrt 5.4.52 #0 SMP Thu Aug 13 00:12:12 2020 armv7l GNU/Linux
```

At the moment, even after several hours and two reboots, I am not able to reproduce this same issue at the moment. Not suggesting the patches are the fix, but I did want to make you aware of this change in my config since my OP. Will report back on what I am seeing as time goes by today.

Fail-Safe, 2020-08-13:
Here is what I am seeing on this latest patched build at "idle" on my primary trouble device (you will recognize this IP from my OP).

Same scenario--pinging my primary trouble device (MBP 15") from my R7800:
```
--- 192.168.9.83 ping statistics ---
60 packets transmitted, 60 packets received, 0% packet loss
round-trip min/avg/max = 1.507/53.457/115.308 ms
```

Second scenario, when I start a ping from my MBP 15" to its gateway (`ping -i.2 192.168.9.5`), then simultaneously ping this MBP from my R7800:
```
--- 192.168.9.83 ping statistics ---
60 packets transmitted, 60 packets received, 0% packet loss
round-trip min/avg/max = 1.529/2.824/5.591 ms
```

graysky2, 2020-08-13:
> I do occasionally, yes. I notice this on my iPhones, especially as distance increases from my R7800 (another complicating factor, I know). But I do see this on my MacBooks occasionally as well.

That is consistent with the bug I reported.  When the bug hits, you see BOTH the high pings+packet loss AND shit response times on the google query.

> Thanks for bringing this up--I actually meant to elaborate more on this in my OP. The answer is yes, absolutely. When this issue is presenting itself with one of my devices, once I get the wireless [...]

This makes me think your issue is the same as mine.  Why?  Once the lag hits my iphones, it does not go away/screen on or screen off, downloading something or not.  I am wondering if the idle nature of your problem is suggestive of some power savings option on the macbooks themselves?  That is just speculation.

Regarding your `/etc/config/wireless`...
You config looks sane to me.  As an aside (likely NOT causing your problem), may want to adjust your dtm_period to a value of 3 as this is what apple recommends.  See [this thread](https://forum.openwrt.org/t/devices-connected-to-wifi-become-sluggish-to-load-data-r7800/64731/23) and the links therein.

Finally, I cannot speak to what your latest snapshot build with extra patches may do for you.  Did you build include https://github.com/openwrt/openwrt/commit/91aab77bf1ce91b0e60e720eb147c94a02c1f2fd?  Unless you're manually overwriting the firmware, this bumps the ath10k-ct version to the .019 versions as you may be aware.

Fail-Safe, 2020-08-13:
> That is consistent with the bug I reported. When the bug hits, you see BOTH the high pings+packet loss AND shit response times on the google query.

Granted, I was not seeing the same level of packet loss as you for the most part. I think that was the primary driver for @greearb wanting me to open this separate issue.

> > Thanks for bringing this up--I actually meant to elaborate more on this in my OP. The answer is yes, absolutely. When this issue is presenting itself with one of my devices, once I get the wireless radio on the device busy with active data flow the ping times actually drop to a consistent <10ms on average. Typically 3-5ms.
> 
> This makes me think your issue is the same as mine. Why? Once the lag hits my iphones, it does not go away/screen on or screen off, downloading something or not. I am wondering if the idle nature of your problem is suggestive of some power savings option on the macbooks themselves? That is just speculation. Do you experience the ping lag when you use the ath10k driver and firmware? In my case, I do not see the bug.

While we do have a number of things in common, my main hold-up in drawing the correlation to your issue is my iPhones were not exhibiting the behavior with the same regularity as yours seem to. But it did happen at times and I wish I had saved some of the historic ping results during those moments.

I was thinking along the exact same lines regarding power saving. I could buy more easily in my case if my phones were the primary trouble devices, but at the moment it's more evident on my MacBooks. I guess it's conceivable that Apple might still be cycling radio power on/off on a MacBook that is plugged into a power supply, but why wouldn't my iPhones be showing this more often given they're 90% of the time running on battery. (Just thinking out loud here...)

I will say, even when my iPhones are not exhibiting the trouble behavior, they do still show consistently higher pings as compared to my kids' Nintendo Switch. It is <5ms (and usually closer to 2-3ms) ALL the time it is powered on.

> Regarding your `/etc/config/wireless`...
> You config looks sane to me. As an aside (likely NOT causing your problem), may want to adjust your dtm_period to a value of 3 as this is what apple [...]

Thanks for the second set of eyes on my config--always glad to have good feedback on things like that. I was not aware of the thread you linked, but checked it out and updated to dtim_period = 3. Many thanks for making me aware of that one!

> Finally, I cannot speak to what your latest snapshot build with extra patches may do for you. Did you build include [openwrt/openwrt@91aab77](https://github.com/openwrt/openwrt/commit/91aab77bf1ce91b0e60e720eb147c94a02c1f2fd)? Unless you're manually overwriting the firmware, this bumps the ath10k-ct version to the .019 versions as you may be aware.

As for trying the ath10k driver + firmware--I haven't used it for months, to be honest. I'm not against building and running that config for a bit, but I generally had tried to just commit to the ath10k-ct setup and grow with it. 👍

At the moment, though, I am just simply not seeing the issue with last night's build using the patches I mentioned. Any chance you would be willing to try a snapshot build with the same patches and see if your issue persists or not?

Fail-Safe, 2020-08-13:
Well... isn't that just how IT stuff goes...

It's back:
```
--- 192.168.9.83 ping statistics ---
91 packets transmitted, 88 packets received, 3% packet loss
round-trip min/avg/max = 1.497/1115.832/4549.649 ms
```

greearb, 2020-08-13:
fail-safe:  Do you have a non-phone that exhibits this?  I need to find a device that I can acquire that will reproduce the issue...this is too complicated to debug from afar I think.

Fail-Safe, 2020-08-13:
> fail-safe: Do you have a non-phone that exhibits this? I need to find a device that I can acquire that will reproduce the issue...this is too complicated to debug from afar I think.

Contrary to @graysky2's reports, my iPhones don't show it consistently. The device that I have the most consistent issue with is my MacBook Pro 15in (2019). After that, my next most troublesome device has been my MacBook Pro 16in (2019).

graysky2, 2020-08-13:
> As for trying the ath10k driver + firmware--I haven't used it for months, to be honest. I'm not against building and running that config for a bit, but I generally had tried to just commit to the ath10k-ct setup and grow with it. +1

The point of the experiment is to see if it's the -ct's driver/firmware.

>At the moment, though, I am just simply not seeing the issue with last night's build using the patches I mentioned. Any chance you would be willing to try a snapshot build with the same patches and see if your issue persists or not?

I am running a snapshot I build several days ago when the 5.4.58 kernel was released.

Fail-Safe, 2020-08-13:
> The point of the experiment is to see if it's the -ct's driver/firmware.

Yeah, I get it--sorry if it sounded like I wasn't tracking with the intent there. I am actually in the process of an ath10k build as I type this.

FYI, I just noticed the three AQL related patches I referred to previously have been committed to the current snapshot branch now.

> > At the moment, though, I am just simply not seeing the issue with last night's build using the patches I mentioned. Any chance you would be willing to try a snapshot build with the same patches and see if your issue persists or not?
> 
> I am running a snapshot I build several days ago when the 5.4.58 kernel was released. I am running the the ath10k driver/firmware until @greearb has a new set of firmware for me test.

Apart from the absence of this issue, are you noticing any other major differences between the ath10k and ath10k-ct drivers/fw at the moment? Decreased range? Increased latency? Anything like that?

Out of curiosity, too, where are you getting a 5.4.58 kernel from? The latest I am seeing in snapshot is 5.4.52.

graysky2, 2020-08-13:
>Yeah, I get it--sorry if it sounded like I wasn't tracking with the intent there. I am actually in the process of an ath10k build as I type this.

OK, see what you get with it.

>Apart from the absence of this issue, are you noticing any other major differences between the ath10k and ath10k-ct drivers/fw at the moment? Decreased range? Increased latency? Anything like that?

I did not compare range or latency.  Just the ping test.

> Out of curiosity, too, where are you getting a 5.4.58 kernel from? The latest I am seeing in snapshot is 5.4.52.

https://github.com/openwrt/openwrt/pull/3285

notmyrealhandle, 2020-08-13:
> Well... isn't that just how IT stuff goes...
> 
> It's back:

Between the time when you were symptom free and the symptoms appears, did the client disassociate/re-associate to the wireless?  If so, you may want to see if you can trigger it by intentionally disconnecting/reconnecting after rebooting the client.

greearb, 2020-09-16:
Any chance you can try some openwrt releases from earlier this year to see if they work any better for this particular problem?

closed 2025-05-24 by Fail-Safe