greearb/ath10k-ct #152: on linux-ct-5.8 ping is terrible

gsustek, 2020-09-03:
Please provide this info. See this link for more info on how to gather debug
info: http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
Wifi is up. i compile your kernel, install on Ubuntu 20.04 after i change regdomain in advanced configuration
cat /lib/firmware/ath10k/fwcfg-pci-0000\:06\:00.0.txt 

regdom = 840

root@u1804:~# 

error:

ping is terrible:
Request timeout for icmp_seq 0
64 bytes from 192.168.168.1: icmp_seq=1 ttl=64 time=2.034 ms
64 bytes from 192.168.168.1: icmp_seq=2 ttl=64 time=16.949 ms
Request timeout for icmp_seq 3
Request timeout for icmp_seq 4
Request timeout for icmp_seq 5
64 bytes from 192.168.168.1: icmp_seq=6 ttl=64 time=323.899 ms
Request timeout for icmp_seq 7
64 bytes from 192.168.168.1: icmp_seq=8 ttl=64 time=381.371 ms
Request timeout for icmp_seq 9
Request timeout for icmp_seq 10
Request timeout for icmp_seq 11
Request timeout for icmp_seq 12
Request timeout for icmp_seq 13
64 bytes from 192.168.168.1: icmp_seq=14 ttl=64 time=109.302 ms
Request timeout for icmp_seq 15
64 bytes from 192.168.168.1: icmp_seq=16 ttl=64 time=219.850 ms
64 bytes from 192.168.168.1: icmp_seq=17 ttl=64 time=89.720 ms
Request timeout for icmp_seq 18
64 bytes from 192.168.168.1: icmp_seq=19 ttl=64 time=1.367 ms

Software (OS, Firmware version, kernel, driver, etc)
ubuntu 20.04
https://github.com/greearb/linux-ct-5.8

Hardware (NIC chipset, platform, etc)
Atheros QCA9984

Logs (dmesg, maybe supplicant and/or hostap)

[Thu Sep  3 18:56:16 2020] ath10k 5.7 driver, optimized for CT firmware, probing pci device: 0x46.
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: vdevs  val: 8
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: peers  val: 10
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: active_peers  val: 10
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: stations  val: 10
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: rate_ctrl_objs  val: 7
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: regdom  val: 840
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: fwver  val: 5
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: nohwcrypt  val: 0
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: ct_sta_mode  val: 0
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: tx_desc  val: 2200
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: tids  val: 450
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: skid_limit  val: 360
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: fwcfg key: max_amsdus  val: 3
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: kconfig debug 0 debugfs 1 tracing 1 dfs 0 testmode 0
[Thu Sep  3 18:56:16 2020] ath10k_pci 0000:06:00.0: firmware ver 10.4b-ct-9984-fH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 123549a3

Sep 03 18:54:11 u1804 hostapd[4175]: Configuration file: /etc/hostapd/hostapd.conf

cat /etc/hostapd/hostapd.conf
driver=nl80211
logger_syslog=127
logger_syslog_level=2
logger_stdout=127
[... 220 lines cut ...]
////hostapd.conf

interface=wlp6s0
bridge=br-lan
ctrl_interface=/var/run/hostapd
ap_isolate=1
bss_load_update_period=60
#chan_util_avg_period=600
disassoc_low_ack=1
preamble=1
wmm_enabled=1
ignore_broadcast_ssid=0
uapsd_advertisement_enabled=1
utf8_ssid=1
#multi_ap=0
wpa_passphrase=xxxxxxxx
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=test
#bridge=br-lan
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK
okc=0
disable_pmksa_caching=1

greearb, 2020-09-03:
Did it work better before on some earlier version of ath10k-ct driver/firmware?

gsustek, 2020-09-03:
it works stable on same ubuntu 20.04 and 
https://github.com/greearb/**linux-ct-5.4**/commits/master

dmesg -T|grep ath10
[Thu Sep  3 19:28:09 2020] ath10k 5.1 driver, optimized for CT firmware, probing pci device: 0x46.
[Thu Sep  3 19:28:09 2020] ath10k_pci 0000:06:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[Thu Sep  3 19:28:09 2020] ath10k_pci 0000:06:00.0: fwcfg key: regdom  val: 332
[Thu Sep  3 19:28:09 2020] ath10k_pci 0000:06:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[Thu Sep  3 19:28:09 2020] ath10k_pci 0000:06:00.0: kconfig debug 0 debugfs 1 tracing 1 dfs 0 testmode 0
[Thu Sep  3 19:28:09 2020] ath10k_pci 0000:06:00.0: firmware ver 10.4b-ct-9984-fH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 123549a3
[Thu Sep  3 19:28:10 2020] ath10k_pci 0000:06:00.0: board_file api 2 bmi_id 0:31 crc32 e807b522
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0: unsupported HTC service id: 1536
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0: DANGER! You're overriding EEPROM-defined regulatory domain
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0: from: 0x0 to 0x14c (svc-ready-work)
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0: Your card was not certified to operate in the domain you chose.
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0: This might result in a violation of your local regulatory rules.
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0: Do not ever do this unless you really know what you are doing!
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0: msdu-desc: 2500  skid: 32
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0: wmi print 'free: 84936 iram: 13572 sram: 11224'
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[Thu Sep  3 19:28:13 2020] ath10k_pci 0000:06:00.0 wlp6s0: renamed from wlan0
[Thu Sep  3 19:28:16 2020] ath10k_pci 0000:06:00.0: unsupported HTC service id: 1536
[Thu Sep  3 19:28:16 2020] ath10k_pci 0000:06:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[Thu Sep  3 19:28:16 2020] ath10k_pci 0000:06:00.0: msdu-desc: 2500  skid: 32
[Thu Sep  3 19:28:16 2020] ath10k_pci 0000:06:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[Thu Sep  3 19:28:16 2020] ath10k_pci 0000:06:00.0: wmi print 'free: 84936 iram: 13572 sram: 11224'
[Thu Sep  3 19:28:16 2020] ath10k_pci 0000:06:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[Thu Sep  3 19:28:17 2020] ath10k_pci 0000:06:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[Thu Sep  3 19:28:29 2020] ath10k_pci 0000:06:00.0: unsupported HTC service id: 1536
[Thu Sep  3 19:28:29 2020] ath10k_pci 0000:06:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[Thu Sep  3 19:28:29 2020] ath10k_pci 0000:06:00.0: msdu-desc: 2500  skid: 32
[Thu Sep  3 19:28:29 2020] ath10k_pci 0000:06:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[Thu Sep  3 19:28:29 2020] ath10k_pci 0000:06:00.0: wmi print 'free: 84936 iram: 13572 sram: 11224'
[Thu Sep  3 19:28:29 2020] ath10k_pci 0000:06:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[Thu Sep  3 19:29:37 2020] ath10k_pci 0000:06:00.0: Invalid peer id 0 or peer stats buffer, peer: 00000000768c8cf6  sta: 00000000b0f4f27f
[Thu Sep  3 20:32:44 2020] ath10k_pci 0000:06:00.0: Invalid VHT mcs 15 peer stats

gsustek, 2020-09-03:
wanna use 5.8 because it is the next LTS kernel

greearb, 2020-09-03:
Ok, so 5.4 ath10k-ct driver works fine, but the 5.8 version works poorly?  Can you try stock 5.8 kernel to see if it is specific to my driver or if 5.8 is broken in other ways?

greearb, 2020-09-03:
And, did you compile my 5.4 driver into your 5.8 kernel and that works OK?

gsustek, 2020-09-03:
> Ok, so 5.4 ath10k-ct driver works fine, but the 5.8 version works poorly? Can you try stock 5.8 kernel to see if it is specific to my driver or if 5.8 is broken in other ways?

Ok, so i tried almost every stock kernel driver from kernel 5.0 and up 5.8.5 with kvalo board.bin and latest firmware 114. Every version of 5 kernel is broken in such a way that maximal (up and down)-load is max about 40MB/s and there is 10% of ping loss. Configuration on ubuntu of 18.04 is the same as on 20.04 which is just regular upgrade on the same kernel version. The latest stable  ping is on ubuntu 18.04 , kernel 4.15.0-116.117 | Ubuntu-4.15.0-116.117 | 4.15.18 on stock driver. I compile on that same ubuntu 18.04 your linux-5.4-ct kernel which is also stable there and on migrated 20.04. so i tried to compile your last 5.8-ct kernel on ubuntu 20.04 with the same expectation for ping as on 5.4-ct kernel...but without of ping luck :-)

gsustek, 2020-09-03:
> And, did you compile my 5.4 driver into your 5.8 kernel and that works OK?

will try it now against your linux-5.8-ct kernel

gsustek, 2020-09-03:
can't compile..

please help.

`  CC [M]  /home/gs/ath10k-ct/tmp/ath10k.build/spectral.o
  CC [M]  /home/gs/ath10k-ct/tmp/ath10k.build/trace.o
In file included from ./include/trace/define_trace.h:102,
                 from /home/gs/ath10k-ct/tmp/ath10k.build/trace.h:537,
                 from /home/gs/ath10k-ct/tmp/ath10k.build/trace.c:9:
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:242:23: error: expected expression before ‘;’ token
  242 |   __field(u8, hw_type);
      |                       ^
./include/trace/trace_events.h:443:2: note: in definition of macro ‘DECLARE_EVENT_CLASS’
  443 |  tstruct        \
      |  ^~~~~~~
./include/trace/trace_events.h:78:9: note: in expansion of macro ‘PARAMS’
   78 |         PARAMS(tstruct),         \
      |         ^~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:234:1: note: in expansion of macro ‘TRACE_EVENT’
  234 | TRACE_EVENT(ath10k_wmi_dbglog,
      | ^~~~~~~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:239:2: note: in expansion of macro ‘TP_STRUCT__entry’
  239 |  TP_STRUCT__entry(
      |  ^~~~~~~~~~~~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:272:23: error: expected expression before ‘;’ token
  272 |   __field(u8, hw_type);
      |                       ^
./include/trace/trace_events.h:443:2: note: in definition of macro ‘DECLARE_EVENT_CLASS’
  443 |  tstruct        \
      |  ^~~~~~~
./include/trace/trace_events.h:78:9: note: in expansion of macro ‘PARAMS’
   78 |         PARAMS(tstruct),         \
      |         ^~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:264:1: note: in expansion of macro ‘TRACE_EVENT’
  264 | TRACE_EVENT(ath10k_htt_pktlog,
      | ^~~~~~~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:269:2: note: in expansion of macro ‘TP_STRUCT__entry’
  269 |  TP_STRUCT__entry(
      |  ^~~~~~~~~~~~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:272:23: error: initialization of ‘const char *’ from incompatible pointer type ‘struct trace_event_fields *’ [-Werror=incompatible-pointer-types]
  272 |   __field(u8, hw_type);
      |                       ^
./include/trace/trace_events.h:443:2: note: in definition of macro ‘DECLARE_EVENT_CLASS’
  443 |  tstruct        \
      |  ^~~~~~~
./include/trace/trace_events.h:78:9: note: in expansion of macro ‘PARAMS’
   78 |         PARAMS(tstruct),         \
      |         ^~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:264:1: note: in expansion of macro ‘TRACE_EVENT’
  264 | TRACE_EVENT(ath10k_htt_pktlog,
      | ^~~~~~~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:269:2: note: in expansion of macro ‘TP_STRUCT__entry’
  269 |  TP_STRUCT__entry(
      |  ^~~~~~~~~~~~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:272:23: note: (near initialization for ‘trace_event_fields_ath10k_htt_pktlog[3].type’)
  272 |   __field(u8, hw_type);
      |                       ^
./include/trace/trace_events.h:443:2: note: in definition of macro ‘DECLARE_EVENT_CLASS’
  443 |  tstruct        \
      |  ^~~~~~~
./include/trace/trace_events.h:78:9: note: in expansion of macro ‘PARAMS’
   78 |         PARAMS(tstruct),         \
      |         ^~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:264:1: note: in expansion of macro ‘TRACE_EVENT’
  264 | TRACE_EVENT(ath10k_htt_pktlog,
      | ^~~~~~~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:269:2: note: in expansion of macro ‘TP_STRUCT__entry’
  269 |  TP_STRUCT__entry(
      |  ^~~~~~~~~~~~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:272:23: error: initializer element is not constant
[... 102 lines cut ...]
  443 |  tstruct        \
      |  ^~~~~~~
./include/trace/trace_events.h:78:9: note: in expansion of macro ‘PARAMS’
   78 |         PARAMS(tstruct),         \
      |         ^~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:430:1: note: in expansion of macro ‘TRACE_EVENT’
  430 | TRACE_EVENT(ath10k_htt_rx_desc,
      | ^~~~~~~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:435:2: note: in expansion of macro ‘TP_STRUCT__entry’
  435 |  TP_STRUCT__entry(
      |  ^~~~~~~~~~~~~~~~
./include/trace/trace_events.h:442:64: warning: missing braces around initializer [-Wmissing-braces]
  442 | static struct trace_event_fields trace_event_fields_##call[] = { \
      |                                                                ^
./include/trace/trace_events.h:75:2: note: in expansion of macro ‘DECLARE_EVENT_CLASS’
   75 |  DECLARE_EVENT_CLASS(name,          \
      |  ^~~~~~~~~~~~~~~~~~~
/home/gs/ath10k-ct/tmp/ath10k.build/./trace.h:430:1: note: in expansion of macro ‘TRACE_EVENT’
  430 | TRACE_EVENT(ath10k_htt_rx_desc,
      | ^~~~~~~~~~~
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:281: /home/gs/ath10k-ct/tmp/ath10k.build/trace.o] Error 1
make: *** [Makefile:1756: /home/gs/ath10k-ct/tmp/ath10k.build] Error 2
make: Leaving directory '/home/gs/linux-ct-5.8'
gs@u`

greearb, 2020-09-03:
Sorry, I don't have time to work on that right now.  Maybe you can try the 5.7 driver and see if it will compile?

gsustek, 2020-09-03:
5.7 is compiling. sou should i try with that?

greearb, 2020-09-03:
yes

gsustek, 2020-09-04:
ok, i did some testing on your 5.8.0-ct kernel with wifi driver version ct5.7 and ct5.4.
On version 5.7, ping is terrible as on 5.8. With driver 5.4 and kernel 5.4 it is much better but not as 5.4 on 5.4 kernel which is flawless.
So on 5.4 i got fair dowload and upload with 5% ping loss.

dmesg of driver 5.4-ct on 5.8-ct kernel:
 [Fri Sep  4 19:49:24 2020] ath10k_core: loading out-of-tree module taints kernel.
[Fri Sep  4 19:49:24 2020] ath10k_core: module verification failed: signature and/or required key missing - tainting kernel
[Fri Sep  4 19:49:24 2020] ath10k 5.4 driver, optimized for CT firmware, probing pci device: 0x46.
[Fri Sep  4 19:49:24 2020] ath10k_mac_create, priv_size: 1624  hw: 000000000915a888  hw->priv: 0000000032d29ddd
[Fri Sep  4 19:49:24 2020] ath10k_pci 0000:06:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[Fri Sep  4 19:49:24 2020] ath10k_pci 0000:06:00.0: fwcfg key: regdom  val: 332
[Fri Sep  4 19:49:24 2020] ath10k_pci 0000:06:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[Fri Sep  4 19:49:24 2020] ath10k_pci 0000:06:00.0: kconfig debug 0 debugfs 1 tracing 1 dfs 0 testmode 0
[Fri Sep  4 19:49:24 2020] ath10k_pci 0000:06:00.0: firmware ver 10.4b-ct-9984-fH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 123549a3
[Fri Sep  4 19:49:25 2020] ath10k_pci 0000:06:00.0: board_file api 2 bmi_id 0:31 crc32 e807b522
[Fri Sep  4 19:49:28 2020] ath10k_pci 0000:06:00.0: unsupported HTC service id: 1536
[Fri Sep  4 19:49:28 2020] ath10k_pci 0000:06:00.0: DANGER! You're overriding EEPROM-defined regulatory domain
[Fri Sep  4 19:49:28 2020] ath10k_pci 0000:06:00.0: from: 0x0 to 0x14c (svc-ready-work)
[Fri Sep  4 19:49:28 2020] ath10k_pci 0000:06:00.0: Your card was not certified to operate in the domain you chose.
[Fri Sep  4 19:49:28 2020] ath10k_pci 0000:06:00.0: This might result in a violation of your local regulatory rules.
[Fri Sep  4 19:49:28 2020] ath10k_pci 0000:06:00.0: Do not ever do this unless you really know what you are doing!
[Fri Sep  4 19:49:28 2020] ath10k_pci 0000:06:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[Fri Sep  4 19:49:28 2020] ath10k_pci 0000:06:00.0: msdu-desc: 2500  skid: 32
[Fri Sep  4 19:49:28 2020] ath10k_pci 0000:06:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[Fri Sep  4 19:49:28 2020] ath10k_pci 0000:06:00.0: wmi print 'free: 84936 iram: 13572 sram: 11224'
[Fri Sep  4 19:49:28 2020] ath10k_pci 0000:06:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal otp max-sta 32 raw 0 hwcrypto 1
[Fri Sep  4 19:49:29 2020] ath10k_pci 0000:06:00.0 wlp6s0: renamed from wlan0
[Fri Sep  4 19:49:31 2020] ath10k_pci 0000:06:00.0: unsupported HTC service id: 1536
[Fri Sep  4 19:49:31 2020] ath10k_pci 0000:06:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[Fri Sep  4 19:49:31 2020] ath10k_pci 0000:06:00.0: msdu-desc: 2500  skid: 32
[Fri Sep  4 19:49:31 2020] ath10k_pci 0000:06:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[Fri Sep  4 19:49:31 2020] ath10k_pci 0000:06:00.0: wmi print 'free: 84936 iram: 13572 sram: 11224'
[Fri Sep  4 19:49:32 2020] ath10k_pci 0000:06:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[Fri Sep  4 19:49:32 2020] ath10k_pci 0000:06:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[Fri Sep  4 19:49:41 2020] ath10k_pci 0000:06:00.0: unsupported HTC service id: 1536
[Fri Sep  4 19:49:41 2020] ath10k_pci 0000:06:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[Fri Sep  4 19:49:41 2020] ath10k_pci 0000:06:00.0: msdu-desc: 2500  skid: 32
[Fri Sep  4 19:49:41 2020] ath10k_pci 0000:06:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[Fri Sep  4 19:49:41 2020] ath10k_pci 0000:06:00.0: wmi print 'free: 84936 iram: 13572 sram: 11224'
[Fri Sep  4 19:49:42 2020] ath10k_pci 0000:06:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[Fri Sep  4 19:49:52 2020] ath10k_pci 0000:06:00.0: Invalid peer id 0 or peer stats buffer, peer: 00000000c5e13b55  sta: 0000000000000000
[Fri Sep  4 19:50:08 2020] ath10k_pci 0000:06:00.0: wmi: fixing invalid VHT TX rate code 0xff

greearb, 2020-09-04:
What works flawless?  Can you find the smallest change between flawless and not so flawless?

gsustek, 2020-09-05:
ok, so for me flawless is ping 100% success and upload & download at least 70MB/s.
5.4.0-ct linux - flawless
5.8.0- ct- linux - 5.4-ct driver 3%-5%  loss in ping, ok up/down
5.8.0- ct- linux - 5.8-ct driver terrible ping, up&down

greearb, 2020-09-05:
Maybe just stay with 5.4 now.  We'll do some testing on our 5.8 (or maybe 5.9) kernel as part of this dev cycle and maybe we can find the issues.  If you feel like poking, try disabling ATF and related features and see if that improves things?

gsustek, 2020-09-05:
Yes, i'm open for testing, eg. poking. 
How can i control airtime fairness on QCA9984? please advise.
Thnx, 
Goran.

gsustek, 2020-09-06:
ok, i figure out, compile hostapd-ct with airtime config enabled. I  disable ATF on kernel 5.8 with driver 5.4, but ping is still terrible, 5% of loss.
Any other suggestion?

greearb, 2020-09-08:
See bug 153 and try to revert that driver change related to DMA and see if that helps?