greearb/ath10k-ct #153: Issues with latest commit

julianocss, 2020-09-03:
Hello @greearb 

I am testing the commit 13319ff0b5dc3c86e8313f49fca583b829878736 for one week.
My router (Arcer C60 V2, QCA9886) became instable after install this commit.
On clients side, the connection drops for a few seconds (10-30s) and then is reestablished.
This happens several times a day.
The router log did not help much.

Was anything changed in this commit that could be causing this?
If you need some tests do debug, please, let me know.

Juliano

greearb, 2020-09-03:
What driver are you actually using out of there?  That tree holds drivers for many different kernels.  If you are trying the 5.8 version, then it is almost completely untested and may be full of bugs.  Earlier kernels should be better tested.  Did you update other things at the same time (like openwrt itself?)

julianocss, 2020-09-03:
I´m using 5.4 driver. Yes, i update Openwrt. 
But on the same day I also updated my other router (ramips, mt76 driver) and I had no problems with it.

Juliano

greearb, 2020-09-03:
If you go back to an earlier ath10k-ct driver commit, does it get better?  Updating openwrt would probably also update the ath10k-ct firmware, you could try going to an earlier version of that as well?

julianocss, 2020-09-03:
I will revert to commit edfbf9161ab75e4831981284ab2d5b1b9fb6ef72 and test for 24h.
Tomorrow i let you know the results.

Thanks!

julianocss, 2020-09-06:
Hello @greearb.

I reverted to commit edfbf9161ab75e4831981284ab2d5b1b9fb6ef72 and the router is stable again.
I used the same version of OpenWRT to test.
Is there anything that i can do to help debug this?

Cheers.
Juliano

greearb, 2020-09-06:
That's quite interesting.  Just to confirm, you reverted that and only that, and things go better?

julianocss, 2020-09-07:
Hi @greearb.

Yes, only ct-driver.
OpenWRT version remained the same, hardware too.
For now, i will continue using commit edfbf9161ab75e4831981284ab2d5b1b9fb6ef72.
if you need me to test something else, let me know.

Regards,
Juliano

greearb, 2020-09-07:
My guess is that this patch may help some radios, but certainly not all.  Can you past dmesg of your driver boot, I'll try to add a patch to disable that bad commit for your particular chipset.

julianocss, 2020-09-07:
Hi @greearb.

Yes, sure.
I sent dmesg output to your email.

Juliano

greearb, 2020-09-08:
Hello, can you upgrade your ath10k-ct owrt makefile to use this commit id and the 5.4 driver
0c2949e3abd7d2811b001bd35961c7e0a6d08b06
and then create a fwcfg file and set dma_burst = 0
and make sure that resolves the issue?  I can help with the fwcfg details as needed.

julianocss, 2020-09-08:
Hi Ben.

Thanks for the patch.
I will test it tomorrow. And i will need help with fwcfg.

Juliano

greearb, 2020-09-08:
Please see if this is enough info to make fwcfg work for you:
To start with, don't add any options but the dma_burst setting.  You can play with
others later as desired.
http://www.candelatech.com/ath10k-10.4.php#config
search for dma_burst down that link a bit.

julianocss, 2020-09-08:
Hi Ben.

That´s the idea?
Or should I delete de .bin file?

Juliano

[image]

greearb, 2020-09-08:
that looks right, don't delete anything!

julianocss, 2020-09-09:
Done.

I will test it tomorrow.

Thanks.
Juliano

julianocss, 2020-09-10:
Hi @greearb.

Should i close the issue?
Or leave it open for a few more days?

Juliano

greearb, 2020-09-10:
I think the best fix is to somehow have owrt use hotplug logic to automatically create that fwcfg file line on your particular platform (and I assume others that need it).  Please leave this open a bit...

Fail-Safe, 2020-09-13:
@greearb I have tried modifying the dma_burst value on my R7800 (QCA9884), but it does not appear to be accepted for me:

```
root@OpenWrt:~# dmesg | grep firmware
[    2.779343] qcom_rpm 108000.rpm: RPM firmware 3.0.16777364
[   14.166673] ath10k 5.4 driver, optimized for CT firmware, probing pci device: 0x46.
[   14.736516] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-tH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7d16c2a5
[   23.263975] ath10k 5.4 driver, optimized for CT firmware, probing pci device: 0x46.
[   23.684504] ath10k_pci 0001:01:00.0: firmware ver 10.4b-ct-9984-tH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7d16c2a5
```

```
root@OpenWrt:~# dmesg | grep dma
[   14.345248] ath10k_pci 0000:01:00.0: fwcfg key: dma_burst  val: 0
[   14.345275] ath10k_pci 0000:01:00.0: Unknown fwcfg key name -:dma_burst:-, val: 0
[   23.474619] ath10k_pci 0001:01:00.0: fwcfg key: dma_burst  val: 0
[   23.474644] ath10k_pci 0001:01:00.0: Unknown fwcfg key name -:dma_burst:-, val: 0
```

```
root@OpenWrt:~# cat /lib/firmware/ath10k/fwcfg-pci-0000:01:00.0.txt
dma_burst = 0
root@OpenWrt:~# cat /lib/firmware/ath10k/fwcfg-pci-0001:01:00.0.txt
dma_burst = 0
```

Am I missing anything or should this be working?

greearb, 2020-09-13:
You have to patch the Makefile in ath10k-ct driver to have it pull down the latest commit, it is not in owrt yet.  I'll send a pull request for that next week.

Fail-Safe, 2020-09-13:
Sorry--I missed that step in the conversation above. Just ran a new build and confirmed the `dma_burst` setting was accepted now. Thanks for the quick reply!

Time for testing now...

julianocss, 2020-09-15:
Hi @greearb.

I just checked the Candelatech download page and found a different file: 
firmware-5-ct-htt-mgt-community-qcache-12.bin-lede.020 

What is the difference of this file? Can I test on my router? Is there any place that I could find this information without having to bother you?

Thanks,
Juliano

greearb, 2020-09-15:
I updated the firmware links and descriptions in the beta downloads section, let me know if that is clear enough?

http://www.candelatech.com/ath10k-10.4.php

julianocss, 2020-09-15:
Hi.

Thanks for the update.

A doubt:
"These htt-mgt firmwares require a driver patch. See release notes."

Which patch are you referring to? Isn't it enough that I use the correct .bin and select the driver in OpenWRT?

Juliano

greearb, 2020-09-15:
If you use openwrt's ath10k-ct then it has all needed patches and will work fine.  Stock upstream kernels w/out ath10k-ct will need patching.

julianocss, 2020-09-16:
Hi @greearb.

There is a problem with the latest .bin files generated in Candelatech site.
My 5Ghz radio does not start when i use this file.

I tried to revert to "firmware-5-ct-htt-mgt-community-12.bin-lede.019 (downloaded now)", and also does not worked.
The only way to start 5Ghz radio again was replacing the firmware-5.bin file from my previous OpenWRT build.

I noticed that the current file size is very different.

Old (019): 555k
New (019): 477k

Can you take a look?

Regards,
Juliano

greearb, 2020-09-16:
This sounds like a new issue, please open a new bug, and include dmesg and such.

Ansuel, 2020-09-16:
Wonder if this value can be passed as a module args too so we can skip the creation of the txt file