greearb/ath10k-ct #155: Unable to compile ath10k-ct for OpenWrt

ahoppen, 2020-09-07:
I am trying to compile a recent-ish version of ath10k-ct for OpenWrt 19.07 in order to backport https://github.com/greearb/ath10k-ct/pull/119 to 19.07 so I can use dynamic VLANs on TP-Link Archer C7 v5 devices (please correct me if my assumption that this should work is wrong).

For this I have compiled OpenWrt 19.07.2 and am now trying to compile the `master` version of ath10k-ct. However, AFAICT nothing is being built (at least the build command finishes in < 1 second and I can’t find any output files).

Could someone give me a hint what I might be doing wrong? I assume it’s something fairly trivial but I’m unable to find it.

```
$ ./build_me.sh 
Enter your kernel's build directory:
/home/alex/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.171
Enter kernel version to build for: 4.4, 4.7, 4.9, 4.13  (or enter for default which is 4.7 currently):
4.13
 * These flags are used with the ampdu_action() callback in
	 * @ampdu_action:
	int (*ampdu_action)(struct ieee80211_hw *hw,
 * @RX_FLAG_ONLY_MONITOR: Report frame only to monitor interfaces without
	RX_FLAG_ONLY_MONITOR		= BIT(17),
Running tmp/ath10k.build/build_me.sh
make: Entering directory '/home/alex/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.171'
  Building modules, stage 2.
  MODPOST 0 modules
make: Leaving directory '/home/alex/openwrt/build_dir/target-mips_24kc_musl/linux-ar71xx_generic/linux-4.14.171'
```

greearb, 2020-09-08:
This is wired into owrt build system already, please ask on owrt mailing list and/or check update to Makefile in openwrt.

closed 2020-09-08 by greearb

ahoppen, 2020-09-14:
Thanks for the reply! For anyone hitting this, I managed to get ath10k-ct building by pointing the OpenWrt build to a different commit hash of ath10k-ct.