greearb/ath10k-ct #156: firmware ver 10.1-ct-8x-__fW-022-538f0906 crash repeatedly.

real-t0mg, 2020-09-07:
CT firmware keeps crashing shortly after fresh boot. Especially bad when connecting ASUS TUF-506 laptop with RealtekRTL8822CE 802.11ac PCIe Wireless Network Adapter. 

hardware: TP-LINK Archer C7 V5
Firmware: OpenWRT 19.07 
[ath10k_pci_keeps_crashing.txt](https://github.com/greearb/ath10k-ct/files/5184220/ath10k_pci_keeps_crashing.txt)

I can compile my own openwrt, what can I do to help debug this?

real-t0mg, 2020-09-07:
Access point harwdare details:

`root@netcholo:~# date` 
`Mon Sep  7 17:44:00 UTC 2020`
`root@netcholo:~# ethtool -i wlan0`
`driver: ath10k_pci`
`version: 4.14.195`
`firmware-version: 10.1-ct-8x-__fW-022-538f0906`
`expansion-rom-version: `
`bus-info: 0000:00:00.0`
`supports-statistics: yes`
`supports-test: no`
`supports-eeprom-access: no`
`supports-register-dump: no`
`supports-priv-flags: no`

client hardware: 

`root@tufboxen:~# date `
`Mon Sep  7 14:44:12 -03 2020`
`root@tufboxen:~# ethtool -i wlan0`
`driver: rtw_8822ce`
`version: 5.8.7-ZEN`
`firmware-version: N/A`
`expansion-rom-version:` 
`bus-info: 0000:03:00.0`
`supports-statistics: yes`
`supports-test: no`
`supports-eeprom-access: no`
`supports-register-dump: no`
`supports-priv-flags: no`

refer to attached dmesg log for more info. 
[ath10kdmesg.txt](https://github.com/greearb/ath10k-ct/files/5184379/ath10kdmesg.txt)

real-t0mg, 2020-09-07:
more info, another crash within minutes of my last update: 

> [ 2735.714295] ath10k_pci 0000:00:00.0: failed to submit frame: -19
> [ 2735.714305] ath10k_pci 0000:00:00.0: failed to push frame: -19
> [ 7114.711577] ath10k_pci 0000:00:00.0: firmware crashed! (guid n/a)
> [ 7114.717938] ath10k_pci 0000:00:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
> [ 7114.727482] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
> [ 7114.739964] ath10k_pci 0000:00:00.0: firmware ver 10.1-ct-8x-__fW-022-538f0906 api 2 features wmi-10.x,has-wmi-mgmt-tx,mfp,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,get-temp-CT,tx-rc-CT,cust-stats-CT,retry-gt2-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT crc32 e1c91a74
> [ 7114.771312] ath10k_pci 0000:00:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
> [ 7114.778874] ath10k_pci 0000:00:00.0: htt-ver 2.1 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
> [ 7114.790718] ath10k_pci 0000:00:00.0: firmware register dump:
> [ 7114.796575] ath10k_pci 0000:00:00.0: [00]: 0x4100016C 0x00000000 0x00993237 0x0059DF69
> [ 7114.804764] ath10k_pci 0000:00:00.0: [04]: 0x00993237 0x00060530 0x0000001B 0x00003791
> [ 7114.812955] ath10k_pci 0000:00:00.0: [08]: 0x00000001 0x0059DEAD 0x004138C5 0x00000001
> [ 7114.821145] ath10k_pci 0000:00:00.0: [12]: 0x0000000D 0x00000000 0x00958360 0x0095836B
> [ 7114.829333] ath10k_pci 0000:00:00.0: [16]: 0x1214FC40 0x06C80000 0x00000004 0xFFFFFFFF
> [ 7114.837524] ath10k_pci 0000:00:00.0: [20]: 0x0040ED68 0x000000E6 0x0040AD14 0x00413894
> [ 7114.845713] ath10k_pci 0000:00:00.0: [24]: 0x8099276C 0x0040AD84 0x004390F8 0x00430EB0
> [ 7114.853909] ath10k_pci 0000:00:00.0: [28]: 0x00000002 0x00000000 0x00430F90 0x004390F8
> [ 7114.862103] ath10k_pci 0000:00:00.0: [32]: 0x000001D2 0x00000008 0x00000094 0x00000000
> [ 7114.870291] ath10k_pci 0000:00:00.0: [36]: 0x00000001 0x00000000 0x009BDAA0 0x00430F90
> [ 7114.878491] ath10k_pci 0000:00:00.0: [40]: 0x004138C4 0x0040AD84 0x00000000 0x0040ADEC
> [ 7114.886681] ath10k_pci 0000:00:00.0: [44]: 0x0040EDA0 0x0040D0B8 0xFFFFFFFF 0x0041F194
> [ 7114.894867] ath10k_pci 0000:00:00.0: [48]: 0x809927B8 0x0040ADA4 0x00000000 0x0040C390
> [ 7114.903056] ath10k_pci 0000:00:00.0: [52]: 0x004138C4 0x00000001 0x00000000 0x00000001
> [ 7114.911246] ath10k_pci 0000:00:00.0: [56]: 0x80999C9D 0x0040AEC4 0x0041F194 0x00000000
> [ 7114.919432] ath10k_pci 0000:00:00.0: Copy Engine register dump:
> [ 7114.925569] ath10k_pci 0000:00:00.0: [00]: 0x00057400   2   2   3   3
> [ 7114.932237] ath10k_pci 0000:00:00.0: [01]: 0x00057800  18  18 404 405
> [ 7114.938911] ath10k_pci 0000:00:00.0: [02]: 0x00057c00  56  56 119 120
> [ 7114.945576] ath10k_pci 0000:00:00.0: [03]: 0x00058000  11  11  15  11
> [ 7114.952243] ath10k_pci 0000:00:00.0: [04]: 0x00058400 8011 8011 207 167
> [ 7114.959089] ath10k_pci 0000:00:00.0: [05]: 0x00058800  24  24 503 504
> [ 7114.965754] ath10k_pci 0000:00:00.0: [06]: 0x00058c00  13  13  13  13
> [ 7114.972420] ath10k_pci 0000:00:00.0: [07]: 0x00059000   0   0   0   0
> [ 7114.981119] ath10k_pci 0000:00:00.0: debug log header, dbuf: 0x412708  dropped: 0
> [ 7114.989877] ath10k_pci 0000:00:00.0: [0] next: 0x412720 buf: 0x41056c sz: 1500 len: 28 count: 1 free: 0
> [ 7115.000617] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
> [ 7115.007007] ath10k: [0000]: 94416F00 0100FC17 00000000 F61D0000 B4AB4000 6C010041 03000000
> [ 7115.015553] ath10k_pci 0000:00:00.0: ATH10K_END
> [ 7115.021265] ath10k_pci 0000:00:00.0: [1] next: 0x412708 buf: 0x410b5c sz: 1500 len: 0 count: 0 free: 0
> [ 7115.033929] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 8729e200 vdev: 0 addr: 70:66:55:57:58:73 
> [ 7115.045302] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 85d8da00 vdev: 0 addr: cc:9f:7a:aa:cb:84 
> [ 7115.056640] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 85d98800 vdev: 0 addr: 54:60:09:d7:3c:16 
> [ 7115.067970] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 85d98400 vdev: 0 addr: 30:f7:72:6b:33:bf 
> [ 7115.079303] ath10k_pci 0000:00:00.0: removing peer, cleanup-all, deleting: peer 86d0fa00 vdev: 0 addr: cc:32:e5:ed:a3:ca 
> [ 7115.174488] ieee80211 phy0: Hardware restart was requested
> [ 7116.161562] ath10k_pci 0000:00:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
> [ 7116.180079] ath10k_pci 0000:00:00.0: wmi print 'P 128 V 8 T 410'
> [ 7116.186320] ath10k_pci 0000:00:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0 ct-sta: 0'
> [ 7116.195041] ath10k_pci 0000:00:00.0: wmi print 'alloc rem: 20984 iram: 25656'
> [ 7116.269319] ath10k_pci 0000:00:00.0: pdev param 0 not supported by firmware
> [ 7116.296126] ath10k_pci 0000:00:00.0: device successfully recovered

real-t0mg, 2020-09-10:
the crashes are getting more frequent and when the firmware "recovers", the connection is slower. It's very frustrating using an ath10k for production. Let me know what I may test to help. Attached dmesg register's messages when failing. 
[ath10k_fails.txt](https://github.com/greearb/ath10k-ct/files/5203475/ath10k_fails.txt)

greearb, 2020-09-10:
hello, first crash I decoded looks a bit weird, so I'll likely need to add debugging.  I'll look at other crashes to see if they are the same and attach an image for more debugging soon.

greearb, 2020-09-13:
If you try older releases (from a month ago or so), does that work any better?

real-t0mg, 2020-09-14:
hello, it did happened with older releases too, thats why I kept updating each time I saw new Candela tech firmware was added on the Openwrt commits. 

The only thing I've been able to prove is that combination of  Realtek RTL8822CE 802.11ac PCIe Wireless Network Adapter (kernel driver: rtw_8822ce) with ath10k (candelatech firmw or standard firmware) is fatal on my all of my tp-link Archer C7 V5. The experience was so bad that I opened another bug on realtek [realtek rtl8822 bug](https://github.com/rtlwifi-linux/rtk_wifi_driver_rtl8822ce/issues/3)
 
I cannot make ath10k crash with my iwilfi (Intel AC wifi) laptops or with my android phones. The only fix I've found is to have the laptop with the rtw_8822ce wifi driver connect to an AR71XX based router (Openwrt 19.07.4 running on TP-LINK 3500) instead of ATH79 based build. 
[image]

greearb, 2020-09-16:
Do you happen to have some wave-2 ath10k AP that you can test with?  I'm curious if it has similar issues or not...  Any idea if that realtek wifi NIC works on Linux enough to trigger the same problem?

real-t0mg, 2020-09-16:
hi @greearb, I dont have any wave ath10k hardware to validate if it triggers the same crash. The network card is supported in linux for some time now and works fine with ath9k based access points. Its the interaction between the kernel module rtw88_8822ce and ath10k module with OEM or CandelaTech firmware that is fatal for me. I made my 3 TP-LINK Archer C7 V5 crash the same way, just by having the Asus laptop with the Network controller: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter connect to the ath10k based AP. Guaranteed 5GHz pandemonium.

greearb, 2020-09-16:
Can you point me to the specific model number of your NIC?  I'd like to get one to see if I can reproduce the problem locally.

real-t0mg, 2020-09-16:
Network card: 

```
Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter
AzureWave Device 3750
```

as found in this laptop: 

```
 Manufacturer: ASUSTeK COMPUTER INC.
 Product Name: TUF Gaming FA506IU_TUF506IU
```

more info: https://www.asus.com/us/supportonly/TUF506IU/HelpDesk_Manual/

lspci -vv: 

```
03:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter                                                                                                                                                                                                               
        Subsystem: AzureWave Device 3750                                                                                                                                                                                                                                                                                   
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+                                                                                                                                                                                                              
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-                                                                                                                                                                                                               
        Latency: 0, Cache Line Size: 64 bytes                                                                                                                                                                                                                                                                              
        Interrupt: pin A routed to IRQ 81                                                                                                                                                                                                                                                                                  
        IOMMU group: 11                                                                                                                                                                                                                                     
        Region 0: I/O ports at d000 [size=256]                                                                                                                                                                                                                                                                             
        Region 2: Memory at fc800000 (64-bit, non-prefetchable) [size=64K]                                                                                                                                                                                                                                                 
        Capabilities: [40] Power Management version 3                                                                                                                                                                                                                                                                      
                Flags: PMEClk- DSI- D1+ D2+ AuxCurrent=375mA PME(D0+,D1+,D2+,D3hot+,D3cold+)                                                                                                                                                                                                                               
                Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-                                                                                                                                                                                                                                                     
        Capabilities: [50] MSI: Enable+ Count=1/1 Maskable- 64bit+                                                                                                                                                                                                                                                         
                Address: 00000000fee00000  Data: 0000                                                                                                                                                                                                                                                                      
        Capabilities: [70] Express (v2) Endpoint, MSI 00                                                                                                                                                                                                                                                                   
                DevCap: MaxPayload 128 bytes, PhantFunc 0, Latency L0s <4us, L1 <64us                                                                                                                                                                                                                                      
[... cut ...]                                                                                                      
        Kernel driver in use: rtw_8822ce                                                                                                                                                                                                                                                                                   
        Kernel modules: rtw88_8822ce                                          
```

real-t0mg, 2020-09-17:
freash reboot, connect asus laptop, immediate crash. Hopefully there's some register info there that is useful to get this addressed. Let me know how I can help. 

[more_crash.txt](https://github.com/greearb/ath10k-ct/files/5242146/more_crash.txt)

greearb, 2020-09-18:
Please try this firmware.  It has some extra debugging that may help me understand the issue.
[firmware-2-full-community.bin.gz](https://github.com/greearb/ath10k-ct/files/5245911/firmware-2-full-community.bin.gz)

Send me the dmesg when it crashes...

EduApps-CDG, 2022-03-01:
@greearb I just want to share that I'm having this issue in 2022:

<details>
<pre><font color="#00AA00"><b>[eduardo]:</b></font><font color="#0000AA"><u style="text-decoration-style:single"><b>~</b></u></font><font color="#AA5500"><b>$</b></font> lspci -vv
00:00.0 Host bridge: Intel Corporation Gemini Lake Host Bridge (rev 03)
	Subsystem: ASUSTeK Computer Inc. Gemini Lake Host Bridge
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast &gt;TAbort- &lt;TAbort- &lt;MAbort- &gt;SERR- &lt;PERR- INTx-
	Latency: 0

00:00.1 Signal processing controller: Intel Corporation Celeron/Pentium Silver Processor Dynamic Platform and Thermal Framework Processor Participant (rev 03)
	Subsystem: ASUSTeK Computer Inc. Celeron/Pentium Silver Processor Dynamic Platform and Thermal Framework Processor Participant
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B+ ParErr- DEVSEL=fast &gt;TAbort- &lt;TAbort- &lt;MAbort- &gt;SERR- &lt;PERR- INTx+
	Latency: 0
	Interrupt: pin B routed to IRQ 24
	Region 0: Memory at 80000000 (64-bit, non-prefetchable) [size=32K]
	Capabilities: &lt;access denied&gt;
	Kernel driver in use: proc_thermal
	Kernel modules: processor_thermal_device

00:00.3 System peripheral: Intel Corporation Celeron/Pentium Silver Processor Gaussian Mixture Model (rev 03)
	Subsystem: ASUSTeK Computer Inc. Celeron/Pentium Silver Processor Gaussian Mixture Model
	Control: I/O- Mem- BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast &gt;TAbort- &lt;TAbort- &lt;MAbort- &gt;SERR- &lt;PERR- INTx-
	Interrupt: pin A routed to IRQ 23
	Region 0: Memory at a1324000 (64-bit, non-prefetchable) [disabled] [size=4K]
	Capabilities: &lt;access denied&gt;

00:02.0 VGA compatible controller: Intel Corporation GeminiLake [UHD Graphics 600] (rev 03) (prog-if 00 [VGA controller])
	DeviceName: VGA
	Subsystem: ASUSTeK Computer Inc. GeminiLake [UHD Graphics 600]
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast &gt;TAbort- &lt;TAbort- &lt;MAbort- &gt;SERR- &lt;PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 126
	Region 0: Memory at a0000000 (64-bit, non-prefetchable) [size=16M]
	Region 2: Memory at 90000000 (64-bit, prefetchable) [size=256M]
	Region 4: I/O ports at f000 [size=64]
	Expansion ROM at 000c0000 [virtual] [disabled] [size=128K]
	Capabilities: &lt;access denied&gt;
	Kernel driver in use: i915
	Kernel modules: i915

00:0e.0 Audio device: Intel Corporation Celeron/Pentium Silver Processor High Definition Audio (rev 03)
	Subsystem: ASUSTeK Computer Inc. Celeron/Pentium Silver Processor High Definition Audio
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast &gt;TAbort- &lt;TAbort- &lt;MAbort- &gt;SERR- &lt;PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 128
	Region 0: Memory at a1310000 (64-bit, non-prefetchable) [size=16K]
	Region 4: Memory at a1200000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: &lt;access denied&gt;
	Kernel driver in use: snd_hda_intel
	Kernel modules: snd_hda_intel, snd_soc_skl, snd_sof_pci_intel_apl

00:0f.0 Communication controller: Intel Corporation Celeron/Pentium Silver Processor Trusted Execution Engine Interface (rev 03)
	Subsystem: ASUSTeK Computer Inc. Celeron/Pentium Silver Processor Trusted Execution Engine Interface
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast &gt;TAbort- &lt;TAbort- &lt;MAbort- &gt;SERR- &lt;PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 127
	Region 0: Memory at a1323000 (64-bit, non-prefetchable) [size=4K]
	Capabilities: &lt;access denied&gt;
	Kernel driver in use: mei_me
	Kernel modules: mei_me

00:12.0 SATA controller: Intel Corporation Celeron/Pentium Silver Processor SATA Controller (rev 03) (prog-if 01 [AHCI 1.0])
	Subsystem: ASUSTeK Computer Inc. Celeron/Pentium Silver Processor SATA Controller
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
[... 133 lines cut ...]
	Subsystem: ASUSTeK Computer Inc. Celeron/Pentium Silver Processor Gaussian Mixture Model
	Control: I/O+ Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap- 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium &gt;TAbort- &lt;TAbort- &lt;MAbort- &gt;SERR- &lt;PERR- INTx-
	Interrupt: pin A routed to IRQ 20
	Region 0: Memory at a1316000 (64-bit, non-prefetchable) [size=256]
	Region 4: I/O ports at f040 [size=32]
	Kernel driver in use: i801_smbus
	Kernel modules: i2c_i801

02:00.0 Network controller: Qualcomm Atheros QCA9377 802.11ac Wireless Network Adapter (rev 31)
	DeviceName: WLAN
	Subsystem: Lite-On Communications Inc QCA9377 802.11ac Wireless Network Adapter
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast &gt;TAbort- &lt;TAbort- &lt;MAbort- &gt;SERR- &lt;PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 129
	Region 0: Memory at a1000000 (64-bit, non-prefetchable) [size=2M]
	Capabilities: &lt;access denied&gt;
	Kernel driver in use: ath10k_pci
	Kernel modules: ath10k_pci

<font color="#00AA00"><b>[eduardo]:</b></font><font color="#0000AA"><u style="text-decoration-style:single"><b>~</b></u></font><font color="#AA5500"><b>$</b></font> 

</pre>
</details>