greearb/ath10k-ct #159: Documentation clarification (and sp fix), plz

systemcrash, 2020-09-17:
https://www.candelatech.com/ath10k-ug.php

Description of the problem (how to configure, how to reproduce, how often it happens).

Seems like there is a problem with formatting under this heading:

Set multicast, broadcast, beacon tx rates.

Contradiction in docu above

e.g.

Which one is actually disable? 
```
# Enable AMSDU for IBSS connections on wave-1 and wave-2.
echo 0x500000001 > /sys/kernel/debug/ieee80211/phy0/ath10k/ct_special

# Enable AMSDU for IBSS connections on wave-1 and wave-2 (default)
echo 0x500000000 > /sys/kernel/debug/ieee80211/phy0/ath10k/ct_special
```

which one should disable? Or same string works for both, as a toggle?
```
# Enable COLD resets in tx-hang work-around.
echo 0x1100000001 > /sys/kernel/debug/ieee80211/phy0/ath10k/ct_special

# Disable COLD resets in tx-hang work-around.
echo 0x1100000001 > /sys/kernel/debug/ieee80211/phy0/ath10k/ct_special
```

Is this docu old, or better maintained elsewhere?

Under the heading `Enable IBSS CCA hack`, this string (that = relative object pronoun):

` At one point, I suspected that decreased performance, but futher testing was inconclusive.`
is clearer as
` At one point, I suspected that this decreased performance, but further testing was inconclusive.`

also

`Currently the antenna_pattern is limitted to 5 bits, but that might change as needed.`
as
`Currently the antenna_pattern is limited to 5 bits, but that might change as needed.`

TIA

greearb, 2020-09-17:
Thanks for the detailed review.  I just updated the page, please let me know if you see any issues remaining.

systemcrash, 2020-09-17:
This code block is still confusing, and I think needs delineating:

```
# See help info on this option
]# cat /sys/kernel/debug/ieee80211/wiphy0/ath10k/set_rates 
This is to set fixed bcast, mcast, and beacon rates.  Normal rate-ctrl
is handled through normal API using 'iw', etc.
To set a value, you specify the dev-name, type, band and rate-code:
types: bcast, mcast, beacon
bands: 2, 5, 60
rate-codes: 0x43 1M, 0x42 2M, 0x41 5.5M, 0x40 11M, 0x3 6M, 0x7 9M, 0x2 12M, 0x6 18M, 0x1 24M, 0x5 36M, 0x0 48M, 0x4 54M, 0xFF default
 For example, to set beacon to 18Mbps on wlan0:  echo "wlan0 beacon 2 0x6" > /debug/..../set_rates

# Set beacons to use 18Mbps encoding
echo "wlan0 beacon 2 0x6" >  /sys/kernel/debug/ieee80211/wiphy0/ath10k/set_rates

For the more adventurous, you can figure out HT and VHT rate codes by looking
at how the ath10k driver builds/decodes them.  I am not sure HT or VHT rates can
be used successfully for these frames currently, however.
```

e.g.

```
# See help info on this option
]# cat /sys/kernel/debug/ieee80211/wiphy0/ath10k/set_rates 
```
This is to set fixed bcast, mcast, and beacon rates.  Normal rate-ctrl
is handled through normal API using 'iw', etc.
To set a value, you specify the dev-name, type, band and rate-code:
```
types: bcast, mcast, beacon
bands: 2, 5, 60
rate-codes: 0x43 1M, 0x42 2M, 0x41 5.5M, 0x40 11M, 0x3 6M, 0x7 9M, 0x2 12M, 0x6 18M, 0x1 24M, 0x5 36M, 0x0 48M, 0x4 54M, 0xFF default
```
 For example, to set beacon to 18Mbps on wlan0:  
```
echo "wlan0 beacon 2 0x6" > /debug/..../set_rates
```

EDIT: is the above a dupe?

```
# Set beacons to use 18Mbps encoding
echo "wlan0 beacon 2 0x6" >  /sys/kernel/debug/ieee80211/wiphy0/ath10k/set_rates
```
For the more adventurous, you can figure out HT and VHT rate codes by looking
at how the ath10k driver builds/decodes them.  I am not sure HT or VHT rates can
be used successfully for these frames currently, however.

systemcrash, 2020-09-17:
Also:
` Due to some register setting commands which I do not full understand, `
as
` Due to some register setting commands which I do not fully understand, `

😄

greearb, 2020-09-17:
Please try now.

systemcrash, 2020-09-17:
Last
```

For the more adventurous, you can figure out HT and VHT rate codes by looking
at how the ath10k driver builds/decodes them.  I am not sure HT or VHT rates can
be used successfully for these frames currently, however.
```

can be un-commented, otherwise great! Thank you for the clarifications.

greearb, 2020-09-17:
ok, I tweaked that.  Let me know if you see anything else.

systemcrash, 2020-09-17:
OK - will comment here if I do. Otherwise closing.

closed 2020-09-17 by systemcrash