greearb/ath10k-ct #161: ipq4019 5.8G wifi  The rate is too slow

zimiao815, 2020-09-24:
ipq4019 5.8G wifi  The rate is too slow How to increase speed ？

<*> kmod-ath10k-ct......... ath10k-ct driver optimized for CT ath10k firmware    
 [*] Enable LED support                                                                                       
 < > kmod-ath10k-ct-smallbuffers  
<*> ath10k-firmware-qca4019-ct......... ath10k CT 10.4 firmware for QCA4018/9 
                           
Software (OS, Firmware version, kernel, driver, etc)
 OS,:OpenWrt 19.07.2, r10947-65030d81f3

Hardware 
IPQ4019 Generic version

Logs (dmesg, maybe supplicant and/or hostap)
pc[wifi5.8]--->ipq4019route[openwrt19]--->Gigabit network cable--->> PC [iperf -s]

root@ubuntu:~# iperf -s 
------------------------------------------------------------
Server listening on TCP port 5001
TCP window size: 85.3 KByte (default)
------------------------------------------------------------
[  4] local 192.168.8.1 port 5001 connected with 192.168.8.96 port 59726
[ ID] Interval       Transfer     Bandwidth
[  4]  0.0-10.0 sec   179 MBytes   150 Mbits/sec
[  4] local 192.168.8.1 port 5001 connected with 192.168.8.96 port 59727
[  4]  0.0-10.0 sec   229 MBytes   191 Mbits/sec

dmesg 
....
[   21.762393] ath10k_ahb a800000.wifi: Direct firmware load for ath10k/fwcfg-ahb-a800000.wifi.txt failed with error -2
[   21.762441] ath10k_ahb a800000.wifi: Falling back to user helper
[   22.035032] firmware ath10k!fwcfg-ahb-a800000.wifi.txt: firmware_loading_store: map pages failed
[   22.039647] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   22.042979] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   22.055490] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-fH-012-17ba98334 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 7049b0d7
[   22.113102] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:23 crc32 1b3251b3
[   23.422323] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   23.422374] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[   23.471377] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   23.474507] ath10k_ahb a800000.wifi: wmi print 'free: 56544 iram: 23688 sram: 32520'
[   23.587575] ath10k_ahb a800000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[   23.715082] ath: EEPROM regdomain: 0x0
[   23.715094] ath: EEPROM indicates default country code should be used
[   23.715099] ath: doing EEPROM country->regdmn map search
[   23.715110] ath: country maps to regdmn code: 0x3a
[   23.715116] ath: Country alpha2 being used: US
[   23.715121] ath: Regpair used: 0x3a
....
[25903.993260] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[25903.993303] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[25904.044237] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[25904.044645] ath10k_ahb a800000.wifi: wmi print 'free: 56544 iram: 23688 sram: 32520'
[25904.347374] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
.....

Looking forward to and thank you for your reply

matejkupljen, 2020-12-18:
We are experiencing similar issue, noticing that throughput varies a lot during testing with iperf.
We are running this on IPQ4019 device, similar to Qualcomm reference design AP-DK01.1-C1.

We were testing this on OpenWRT 19.07, but now we have compiled OpenWRT master branch with the latest drivers and firmware available.
in the atk10k-ct driver package the CT_KVER="-5.8" is selected and in the firmware package the QCA4019_FIRMWARE_FILE_CT:=firmware-5-ct-full-community-12.bin-lede.019 is selected.

Here is the output when ath10k-ct driver is loaded:
```
[    9.594981] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[    9.595030] ath10k_ahb a000000.wifi: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[    9.606686] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-fW-13-8c5b2baa2 api 5 features mfp,peer-flow-ctrl,txstatus->noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 1f3ed50f
[    9.664887] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:20 crc32 e2dfaa91
[   10.953537] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96 
[   10.953588] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[   11.003047] ath10k_ahb a000000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   11.004019] ath10k_ahb a000000.wifi: wmi print 'free: 53252 iram: 13576 sram: 35752'
[   11.146693] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[   11.313067] ath: EEPROM regdomain sanitized
[   11.313079] ath: EEPROM regdomain: 0x64
[   11.313083] ath: EEPROM indicates we should expect a direct regpair map
[   11.313097] ath: Country alpha2 being used: 00
[   11.313101] ath: Regpair used: 0x64
[   11.983563] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   11.983616] ath10k_ahb a800000.wifi: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[   11.995770] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-fW-13-8c5b2baa2 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-alll-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 1f3ed50f
[   12.050936] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:21 crc32 e2dfaa91
[   13.338691] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   13.338744] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[   13.387849] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   13.388823] ath10k_ahb a800000.wifi: wmi print 'free: 53252 iram: 13576 sram: 35752'
[   13.529374] ath10k_ahb a800000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[   13.659858] ath: EEPROM regdomain sanitized
[   13.659873] ath: EEPROM regdomain: 0x64
[   13.659878] ath: EEPROM indicates we should expect a direct regpair map
[   13.659889] ath: Country alpha2 being used: 00
[   13.659894] ath: Regpair used: 0x64
```

We set up one SSID on both radios and on 5GHz we set channel width to VHT80 and select channel 44.
Our AP is connected to switch, and on that switch we have a Desktop PC, running Ubuntu 18.04 and iperf3 server.
All the Ethernet links are at 1GB/s.

We use a laptop (also running Ubuntu 18.04) and connect to this SSID using Intel Corporation Dual Band Wireless-AC 8265 card:
```
$ iw dev wlp2s0 info
Interface wlp2s0
	ifindex 3
	wdev 0x1
	addr 00:28:xx:xx:xx:xx
	ssid Testing
	type managed
	wiphy 0
	channel 44 (5220 MHz), width: 80 MHz, center1: 5210 MHz
	txpower 22.00 dBm
```

When we run iper3 client on the laptop, we see different speed results:
```
$ iperf3 -c 192.168.50.100 -t 30
Connecting to host 192.168.50.100, port 5201
[  4] local 192.168.50.214 port 59910 connected to 192.168.50.100 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  30.7 MBytes   257 Mbits/sec    0    987 KBytes       
[  4]   1.00-2.00   sec  30.8 MBytes   258 Mbits/sec    0   1.18 MBytes       
[  4]   2.00-3.00   sec  33.4 MBytes   280 Mbits/sec    0   1.42 MBytes       
[  4]   3.00-4.00   sec  35.0 MBytes   294 Mbits/sec    0   1.42 MBytes       
[  4]   4.00-5.00   sec  32.1 MBytes   269 Mbits/sec    0   1.42 MBytes       
[  4]   5.00-6.00   sec  41.1 MBytes   345 Mbits/sec    0   1.56 MBytes       
[  4]   6.00-7.00   sec  38.3 MBytes   321 Mbits/sec    0   1.56 MBytes       
[  4]   7.00-8.00   sec  42.7 MBytes   358 Mbits/sec    0   1.59 MBytes       
[... 24 lines cut ...]
[  4]   0.00-30.00  sec  1.02 GBytes   293 Mbits/sec    0             sender
[  4]   0.00-30.00  sec  1.02 GBytes   292 Mbits/sec                  receiver

iperf Done.
```

As you can see it drops bellow 200Mb/s (153 in this case) and goes almost to 400Mb/s (364 in this case). We expect that the results will vary, but not at such high values.
Note, that using WiFi analyzer on my phone, I can see that our SSID is the only one available on this channel.

One more thing. If I run iwinfo on AP during iperf3 test, I can see that TX VHT-MCS is constantly changing from 4 up to 8:
```
root@OpenWrt:/# iwinfo wlan1 as
00:28:F8:43:42:70  -51 dBm / -103 dBm (SNR 52)  0 ms ago
        RX: 780.0 MBit/s, VHT-MCS 8, 80MHz, VHT-NSS 2   1260975 Pkts.
        TX: 866.7 MBit/s, VHT-MCS 9, 80MHz, VHT-NSS 2    238756 Pkts.
        expected throughput: unknow
```

Any ideas and hints what could be causing so different speed tests?
I can try different versions of the driver, different firmwares and/or different tests.

Maybe @greearb have some ideas?

Thanks,
Matej

greearb, 2020-12-18:
Does it work better with QCA firmware?

matejkupljen, 2020-12-18:
Tested with QCA firmware. Just copied new firmware to the AP and did:
```
wifi down
rmmod ath10k_pci
modprobe ath10k_pci
wifi up
```

Here is the output when I loaded the module:
```
[12942.225233] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[12942.225284] ath10k_ahb a000000.wifi: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[12942.237439] ath10k_ahb a000000.wifi: firmware ver 10.4-3.6-00140 api 5 features no-p2p,mfp,peer-flow-ctrl,btcoex-param,allows-mesh-bcast,no-ps crc32 ba79b746
[12942.287868] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:20 crc32 e2dfaa91
[12943.631197] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 528  tid: 102
[12943.631243] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[12943.712224] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
[12944.390679] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[12944.390727] ath10k_ahb a800000.wifi: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[12944.402528] ath10k_ahb a800000.wifi: firmware ver 10.4-3.6-00140 api 5 features no-p2p,mfp,peer-flow-ctrl,btcoex-param,allows-mesh-bcast,no-ps crc32 ba79b746
[12944.452444] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:21 crc32 e2dfaa91
[12945.794922] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 528  tid: 102
[12945.794969] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[12945.876837] ath10k_ahb a800000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 512 raw 0 hwcrypto 1
```

And here are the results:
```
$ iperf3 -c 192.168.50.100 -t 30
Connecting to host 192.168.50.100, port 5201
[  4] local 192.168.50.214 port 60662 connected to 192.168.50.100 port 5201
[ ID] Interval           Transfer     Bandwidth       Retr  Cwnd
[  4]   0.00-1.00   sec  33.3 MBytes   279 Mbits/sec    0   1.19 MBytes       
[  4]   1.00-2.00   sec  39.0 MBytes   328 Mbits/sec    0   1.77 MBytes       
[  4]   2.00-3.00   sec  39.5 MBytes   331 Mbits/sec    0   2.12 MBytes       
[  4]   3.00-4.00   sec  43.8 MBytes   367 Mbits/sec    0   2.46 MBytes       
[  4]   4.00-5.00   sec  44.4 MBytes   372 Mbits/sec    0   2.46 MBytes       
[  4]   5.00-6.00   sec  44.8 MBytes   376 Mbits/sec    0   2.46 MBytes       
[  4]   6.00-7.00   sec  39.0 MBytes   327 Mbits/sec    0   2.46 MBytes       
[  4]   7.00-8.00   sec  39.5 MBytes   331 Mbits/sec    0   2.46 MBytes       
[  4]   8.00-9.00   sec  39.9 MBytes   335 Mbits/sec    0   2.46 MBytes       
[  4]   9.00-10.00  sec  39.5 MBytes   331 Mbits/sec    0   2.46 MBytes       
[  4]  10.00-11.00  sec  40.0 MBytes   335 Mbits/sec    0   2.46 MBytes       
[  4]  11.00-12.00  sec  41.5 MBytes   349 Mbits/sec    0   2.46 MBytes       
[  4]  12.00-13.00  sec  42.5 MBytes   356 Mbits/sec    0   2.72 MBytes       
[  4]  13.00-14.00  sec  42.2 MBytes   354 Mbits/sec    0   2.72 MBytes       
[  4]  14.00-15.00  sec  43.2 MBytes   362 Mbits/sec    0   2.72 MBytes       
[  4]  15.00-16.00  sec  43.8 MBytes   367 Mbits/sec    0   2.72 MBytes       
[  4]  16.00-17.00  sec  44.9 MBytes   376 Mbits/sec    0   2.72 MBytes       
[  4]  17.00-18.00  sec  44.9 MBytes   377 Mbits/sec    0   2.72 MBytes       
[  4]  18.00-19.00  sec  42.6 MBytes   357 Mbits/sec    0   2.72 MBytes       
[  4]  19.00-20.00  sec  44.3 MBytes   372 Mbits/sec    0   2.72 MBytes       
[  4]  20.00-21.00  sec  44.5 MBytes   373 Mbits/sec    0   2.72 MBytes       
[  4]  21.00-22.00  sec  43.7 MBytes   367 Mbits/sec    0   2.72 MBytes       
[  4]  22.00-23.00  sec  45.0 MBytes   378 Mbits/sec    4   1.99 MBytes       
[  4]  23.00-24.00  sec  46.2 MBytes   387 Mbits/sec    0   2.18 MBytes       
[  4]  24.00-25.00  sec  45.6 MBytes   383 Mbits/sec    0   2.30 MBytes       
[  4]  25.00-26.00  sec  44.1 MBytes   370 Mbits/sec    0   2.30 MBytes       
[  4]  26.00-27.00  sec  45.4 MBytes   381 Mbits/sec    0   2.52 MBytes       
[  4]  27.00-28.00  sec  43.5 MBytes   365 Mbits/sec    0   2.57 MBytes       
[  4]  28.00-29.00  sec  42.4 MBytes   356 Mbits/sec    0   2.57 MBytes       
[  4]  29.00-30.00  sec  44.4 MBytes   372 Mbits/sec    0   2.57 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-30.00  sec  1.25 GBytes   357 Mbits/sec    4             sender
[  4]   0.00-30.00  sec  1.25 GBytes   357 Mbits/sec                  receiver

iperf Done.
```

It seems the speed still varies, but not as much.

BR,
Matej

greearb, 2020-12-18:
I can build you a series of firmware images if you'd like to try bisecting the problem?

matejkupljen, 2020-12-18:
Of course.
Just upload them and I will do the tests.

greearb, 2020-12-18:
ok, I will start the builds.  It will take a day or two to build all of them...

zimiao815, 2021-01-13:
thank you for your reply. 

The maximum speed is about 300MB / S. is it the current openwrt master branch? Is there no way to improve it? We test that the  qsdk of Qualcomm is   600 + MB / s.

I use 9653 + 9886 board, 5.8G can reach 400m or so. 
The problem should be firmware-5.bin instead of ath10k. Is that correct?

matejkupljen, 2021-01-13:
Hi,

using ath10k-ct driver (PKG_SOURCE_DATE = 2020-12-24 as in master of OpenWRT), but using version 4.19 of the code, since 5.8 doesn't compile against my version of the kernel and using firmware firmware-5-ct-full-community-12.bin-lede.022 I get the following results:
```
$ iperf3 -c 192.168.50.100 -Z
Connecting to host 192.168.50.100, port 5201
[  5] local 192.168.50.221 port 62783 connected to 192.168.50.100 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  54.2 MBytes   454 Mbits/sec
[  5]   1.00-2.00   sec  57.4 MBytes   482 Mbits/sec
[  5]   2.00-3.00   sec  66.4 MBytes   556 Mbits/sec
[  5]   3.00-4.00   sec  66.6 MBytes   559 Mbits/sec
[  5]   4.00-5.00   sec  66.8 MBytes   560 Mbits/sec
[  5]   5.00-6.00   sec  66.6 MBytes   559 Mbits/sec
[  5]   6.00-7.00   sec  66.8 MBytes   561 Mbits/sec
[  5]   7.00-8.00   sec  67.3 MBytes   564 Mbits/sec
[  5]   8.00-9.00   sec  66.9 MBytes   561 Mbits/sec
[  5]   9.00-10.00  sec  66.8 MBytes   560 Mbits/sec
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-10.00  sec   646 MBytes   542 Mbits/sec                  sender
[  5]   0.00-10.00  sec   646 MBytes   542 Mbits/sec                  receiver

iperf Done
```
This was tested on my MacBook Pro on 5GHz, channel 44 and VHT 80. Sometimes the throughput can go above 600Mb/s.

But I saw that the results can be quite different during testing. They can go from -400MB/s to +600MB/s using the same setup. I din't find out yet why is this happening. Any hint would be appreciated.

What is **VERY IMPORTANT** is that if you are using multi core SoC, like IPQ40xx is, you have to install irqbalance package, enable it and start it. This will spread the IRQ work across different cores, otherwise all the work will be done on a single core only. Just compare todays results with the ones I posted earlier.

@greearb you said that you would like me to do some tests.
Is this still the case?
Did you do any iper3 measurements? What throughputs did you get? 

Thanks and BR,
Matej

notgood, 2021-04-07:
D-Link DAP-2610, featuring IPQ4019, running latest Openwrt master, dumb AP mode bridging 5Ghz WiFi to Ethernet.

Pretty much the same problem: client is connected at max TX/RX rate 866.7 MBit/s, yet the real TCP iperf3 speed is around 200+ MBit/sec. One CPU core  is bottlenecked at 100% sirq, other 3 cores are idle.

Running irqbalance helps, iperf3 goes up to 500+ Mbit, although core load is still uneven, one is still around 100%.

Basicly two issues:
1) WiFi is consuming way too much CPU. I mean, IPQ40xx is a quad-core ARM, the are tons of lower spec MIPS cpus able to push such traffic in dump AP mode. Are there any firmware/driver options to lower CPU usage for simple home scenarios?
2) Driver isn't utilising multiple threads/cores, at least not on IPQ40xx boards. Can't really expect people to manually install and run irqbalance.

Netgear R7800 with QCA9984 running stable OpenWRT 19.07.7 with ath10k-ct in the same room pushes 600+ MBit, with the same client. CPU load is around 50%, evenly split between both cores. Still too high for a rather high-end home router, but it least it doesn't bottleneck and reaches almost maximum possible TCP speed for 866.7 MBit/s rate.

```
[   12.384870] ath10k_ahb a000000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   12.384918] ath10k_ahb a000000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 0 testmode 0
[   12.396083] ath10k_ahb a000000.wifi: firmware ver 10.4b-ct-4019-tH-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 ccaa1586
[   12.450826] ath10k_ahb a000000.wifi: board_file api 2 bmi_id 0:16 crc32 09aab8f9
[   13.666771] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   13.666817] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[   13.703105] ath10k_ahb a000000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   13.704032] ath10k_ahb a000000.wifi: wmi print 'free: 91592 iram: 18152 sram: 41260'
[   13.840381] ath10k_ahb a000000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[   14.559631] ath10k_ahb a800000.wifi: qca4019 hw1.0 target 0x01000000 chip_id 0x003b00ff sub 0000:0000
[   14.559687] ath10k_ahb a800000.wifi: kconfig debug 0 debugfs 1 tracing 0 dfs 0 testmode 0
[   14.572444] ath10k_ahb a800000.wifi: firmware ver 10.4b-ct-4019-tH-13-5ae337bb1 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,no-bmiss-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 ccaa1586
[   14.625319] ath10k_ahb a800000.wifi: board_file api 2 bmi_id 0:17 crc32 09aab8f9
[   15.841363] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   15.841405] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[   15.877127] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   15.877992] ath10k_ahb a800000.wifi: wmi print 'free: 91592 iram: 18152 sram: 41260'
[   16.003138] ath10k_ahb a800000.wifi: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[   21.926175] ath10k_ahb a000000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   21.926229] ath10k_ahb a000000.wifi: msdu-desc: 2500  skid: 32
[   21.962250] ath10k_ahb a000000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   21.963176] ath10k_ahb a000000.wifi: wmi print 'free: 91592 iram: 18152 sram: 41260'
[   22.276426] ath10k_ahb a000000.wifi: rts threshold -1
[   22.279973] ath10k_ahb a000000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[   22.503075] ath10k_ahb a000000.wifi: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
[   24.107920] ath10k_ahb a800000.wifi: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   24.107964] ath10k_ahb a800000.wifi: msdu-desc: 2500  skid: 32
[   24.143594] ath10k_ahb a800000.wifi: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   24.144948] ath10k_ahb a800000.wifi: wmi print 'free: 91592 iram: 18152 sram: 41260'
[   24.431452] ath10k_ahb a800000.wifi: rts threshold -1
[   24.432367] ath10k_ahb a800000.wifi: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4

```

```
driver: ath10k_ahb
version: 5.4.109
firmware-version: 10.4b-ct-4019-tH-13-5ae337bb1
expansion-rom-version: 
bus-info: a800000.wifi
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no

```

notgood, 2021-05-04:
Any updates on this one?

paladin4fan, 2021-12-22:
Any updates?
Got Netgear EX6150 v2 with ipq4019
Same issue here.
Says 867Mbps TX/RX
But only 280Mbps shows in SpeedTest.net (400Mbps with other extender, 500Mbps with wired device)
iperf3 -c could reach 400Mbps

SuperKali, 2022-06-30:
I have the same problem, wifi is slow..

supersebbo, 2024-10-04:
I started reading this thread with some hope, then saw it never got resolved.  The issue persists, even with irqbalance spreading the load across cores the WiFi throughput tops out at around 350-350Mbps.  Did anyone make any progress?