greearb/ath10k-ct #163: Support of WLE1216VX

hansekiter, 2020-10-03:
Hello,

i bought a WLE1216VX with the QCA9984 for my Turris Omnia. 

When i searched for compatiblity i got several Sites and also Compex itself providing a Guide how to fix the Driver issue. AFAIK the QCA9984 Chipset is compatible with ath10k. 

When will this this fix be implemented in the driver? 
[wle1216vx-ath10k-ct-support.zip](https://github.com/greearb/ath10k-ct/files/5321336/wle1216vx-ath10k-ct-support.zip)

Best regards
Nik

greearb, 2020-10-03:
I don't currently have access to the upstream fw source that enables this feature, so cannot support this NIC at this time.

AndyWas, 2020-11-22:
Hello Nik,

I've just reported an issue with loading firmware for my Compex WLE1216VX QCA9984 802.11ac wireless module (https://github.com/greearb/ath10k-ct/issues/172), and is saw you had raised this issue recently with the same product.  Does my problem look anything like yours?  I've also raised this with the main Linux Wireless dev mailing list but no one has replied yet.  I suspect it may simply be the case that this is a new product and the board parameters haven't been incorporated yet, but I'm not familiar enough with this sort of hardware to build the new correct version - nor do I know where the source is.

Let me know if you think there's anything we can do to help each other on this.

Regards,
Andrew