greearb/ath10k-ct #168: Question: QBSS Load Element invalid values

MartB, 2020-11-01:
The QBSS Load element randomly shows the channel utilization to be 220 (max is 255) with 0 attached clients on an ath10k 988x device it then goes down to low values again on the next beacon frame.

Is this value fed by the driver or where is it coming from?
An identical AP on the same channel does report < 10 at all time.

The Available Admission capacity is also always 0, it does not change at all, but that behavior is also found on some other aps.

nicklowe, 2021-05-12:
The QBSS Lowe IE reports the condition of the channel as CU. It is not related to the number of clients.

However, I do see this reporting suspiciously high and wildly fluctating values that are not in keeping with what I see happening to the channel with a spectrum analyser on an Netgear R7800 running the latest developer snapshot of OpenWRT.
For spectrum analysis, I used a Cisco 3702i with Metageek.

Considering that some popular STA will roam based on this as a factor in the decision, I think this should be checked carefully.

https://cdn2.hubspot.net/hubfs/353374/CWNP%207signal%20Roaming%20Revealed%20v10.pdf

https://docs.samsungknox.com/admin/knox-platform-for-enterprise/kbas/kba-115013403768.htm

greearb, 2021-05-12:
It would be hostapd (I guess) that populates this value.  I think that whatever is sampling the channel usage time should do a running average or otherwise smooth out results a bit.  Probably the raw values come from a survey dump.  Maybe you can run survey dumps periodically and see if you can reproduce bogus values that way?

nicklowe, 2021-05-12:
Indeed it is hostapd that adds the IE:

https://w1.fi/cgit/hostap/commit/?id=ec8f36afca731aa6a30a314c54bb83441180719d
https://w1.fi/cgit/hostap/commit/?id=802c0fd0c328bc634fd2c76e82f53c0e35abaff1

It comments "Channel Utilization is calculated based on the channel survey information from the driver and as such, requires a driver that supports providing that information for the current operating channel."

It does not appear as if the driver or underlying firmware is providing accurate information to hostapd here.

greearb, 2021-05-12:
Please do manual testing to gather survey results, and let me know if you see inaccurate information.  Either way, hostapd should smooth these values a bit so that a few bad samples do not cause undue influence on the value reported by hostapd.

nicklowe, 2021-05-12:
Yes, will do.

closed 2024-11-11 by MartB