greearb/ath10k-ct #169: 5ghz network drops on R7800

theAeon, 2020-11-01:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
5Ghz radio stops broadcasting around once a day or so, module doesn't crash and 2.4ghz radio continues working. Only noticed this w/ DFS VHT80/160, but I have not tried other combinations.

Software (OS, Firmware version, kernel, driver, etc)
Openwrt r14694-e788e9bd87, hnyman's R7800 build, kernel/driver 5.4.70, firmware 10.4b-ct-9984-fW-13-795874169

Hardware (NIC chipset, platform, etc)

R7800, QCA8894/IPQ8065

Logs (dmesg, maybe supplicant and/or hostap)
dmesg:
`[   16.223266] ath10k_pci 0000:01:00.0: assign IRQ: got 35
[   16.223287] ath10k 5.4 driver, optimized for CT firmware, probing pci device: 0x46.
[   16.223344] ath10k_mac_create, priv_size: 868  hw: 32dee5a9  hw->priv: a2e7a7a7
[   16.230133] ath10k_pci 0000:01:00.0: enabling device (0140 -> 0142)
[   16.237188] ath10k_pci 0000:01:00.0: enabling bus mastering
[   16.237692] ath10k_pci 0000:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   16.720940] ath10k_pci 0000:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[   16.720973] ath10k_pci 0000:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   16.731482] ath10k_pci 0000:01:00.0: firmware ver 10.4b-ct-9984-fW-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 f3d4b769
[   19.063857] ath10k_pci 0000:01:00.0: board_file api 2 bmi_id 0:1 crc32 85498734
[   24.912547] ath10k_pci 0000:01:00.0: unsupported HTC service id: 1536
[   24.913706] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   24.917965] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[   25.001000] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   25.001837] ath10k_pci 0000:01:00.0: wmi print 'free: 84920 iram: 13316 sram: 11224'
[   25.290161] ath10k_pci 0000:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[   25.384578] ath: EEPROM regdomain sanitized
[   25.384595] ath: EEPROM regdomain: 0x64
[   25.384606] ath: EEPROM indicates we should expect a direct regpair map
[   25.384630] ath: Country alpha2 being used: 00
[   25.384640] ath: Regpair used: 0x64
[   25.390777] ath10k_pci 0001:01:00.0: assign IRQ: got 37
[   25.390820] ath10k 5.4 driver, optimized for CT firmware, probing pci device: 0x46.
[   25.390967] ath10k_mac_create, priv_size: 868  hw: 4686a520  hw->priv: 069f46d5
[   25.398452] ath10k_pci 0001:01:00.0: enabling device (0140 -> 0142)
[   25.404727] ath10k_pci 0001:01:00.0: enabling bus mastering
[   25.405310] ath10k_pci 0001:01:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
[   25.735445] ath10k_pci 0001:01:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[   25.735476] ath10k_pci 0001:01:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   25.746109] ath10k_pci 0001:01:00.0: firmware ver 10.4b-ct-9984-fW-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 f3d4b769
[   28.065582] ath10k_pci 0001:01:00.0: board_file api 2 bmi_id 0:2 crc32 85498734
[   33.896536] ath10k_pci 0001:01:00.0: unsupported HTC service id: 1536
[   33.897332] ath10k_pci 0001:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   33.901951] ath10k_pci 0001:01:00.0: msdu-desc: 2500  skid: 32
[   33.986948] ath10k_pci 0001:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   33.987819] ath10k_pci 0001:01:00.0: wmi print 'free: 84920 iram: 13316 sram: 11224'
[   34.316708] ath10k_pci 0001:01:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[   34.414582] ath: EEPROM regdomain sanitized
[   34.414600] ath: EEPROM regdomain: 0x64
[   34.414611] ath: EEPROM indicates we should expect a direct regpair map
[   34.414635] ath: Country alpha2 being used: 00
[   34.414645] ath: Regpair used: 0x64
[   34.425200] kmodloader: done loading kernel modules from /etc/modules.d/*
[   38.640089] dwmac1000: Master AXI performs any burst length
[   38.640118] ipq806x-gmac-dwmac 37400000.ethernet eth1: No Safety Features support found
[   38.644543] ipq806x-gmac-dwmac 37400000.ethernet eth1: IEEE 1588-2008 Advanced Timestamp supported
[   38.652560] ipq806x-gmac-dwmac 37400000.ethernet eth1: registered PTP clock
[   38.661571] ipq806x-gmac-dwmac 37400000.ethernet eth1: configuring for fixed/sgmii link mode
[   38.668549] ipq806x-gmac-dwmac 37400000.ethernet eth1: Link is Up - 1Gbps/Full - flow control off
[   38.673585] br-lan: port 1(eth1.1) entered blocking state
[   38.685876] br-lan: port 1(eth1.1) entered disabled state
[   38.691374] device eth1.1 entered promiscuous mode
[   38.696630] device eth1 entered promiscuous mode
[   38.702860] br-lan: port 1(eth1.1) entered blocking state
[   38.706012] br-lan: port 1(eth1.1) entered forwarding state
[... 98 lines cut ...]
[27628.989608] device wlan1 entered promiscuous mode
[27628.992800] br-lan: port 2(wlan1) entered blocking state
[27628.997158] br-lan: port 2(wlan1) entered forwarding state
[27629.073179] br-lan: port 3(wlan0) entered blocking state
[27629.073209] br-lan: port 3(wlan0) entered disabled state
[27629.078472] device wlan0 entered promiscuous mode
[27629.083072] br-lan: port 3(wlan0) entered blocking state
[27629.087525] br-lan: port 3(wlan0) entered forwarding state
[62859.235043] ath10k_pci 0000:01:00.0: NIC rx-max-rate: 0 calculated-max: 0 rxnss_override: 0x80000000  nss160: 1  spatial-streams: 2
[82401.404737] device wlan0 left promiscuous mode
[82401.404873] br-lan: port 3(wlan0) entered disabled state
[82401.456316] ath10k_pci 0000:01:00.0: mac flush null vif, drop 0 queues 0xffff
[82401.458783] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
[82401.462479] ath10k_pci 0000:01:00.0: peer-unmap-event: unknown peer id 0
[82407.458323] ath10k_pci 0000:01:00.0: unsupported HTC service id: 1536
[82407.458560] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[82407.463740] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[82407.546790] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[82407.547642] ath10k_pci 0000:01:00.0: wmi print 'free: 84920 iram: 13316 sram: 11224'
[82407.933929] ath10k_pci 0000:01:00.0: Firmware lacks feature flag indicating a retry limit of > 2 is OK, requested limit: 4
[82407.941444] br-lan: port 3(wlan0) entered blocking state
[82407.943900] br-lan: port 3(wlan0) entered disabled state
[82407.950169] device wlan0 entered promiscuous mode`

I just bumped the debug verbosity and enabled hostapd debug logging, will update when it happens again.

theAeon, 2020-11-03:
just happened again

hostapd:
```
Mon Nov  2 19:12:19 2020 daemon.notice hostapd: wlan1: STA c0:b8:83:2d:b8:42 IEEE 802.11: did not acknowledge authentication response

Mon Nov  2 19:12:19 2020 daemon.notice hostapd: wlan1: STA 48:5f:99:51:55:02 IEEE 802.11: did not acknowledge authentication response

Mon Nov  2 19:12:19 2020 daemon.notice hostapd: wlan1: STA c0:b8:83:2d:b8:42 IEEE 802.11: did not acknowledge authentication response

Mon Nov  2 19:12:19 2020 daemon.notice hostapd: wlan1: STA c0:b8:83:2d:b8:42 IEEE 802.11: did not acknowledge authentication response

Mon Nov  2 19:12:20 2020 daemon.err hostapd: could not get valid channel

Mon Nov  2 19:12:21 2020 daemon.info hostapd: wlan1: STA c0:b8:83:2d:b8:42 IEEE 802.11: authenticated

Mon Nov  2 19:12:21 2020 daemon.info hostapd: wlan1: STA c0:b8:83:2d:b8:42 IEEE 802.11: associated (aid 1)

Mon Nov  2 19:12:21 2020 daemon.info hostapd: wlan1: STA c0:b8:83:2d:b8:42 RADIUS: starting accounting session 777EFE352E749016

Mon Nov  2 19:12:21 2020 daemon.info hostapd: wlan1: STA c0:b8:83:2d:b8:42 WPA: pairwise key handshake completed (RSN)

Mon Nov  2 19:12:24 2020 daemon.info hostapd: wlan1: STA 48:5f:99:51:55:02 IEEE 802.11: authenticated

Mon Nov  2 19:12:24 2020 daemon.info hostapd: wlan1: STA 48:5f:99:51:55:02 IEEE 802.11: associated (aid 4)

Mon Nov  2 19:12:25 2020 daemon.info hostapd: wlan1: STA 48:5f:99:51:55:02 RADIUS: starting accounting session 11FABBE0B146F74F

Mon Nov  2 19:12:25 2020 daemon.info hostapd: wlan1: STA 48:5f:99:51:55:02 WPA: pairwise key handshake completed (RSN)
```

and dmesg 
```

[97958.570741] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:

[97958.570808] ath10k: [0000]: 05F9C65D 10005881 00003112 00430C04 0000000A 00000000 05F9C65D 10005881

[97958.575905] ath10k: [0008]: 00003112 00430C04 0000000E 00000001

[97958.584831] ath10k_pci 0001:01:00.0: ATH10K_END

[97961.920948] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:

[97961.921011] ath10k: [0000]: 05F9B82A 10005881 00003112 00430C04 0000000A 00000000 05F9B82A 10005881

[97961.926133] ath10k: [0008]: 00003112 00430C04 0000000E 00000000

[97961.934982] ath10k_pci 0000:01:00.0: ATH10K_END

[97964.571674] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:

[97964.571701] ath10k: [0000]: 05F9DE5E 10005881 00003112 00430C04 0000000A 00000000 05F9DE5E 10005881

[97964.576844] ath10k: [0008]: 00003112 00430C04 0000000E 00000001

[97964.585746] ath10k_pci 0001:01:00.0: ATH10K_END

[97967.922054] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:

[97967.922100] ath10k: [0000]: 05F9CFD1 08005824 00009085 00000000 05F9CFD3 08005824 00009084 00000000

[97967.927236] ath10k: [0008]: 05F9D02B 10005881 00003112 00430C04 0000000A 00000000 05F9D02B 10005881

[97967.936197] ath10k: [0016]: 00003112 00430C04 0000000E 00000000

[97967.945169] ath10k_pci 0000:01:00.0: ATH10K_END

[97970.572954] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:

[... 222 lines cut ...]

[97983.688619] ath10k: [0104]: 71108880 00030000 00D80012 00000000 00000000 05FA2B89 14004C01 71108880

[97983.697647] ath10k: [0112]: 00040000 00D80012 00000000 00000000 05FA2B89 14004C01 71108880 00050000

[97983.706676] ath10k: [0120]: 00D80012 00000000 00000000 05FA2B89 14004C01 71108880 00060000 00D80012

[97983.715704] ath10k: [0128]: 00000000 00000000

[97983.724743] ath10k_pci 0001:01:00.0: ATH10K_END

[97984.575394] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:

[97984.575456] ath10k: [0000]: 05FA3002 1400581D 000F72EC 00456BBC 000F72EC 00000006 00004000 05FA3002

[97984.580508] ath10k: [0008]: 1000581B 00005502 00000000 00000006 000F72EC

[97984.589507] ath10k_pci 0001:01:00.0: ATH10K_END

[97985.575538] ath10k_pci 0001:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:

[97985.575585] ath10k: [0000]: 05FA31F3 10005881 00003112 00430C04 0000000A 00000000 05FA31F3 10005881

[97985.580685] ath10k: [0008]: 00003112 00430C04 0000000E 00000001
```

theAeon, 2020-11-03:
on further investigation, this appears to be a DFS event handling issue. Is that a hostapd problem?

greearb, 2020-11-03:
Maybe your system is just detecting radar and acting appropriately?  Try using a non-dfs channel?  Use smaller bandwidth in hopes that dodges whatever frequency DFS is detecting?

theAeon, 2020-11-03:
I was on a static channel-which apparently kills the dynamic switching.
Seems to be functional on auto/160 as of right now. Will keep updated

theAeon, 2020-11-17:
Yeah. that was the issue. My bad.

closed 2020-11-17 by theAeon

jonozzz, 2021-03-26:
@theAeon I'm having this exact same issue happening once in a blue moon on a similar hardware QCA9880. I have 3 mesh APs all on a non-DFS channel (149 in US) with 80Mhz width, running the latest Openwrt with the CT firmware. I'm not seeing that " hostapd: could not get valid channel" in my logs. 

How did you conclude that it was a hostapd issue? Switching to auto channel selection is not an option for me in this case.

theAeon, 2021-03-26:
If I remember correctly I needed to enable verbose logging for hostapd.