greearb/ath10k-ct #170: [QCA9984] Wireless AC 5GHZ cannot start

gwlim, 2020-11-13:
Please provide this info.  See this link for more info on how to gather debug
info:  http://www.candelatech.com/ath10k-bugs.php

Description of the problem (how to configure, how to reproduce, how often it happens).
Unable to start AP Mode on QCA9984.
On certain channels scanning works.
Reverted to OEM Firmware 5GHZ works fine hence no issue with WiFi hardware

Software (OS, Firmware version, kernel, driver, etc)
OpenWrt Trunk
Kernel 5.4
firmware ver 10.4b-ct-9984-fH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 123549a3

Hardware (NIC chipset, platform, etc)
OpenWrt Trunk
TP-LINK WDR8500v2
ath10k_pci 0000:00:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe

Logs (dmesg, maybe supplicant and/or hostap)

```
[   39.145236] ath10k 5.4 driver, optimized for CT firmware, probing pci device: 0x46.
[   39.153307] ath10k_mac_create, priv_size: 804  hw: 73bea779  hw->priv: e4d16f89
[   39.177739] ath10k_pci 0000:00:00.0: enabling device (0000 -> 0002)
[   39.184535] ath10k_pci 0000:00:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   40.359346] ath10k_pci 0000:00:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
[   40.369628] ath10k_pci 0000:00:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 1 testmode 0
[   40.383047] ath10k_pci 0000:00:00.0: firmware ver 10.4b-ct-9984-fH-13-795874169 api 5 features mfp,peer-flow-ctrl,txstatus-noack,wmi-10.x-CT,ratemask-CT,regdump-CT,txrate-CT,flush-all-CT,pingpong-CT,ch-regs-CT,nop-CT,htt-mgt-CT,set-special-CT,tx-rc-CT,cust-stats-CT,txrate2-CT,beacon-cb-CT,wmi-block-ack-CT,wmi-bcn-rc-CT crc32 123549a3
[   42.860874] ath10k_pci 0000:00:00.0: board_file api 2 bmi_id 0:1 crc32 85498734
[   48.939678] ath10k_pci 0000:00:00.0: unsupported HTC service id: 1536
[   48.950727] ath10k_pci 0000:00:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   48.958500] ath10k_pci 0000:00:00.0: msdu-desc: 2500  skid: 32
[   49.040300] ath10k_pci 0000:00:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0 ct-sta: 0'
[   49.051608] ath10k_pci 0000:00:00.0: wmi print 'free: 84936 iram: 13572 sram: 11224'
[   49.338599] ath10k_pci 0000:00:00.0: htt-ver 2.2 wmi-op 6 htt-op 4 cal pre-cal-file max-sta 32 raw 0 hwcrypto 1
[   49.421852] ath: EEPROM regdomain sanitized
[   49.421861] ath: EEPROM regdomain: 0x64
[   49.421864] ath: EEPROM indicates we should expect a direct regpair map
[   49.421879] ath: Country alpha2 being used: 00
[   49.421881] ath: Regpair used: 0x64
[   49.481259] l2tp_ppp: PPPoL2TP kernel driver, V2.0
[   49.565105] ath: EEPROM regdomain sanitized
[   49.565114] ath: EEPROM regdomain: 0x64
[   49.565117] ath: EEPROM indicates we should expect a direct regpair map
[   49.565134] ath: Country alpha2 being used: 00
[   49.565136] ath: Regpair used: 0x64
[   49.576767] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[   49.578490] ieee80211 phy1: Atheros AR9561 Rev:0 mem=0xb8100000, irq=2
[   49.587940] kmodloader: done loading kernel modules from /etc/modules.d/*
[   60.830573] eth0: link up (1000Mbps/Full duplex)
[   60.846176] br-lan: port 1(eth0.1) entered blocking state
[   60.851803] br-lan: port 1(eth0.1) entered disabled state
[   60.857640] device eth0.1 entered promiscuous mode
[   60.862622] device eth0 entered promiscuous mode
[   60.936194] br-lan: port 1(eth0.1) entered blocking state
[   60.941822] br-lan: port 1(eth0.1) entered forwarding state
[   61.841183] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   61.848165] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   63.964133] ath: EEPROM regdomain: 0x8348
[   63.968318] ath: EEPROM indicates we should expect a country code
[   63.974630] ath: doing EEPROM country->regdmn map search
[   63.980122] ath: country maps to regdmn code: 0x3a
[   63.985078] ath: Country alpha2 being used: US
[   63.989670] ath: Regpair used: 0x3a
[   63.993273] ath: regdomain 0x8348 dynamically updated by user
[   63.999273] ath: EEPROM regdomain: 0x8348
[   64.003419] ath: EEPROM indicates we should expect a country code
[   64.009717] ath: doing EEPROM country->regdmn map search
[   64.015207] ath: country maps to regdmn code: 0x3a
[   64.020161] ath: Country alpha2 being used: US
[... 233 lines cut ...]
sae_require_mfp=1
wpa_passphrase=xxxxxxxxx
wpa_psk_file=/var/run/hostapd-wlan0.psk
auth_algs=1
wpa=2
wpa_pairwise=CCMP
ssid=XXXXXXXXXXXX
bridge=br-lan
mobility_domain=xxxx
ft_psk_generate_local=1
ft_over_ds=1
reassociation_deadline=1000
nas_identifier=xxxxxxxxx
wpa_disable_eapol_key_retries=0
wpa_key_mgmt=WPA-PSK FT-PSK WPA-PSK-SHA256 SAE FT-SAE
okc=1
ieee80211w=1
group_mgmt_cipher=AES-128-CMAC
dynamic_vlan=0
vlan_naming=1
vlan_file=/var/run/hostapd-wlan0.vlan
config_id=fc34eda45a59237c00dd838f3c98ab70
bssid=xx:xx:xx:xx:xx:xx

```

gwlim, 2020-11-13:
Original Board Data
[board.zip](https://github.com/greearb/ath10k-ct/files/5538021/board.zip)

qosmio, 2020-11-18:
`iw dev wlan0 info` is showing the access point is up and broadcasting. This would not be related to the firmware. It's a combination of your authentication and channel choices likely.

gwlim, 2020-11-19:
Wifi Analyzer is running but the SSID does not appear in the scan.
Reverting to factory firmware shows the radios are working

gwlim, 2020-11-21:
Never mind I think the tx aspect of the radios are damaged hence signal strengh is so low

closed 2020-11-21 by gwlim