greearb/ath10k-ct #172: Problem loading CT ath10k firmware for QCA9984 device (PCI ID 168c:0046)

AndyWas, 2020-11-22:
I'm having problems loading the ath10k firmware for a Qualcomm Atheros QCA9984 802.11ac wireless module (Compex WLE1216VX) in a x86_64 PC.  This particular unit is 2.4/5 GHz dual band.

When I modprobe ath10k_pci I get the following syslog error messages:

Nov 22 12:02:56 localhost.localdomain kernel: cfg80211: Loading compiled-in X.509 certificates for regulatory database
Nov 22 12:02:56 localhost.localdomain kernel: cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
Nov 22 12:02:56 localhost.localdomain kernel: ath10k_pci 0000:03:00.0: pci irq msi oper_irq_mode 2 irq_mode 0 reset_mode 0
Nov 22 12:02:57 localhost.localdomain kernel: ath10k_pci 0000:03:00.0: Unknown FW IE: 30
Nov 22 12:02:57 localhost.localdomain kernel: ath10k_pci 0000:03:00.0: qca9984/qca9994 hw1.0 target 0x01000000 chip_id 0x00000000 sub 168c:cafe
Nov 22 12:02:57 localhost.localdomain kernel: ath10k_pci 0000:03:00.0: kconfig debug 0 debugfs 1 tracing 0 dfs 0 testmode 0
Nov 22 12:02:57 localhost.localdomain kernel: ath10k_pci 0000:03:00.0: firmware ver 10.4b-ct-9984-fW-13-f2b8ccbd5 api 5 features mfp,peer-flow-ctrl crc32 e7a1e8d7
Nov 22 12:02:58 localhost.localdomain kernel: ath10k_pci 0000:03:00.0: failed to fetch board data for bus=pci,bmi-chip-id=0,bmi-board-id=11 from ath10k/QCA9984/hw1.0/board-2.bin
Nov 22 12:02:58 localhost.localdomain kernel: ath10k_pci 0000:03:00.0: failed to fetch board-2.bin or board.bin from ath10k/QCA9984/hw1.0
Nov 22 12:02:58 localhost.localdomain kernel: ath10k_pci 0000:03:00.0: failed to fetch board file: -2
Nov 22 12:02:58 localhost.localdomain kernel: ath10k_pci 0000:03:00.0: could not probe fw (-2)

This is with Ath10k CT 10.4.3 Beta 12 firmware-5-ct-full-community.bin on kernel 5.9.8-200 (Fedora 33), though I get the same outcome with "community" and "non-commercial-full".  I also get essentially the same outcome with the "official" firmware - ver. 10.4-3.9.0.2-00131, which is the most recent I can find on Git, and 10.4-3.9.0.2-00070 which is installed by linux-firmware-20201022-113 (latest Fedora package) - as well as on earlier kernels and firmware versions, and other x86_64 hardware.The board-2.bin and firmware-5.bin files do exist in the /lib/firmware/ath10k/QCA9984/hw1.0 directory, which matches the exact chipset ID.

lspci -nn output reports PCI ID 168c:0046.

I've reported this on the Linux Wireless dev mailing list but no reply yet (https://marc.info/?l=linux-wireless&m=160574568703161&w=2).  On reflection could it be the case that board-2.bin simply doesn't contain data for these board parameters (bmi-chip-id=0,bmi-board-id=11 etc.) e.g. because this is a new variant and they haven't been added yet?  More lspci and modinfo output appended below.

Many thanks,
Andrew

[root@localhost QCA9984]# lspci -s 03:00.0 -vvv
03:00.0 Network controller: Qualcomm Atheros QCA9984 802.11ac Wave 2 Wireless Network Adapter
        Subsystem: Qualcomm Atheros Device cafe
        Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 32 bytes
        Interrupt: pin A routed to IRQ 34
        Region 0: Memory at f7800000 (64-bit, non-prefetchable) [size=2M]
        Capabilities: [40] Power Management version 3
                Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
                Status: D0 NoSoftRst- PME-Enable- DSel=0 DScale=0 PME-
        Capabilities: [50] MSI: Enable+ Count=1/32 Maskable+ 64bit+
                Address: 00000000fee04004  Data: 4025
                Masking: fffffffe  Pending: 00000000
        Capabilities: [70] Express (v2) Endpoint, MSI 00
                DevCap: MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
                        ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset- SlotPowerLimit 10.000W
                DevCtl: CorrErr- NonFatalErr- FatalErr- UnsupReq-
                        RlxdOrd- ExtTag- PhantFunc- AuxPwr- NoSnoop-
                        MaxPayload 128 bytes, MaxReadReq 512 bytes
                DevSta: CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
                LnkCap: Port #0, Speed 5GT/s, Width x1, ASPM not supported
                        ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
                LnkCtl: ASPM Disabled; RCB 64 bytes Disabled- CommClk+
                        ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
                LnkSta: Speed 2.5GT/s (downgraded), Width x1 (ok)
                        TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
                DevCap2: Completion Timeout: Range ABCD, TimeoutDis+, NROPrPrP-, LTR+
                         10BitTagComp-, 10BitTagReq-, OBFF Not Supported, ExtFmt-, EETLPPrefix-
                         EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
                         FRS-, TPHComp-, ExtTPHComp-
                         AtomicOpsCap: 32bit- 64bit- 128bitCAS-
                DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis-, LTR-, OBFF Disabled
                         AtomicOpsCtl: ReqEn-
                LnkCtl2: Target Link Speed: 5GT/s, EnterCompliance- SpeedDis-
                         Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
                         Compliance De-emphasis: -6dB
                LnkSta2: Current De-emphasis Level: -3.5dB, EqualizationComplete-, EqualizationPhase1-
                         EqualizationPhase2-, EqualizationPhase3-, LinkEqualizationRequest-
        Capabilities: [100 v2] Advanced Error Reporting
                UESta:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UEMsk:  DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
                UESvrt: DLP+ SDES+ TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF+ MalfTLP+ ECRC- UnsupReq- ACSViol-
                CESta:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr-
                CEMsk:  RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
[... 61 lines cut ...]
                1B:F4:54:80:48:E0:26:59:4B:D7:97:3E:85:8E:4E:A6:77:8A:CC:28:
                64:35:4F:06:EC:19:E3:3A:63:DA:54:31:58:39:F6:38:89:93:DA:0F:
                5B:63:5F:A9:52:A0:11:47:28:32:6C:F5:F2:24:BB:A3:48:02:10:A1:
                5B:DF:4C:ED:EE:63:0E:32:EE:A0:2C:2D:13:D4:B4:40:CA:9B:CA:52:
                0A:15:BA:0B:40:CB:58:C7:9F:6D:FF:B6:E0:D7:4C:B6:80:8C:D5:25:
                82:51:21:59:17:5E:84:72:DA:A8:75:AF:B6:2D:51:E3:12:5D:AD:71:
                82:11:AA:BF:74:7A:EA:18:EF:F7:EC:76:76:B5:3A:FC:04:FF:44:E7:
                16:50:EA:1C:C2:BF:89:FC:A4:A1:04:A3:89:7C:1D:21:A2:02:08:1E:
                D3:0B:42:B4:2E:24:0F:F8:73:BC:04:C0:60:25:84:21:2D:DE:6E:9E:
                87:37:20:DE:6F:9E:A7:52:BC:39:8A:63:EF:78:A8:4C:BE:19:FD:F8:
                6B:9D:23:24:82:7D:8B:8E:A8:6E:D5:CA:E2:34:B1:8F:71:79:17:4F:
                2A:30:54:47:98:85:E1:2E:4A:EA:C4:98:14:86:86:97:05:D2:1D:5B:
                74:68:BF:DE:E8:DE:B8:0F:CC:36:00:5C:FC:B3:3D:07:9F:39:AC:12:
                9E:FE:8A:2A:86:67:3D:6F:6E:A0:D4:41:76:5D:55:2F:D7:E7:64:A0:
                96:34:BD:40:F4:2A:6E:D7:65:F8:92:21:82:A7:3C:D1:7C:56:A6:3D:
                81:2B:2C:7F:38:0B:6E:D2:F7:6B:18:59:7E:21:2B:F0:CF:2E:99:38:
                4C:94:DF:6E:82:DA:C5:56:2B:5B:78:CB:8E:D7:9E:A5:37:1B:FE:BB:
                C6:3E:5E:33:11:5D:64:3A:BF:94:90:E6:75:ED:4F:96:F6:CA:A1:1E:
                5B:A6:97:93:62:22:7A:8F:53:5B:33:E3:3C:6A:DD:43:A9:8A:A5:15:
                5B:8A:C5:D2:BE:FF:FF:29:4C:DC:77:85:2E:90:33:A8:7B:5A:CB:56:
                72:65:97:50:E8:D4:68:A3:ED:C5:55:AE:9B:5A:84:4E:84:6C:DD:D9:
                12:24:AA:73:77:EF:78:E2:75:01:5C:C6:DD:BF:88:F2:F9:11:3D:5C:
                8E:30:41:FB:39:73:C8:D7:3F:CE:F1:46
parm:           irq_mode:0: auto, 1: legacy, 2: msi (default: 0) (uint)
parm:           reset_mode:0: auto, 1: warm only (default: 0) (uint)

greearb, 2020-11-22:
This board is not supported by ath10k-ct driver or firmware at this time.  It is not supported by upstream ath10k driver either as far as I know.

AndyWas, 2020-11-22:
Thanks for clarifying this Ben.

So who do we need to ask to get it fixed - Qualcomm, Compex or the Linux wireless devs?  Do Qualcomm / Compex provide anything either as source code or compiled binaries?  As it's only supported on Linux AFAIK it's in their commercial interests for it to work.  On the ath10k-firmware Git I can only see firmware binaries, and the code on here looks like it is for the CT ath10k driver rather than firmware.

Apologies for any dumb questions...if there's anything I can do to help move things along I'll do it.

Andrew

jblankfeld, 2020-11-25:
Hi guys,
I'm in the same situation.
Compex ignores my request for new firmware.
But I found an old thread about this card: https://forum.openwrt.org/t/rebuild-firmware-with-a-modified-patch/63155
I tried it and it seems good, at least I managed to get a new interface. But haven't tried AP yet.

AndyWas, 2020-11-25:
Hey J,

I saw a link to that OpenWrt forum post from an issue raised on the Turris Omnia forum by the same Nik (I'm pretty sure) who also raised #163 on here...all very Six Degrees of Separation! However I'm not clear whether that patch applies to the build of board-2.bin, firmware-5.bin, ath10k_pci.ko, all of them or something else.  And in any event where do you download the original source code from?  I'm aware of OpenWrt but not very familiar or directly involved.  I'm not a system programmer but I can build packages when I need to so I should be able to build on x86_64...I'll have a go anyway!

I was able to create an account for the Compex download portal but there's nothing remotely recent on there.  I doubt I could even complete the contact form as I don't have my own company or website...I'm just a guy who builds his own firewall/router.

Regards,
Andrew

jblankfeld, 2020-11-27:
Hi,
In this thread, there is a dropbox folder with QCA9984 firmwares and a patch of ath10k core.c
I'm building my own kernel on my router and I confirm that the whole patch works, I have now a single interface that works in 5GHz mode. I don't know how the 2GHz is supposed to work, it looks like you must select the mode when loading ath10k kernel module but I was hoping to get 2 distinct interfaces so both modes can work simultaneously.
Now I am not sure the whole patch is needed, maybe applying just the firmware with mainline ath10k works too.
I hope you can sort this out, I believe this card will eventually be supported in upstream linux.

DanielRIOT, 2020-12-08:
Hi @jblankfeld 
Last I heard from Compex was that the WLE1216VX does not do simultaneous dual band and works like the WLE900VX where you set the mode as 11a, or 11g  in /etc/config/wireless for 5GHz or 2GHz modes  ( or select them in LuCi )

I'm waiting for mine to arrive to verify that as there is a shortage of the V2 and V5 cards :(

AndyWas, 2020-12-09:
Hi @jblankfeld and @DanielRIOT 

I cloned the Linux Wireless Git (https://git.kernel.org/cgit/linux/kernel/git/kvalo/ath.git/) onto a x86_64 PC running Fedora 33 (kernel 5.9.11-200).  I had to build the entire kernel (which is labelled "5.10.0-rc4-wt-ath+") as I couldn't build the ath10k module by itself due to missing symbol definitions, though I can subsequently rebuild just that module.  

I also installed separately the Fedora kernel-devel and kernel-headers packages, although the kernel source tree there doesn't contain the full ath10k source so I had to symlink from the source tree root to ./drivers in the Git source tree (as well as ./include, ./arch and ./tools in order to avoid dependency errors - there seem to be a lot of commits in other areas still to make it into the release kernel).  However I couldn't get that to build either the ath10k module or the full kernel.

I was able to apply the patch to the Linux Wireless Git source cleanly, and to rebuild the ath10k module successfully.  On the 5.10.0-rc4-wt-ath+ kernel the patched module loads with the ath10k-ct firmware in the Dropbox package.  However the same happens with the unpatched module and indeed the Fedora 5.9.11 kernel, so long as the ath10k-ct firmware is used.  This is "success" in the sense that the module loads the firmware (which is identified as "10.4b-ct-9984-fW-012-17ba98334") without any obvious error and a wireless LAN interface is created, which Network Manager says supports AP mode - I haven't gone any further, and in particular to check out the functionality implemented in the patch.  It looks like that allows you to select frequency band (as opposed to default?) with a module load parameter but I am open to correction on that.

I would be interested find out how and from where Candela Tech get the firmware source so I can try and get it merged into the official Linux Wireless firmware.  I don't know where their firmware comes from, and haven't seen any firmware source code anywhere - just binaries.  It looks like the right firmware is the key thing.  If the patch does just facilitate selecting frequency band that is less vital assuming it defaults to 5 GHz - most of the new features in the WLE1216VX only work there anyway.

AndyWas, 2020-12-10:
OK so I have had time to have a little play...

- I can't work out how to apply the startup parameters to select frequency band so it starts by default in 5 GHz.  No great problem as that is where I aim to use it, especially as it seems it doesn't support dual-band simultaneously.
- It scan 5 GHz APs in wavemon or Network Manager GUI.  I can't seehow it performs as a client as I don't have credentials for any 5 GHz APs in range...
- It runs as an AP for a few seconds with a fairly minimal 5 GHz hostapd config, but then crashes due to "peer-unmap-event: unknown peer id 0"
```
- Dec 10 23:06:56 localhost.localdomain systemd[1]: Starting Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator...
- Dec 10 23:06:57 localhost.localdomain hostapd[61766]: Configuration file: /etc/hostapd/hostapd.conf
- Dec 10 23:06:59 localhost.localdomain hostapd[61766]: wlp3s0: interface state UNINITIALIZED->COUNTRY_UPDATE
- Dec 10 23:06:59 localhost.localdomain hostapd[61766]: ACS: Automatic channel selection started, this may take a bit
- Dec 10 23:06:59 localhost.localdomain hostapd[61766]: wlp3s0: interface state COUNTRY_UPDATE->ACS
- Dec 10 23:06:59 localhost.localdomain hostapd[61766]: wlp3s0: ACS-STARTED
- Dec 10 23:06:59 localhost.localdomain audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=hostapd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
- Dec 10 23:06:59 localhost.localdomain systemd[1]: Started Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
- Dec 10 23:07:01 localhost.localdomain kernel: ath10k_pci 0000:03:00.0: peer-unmap-event: unknown peer id 0
- Dec 10 23:07:01 localhost.localdomain systemd[1]: hostapd.service: Succeeded.
- Dec 10 23:07:01 localhost.localdomain audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=hostapd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
- Dec 10 23:07:01 localhost.localdomain systemd[1]: hostapd.service: Consumed 2.445s CPU time.
```

DanielRIOT, 2020-12-18:
I've received my cards and can also only get them to work in 5GHz mode..
IW reports that they both only support 5GHz , 
no matter what config i set in /etc/modules.d/ath10k-ct as the patch readme suggests..
I wonder if there is a parameter we can set from ath10k firmware ( in /lib/firmware/ath10k/ ) like [This page ](http://www.candelatech.com/ath10k-10.4.php) suggests in the "CT Ath10k Advanced configuration" section

youxiaojie, 2021-02-24:
the same question!

greearb, 2021-02-25:
if you have a patch for ath10k-ct driver, please post it here and I'll review it.

Black6spdZ, 2021-06-08:
I'm also having issues with ath10k-ct on OpenWRT 19.07.7 with a SparkLan WPEQ-450AC, same dev ID https://pastebin.com/QqZJfEDx

youxiaojie, 2021-06-16:
I get pci-pre-cal*.bin from netgear art , but it runs only in sta mode, why?